MichaIng/DietPi #6174: bash enters 'help' by itself.

Ian-Drunkenson, 2023-02-18:
When I log in to Dietpi on a Raspberry Pi zero over ssh, bash enters automatic 'help' and shows a long list with all bash commands. 
Since I run Dietpi on other devises and dont have such a behaviour I suppose it might be a bug.

Joulinar, 2023-02-18:
Can you share the output or error message? Actually we are not aware on such behaviour.

Ian-Drunkenson, 2023-02-18:
Thats what the login looks like (I dont enter 'help', it does all by itself):

login as: root
root@192.168.1.3's password:
help
Linux pihole-2 5.15.84+ #1613 Thu Jan 5 11:58:09 GMT 2023 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v8.14.2 : 09:32 - Sa 18.02.2023
 ─────────────────────────────────────────────────────
 - Device model : RPi Zero W (armv6l)
 - CPU temp : 48 °C / 118 °F : Optimal temperature
 - LAN IP : 192.168.1.3 (eth0)
 - MOTD : Check out the DietPi v8.14 release notes:
          https://dietpi.com/docs/releases/v8_14/
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Image by        : DietPi Core Team (pre-image: from scratch)
 Patreon Legends : Camry2731, Chris Gelatt
 Website         : https://dietpi.com/ | https://twitter.com/DietPi_
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@pihole-2:~# help
GNU bash, Version 5.1.4(1)-release (arm-unknown-linux-gnueabihf)
Diese Shellkommandos sind intern definiert. Geben Sie »help« ein, um diese
Liste zu sehen. Geben Sie »help Name« ein, um die Beschreibung der Funktion
»Name« zu sehen. Geben Sie »info bash« ein, um die vollständige Dokumentation
zu sehen. Geben Sie »man -k« oder »info« ein, um detaillierte Beschreibungen
der Shellkommandos zu sehen.

Ein Stern (*) neben dem Namen kennzeichnet deaktivierte Kommandos.

 Jobbezeichnung [&]                                                                                                    history [-c] [-d Offset] [n] oder history -anrw [Dateiname] oder history -ps Argument [Argument...]
 (( Ausdruck ))                                                                                                        if Kommandos; then Kommandos; [ elif Kommandos; then Kommandos; ]... [ else Kommandos; ] fi
 . Dateiname [Argumente]                                                                                               jobs [-lnprs] [Jobbez. ...] or jobs -x Kommando [Arg]
 :                                                                                                                     kill [-s Signalname | -n Signalnummer | -Signalname] pid | jobspec ... oder kill -l [Signalname]
 [ Argument... ]                                                                                                       let Argument [Argument ...]
 [[ Ausdruck ]]                                                                                                        local [Option] Name[=Wert] ...
 alias [-p] [Name[=Wert] ... ]                                                                                         logout [n]
 bg [Jobbezeichnung ...]                                                                                               mapfile [-d Begrenzer] [-n Anzahl] [-O Quelle] [-s Anzahl] [-t] [-u fd]         [-C Callback] [-c Menge] [Feldvari>
 bind [-lpsvPSVX] [-m Tastaturtabelle] [-f Dateiname] [-q Name] [-u Name]  [-r Tastenfolge] [-x Tastenfolge:Shell Ko>  popd [-n] [+N | -N]
 break [n]                                                                                                             printf [-v var] Format [Argumente]
 builtin [Shellkommando [Argument ...]]                                                                                pushd [-n] [+N | -N | Verzeichnis]
 caller [Ausdruck]                                                                                                     pwd [-LP]
 case Wort in [Muster [| Muster]...) Kommandos ;;]... esac                                                             read [-ers] [-a Feld] [-d Begrenzer] [-i Text] [-n Zeichenanzahl] [-N Zeichenanzahl] [-p Prompt] [-t Zeitlimit] [->
 cd [-L|[-P [-e]] [-@]] [Verzeichnis]                                                                                  readarray [-d Begrenzer] [-n Anzahl] [-O Quelle] [-s Anzahl] [-t]           [-u fd] [-C Callback] [-c Menge] [Feld>
 command [-pVv] Kommando [Argument ...]                                                                                readonly [-aAf] [Name[=Wert] ...] oder readonly -p
 compgen [-abcdefgjksuv] [-o option] [-A action] [-G globpat] [-W wordlist] [-F function] [-C command] [-X filterpat>  return [n]
 complete [-abcdefgjksuv] [-pr] [-DEI] [-o option] [-A action] [-G globpat] [-W wordlist] [-F function] [-C command]>  select Name [in Wort ... ;] do Kommandos; done
 compopt [-o|+o Option] [-DEI] [Name ...]                                                                              set [-abefhkmnptuvxBCHP] [-o Option] [--] [Argument ...]
 continue [n]                                                                                                          shift [n]
 coproc [Name] Kommando [Umleitungen]                                                                                  shopt [-pqsu] [-o] [Optionsname ...]
[... cut ...]
 function Name { Kommandos ; } oder Name () { Kommandos ; }                                                            variables - Namen und Bedeutung einiger Shellvariablen
 getopts optstring name [arg ...]                                                                                      wait [-fn] [-p var] [id ...]
 hash [-lr] [-p Pfadname] [-dt] [Name ...]                                                                             while Kommandos; do Kommandos; done
 help [-dms] [Muster ...]                                                                                              { Kommandos ; }
root@pihole-2:~#

Joulinar, 2023-02-18:
> login as: root
root@192.168.1.3's password:
help

looks like `help `command is entered during login process

Ian-Drunkenson, 2023-02-18:
Yes it looks like but its not and the passwotd is also not 'help'. It enters 'help' by itself. As I mentioned before on all other 5 or 6 Dietpi devices with the same Dietpi version and the same root password I dont have this.
I looked up if there ist something in bashrc and other startup scripts but I coulnd find anything.

Joulinar, 2023-02-18:
Did you tried connection from a different client device using a different SSH client?

Ian-Drunkenson, 2023-02-18:
Yes, I logged the ssh raw data and Putty types in 'help' by itself. I removed the stored Putty config for the device and put it back in and the 'help' issue is gone. Sorry guys. Its not a Dietpi, its a Putty bug.

Joulinar, 2023-02-18:
ok good you found it. Going to close the issue for now.

closed 2023-02-18 by Joulinar