MichaIng/DietPi #6177: SSH on Hyper-V not operational

zarigata, 2023-02-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.14.12
- Distro version | Hyper-V
- Kernel version | 5.10.0

#### Steps to reproduce

1. try to log in via putty

#### Expected behaviour

- SSH OPEN ON PUTTY

#### Actual behaviour

- Access denied

#### Extra details

- ...

Joulinar, 2023-02-18:
Are you able to login locally and check SSH server, if it has been started correctly.

zarigata, 2023-02-18:
yes i am, no problems on that

Joulinar, 2023-02-18:
SSH server is running a listen to correct port? Are you able to ping anything on the web? You get correct IP address assigned?

```
journalctl -u dropbear.service
ss -tulpn | grep LISTEN
ping google.com
ip a, ip r
```

zarigata, 2023-02-18:
yes it can leave the signal
journalctl -u dropbear.service = no records
it is a IP fixed

Joulinar, 2023-02-18:
I would be appreciated if you can share the output of the commands I request. Did you changed to OpenSSH server?

zarigata, 2023-02-18:
[image]
[image]

MichaIng, 2023-02-18:
and
```sh
ss -tlpn
```
?

Joulinar, 2023-02-18:
Is `192.168.15.0` your local network range?

zarigata, 2023-02-18:
yes

zarigata, 2023-02-18:
> 

ethernet

Joulinar, 2023-02-18:
To be honest, it's not that easy to communicate with you. Some more words would be appreciated. As well, please answer our questions. Can you post the output as requested by @MichaIng . Did you disabled IPv6 in meantime? What interface type you setup on Hyper-V? What is the IP range of your local network?

zarigata, 2023-02-18:
sorry not a good response, i have to do this while doing some other stuff at the company and they sell what they cant demand, so i have to keep, 

SS - tlpn 
[image]
ipv6 is not enabled, it can  be activated ath the modem, but it is not recommended,  it is on a 100mbts ethernet, it has a IP locked on it.

Joulinar, 2023-02-18:
disable IPv6 on your DietPi device. And still an outstanding question, what is the IP range of your local network? Is it `192.168.15.0`?

zarigata, 2023-02-18:
it is 192,168,15,0 to 192,168,15,255, being 255 used to end

Joulinar, 2023-02-18:
are you able to reach PiHole web interface located on this VM? Did you installed UFW?

zarigata, 2023-02-18:
> are you able to reach PiHole web interface located on this VM? Did you installed UFW?

[image]
yes i can acees the progtrams i have made available, i have a firewall but is not UFW, it is a modem one

MichaIng, 2023-02-18:
Double check your firewall rules then. The VM is generally connected to LAN and reachable, SSH is listening and not reporting any errors, so there is something else in front of it blocking requests. If there is any firewall, the very first attempt to solve connection issues is disabling the firewall 😉.

zarigata, 2023-02-20:
i checked, it was not, i installed a BIOS version and it went well, perhaps some HYPER-V modification?

closed 2023-02-20 by zarigata