MichaIng/DietPi #6178: Odroid C4 - HDMI - NO Audio

MDAR, 2023-02-18:
#### Required Information
- DietPi version | **V8.4.14.2**
- Distro version | **Bulleye**
- Kernel version | **Linux LiveStream-c4 6.0.13-meson64 #22.11.2 SMP PREEMPT Sun Dec 18 16:52:19 CET 2022 aarch64 GNU/Linux**
- SBC model | **Odroid C4/HC4 (aarch64)**
- Power supply used | **12V 5amp**
- SD card used | **eMMC 32GB**

#### Additional Information (if applicable)
- Software title | **Chromium in Xinit**
- Was the software title installed freshly or updated/migrated? - **Fresh**
- Can this issue be replicated on a fresh installation of DietPi? - **Yes**

#### Steps to reproduce
Flash a fresh image to the Odroid
Use Chromium to play some audio / video content

Video plays wonderfully, but without audio.

if selecting a USB DAC, audio is perfect

#### Expected behaviour
HDMI as an option the Audio options

#### Actual behaviour
```
usb-dac : USB Audio DAC (any)
hw:0,0  : ODROIDC4
hw:0,1  : ODROIDC4
hw:0,2  : ODROIDC4
```

#### Extra details

I can't remember having to make any changes in order to get audio out of the HDMI port on an Odroid C4 (when using Chromium in Xinit)

Today I put a fresh Bullseye config into my main production unit and it doesn't seem to want to output audio from the HDMI.

I have a faint memory of seeing HDMI as an audio option, but all I see now is

```
usb-dac : USB Audio DAC (any)
hw:0,0  : ODROIDC4
hw:0,1  : ODROIDC4
hw:0,2  : ODROIDC4
```

Before I go down a rabbit hole, have you changed something ?

-----------
FYI
I have plugged in a Roland UA-30 USB audio device and can confirm that I get audio out of that, my fear was that Chromium wasn't sending audio.

The Roland is the only device I can get audio out of now.

I do still have the Buster OS on a card that I can pop in, if you need me to check anything,

```
usb-dac : USB Audio DAC (any)
hw:0,0  : ODROIDC4
hw:0,1  : ODROIDC4
hw:0,2  : ODROIDC4
hw1,0   : REC USB Audio
```

MichaIng, 2023-02-18:
Many thanks for your report.

Have you tried with all those `hw:0,*` devices and did you check `alsamixer` for some mute or low volume control?

MDAR, 2023-02-18:
> 

Hi

Indeed I did.

I was hoping to find the solution in there as something as silly as the output level being low

MichaIng, 2023-02-18:
Can you show all controls:
```sh
amixer -c 0 scontents
```

MDAR, 2023-02-19:
> ```shell
> amixer -c 0 scontents
> ```

I hope this means more top you than it does it me.....

```
root@LiveStream-c4:~# amixer -c 0 scontents

Simple mixer control 'FRDDR_A SINK 1 SEL',0
  Capabilities: enum
  Items: 'OUT 0' 'OUT 1' 'OUT 2' 'OUT 3' 'OUT 4' 'OUT 5' 'OUT 6' 'OUT 7'
  Item0: 'OUT 0'
Simple mixer control 'FRDDR_A SINK 2 SEL',0
  Capabilities: enum
  Items: 'OUT 0' 'OUT 1' 'OUT 2' 'OUT 3' 'OUT 4' 'OUT 5' 'OUT 6' 'OUT 7'
  Item0: 'OUT 1'
Simple mixer control 'FRDDR_A SINK 3 SEL',0
  Capabilities: enum
  Items: 'OUT 0' 'OUT 1' 'OUT 2' 'OUT 3' 'OUT 4' 'OUT 5' 'OUT 6' 'OUT 7'
  Item0: 'OUT 2'
Simple mixer control 'FRDDR_A SRC 1 EN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'FRDDR_A SRC 2 EN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'FRDDR_A SRC 3 EN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'FRDDR_B SINK 1 SEL',0
  Capabilities: enum
  Items: 'OUT 0' 'OUT 1' 'OUT 2' 'OUT 3' 'OUT 4' 'OUT 5' 'OUT 6' 'OUT 7'
  Item0: 'OUT 0'
Simple mixer control 'FRDDR_B SINK 2 SEL',0
  Capabilities: enum
  Items: 'OUT 0' 'OUT 1' 'OUT 2' 'OUT 3' 'OUT 4' 'OUT 5' 'OUT 6' 'OUT 7'
  Item0: 'OUT 1'
Simple mixer control 'FRDDR_B SINK 3 SEL',0
  Capabilities: enum
  Items: 'OUT 0' 'OUT 1' 'OUT 2' 'OUT 3' 'OUT 4' 'OUT 5' 'OUT 6' 'OUT 7'
  Item0: 'OUT 2'
Simple mixer control 'FRDDR_B SRC 1 EN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'FRDDR_B SRC 2 EN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'FRDDR_B SRC 3 EN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'FRDDR_C SINK 1 SEL',0
  Capabilities: enum
  Items: 'OUT 0' 'OUT 1' 'OUT 2' 'OUT 3' 'OUT 4' 'OUT 5' 'OUT 6' 'OUT 7'
  Item0: 'OUT 0'
Simple mixer control 'FRDDR_C SINK 2 SEL',0
  Capabilities: enum
  Items: 'OUT 0' 'OUT 1' 'OUT 2' 'OUT 3' 'OUT 4' 'OUT 5' 'OUT 6' 'OUT 7'
  Item0: 'OUT 0'
Simple mixer control 'FRDDR_C SINK 3 SEL',0
  Capabilities: enum
  Items: 'OUT 0' 'OUT 1' 'OUT 2' 'OUT 3' 'OUT 4' 'OUT 5' 'OUT 6' 'OUT 7'
  Item0: 'OUT 0'
Simple mixer control 'FRDDR_C SRC 1 EN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'FRDDR_C SRC 2 EN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'FRDDR_C SRC 3 EN',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'TDMOUT_B Gain Enable',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'TDMOUT_B Lane 0',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 81 [32%]
  Front Right: 81 [32%]
Simple mixer control 'TDMOUT_B Lane 1',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 81 [32%]
  Front Right: 81 [32%]
Simple mixer control 'TDMOUT_B Lane 2',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 81 [32%]
  Front Right: 81 [32%]
Simple mixer control 'TDMOUT_B Lane 3',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 81 [32%]
  Front Right: 81 [32%]
Simple mixer control 'TDMOUT_B SRC SEL',0
  Capabilities: enum
  Items: 'IN 0' 'IN 1' 'IN 2' 'IN 3' 'IN 4'
  Item0: 'IN 0'
Simple mixer control 'TOHDMITX',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'TOHDMITX I2S SRC',0
  Capabilities: enum
  Items: 'I2S A' 'I2S B' 'I2S C'
  Item0: 'I2S A'
Simple mixer control 'TOHDMITX SPDIF SRC',0
  Capabilities: enum
  Items: 'SPDIF A' 'SPDIF B'
  Item0: 'SPDIF A'
```

I saw this, does it help???

```
root@LiveStream-c4:~# dietpi-config
[ SUB1 ] DietPi-Set_hardware > soundcard (hw:1,0)
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: alsa-utils
[ INFO ] DietPi-Set_hardware | Resetting all sound card settings...
[  OK  ] DietPi-Set_hardware | rm /etc/asound.conf
[  OK  ] DietPi-Set_hardware | rm /var/lib/alsa/asound.state
alsa-lib parser.c:2179:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "axg-sound-card" "" "" "" ""
Hardware is initialized using a generic method
alsa-lib parser.c:2179:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
Found hardware: "USB-Audio" "USB Mixer" "USB0582:0006" "" ""
Hardware is initialized using a generic method
[ INFO ] DietPi-Set_hardware | Applying new sound card settings...
[  OK  ] DietPi-Set_hardware | Setting in /boot/dietpi.txt adjusted: CONFIG_SOUNDCARD=hw:1,0
[  OK  ] DietPi-Set_hardware | alsactl -g store
[  OK  ] soundcard hw:1,0 | Completed
```

MDAR, 2023-02-19:
> Can you show all controls:
> 
> ```shell
> amixer -c 0 scontents
> ```

FYI

I have connected to an Odroid C4 with old Buster and diet-config > audio options, shows this

```
usb-dac : USB Audio DAC (any)
hw:0,0  : ODROIDHDMI
```

```
dietpi-config
[ SUB1 ] DietPi-Set_hardware > soundcard (usb-dac)
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: alsa-utils
[  OK  ] DietPi-Set_hardware | All required APT packages are already installed.
[ INFO ] DietPi-Set_hardware | Resetting all sound card settings...
[  OK  ] DietPi-Set_hardware | rm /etc/asound.conf
[  OK  ] DietPi-Set_hardware | rm /var/lib/alsa/asound.state
alsa-lib parser.c:2179:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "ODROID-HDMI" "" "" "" ""
Hardware is initialized using a generic method
[ INFO ] DietPi-Set_hardware | Applying new sound card settings...
[  OK  ] DietPi-Set_hardware | Setting in /boot/dietpi.txt adjusted: CONFIG_SOUNDCARD=usb-dac
[FAILED] DietPi-Set_hardware | Unable to find any USB DAC on your system, reverting to "none"...
[  OK  ] DietPi-Set_hardware | Setting in /boot/dietpi.txt adjusted: CONFIG_SOUNDCARD=none
[FAILED] soundcard usb-dac | Exited with error
[ SUB1 ] DietPi-Set_hardware > soundcard (hw:0,0)
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: alsa-utils
[  OK  ] DietPi-Set_hardware | All required APT packages are already installed.
[ INFO ] DietPi-Set_hardware | Resetting all sound card settings...
alsa-lib parser.c:2179:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "ODROID-HDMI" "" "" "" ""
Hardware is initialized using a generic method
[ INFO ] DietPi-Set_hardware | Applying new sound card settings...
[  OK  ] DietPi-Set_hardware | Setting in /boot/dietpi.txt adjusted: CONFIG_SOUNDCARD=hw:0,0
[  OK  ] DietPi-Set_hardware | alsactl -g store
[  OK  ] soundcard hw:0,0 | Completed
```

```
root@LiveStream-C4:~# amixer -c 0 scontents
Simple mixer control 'Audio In Source',0
  Capabilities: enum
  Items: 'TDMIN_A' 'TDMIN_B' 'TDMIN_C' 'SPDIFIN' 'PDMIN' 'FRATV' 'TDMIN_LB' 'LOOPBACK_A' 'FRHDMIRX' 'LOOPBACK_B' 'SPDIFIN_LB' 'EARCRX_DMAC' 'RESERVED_0' 'RESERVED_1' 'RESERVED_2' 'VAD'
  Item0: 'TDMIN_A'
Simple mixer control 'Audio Out Sink',0
  Capabilities: enum
  Items: 'TDMIN_A' 'TDMIN_B' 'TDMIN_C' 'SPDIFIN' 'PDMIN' 'FRATV' 'TDMIN_LB' 'LOOPBACK_A' 'FRHDMIRX' 'LOOPBACK_B' 'SPDIFIN_LB' 'EARCRX_DMAC' 'RESERVED_0' 'RESERVED_1' 'RESERVED_2' 'VAD'
  Item0: 'TDMIN_A'
Simple mixer control 'Audio hdmi-out mute',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'Audio spdif format',0
  Capabilities: enum
  Items: '2 CH PCM' 'DTS RAW Mode' 'Dolby Digital' 'DTS' 'DD+' 'DTS-HD' 'Multi-channel LPCM' 'TrueHD' 'DTS-HD MA' 'HIGH SR Stereo LPCM'
  Item0: '2 CH PCM'
Simple mixer control 'Audio spdif mute',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'Audio spdifin source',0
  Capabilities: enum
  Items: 'spdifin pad' 'spdifout' 'N/A' 'HDMIRX'
  Item0: 'spdifin pad'
Simple mixer control 'SPDIFIN Audio Type',0
  Capabilities: enum
  Items: 'LPCM' 'AC3' 'EAC3' 'DTS' 'DTS-HD' 'TRUEHD' 'PAUSE'
  Item0: 'LPCM'
Simple mixer control 'SPDIFIN audio samplerate',0
  Capabilities: enum
  Items: 'N/A' '32000' '44100' '48000' '88200' '96000' '176400' '192000'
  Item0: 'N/A'
Simple mixer control 'audio inskew set',0
  Capabilities: enum
  Items: '0' '1' '2' '3' '4' '5' '6'
  Item0: '0'
Simple mixer control 'audio locker enable',0
  Capabilities: enum
  Items: 'Disable' 'Enable'
  Item0: 'Disable'
Simple mixer control 'tdmout_c binv set',0
  Capabilities: enum
  Items: '0' '1'
  Item0: '0'
root@LiveStream-C4:~#
```

MichaIng, 2023-02-19:
> snd_use_case_mgr_open

Interesting, since when does `alsactl init` laudly complain when UCM is (intentionally) not installed 🤔. It is an extra package (since Bullseye) which contains a set of configurations for specific sound cards: https://packages.debian.org/en/bullseye/all/alsa-ucm-conf/filelist
Probably makes sense that we install this. Not many SBCs will be covered with card specific configs, but there seem to be some generic ones as well.

... just tested on VM with emulated onboard AC97 audio:
```console
# alsactl init
alsa-lib parser.c:2179:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "ICH" "Analog Devices AD1980" "AC97a:41445370" "0x1028" "0x0177"
Hardware is initialized using a generic method
# apt install alsa-ucm-conf
# alsactl init
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "ICH" "Analog Devices AD1980" "AC97a:41445370" "0x1028" "0x0177"
Hardware is initialized using a generic method
```
It still fails loudly and the result is the same "initialized using a generic method". So the package only makes sense if one has a matching sound card. It would then set some of the controls automatically.

There is a flag to explicitly skip UCM, which is used from DietPi v8.15 on, if its not installed, to mute the error: https://github.com/MichaIng/DietPi/commit/4b25af1

However, back to your actual issue. Yes the older Buster image uses the legacy vendor kernel, so this cannot be compared. Your controls indeed look exactly like those on Odroid N2, which makes sense as it is the same kernel. Just strange that no one else reported this earlier. I thought whichever Armbian patch is breaking this for Odroid N2, did not apply for C4, e.g. because of different device trees. However, try this:
1. Detach the USB DAC, just in case: https://forum.armbian.com/topic/24882-no-sound/#comment-159451
2. ```sh
	# Enable HDMI output
	G_EXEC amixer -c 0 set 'TOHDMITX' 'on'
	# Use I2S B as source for HDMI output (I2S A does not work)
	G_EXEC amixer -c 0 set 'TOHDMITX I2S SRC' 'I2S B'
	# Use ALSA device 0 as input for I2S B
	G_EXEC amixer -c 0 set 'TDMOUT_B SRC SEL' 'IN 0'
	# Enable I2S B (SRC 2) on device 0 (_A)
	G_EXEC amixer -c 0 set 'FRDDR_A SRC 2 EN' 'on'
	# Set output channels for device 0 (_A)
	G_EXEC amixer -c 0 set 'FRDDR_A SINK 1 SEL' 'OUT 0'
	G_EXEC amixer -c 0 set 'FRDDR_A SINK 2 SEL' 'OUT 1'
	G_EXEC amixer -c 0 set 'FRDDR_A SINK 3 SEL' 'OUT 2'
	# Set master volume to 85%
	G_EXEC amixer -c 0 set 'ACODEC' '85%'
	```

MDAR, 2023-02-19:
> However, try this:

Thanks @MichaIng 

I'll give it a try when I'm back in my office (maybe Thursday)

One question..

Are these settings that I need to put in a configuration file or individual commands that I need to run?

```
# Enable HDMI output
 G_EXEC amixer -c 0 set 'TOHDMITX' 'on'
 # Use I2S B as source for HDMI output (I2S A does not work)
 G_EXEC amixer -c 0 set 'TOHDMITX I2S SRC' 'I2S B'
 # Use ALSA device 0 as input for I2S B
 G_EXEC amixer -c 0 set 'TDMOUT_B SRC SEL' 'IN 0'
 # Enable I2S B (SRC 2) on device 0 (_A)
 G_EXEC amixer -c 0 set 'FRDDR_A SRC 2 EN' 'on'
 # Set output channels for device 0 (_A)
 G_EXEC amixer -c 0 set 'FRDDR_A SINK 1 SEL' 'OUT 0'
 G_EXEC amixer -c 0 set 'FRDDR_A SINK 2 SEL' 'OUT 1'
 G_EXEC amixer -c 0 set 'FRDDR_A SINK 3 SEL' 'OUT 2'
 # Set master volume to 85%
 G_EXEC amixer -c 0 set 'ACODEC' '85%'
```

MichaIng, 2023-02-19:
These are console commands you can copy&paste into your console, at best one by one in case you see an error in between.

MDAR, 2023-02-23:
Hi

I've got terrible news

I ran all the commands you have provided and nothing changed.

FYI, the last command failed, your blue error screen came up, saying that ACODEC wasn't valid.
```
 Error log:                                                                                                        │
                                           │ amixer: Unable to find simple control 'ACODEC',0 
```

So...

I downloaded a brand new image from the website
Flashed it to a spare emmc

Installed Chromium and configured it to auto start and launch
`SOFTWARE_CHROMIUM_AUTOSTART_URL=https://www.boomradiouk.com/player`

(Thinking that has to be a pretty pure no frills web audio source)

When it didn't play any audio, I added Tiger Scrapper VNC and went into the settings of Chromium to set Sound playback permission for all sites, via --
 
**Chromium Settings > Privacy And Security > Site Settings > Additional Content Settings > Sound > "Allowed to play sounds"**

In all the text that Chromium spat out into the console, I did notice this

**ALSA lib pcm.c:8719:(snd_pcm_set_params) Unable to set hw params for PLAYBACK: Invalid argument**

I tried all the Odroid sound options and they all failed.

However, just to double check, I plugged back in my faithful Roland UA-30 and that works perfectly.
Audio playback from any website work.

Just for clarity, I did reboot the machine between each change and allow Chromium to autostart.
The only thing I needed to do was interact via VNC to press the Play button on the website

For the short term, I can route audio via the Roland (or any other USB device I have), so there's no great rush to resolve this.

As always, if there is anything at all that I can do to assist, including a remote connection, please just ask,

_____________

FYI

Console output during a successful USB Audio playback

```
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu Feb 23 12:57:50 2023
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
[2276:2303:0223/125753.760362:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type                                                                                                          (examples of valid types are "tcp" and on UNIX "unix")
[2276:2303:0223/125753.760845:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type                                                                                                          (examples of valid types are "tcp" and on UNIX "unix")
[2276:2303:0223/125753.826065:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type                                                                                                          (examples of valid types are "tcp" and on UNIX "unix")
[2276:2303:0223/125753.826259:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type                                                                                                          (examples of valid types are "tcp" and on UNIX "unix")
[2276:2303:0223/125753.927854:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type                                                                                                          (examples of valid types are "tcp" and on UNIX "unix")
[2276:2303:0223/125753.928322:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type                                                                                                          (examples of valid types are "tcp" and on UNIX "unix")
[2276:2349:0223/125754.197466:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /                                                                                                         org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service                                                                                                          files
[2276:2349:0223/125754.198842:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_pat                                                                                                         h= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .serv                                                                                                         ice files
[2276:2349:0223/125754.199908:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_pat                                                                                                         h= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .serv                                                                                                         ic [... cut ...]
[2276:2303:0223/125818.189433:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2276:2303:0223/125818.279253:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2276:2303:0223/125818.279408:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
```

MDAR, 2023-02-23:
WOULD YOU BELIEVE IT !!!!

I have no idea what I have done differently, but in frustration I set the Audio back to 
hw:0:0

rebooted
VNC in to press play

AND THE DAM THING STARTED PLAYING !!!!!!

(Bloody loud in the room, because I am an idiot and forgot to turn the volume down for the HDMI input of my vision desk and the interim text monitor was blaring out too)

This is the info from the JustBoom console

```
[ SUB1 ] DietPi-JustBoom > ALSA Output Info
access: RW_INTERLEAVED
format: S16_LE
subformat: STD
channels: 2
rate: 48000 (48000/1)
period_size: 1024
buffer_size: 4096
closed
```

I'll put the previous eMMC card and see what I can do.

Failing that, I'll put this card back in and rebuild Node-RED etc (an easy restore process)

MichaIng, 2023-02-23:
> I set the Audio back to hw:0:0

What was it set before? That should be the default and the commands I provided were to configure this exact device 😉.

Btw, you do not need to go all the way through X, Chromium, VNC, whatever. Just test via `speaker-test` from any console. If that doesn't play sound, the default ALSA device is not working or not configured correctly, and nothing else can play sound then (when using the default device).

However, great that it works now! Can you show the mixer controls again?
```sh
amixer -c 0 scontents
```

MichaIng, 2023-02-23:
If the sound card does not support a single channel, you can change the value like:
```sh
speaker-test -c 2
speaker-test -c 3
```
It should play some noise, looping through speakers/channels, if it succeeds. Also when enabling the auto-conversion plugin in audio options, any input stream is converted automatically to a supported output format. This makes quite sense if the sound card itself is strict about the format it supports, because otherwise depending on the audio source, it may play, or may not play.

Okay, all mixer controls are set the way the posted commands would do. Did you do this manually or was it set like that by default?

The `ACODEC` (master volume) control is missing, compared to Odroid N2. There is also no `TOACODEC` to enable 3.5mm output, which makes sense as the C4 has no 3.5mm jack. "ACODEC" seems like meant to be the 3.5mm/analogue master volume, so both are missing together. But there is no HDMI master volume 🤔. The only volume controls are these "TDMOUT_B Lane" 0-3. Does any of these (or all of them) affect HDMI volume, e.g. when you adjust them via `alsamixer` while playing audio?

MDAR, 2023-02-23:
Thank you for paying so much attention to this, I really do appreciate it.

> Does any of these (or all of them) affect HDMI volume, e.g. when you adjust them via `alsamixer` while playing audio?
No, as it happens, they don't.

I noticed a massive imbalance between the L&R channels and went looking, 
The ALSAmixer had no impact on the outputs.

(It turns out I must have had an imbalance on a previous bit of playback, as the levels were adjusted on the Atem Vision desk.

This command does work PERFECTLY, that is very useful.

`speaker-test -c 2`

Useful to know that I can test multiple speaker setups this way, I might just roll my Video rig into my cinema room and see what that outputs with 
`speaker-test -c 8`

```
root@LiveStream-C4:~# speaker-test -c 8

speaker-test 1.2.4

Playback device is default
Stream parameters are 48000Hz, S16_LE, 8 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 2 to 65536
Period size range from 1 to 32768
Using max buffer size 65536
Periods = 4
was set period_size = 16384
was set buffer_size = 65536
 0 - Front Left
 4 - Center
 1 - Front Right
 7 - Side Right
 3 - Rear Right
 2 - Rear Left
 6 - Side Left
 5 - LFE
```
(I actually have 2 more channels in that room, but as they are Dolby Atmos, I don't expect a humble Odroid to handle that)

>Okay, all mixer controls are set the way the posted commands would do. Did you do this manually or was it set like that by default?

This is just as it was from a fresh install and after running your commands.

I think I adjusted the four outputs to ~39, but that didn't make any difference.

>The ACODEC (master volume) control is missing, compared to Odroid N2.
That makes sense.

>There is also no TOACODEC to enable 3.5mm output, which makes sense as the C4 has no 3.5mm jack. "ACODEC" seems like meant to be the 3.5mm/analogue master volume, so both are missing together. 

Yeah, that looks about right.

According to the HardKernel site, we should be seeing HDMI and SPDIF audio outputs
https://www.hardkernel.com/shop/odroid-c4/

<html>
<body>

Audio | 1 x HDMI digital output 1 x Optional SPDIF optical output
-- | --

</body>
</html>

But I can't see a terminal for hooking up an Optical transmitter (certainly no TOSLINK socket on board)

>The only volume controls are these "TDMOUT_B Lane" 0-3. Does any of these (or all of them) affect HDMI volume, e.g. when you adjust them via alsamixer while playing audio?

Nothing changes.
If you want me to do something to test them, please just ask.

Just as a side note, at some point in the future I am going to try plugging in a USB3 to HDMI dongle, to try and get a second Chromium instance out of the machine, so I can have a Source and Fill feed into my Vision desk, to use with [H2R.graphic ](https://h2r.graphics/) server on another machine.

MichaIng, 2023-02-24:
> I think I adjusted the four outputs to ~39, but that didn't make any difference.

Hmm, that means that there is no mixer volume control for HDMI then. At least, usually the HDMI device or software frontend can adjust the volume as well.

> According to the HardKernel site, we should be seeing HDMI and SPDIF audio outputs

The `TOHDMITX SPDIF SRC` control sounds more like an alternative source for HDMI output but not a way to send the audio signal to S/PDIF 🤔.

MichaIng, 2023-03-09:
Solved with: https://github.com/MichaIng/DietPi/commit/bf59e28

MDAR, 2023-03-09:
Thanks @MichaIng 

I'm looking forward to seeing the results of your efforts

MichaIng, 2023-03-10:
Can be tested on beta branch:
```sh
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=beta' /boot/dietpi.txt
dietpi-update
```

MDAR, 2023-05-17:
A tiny update.

I've just installed a fresh V8.17 instance onto a C4

Straight out of the box there was no sound from the HDMI.

But once I run `dietpi-config`

Go into `Audio options`

`sound card` - which did say Odroid HDMI (or something like that)

Click on the top item

Something exciting happens, then reverts to `OdroidC4_HDMI`

Exit out of all that and the audio works **perfectly**

I can't get more than 2 channels out of the HDMI, but having read more about `speaker-test` that doesn't surprise me.

It seems it's designed for 2 channel PCM testing, rather than HDMI multi channel audio.

Multi channel with speaker-test assumes there are multiple jack outlets, rather than a digital stream that supports multiple channels.

FYI
Ubuntu Mate on my desktop machine does support HDMI multi channel, but that's a wildly different beast of a machine.

MichaIng, 2023-05-18:
Perfect 🙂!

> Multi channel with speaker-test assumes there are multiple jack outlets, rather than a digital stream that supports multiple channels.

Interesting 🤔. So e.g. `-c 6` to test 6 channels does not work?

closed 2023-05-18 by MichaIng

MDAR, 2023-05-19:
> So e.g. `-c 6` to test 6 channels does not work?

Indeed.

(Or anything else)

The amp reports

PCM 48KHZ when left is playing

No Audio / Unknown for centre channel

PCM 48KHZ when right is playing

No Audio / Unknown for any other channel

`speaker-test -c 1` just fails, as HDMI doesn't understand "mono"

I tried up to 8 channels, all with the same behaviour.

`speaker-test -c 1` just fails, as HDMI doesn't understand "mono"

Either the HDMI audio driver can't output 8 "direct audio" channels or the `speaker-test` app just sends something and assumes the driver will know what to do with it.

I'm just happy that I've got a stereo pair coming out of the Odroid C4.

If I get adventurous, I'll load up Kodi or something and see what happens with multi channel audio playback.

MichaIng, 2023-05-20:
The auto-conversion plugin btw might help to have whatever strange format sent by speaker-test converted to something the driver supports. However, as long as it's the test only probably doesn't matter.

MDAR, 2023-05-21:
> auto-conversion plugin btw might help to have whatever strange format sent by speaker-test converted to something the driver support

Oh that's interesting.

I'm happy to give it a try.

Is this a feature that can be installed with dietpi-software ?

> However, as long as it's the test only probably doesn't matter.

Indeed, I got two channels of audio out of it, which is plenty for what I'm doing.