MichaIng/DietPi #6180: brand new Plex install on Dietpi fails to add default folder as library.

porrascarlos802018, 2023-02-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
-  cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
-  echo $G_DISTRO_NAME $G_RASPBIAN
bullseye 0

- Kernel version | `uname -a`
- uname -a
Linux DietPi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
-  echo $G_HW_MODEL_NAME
RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
-5v 5A

- SD card used | (EG: SanDisk ultra)
- ADATA usb

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- PLEX
- 
- Was the software title installed freshly or updated/migrated?
-fresh PLEX install, brand new.

- Can this issue be replicated on a fresh installation of DietPi?
YES

- Bug report ID | `echo $G_HW_UUID`
- Reference code: 017ec7c6-5514-4d0a-b5f2-1308d921e676

#### Steps to reproduce

1. ... select plex media software
2. ...add a library using default path /mnt/dietpi_userdata/downloads

#### Expected behaviour

- ... library added to PLEX for folder /mnt/dietpi_userdata/downloads
- 

#### Actual behaviour

- ... PLEX fails to add library,,, eror: "Your changes could not be saved." none of the folders can be added. Problem seems to be with the latest release of the image.

#### Extra details

- ... using old image may work

MichaIng, 2023-02-20:
Many thanks for your report

> add a library using default path /mnt/dietpi_userdata/downloads

I'd see the Music and Video directories as a better default than the downloads directory. But it shouldn't matter. All of them are handed to the `dietpi` group with `0775` mode and Plex uses this as primary group, so it should have R/W access to all three of them.

> "Your changes could not be saved."

Sounds more like settings cannot be changed than an issue with the media directory itself.

> Problem seems to be with the latest release of the image.

There was no change introduced with our Plex implementation or media directories, so if not on your end (setup, configs), then it's a change on Plex end.

Do you see any errors in service logs?
```sh
journalctl -u plexmediaserver
```

MichaIng, 2023-02-20:
Oh, I can replicate. It isn't possible to add any directory, not even the service user's home directory. Looks like a bug in PMS.

MichaIng, 2023-02-20:
Reported already: https://forums.plex.tv/t/trying-to-add-folder-to-library-your-changes-could-not-be-saved-bad-api-call/831700

porrascarlos802018, 2023-02-21:
thanks for checking this one! 
you are right, restarting the server will allow to add the folder where your videos are, but it wont do the scan nor display any content as media server.

systemctl restart plexmediaserver

MichaIng, 2023-02-21:
It works pretty well here, both to scan the library, see all included files, play and delete them, in `/mnt/dietpi_userdata` Music, Video and downloads dirs. Assure that the `dietpi` group has read and in case write permissions to both, the directories themselves as well as the media files.

MichaIng, 2023-12-20:
I cannot replicate the issue anymore. It seems to work now OOTB. I'm closing this. @porrascarlos802018 let me know if you still face this issue.

closed 2023-12-20 by MichaIng