MichaIng/DietPi #6187: Both `.config` and `.local` in `/home/dietpi` are owned by root

slashtechno, 2023-02-26:
#### System Information
- DietPi version | `8.14.2`
- Distro version | `bullseye 0`
- Kernel version | `5.15.84-v8+`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | Canakit USB-C power supply 
- SD card used | Micro Center 64 GB

#### Additional Information
- This can be replicated on a fresh install.

#### Steps to reproduce

-  `ls -la /home/dietpi`

#### Expected behaviour

Expected output:  
```
total x
drwxr-xr-x 6 dietpi dietpi 4096 Feb 26 02:33 .
drwxr-xr-x 3 root   root   4096 Feb 25 20:48 ..
-rw------- 1 dietpi dietpi 4607 Feb 26 01:52 .bash_history
-rw-r--r-- 1 dietpi dietpi  220 Mar 27  2022 .bash_logout
-rw-r--r-- 1 dietpi dietpi  571 Feb 13 20:34 .bashrc
drwxr-xr-x 3 dietpi dietpi 4096 Feb 26 02:01 .cache
drwxr-xr-x 4 dietpi dietpi 4096 Feb 26 02:15 .config
drwxr-xr-x 3 dietpi dietpi 4096 Feb 26 01:13 .local
-rw-r--r-- 1 dietpi dietpi  161 Feb 13 20:34 .profile
drwx------ 2 dietpi dietpi 4096 Feb 25 22:43 .ssh
```

#### Actual behaviour  

Both `.local` and `.config` are owned by root, and not `dietpi`. I am unsure if this is intentional, but it causes problem when installing some software as `dietpi`.
```
drwxr-xr-x 6 dietpi dietpi 4096 Feb 26 02:05 .
drwxr-xr-x 3 root   root   4096 Feb 25 20:48 ..
-rw------- 1 dietpi dietpi 4607 Feb 26 01:52 .bash_history
-rw-r--r-- 1 dietpi dietpi  220 Mar 27  2022 .bash_logout
-rw-r--r-- 1 dietpi dietpi  571 Feb 13 20:34 .bashrc
drwxr-xr-x 3 dietpi dietpi 4096 Feb 26 02:01 .cache
drwxr-xr-x 3 root   root   4096 Feb 26 01:33 .config
drwxr-xr-x 3 root   root   4096 Feb 26 01:13 .local
-rw-r--r-- 1 dietpi dietpi  161 Feb 13 20:34 .profile
drwx------ 2 dietpi dietpi 4096 Feb 25 22:43 .ssh
```

#### Solution  

I fixed the issue with the following comannds:
```sh
sudo chown -R dietpi:dietpi /home/dietpi/.local
sudo chown -R dietpi:dietpi /home/dietpi/.config
```

MichaIng, 2023-02-26:
Many thanks for your report.

Those two directories do not exist on a fresh image. Which software did you install?

slashtechno, 2023-02-26:
I noticed this problem after attempting to run `code-server` via Docker. Looking back, this might have been a result of using `sudo` when mounting volumes to a Docker container. I thought this was a part of DietPi as I installed the operating system the same day, apologies.

Perhaps the Docker install script could run `sudo usermod -aG docker $USER`?

MichaIng, 2023-02-26:
And which user should $USER be? As dietpi-software requires root permissions, it is always root there, which of course does not need to be in any group.

When using sudo, never use -E option, and never use su, to assure the $HOME variable is always reset to the target users home, i.e. root. Passing/preserving the original users environment variables is a dangerous thing and potential security leak, aside of causing such kind of issues 😉.

slashtechno, 2023-02-26:
Okay, thank you.

closed 2023-02-26 by slashtechno