MichaIng/DietPi #6192: DietPi-Software | K3s: Allow to install in agent mode

MichaIng, 2023-02-26:
https://dietpi.com/forum/t/is-anyone-running-dietpi-on-a-pine64-soquartz/14852/39

hughsheehy, 2023-03-30:
Just commented on the bottom of that thread. 

https://dietpi.com/forum/t/is-anyone-running-dietpi-on-a-pine64-soquartz/14852/45?u=hughsheehy

hughsheehy, 2023-03-30:
Happy to help on making it work too!

MichaIng, 2023-04-29:
Are you on latest kernel (DietPi version)? The `conntrack` and `comment` extensions should actually be enabled, or did I miss some needed flag? https://github.com/MichaIng/DietPi/blob/dev/.build/images/Quartz64/quartz64_defconfig#L136

The actual error however might to be
```
error appending rule: exit status 4: Ignoring deprecated --wait-interval option.`
```
This option indeed has been removed with iptables v1.8.8. I see it here: https://manpages.debian.org/bullseye/iptables
But not here: https://manpages.debian.org/bookworm/iptables