MichaIng/DietPi #6193: Invalid volume level and M.A.L.P. cannot connect

yu13740000, 2023-02-27:
armbian 6.1.14 Debian 11 - DietPi v8.14.2 

[image]

MichaIng, 2023-02-27:
Can you give some more details about what you are trying, what M.A.L.P. is (the MPD client?), and fill the bug report issue template please.

yu13740000, 2023-02-27:
https://f-droid.org/en/packages/org.gateshipone.malp/

yu13740000, 2023-02-27:
[image]

yu13740000, 2023-02-27:
root@DietPi:~# journalctl -u mpd
-- Journal begins at Mon 2023-02-27 13:17:11 GMT, ends at Mon 2023-02-27 14:43:41 GMT. --
Feb 27 13:17:16 DietPi systemd[1]: Starting Music Player Daemon...
Feb 27 13:17:18 DietPi systemd[1]: Started Music Player Daemon.
Feb 27 13:17:19 DietPi mpd[843]: avahi: Service 'Music Player @ DietPi' successfully established.
Feb 27 14:06:27 DietPi mpd[843]: alsa_output: Decoder is too slow; playing silence to avoid xrun
Feb 27 14:06:32 DietPi mpd[843]: alsa_output: Decoder is too slow; playing silence to avoid xrun
Feb 27 14:06:37 DietPi mpd[843]: alsa_output: Decoder is too slow; playing silence to avoid xrun
Feb 27 14:06:42 DietPi mpd[843]: alsa_output: Decoder is too slow; playing silence to avoid xrun
Feb 27 14:06:47 DietPi mpd[843]: alsa_output: Decoder is too slow; playing silence to avoid xrun
Feb 27 14:06:52 DietPi mpd[843]: alsa_output: Decoder is too slow; playing silence to avoid xrun
Feb 27 14:06:57 DietPi mpd[843]: alsa_output: Decoder is too slow; playing silence to avoid xrun
Feb 27 14:07:03 DietPi mpd[843]: alsa_output: Decoder is too slow; playing silence to avoid xrun
Feb 27 14:19:39 DietPi systemd[1]: Stopping Music Player Daemon...
Feb 27 14:19:39 DietPi mpd[843]: player: played "http://asiafm.hk:8000/asiahd"
Feb 27 14:19:39 DietPi systemd[1]: mpd.service: Succeeded.
Feb 27 14:19:39 DietPi systemd[1]: Stopped Music Player Daemon.
Feb 27 14:19:39 DietPi systemd[1]: mpd.service: Consumed 1min 28.293s CPU time.
Feb 27 14:19:39 DietPi systemd[1]: Starting Music Player Daemon...
Feb 27 14:19:40 DietPi systemd[1]: Started Music Player Daemon.
Feb 27 14:19:41 DietPi mpd[2512]: avahi: Service 'Music Player @ DietPi' successfully established.
Feb 27 14:19:57 DietPi systemd[1]: Stopping Music Player Daemon...
Feb 27 14:19:57 DietPi mpd[2512]: player: played "http://asiafm.hk:8000/asiahd"
Feb 27 14:19:57 DietPi systemd[1]: mpd.service: Succeeded.
Feb 27 14:19:57 DietPi systemd[1]: Stopped Music Player Daemon.
Feb 27 14:19:57 DietPi systemd[1]: mpd.service: Consumed 4.203s CPU time.
Feb 27 14:19:57 DietPi systemd[1]: Starting Music Player Daemon...
Feb 27 14:19:58 DietPi systemd[1]: Started Music Player Daemon.
Feb 27 14:19:59 DietPi mpd[2645]: avahi: Service 'Music Player @ DietPi' successfully established.
Feb 27 14:20:10 DietPi systemd[1]: Stopping Music Player Daemon...
Feb 27 14:20:10 DietPi mpd[2645]: player: played "http://asiafm.hk:8000/asiahd"
Feb 27 14:20:10 DietPi systemd[1]: mpd.service: Succeeded.
Feb 27 14:20:10 DietPi systemd[1]: Stopped Music Player Daemon.
Feb 27 14:20:10 DietPi systemd[1]: mpd.service: Consumed 3.158s CPU time.
Feb 27 14:20:10 DietPi systemd[1]: Starting Music Player Daemon...
Feb 27 14:20:11 DietPi systemd[1]: Started Music Player Daemon.
Feb 27 14:20:11 DietPi mpd[2790]: avahi: Service 'Music Player @ DietPi' successfully established.
Feb 27 14:20:52 DietPi systemd[1]: Stopping Music Player Daemon...
Feb 27 14:20:52 DietPi mpd[2790]: player: played "http://asiafm.hk:8000/asiahd"
Feb 27 14:20:52 DietPi systemd[1]: mpd.service: Succeeded.
Feb 27 14:20:52 DietPi systemd[1]: Stopped Music Player Daemon.
Feb 27 14:20:52 DietPi systemd[1]: mpd.service: Consumed 6.131s CPU time.
Feb 27 14:20:52 DietPi systemd[1]: Starting Music Player Daemon...
Feb 27 14:20:53 DietPi systemd[1]: Started Music Player Daemon.
Feb 27 14:20:54 DietPi mpd[3070]: avahi: Service 'Music Player @ DietPi' successfully established.
Feb 27 14:20:54 DietPi mpd[3070]: exception: Failed to decode http://asiafm.hk:8000/asiahd; Not an AAC stream
Feb 27 14:20:54 DietPi mpd[3070]: player: played "http://asiafm.hk:8000/asiahd"
Feb 27 14:21:13 DietPi systemd[1]: Stopping Music Player Daemon...
Feb 27 14:21:13 DietPi mpd[3070]: player: played "http://asiafm.hk:8000/asiahd"
Feb 27 14:21:13 DietPi systemd[1]: mpd.service: Succeeded.
Feb 27 14:21:13 DietPi systemd[1]: Stopped Music Player Daemon.
Feb 27 14:21:13 DietPi systemd[1]: mpd.service: Consumed 2.406s CPU time.
Feb 27 14:21:13 DietPi systemd[1]: Starting Music Player Daemon...
Feb 27 14:21:14 DietPi systemd[1]: Started Music Player Daemon.
Feb 27 14:21:15 DietPi mpd[3204]: avahi: Service 'Music Player @ DietPi' successfully established.
Feb 27 14:22:17 DietPi systemd[1]: Stopping Music Player Daemon...
Feb 27 14:22:17 DietPi mpd[3204]: player: played "http://asiafm.hk:8000/asiahd"
Feb 27 14:22:17 DietPi systemd[1]: mpd.service: Succeeded.
Feb 27 14:22:17 DietPi systemd[1]: Stopped Music Player Daemon.
Feb 27 14:22:17 DietPi systemd[1]: mpd.service: Consumed 6.974s CPU time.
Feb 27 14:22:17 DietPi systemd[1]: Starting Music Player Daemon...
Feb 27 14:22:18 DietPi systemd[1]: Started Music Player Daemon.
Feb 27 14:22:19 DietPi mpd[3549]: avahi: Service 'Music Player @ DietPi' successfully established.
Feb 27 14:22:49 DietPi systemd[1]: Stopping Music Player Daemon...
Feb 27 14:22:49 DietPi mpd[3549]: player: played "http://asiafm.com.cn:8000/myradio"
Feb 27 14:22:49 DietPi systemd[1]: mpd.service: Succeeded.
Feb 27 14:22:49 DietPi systemd[1]: Stopped Music Player Daemon.
Feb 27 14:22:49 DietPi systemd[1]: mpd.service: Consumed 4.262s CPU time.
Feb 27 14:22:49 DietPi systemd[1]: Starting Music Player Daemon...
Feb 27 14:22:50 DietPi systemd[1]: Started Music Player Daemon.
Feb 27 14:22:51 DietPi mpd[3689]: avahi: Service 'Music Player @ DietPi' successfully established.
root@DietPi:~#

yu13740000, 2023-02-27:
Can you reach the O!MPD web interface from the phone and play music from there?

The webpage can be remote controlled, but the M.A.L.P. apk cannot

MichaIng, 2023-02-27:
MPD by default listens on localhost only. Change `/etc/mpd.conf`.
- If you want to access it from LAN only, change `bind_to_address "localhost"` to `bind_to_address "192.168.30.245"`, assuming that this is a static IP address.
- If you need to access it remotely from outside your LAN as well, change it to `bind_to_address "0.0.0.0"`. Also set the `password` setting so that not the whole world can control your MPD. But I guess there is rarely a use case for remote MPD control.

yu13740000, 2023-02-28:
succeed your way，Thanks！

closed 2023-02-28 by yu13740000

yu13740000, 2023-02-28:
[image]
Also the volume adjustment is invalid

MichaIng, 2023-02-28:
You mean that volume handles to not work? MPD uses a set of default ALSA mixer controls. It seems based on your hardware and kernel, there is no standard volume control. You can check this with:
```sh
alsamixer
```
or
```
amixer scontrols
```

yu13740000, 2023-02-28:
[image]
[image]
No change in volume adjustment。

MichaIng, 2023-02-28:
> Dietpi 7.93 volume adjustment is normal

But not with Armbian's Linux 6.1. The Linux version is relevant here, not the DietPi or even the Debian version.

yu13740000, 2023-02-28:
Google Translate may be inaccurate, and I don’t understand Linux very well. I just found a bug in the volume adjustment of the newly installed dietpi, mpd, and ompd. Thank you for your work, it is very useful, I like it very much!

MichaIng, 2023-02-28:
> I just found a bug in the volume adjustment of the newly installed dietpi, mpd, and ompd

As said, this is not a bug in DietPi (the distro/userland part), MPD or O!MPD, but a bug in the kernel package, which is maintained by Armbian. There has been a major Linux upgrade recently, and Armbian kernel's are sadly known to break features. Exactly with recent `meson64` board kernel (the one you use), audio is known to be often broken unless you do manual ALSA mixer adjustments. See the issue/requests about Odroid C4 here: #6178
Interestingly also on Odroid C4 there is no volume control available. Can be checked with:
```sh
amixer scontents
```

yu13740000, 2023-03-01:
Got it, thanks a lot! Then wait for Linux to fix it!

MichaIng, 2023-03-01:
It is likely not on Linux (Foundation) to fix it, but on Armbian. I guess it is a faulty patch while it works on mainline kernel. Which SBC do you actually use?

MPD can be actually configured to use a specific control for the volume: https://stackoverflow.com/questions/22283154/configuring-volume-control-for-usb-dac-to-be-used-via-mpd
So the outputs of the `amixer scontents` might help to achieve this, if there is a volume control, just with an uncommon name.

yu13740000, 2023-03-28:
Is there a list of usb wireless network card drivers?
[image]
rtl8821cu cannot drive

MichaIng, 2023-03-28:
As it is completely unrelated to the original topic, can you please open a new issue and fill the bug report template? It depends on the SBC, as this drivers for these newer Realtek chips are not in mainline kernel.