MichaIng/DietPi #6194: OctoPrint not working on fresh DietPi install

MJxMRB, 2023-02-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye

- Kernel version | `uname -a`
Linux DietPi 5.15.93-rockchip64 #23.02.2 SMP PREEMPT Fri Feb 17 23:48:36 UTC 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
ROCK Pi S (aarch64)

- Power supply used | (EG: 5V 1A RAVpower) 
USB C 
- SD card used | (EG: SanDisk ultra)
Kingston 32GB class 10

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) OctoPrint
- Was the software title installed freshly or updated/migrated? Installed freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Fresh dietpi install on rock pi S 
2. octoprint install from dietpi-software
3. Open in browser [yourIP]:5001

#### Expected behaviour
It should start octoprint wizard- ...

#### Actual behaviour
Not working...

#### Extra details
Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application. -message from connecting in browser.

This case is very urgent to me. Please help me to solve the issue. Thanks in advance.

MichaIng, 2023-02-28:
> OSError: [Errno 98] Address already in use

There seem to be an instance running already, respectively the port is used already:
```sh
ss -tlpn
```

MJxMRB, 2023-03-01:
> 

[image]

Joulinar, 2023-03-01:
can you reboot your system and check the system log

```
journalctl -u octoprint
```

if possible, try to avoid creating screen prints. You should be able to copy the whole output from SSH terminal directly.

MichaIng, 2023-03-01:
Works fine here on my ROCK Pi S Bullseye system. Looks like an issue with the automatic language selection in your case. This is selected based on the browser language. Which one is it in your case, and does it work when you change it to e.g. English? This `best_match(LANGUAGES)` seems to not find any match in your case, and a sanity check + fallback to English seems to be missing, to that it returns an exception instead. Might be simple to fix this at OctoPrint end.

MJxMRB, 2023-03-01:
> Works fine here on my ROCK Pi S Bullseye system. Looks like an issue with the automatic language selection in your case. This is selected based on the browser language. Which one is it in your case, [...]

You are completely right, thank you. It was browser issue (Brave). On chrome it loaded. Both have polish language.

Joulinar, 2023-03-01:
Ok I'm going to close the issue for now, feel free to reopen if needed.

closed 2023-03-01 by Joulinar

MichaIng, 2023-03-03:
This has been fixed for next OctoPrint release 🙂: https://github.com/OctoPrint/OctoPrint/commit/588a1c4