MichaIng/DietPi #6196: dietpi "cpu" command request root privileges since release v8.14.2

hansratzinger, 2023-03-01:
### Creating a bug report/issue
dietpi "cpu" command request root privileges since release v8.14.2

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='not applied'
G_LIVE_PATCH_STATUS[2]='not applied'
- Distro versionbullseye 0
- Kernel version | Linux Hermes 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used |15 W DC 5 V, 3 A

- SD card used | SanDisk

#### Steps to reproduce
After logging in via SSH and entering:
dietpi@Hermes:~$ cpu
[FAILED] DietPi-CPU_info | Root privileges required. Please run the command with "sudo" or "G_SUDO".

I`m using "cpu" to check the temperature, because the raspi is located in a distant location.

#### Expected behaviour
In previous DietPi versions Root privileges were NOT required using "cpu"

MichaIng, 2023-03-01:
> In previous DietPi versions Root privileges were NOT required using "cpu"

It was! This script was not touched since v8.4 (one year ago) and the root user check was implemented with DietPi v160 in 2017: https://github.com/MichaIng/DietPi/commit/874383bf3373e1cd105f4dd11176b07f1320797f#diff-4e91ffac40f5d47b68048c896a9dcc5cc97600aad10790f07a5dc786eab075e4

But actually I'm wondering if it is really needed. The related sysfs entries are usually world-readable.

hansratzinger, 2023-03-01:
My Raspi has been setted up in summer 2022 and I updated every release since then. I was checking the cpu temperature daily using "cpu" without "sudo". The user is dietpi. Only since about 2 weeks root privileges are requiered. There must be different reasons for this behavoir when root privileges are required since 2017.

MichaIng, 2023-03-01:
I have no idea how this would be possible. The check was always there and is effective. Probably you did actually login as root user?

However, I removed the root user requirement now, along with an overall cleanup of old unused code: https://github.com/MichaIng/DietPi/commit/0f0c7c3

closed 2023-03-04 by MichaIng