MichaIng/DietPi #6203: Unable to install Python3

Getafix17, 2023-03-03:
#### Details:
- Date           | Fri Mar  3 09:23:31 IST 2023
- DietPi version | v8.10.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero 2 W (armv7l) (ID=3)
- Kernel version | `Linux DietPiMEye 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `ping -6nc 1 -W 10 2620:fe::fe`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
PING 2620:fe::fe(2620:fe::fe) 56 data bytes
From fe80::eead:e0ff:fed0:5dbb%wlan0 icmp_seq=1 Destination unreachable: No route

--- 2620:fe::fe ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms
```

Joulinar, 2023-03-03:
Pls disable IPv6 within network settings.

Getafix17, 2023-03-03:
I checked in network adapters, IPv6 is off. Also apt upgrade is giving an error. Also Err:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf base-files armhf 11.1+rpi1+deb11u6
  Connection failed [IP: 103.195.68.3 80]

Joulinar, 2023-03-03:
Chech that you have a valid network connection.

closed 2023-03-03 by Getafix17