MichaIng/DietPi #6204: running lets encrypt doesnt work

nageshshetty, 2023-03-03:
#### Details:
- Date           | Fri Mar  3 11:25:53 GMT 2023
- DietPi version | v8.14.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux CT-MACHINA 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `a2enmod http2`
- Exit code      | 127
- Software title | DietPi-LetsEncrypt
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 858: a2enmod: command not found
```
---------------------------------------------------------------------
[FAILED] DietPi-LetsEncrypt | Unable to continue, DietPi-LetsEncrypt will now terminate.
 - Press any key to return to DietPi-Launcher...^[[B^[[D

 DietPi-LetsEncrypt
─────────────────────────────────────────────────────
 Mode: Running Certbot

[  OK  ] DietPi-LetsEncrypt | Apache webserver detected
[  OK  ] DietPi-LetsEncrypt | Desired setting in /etc/apache2/sites-available/000-default.conf was already set:        ServerName ct8.uk
[FAILED] DietPi-LetsEncrypt | a2enmod http2

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Fri Mar  3 11:26:26 GMT 2023
- DietPi version | v8.14.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux CT-MACHINA 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `a2enmod http2`
- Exit code      | 127
- Software title | DietPi-LetsEncrypt
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 858: a2enmod: command not found

Joulinar, 2023-03-03:
> a2enmod: command not found

hmm strange, usually `a2enmod` is part of the `apache2` package. Not sure how this could be missing. https://packages.debian.org/search?searchon=contents&keywords=a2enmod&mode=path&suite=stable&arch=any

Can you try to reinstall Apache2?

```
dietpi-software reinstall 83
```

MichaIng, 2023-04-03:
Marking as closed due to outstanding reply. Feel free to reopen if issue persists.

closed 2023-04-03 by MichaIng