MichaIng/DietPi #6205: Activate zswap by default

rekcodocker, 2023-03-03:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
The small devices on which DietPi runs do not have a lot of memory. Swapping is a solution but on SD cards this is gives new meaning to the word 'slow' and it wears out the SD card.

- ...

#### Describe the solution you'd like:
I would like to see zswap (not to be confused with zram!) enabled.
The module is available in the kernel and can be activated on boot; I tried it.
When this is added to /boot/cmdline.txt, zswap is activated:
`zswap.enabled=1`

#### Describe alternatives you've considered:
Most ideal would be if the initramfs would contain zstd and z3fold. 
Those modules make even better use of compressed swapspace. This would need to be added to the commandline:
`zswap.enabled=1 zswap.compressor=zstd zswap.zpool=z3fold`

#### Additional context
* Apparently zswap is better than zram. It does not block part of the ram.
* as I understand it, zswap still needs a disk-backend to work (zram does not). 
* I don't know if zram can act as this backend, it sounds like a snake biting its own tail but perhaps zswap can be in charge of the swapping and a minimal zram can be configured so swap runs with a ram-only configuration.
It would be nice to have the option of having no-SD-swap.
* reference to how this works on ubuntu-on-a-pi:  [(https://www.cnx-software.com/2022/01/13/ubuntu-22-04-zswap-raspberry-pi-4-2gb-ram/)]
* Explanation why z3fold is saving the world: [https://www.kernel.org/doc/html/v5.0/vm/z3fold.html]
* I don't know how to add the zstd and z3fold modules to initramfs on a raspberry - I am pretty sure the people who read this do.

Reference: #6041 is about zram and if this FeatureRequest is adressed, it touches on configuring zram.

MichaIng, 2023-03-03:
> Apparently zswap is better than zram.

That is not true, but it depends on the use case. zswap alone does not avoid disk swapping but is an additional RAM buffer for it. So without a disk swap, which one might not want to have at all, zswap is not used. zram is a way to practically increase the amount of data stored in RAM without any disk swap file. Which one you prefer, highly depends on idle and peak RAM usage vs amount of RAM.
EDIT: I should have read your other points, which state the same, invalidating your first statement 😛.

> I don't know how to add the zstd and z3fold modules to initramfs on a raspberry

RPi does not use/need an initramfs by default. It shouldn't be needed for zswap either, since the swap space is not mounted in initramfs stage, but in userspace, where the modules can be loaded already.
__________
Basically, with any compressed RAM space, you trade RAM size with CPU processing and I/O speed, since of course making use of the compression is/needs to start before the RAM is actually full. So I'm not yet sure whether it is a good default choice. But an option for it does definitely not hurt.

andiohn, 2025-04-24:
I would also put my voice in here, perhaps for it could be added to the diet-pi configuration tool in performance, that would be the best.

andiohn, 2025-04-24:
Ok, so just enabling it in the kernel using the boot/cmdline.txt file works great. It would be nice to have the z3fold and zstd options too but ya, things are working well over here, that I can see.

MichaIng, 2025-04-24:
Just add `zswap.compressor=zstd zswap.zpool=z3fold`.

But note that [z3fold](https://docs.kernel.org/mm/z3fold.html) has been deprecated: https://lore.kernel.org/all/20240904233343.933462-1-yosryahmed@google.com/T/#u

andiohn, 2025-04-25:
I get this on boot, so zstd isn't available. 

[    2.053100] zswap: compressor zstd not available, using default lzo
[    2.053239] zswap: zpool z3fold not available, using default zsmalloc
[    2.053611] zswap: loaded using pool lzo/zsmalloc

MichaIng, 2025-04-27:
Hmm, there is no kernel (build) config to enable/disable any compression and zpool algorithm 🤔. Can you check whether the `zstd` kernel module is loaded?
```sh
lsmod
```
And does it work when you load it manually, and then change the compressor?
```sh
modprobe zstd
echo zstd > /sys/module/zswap/parameters/compressor
```
... actually makes sense that it does not work via cmdline as long as `zstd` is a module (which it is) instead of builtin. Either the module(s) need(s) to be loaded via initramfs (which is not used on RPi by default), or the settings have to be changed in userland. Could be done with a systemd service:
```sh
cat << '_EOF_' > /etc/systemd/system/zswap.service
[Unit]
Description=Configure zswap

[Service]
Type=oneshot
ExecStart=/bin/dash -c 'echo zstd > /sys/module/zswap/parameters/compressor; echo 1 > /sys/module/zswap/parameters/enabled'

[Install]
WantedBy=multi-user.target
_EOF_
systemctl daemon-reload
systemctl enable zswap
```