MichaIng/DietPi #6207: Encountered Error

kmreed55, 2023-03-03:
---------------------------------------------------------------------
#### Details:
- Date           | Fri Mar  3 10:40:53 PST 2023
- DietPi version | v8.14.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux reedsl-net-svcs01 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 1
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.
```
---------------------------------------------------------------------

Joulinar, 2023-03-03:
Are you using any onw time sync service? Or did you have done an upgrade from Buster to Bullseye recently?

kmreed55, 2023-03-03:
Yes, I've setup ntpd on this machine to use as a local ntp server. Is there
an easier way to do this? I couldn't find any info on making dietpi a ntp
server.
Thanks,
Kyle

MichaIng, 2023-03-03:
> Yes, I've setup ntpd on this machine to use as a local ntp server

Then disable the DietPi/systemd time sync, as it is conflicting, e.g. via `dietpi-config` Advanced Options.

You can setup an NTP server on DietPi like on every other Debian or Ubuntu system. Have a look into `/etc/ntp.conf` and `/etc/default/ntp`. If the daemon listens on port 123, it can be used by other clients:
```sh
ss -tulpn | grep ':123'
```

kmreed55, 2023-03-03:
Thanks so much, I appreciate it. I will give it a try.
Kyle

MichaIng, 2023-04-03:
Marking as closed as the original error should be solved. Feel free to reopen if required.

closed 2023-04-03 by MichaIng