MichaIng/DietPi #6208: [DietpPi-RAMlog] /run/log/ (system.journal) in RAM?

GvY85, 2023-03-04:
Hello,

With DietPi-RAMlog option 1, are the files in `/run/log/` also stored in RAM? I am asking because I am getting messages from `systemd-journald` that the log is filling and needs rotating.
This is probably due to `auditd` or `apparmor` but I have to find out how  to look into that.
In the meantime, I would like to know if this is just happening in RAM or that my eMMC is being overwritten constantly which is more serious.

Also, any tips on how to debug?

Thanks!

GvY85, 2023-03-04:
Found out it are indeed logs from `apparmor`. No problem but can I make sure `/run/log/` is kept in RAM or can I change the log location of `journalctl` from `/run/log/` to `/var/log`?

GvY85, 2023-03-04:
In `/etc/systemd/journald.conf` I see the option `#Storage=auto`

```
`Storage=
Controls where to store journal data. One of "volatile", "persistent", "auto" and "none". If "volatile", journal log data will be stored only in memory, i.e. below the /run/log/journal hierarchy (which is created if needed). If "persistent", data will be stored preferably on disk, i.e. below the /var/log/journal hierarchy (which is created if needed), with a fallback to /run/log/journal (which is created if needed), during early boot and if the disk is not writable. "auto" behaves like "persistent" if the /var/log/journal directory exists, and "volatile" otherwise (the existence of the directory controls the storage mode). "none" turns off all storage, all log data received will be dropped (but forwarding to other targets, such as the console, the kernel log buffer, or a syslog socket will still work). Defaults to "auto" in the default journal namespace, and "persistent" in all others.

Note that journald will initially use volatile storage, until a call to journalctl --flush (or sending SIGUSR1 to journald) will cause it to switch to persistent logging (under the conditions mentioned above). This is done automatically on boot via "systemd-journal-flush.service".

Note that when this option is changed to "volatile", existing persistent data is not removed. In the other direction, [journalctl(1)](https://www.freedesktop.org/software/systemd/man/journalctl.html#) with the --flush option may be used to move volatile data to persistent storage.

When journal namespacing (see LogNamespace= in [systemd.exec(5)](https://www.freedesktop.org/software/systemd/man/systemd.exec.html#)) is used, setting Storage= to "volatile" or "auto" will not have an effect on the creation of the per-namespace logs directory in /var/log/journal/, as the systemd-journald@.service service file by default carries LogsDirectory=. To turn that off, add a unit file drop-in file that sets LogsDirectory= to an empty string.
```

Should this be set to `Storage=volatile` to prevent logs being written to `/run/log/` but only to '/var/log/`?

GvY85, 2023-03-04:
Ah, found out that `/run` is mounted as `tmpfs` anyway.

closed 2023-03-04 by GvY85

MichaIng, 2023-03-04:
Yes, `/run` implicitly is a tmpfs on Linux systems.