MichaIng/DietPi #6209: MiniDLNA Media Path Bug

AdamF100001, 2023-03-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | v8.14.2
- Kernel version | Linux DietPi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
- SBC model | Raspberry Pi 4 4GB Model 
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Mini DLNA)
- Was the software title installed freshly or updated/migrated? Updated
- Can this issue be replicated on a fresh installation of DietPi? Yes
 13d7588b-63bd-4089-b2d6-df575745ccad

#### Steps to reproduce

1. ... Uninstall minidlna server 
2. ... Reinstall minidlna sever 
3. ... Run sudo nano /etc/minidlna.conf
3. ... Remove media_dir=V,/mnt/dietpi_userdata/Video 
4. ... Add custom path media_dir=V,/mnt/2TB_SSD/dietpi_userdata/Video
5. ... Save config file
6. ... Reboot system 

#### Expected behaviour

Custom media path media_dir=V,/mnt/2TB_SSD/dietpi_userdata/Video

Should be added by minidlna and be visible by media network share in VLC

#### Actual behaviour
On reboot custom media path that was added has reverted back to default path media_dir=V,/mnt/dietpi_userdata/Video

#### Extra details

I was watching a video file last night and  the minidlna server on the Raspberry Pi crashed.

Reboot pi and amazon fire tv box several times. Tried backup restore still have the issue. Uninstalled and reinstalled minidlna server on pi not fixed issue.

Thanks

Joulinar, 2023-03-04:
I'm not able to replicate your behavior. Changes done to `/etc/minidlna.conf` stay persistent on my system. Can you try following:

```
G_CONFIG_INJECT 'media_dir=V' 'media_dir=V,/mnt/2TB_SSD/dietpi_userdata/Video' /etc/minidlna.conf
cat /etc/minidlna.conf | grep media_dir
reboot
cat /etc/minidlna.conf | grep media_dir
```

One more question: Did you moved `dietpi_userdata` to the external drive? Can you show

```
ls -la /mnt/
df -h
lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
```

AdamF100001, 2023-03-04:
Please resolve the ticket, fixed it now. Thanks

Joulinar, 2023-03-04:
Maybe you like to share what you have done.

AdamF100001, 2023-03-04:
I just re-did the process as before and it worked this time.

Joulinar, 2023-03-04:
Ok I'm going to close the issue for now. Feel free to reopen if needed.

closed 2023-03-04 by Joulinar