MichaIng/DietPi #6212: RISC-V and VisionFive 2 testing issue

MichaIng, 2023-03-04:
Some software titles may not be supported on RISC-V in general, or because no RISC-V binaries or packages are provided yet. Most of these cases have been identified already and related software titles have been disabled for the RISC-V architecture for now.

Other software titles may only be unsupported on the particular VisionFive 2 SBC, or the kernel build we use. This is known for Docker, likely other container engines, the NFS server and the SMB client, but there may be more. We aim to generate own kernel builds, with added features the next week(s). Any help to create an rebase a fork on latest upstream Linux 5.15 is highly appreciated.

Again other software titles may not be supported with latest PHP, Python and other runtime environments and libraries, provided by Debian Sid. The RISC-V architecture is not yet officially supported by Debian, but only with the dedicated [Debian ports](https://www.debian.org/ports/) repository and Debian Sid/unstable suite. Currently, this mostly equals Bookworm, which is the reason DietPi identifies as this. A list of software titles tested on Bookworm can be found here: https://github.com/MichaIng/DietPi/wiki/Debian-Bookworm-testing

lgray, 2023-03-10:
Hi, just updated the bootloader and got dietpi running on my VF2, thanks a lot for this! One first major issue is that I only see half the amount of RAM that's available on my board (8 GB model). If I use the starfivetech debian image the full amount shows up, but both armbian and dietpi show half that! Any pointers would be greatly appreciated!

MichaIng, 2023-03-10:
Have you updated the SPI bootloader? Some open issues state that it has been fixed since a while:
```sh
apt install mtd-utils
curl -fO 'https://github.com/starfive-tech/VisionFive2/releases/download/VF2_v2.10.4/u-boot-spl.bin.normal.out'
curl -fO 'https://github.com/starfive-tech/VisionFive2/releases/download/VF2_v2.10.4/visionfive2_fw_payload.img'
flashcp -v u-boot-spl.bin.normal.out /dev/mtd0
flashcp -v visionfive2_fw_payload.img /dev/mtd1
rm u-boot-spl.bin.normal.out visionfive2_fw_payload.img

lgray, 2023-03-10:
Yes I updated the bootloader before installing dietpi. What's the best way I can check the versions of uboot / fw-payload just to make sure?

MichaIng, 2023-03-10:
Ah okay, indeed not fixed when not using the 4 partitions layout and `uEnv.txt` to run different boot commands: https://github.com/starfive-tech/VisionFive2/issues/20#issuecomment-1426529552

I'll check where the U-Boot environment is located, so we can adjust it to apply the needed `visionfive2_mem_set` and also enable eMMC boot (which AFAIK currently does not work).

MichaIng, 2023-03-13:
Many thanks for testing and reporting. Did you test both Ethernet ports? It works fine on revision 1.3 here, but not sure whether on both Ethernet ports. @StephanStS could you test switching the port, after
```sh
sed -i 's/eth0/eth1/' /etc/network/interfaces
```

congocongo, 2023-03-13:
A second eth is YT8512B, not YT8531 on rev 1.2a

```
Linux DietPi 5.15.98 #1 SMP Fri Mar 10 18:44:22 CET 2023 riscv64 GNU/Linux

[    0.513157] starfive-eth-plat 16030000.ethernet: User ID: 0x41, Synopsys ID: 0x52
[    0.513196] starfive-eth-plat 16030000.ethernet: 	DWMAC4/5
[    0.513219] starfive-eth-plat 16030000.ethernet: DMA HW capability register supported
[    0.513245] starfive-eth-plat 16030000.ethernet: RX Checksum Offload Engine supported
[    0.513271] starfive-eth-plat 16030000.ethernet: Wake-Up On Lan supported
[    0.513363] starfive-eth-plat 16030000.ethernet: TSO supported
[    0.513387] starfive-eth-plat 16030000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    0.513416] starfive-eth-plat 16030000.ethernet: Enabled L3L4 Flow TC (entries=1)
[    0.513444] starfive-eth-plat 16030000.ethernet: Enabled RFS Flow TC (entries=8)
[    0.513470] starfive-eth-plat 16030000.ethernet: TSO feature enabled
[    0.513492] starfive-eth-plat 16030000.ethernet: Using 40 bits DMA width
[    0.746781] starfive-eth-plat 16040000.ethernet: User ID: 0x41, Synopsys ID: 0x52
[    0.746818] starfive-eth-plat 16040000.ethernet: 	DWMAC4/5
[    0.746840] starfive-eth-plat 16040000.ethernet: DMA HW capability register supported
[    0.746866] starfive-eth-plat 16040000.ethernet: RX Checksum Offload Engine supported
[    0.746891] starfive-eth-plat 16040000.ethernet: Wake-Up On Lan supported
[    0.746975] starfive-eth-plat 16040000.ethernet: TSO supported
[    0.746997] starfive-eth-plat 16040000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    0.747025] starfive-eth-plat 16040000.ethernet: Enabled L3L4 Flow TC (entries=1)
[    0.747052] starfive-eth-plat 16040000.ethernet: Enabled RFS Flow TC (entries=8)
[    0.747079] starfive-eth-plat 16040000.ethernet: TSO feature enabled
[    0.747100] starfive-eth-plat 16040000.ethernet: Using 40 bits DMA width
[   11.065020] starfive-eth-plat 16030000.ethernet eth0: PHY [stmmac-0:00] driver [YT8531 Gigabit Ethernet] (irq=POLL)
[   11.065868] starfive-eth-plat 16030000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   11.066855] starfive-eth-plat 16030000.ethernet eth0: No Safety Features support found
[   11.066907] starfive-eth-plat 16030000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   11.068055] starfive-eth-plat 16030000.ethernet eth0: configuring for phy/rgmii-id link mode
[   14.364477] starfive-eth-plat 16030000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[  233.172002] starfive-eth-plat 16030000.ethernet eth0: Link is Down
[  246.153305] starfive-eth-plat 16030000.ethernet eth0: PHY [stmmac-0:00] driver [YT8531 Gigabit Ethernet] (irq=POLL)
[  246.153580] starfive-eth-plat 16030000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[  246.153882] starfive-eth-plat 16030000.ethernet eth0: No Safety Features support found
[  246.153901] starfive-eth-plat 16030000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[  246.153917] starfdive-eth-plat 16030000.ethernet eth0: configuring for phy/rgmii-id link mode

root@DietPi:/etc/network# ethtool eth0
Settings for eth0:
	Supported ports: [ TP	 MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Auto-negotiation: on
	master-slave cfg: preferred slave
	master-slave status: slave
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
	MDI-X: Unknown
	Supports Wake-on: ug
	Wake-on: d
        Current message level: 0x0000003f (63)
                               drv probe link timer ifdown ifup
	Link detected: yes

root@DietPi:/etc/network# ethtool eth1
netlink error: Device or resource busy
netlink error: Device or resource busy
netlink error: Device or resource busy
netlink error: Device or resource busy
netlink error: Device or resource busy
No data available

root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 6c:cf:39:00:21:83 brd ff:ff:ff:ff:ff:ff
    inet 192.168.88.60/24 brd 192.168.88.255 scope global eth0
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 6c:cf:39:00:21:84 brd ff:ff:ff:ff:ff:ff

root@DietPi:~# ip r
default via 192.168.88.1 dev eth0 onlink 
192.168.88.0/24 dev eth0 proto kernel scope link src 192.168.88.60 

root@DietPi:~# ping 192.168.88.1
PING 192.168.88.1 (192.168.88.1) 56(84) bytes of data.
From 192.168.88.60 icmp_seq=1 Destination Host Unreachable
From 192.168.88.60 icmp_seq=2 Destination Host Unreachable
From 192.168.88.60 icmp_seq=3 Destination Host Unreachable
^C
--- 192.168.88.1 ping statistics ---
5 packets transmitted, 0 received, +3 errors, 100% packet loss, time 4120ms
pipe 3
```

congocongo, 2023-03-13:
compare it with vendor's build:

```
Linux starfive 5.15.0-starfive #1 SMP Mon Feb 27 14:03:14 EST 2023 riscv64 GNU/Linux

[    1.278085] starfive-eth-plat 16030000.ethernet: force_sf_dma_mode is ignored if force_thresh_dma_mode is set.
[    1.289466] starfive-eth-plat 16030000.ethernet: User ID: 0x41, Synopsys ID: 0x52
[    1.297753] starfive-eth-plat 16030000.ethernet: 	DWMAC4/5
[    1.303804] starfive-eth-plat 16030000.ethernet: DMA HW capability register supported
[    1.312449] starfive-eth-plat 16030000.ethernet: RX Checksum Offload Engine supported
[    1.321094] starfive-eth-plat 16030000.ethernet: Wake-Up On Lan supported
[    1.328585] starfive-eth-plat 16030000.ethernet: TSO supported
[    1.335016] starfive-eth-plat 16030000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    1.344241] starfive-eth-plat 16030000.ethernet: Enabled Flow TC (entries=1)
[    1.352023] starfive-eth-plat 16030000.ethernet: TSO feature enabled
[    1.359036] starfive-eth-plat 16030000.ethernet: Using 40 bits DMA width
[    1.647333] starfive-eth-plat 16040000.ethernet: force_sf_dma_mode is ignored if force_thresh_dma_mode is set.
[    1.658771] starfive-eth-plat 16040000.ethernet: User ID: 0x41, Synopsys ID: 0x52
[    1.667055] starfive-eth-plat 16040000.ethernet: 	DWMAC4/5
[    1.673105] starfive-eth-plat 16040000.ethernet: DMA HW capability register supported
[    1.681747] starfive-eth-plat 16040000.ethernet: RX Checksum Offload Engine supported
[    1.690393] starfive-eth-plat 16040000.ethernet: Wake-Up On Lan supported
[    1.697881] starfive-eth-plat 16040000.ethernet: TSO supported
[    1.704313] starfive-eth-plat 16040000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    1.713547] starfive-eth-plat 16040000.ethernet: Enabled Flow TC (entries=1)
[    1.721327] starfive-eth-plat 16040000.ethernet: TSO feature enabled
[    1.728342] starfive-eth-plat 16040000.ethernet: Using 40 bits DMA width
[    6.890982] starfive-eth-plat 16030000.ethernet end0: renamed from eth0
[    6.946865] starfive-eth-plat 16040000.ethernet end1: renamed from eth1
[   13.887244] starfive-eth-plat 16030000.ethernet end0: PHY [stmmac-0:00] driver [YT8531 Gigabit Ethernet] (irq=POLL)
[   13.898627] starfive-eth-plat 16030000.ethernet end0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   13.912425] starfive-eth-plat 16030000.ethernet end0: No Safety Features support found
[   13.920408] starfive-eth-plat 16030000.ethernet end0: IEEE 1588-2008 Advanced Timestamp supported
[   13.943766] starfive-eth-plat 16030000.ethernet end0: configuring for phy/rgmii-id link mode
[   13.974442] starfive-eth-plat 16040000.ethernet end1: PHY [stmmac-1:00] driver [YT8512B Ethernet] (irq=POLL)
[   13.985089] starfive-eth-plat 16040000.ethernet end1: Register MEM_TYPE_PAGE_POOL RxQ-0
[   13.993446] starfive-eth-plat 16040000.ethernet end1: No Safety Features support found
[   13.993461] starfive-eth-plat 16040000.ethernet end1: IEEE 1588-2008 Advanced Timestamp supported
[   14.034533] starfive-eth-plat 16040000.ethernet end1: configuring for phy/rgmii-id link mode
[   18.088672] starfive-eth-plat 16030000.ethernet end0: Link is Up - 1Gbps/Full - flow control off

root@starfive:~# ethtool end0
Settings for end0:
	Supported ports: [ TP	 MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Auto-negotiation: on
	master-slave cfg: preferred slave
	master-slave status: slave
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
[... 34 lines cut ...]
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: end0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 6c:cf:39:00:21:83 brd ff:ff:ff:ff:ff:ff
    inet 192.168.88.60/24 brd 192.168.88.255 scope global dynamic noprefixroute end0
       valid_lft 540sec preferred_lft 540sec
3: end1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 6c:cf:39:00:21:84 brd ff:ff:ff:ff:ff:ff

root@starfive:~# ip r
default via 192.168.88.1 dev end0 proto dhcp src 192.168.88.60 metric 100 
192.168.88.0/24 dev end0 proto kernel scope link src 192.168.88.60 metric 100 

root@starfive:~# ping 192.168.88.1
PING 192.168.88.1 (192.168.88.1) 56(84) bytes of data.
64 bytes from 192.168.88.1: icmp_seq=1 ttl=64 time=0.360 ms
64 bytes from 192.168.88.1: icmp_seq=2 ttl=64 time=0.363 ms
64 bytes from 192.168.88.1: icmp_seq=3 ttl=64 time=0.397 ms
^C
--- 192.168.88.1 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2052ms
rtt min/avg/max/mdev = 0.360/0.373/0.397/0.016 ms
```

MichaIng, 2023-03-13:
And did you test using `eth1` on your image?

Yours and our logs match nearly 100%, the only difference is that somehow in your case `eth0` is identified as 100Mbps adapter and in our case as well as in your case with vendor kernel as 1Gbps adapter (which is of course correct). I have somehow the impression that there is an issue with dealing our link attributes with the router. Also the link goes down after some minutes, while is is up at first. Just to rule that out, does it help to detach and reattach the cable, then:
```sh
ifdown eth0
ifup eth0
sleep 5
ping 192.168.88.1
```
And another test: Does it help to add `stmmaceth=chain_mode:1` to the end of the kernel command-line (`append` line) in `/boot/extlinux/extlinux.conf` and reboot to apply the change? I didn't fully understand the purpose of this option and on our test system it didn't make any difference (I also ran benchmarks), so I removed it compared to StarFive's Debian images.

congocongo, 2023-03-13:
I just copied fresh device tree from vendor's image and now it works. Only first eth is avaliable, looks like there is no driver for YT8512B in kernel. rev 1.2a was supplied with YT8531(1Gb)/YT8512B(100Mb) PHY on board.

MichaIng, 2023-03-14:
Is this the same with the StarFive image?

Support for this chip was added here: https://github.com/MichaIng/linux/commit/e036d0777cbbc975615105f37b6d6b82257ed62d
Then, without removing the support declaration at the top of the file, inverted here: https://github.com/MichaIng/linux/commit/3c9ef2d35c69c6f388701bd4b019817d83f3cd99

These commits are from StarFive kernel repository, so should be the same there.

congocongo, 2023-03-15:
It works on Starfive. And I copied dtb from this release https://forum.rvspace.org/t/visionfive-2-debian-image-202302-released/2132

MichaIng, 2023-03-15:
I've no idea then. The device tree source and the Ethernet driver source are 100% identical and it works well on V1.3. If neither of both Ethernet adapters work with the device tree from our build, something in mainline Linux 5.15.y must have changed, ourside of those drivers and device tree, but affecting then on build time. And this must affect V1.2 only, V1.3 not, even for the Ethernet adapter which uses the same chip. Very strange.

Just to be sure: You copied the `jh7110-visionfive-v2.dtb` (and only this!) from the latest StarFive Debian image and installed it on our image to `/usr/lib/linux-image-visionfive2/starfive/jh7110-visionfive-v2.dtb`? Did you run a `diff` on them to see whether there is an actual difference? Not sure whether builds are reproducible and may contain compiler/environment related differences though.

And also please assure you can reliably trigger the issue by swapping the device trees back and forth and that it is not a coincidence while the issue is unrelated to the device tree, as searching through possibly affecting upstream 5.15.y commits is time consuming as I have no idea where to start 🤔.

MichaIng, 2023-03-15:
### I moved a new kernel build in place
Update can be applied via:
```sh
cd /tmp
curl -O 'https://dietpi.com/downloads/binaries/testing/linux-image-visionfive2.deb'
dpkg -i linux-image-visionfive2.deb
rm linux-image-visionfive2.deb
```
### Added builtin features
- Framebuffer device for console on HDMI
- Fixed DRM for X11/desktop sessions on HDMI
- IPv6
- zswap with zstd as default compression
- binfmt_misc: automatic emulation handler selection, e.g. using Box64 (if installed) when executing x86_64 binaries

### Added kernel modules
- F2FS and Brtfs filesystem types
- NFS kernel server
- SMB/CIFS client/mounts
- tun/OpenVPN
- WireGuard
- zram with zstd as default compression

### Turned builtins into modules
- Bluetooth
- WiFi
- Plan 9
- Camera drivers
- overlayfs
- exFAT
- NTFS
- NFS client/mounts
- nf_tables packet duplication and FIB lookups
- Goodix and Tinker FT5406 touchscreens

### Removed kernel modules
- StarFive mailbox

### Removed builtin features
- VirtIO drives, used by virtual machines
- `/dev/pty*` legacy pseudo terminal devices
- `/dev/hvc*` hypervisor virtual console devices

### Other changes
- Kernel modules are now xz-compressed
- Default CPU governor is now schedutil (only relevant until `dietpi-preboot` or `cpufrequtils` change it in userland)
- Default hostname is now `DietPi` (only relevant until the init system changes it in userland)
- Solved a kernel error at boot, happened as we do not use an initramfs
- Fixed U-Boot env SPI address in device tree, so `fw_printenv` now generally can find it at `/dev/mtd2`

Check out the source code here: https://github.com/MichaIng/linux
I'll keep rebasing all changes onto latest upstream Linux 5.15.y release as well merging commits from the StarFive repo.

congocongo, 2023-03-16:
Will try this kernel. Maybe better to wait when they send patches to upstream? They are actively do it now:

https://lore.kernel.org/lkml/?q=JH71x0

btw I decompiled dtb files:

```
--- jh7110-visionfive-v2.dtb.dietpi.dts	2023-03-16 16:26:13.199095247 +0300
+++ jh7110-visionfive-v2.dtb.starfive.dts	2023-03-16 16:28:22.569130546 +0300
@@ -347,18 +347,6 @@
 			phandle = <0x2f>;
 		};
 
-		multi-phyctrl@10210000 {
-			compatible = "starfive,phyctrl";
-			reg = <0x00 0x10210000 0x00 0x10000>;
-			phandle = <0x3a>;
-		};
-
-		pcie1-phyctrl@10220000 {
-			compatible = "starfive,phyctrl";
-			reg = <0x00 0x10220000 0x00 0x10000>;
-			phandle = <0x3f>;
-		};
-
 		stg_syscon@10240000 {
 			compatible = "syscon";
 			reg = <0x00 0x10240000 0x00 0x1000>;
@@ -414,7 +402,7 @@
 			#clock-cells = <0x01>;
 			power-domains = <0x1d 0x04>;
 			status = "okay";
-			phandle = <0x47>;
+			phandle = <0x45>;
 		};
 
 		clock-controller@19810000 {
@@ -822,7 +810,7 @@
 			};
 
 			pcie0_perst_default {
-				phandle = <0x3d>;
+				phandle = <0x3c>;
 
 				perst-pins {
 					starfive,pins = <0x1a>;
@@ -834,7 +822,7 @@
 			};
 
 			pcie0_perst_active {
-				phandle = <0x3e>;
+				phandle = <0x3d>;
 
 				perst-pins {
 					starfive,pins = <0x1a>;
@@ -846,7 +834,7 @@
 			};
 
 			pcie0_wake_default {
-				phandle = <0x3b>;
+				phandle = <0x3a>;
 
 				wake-pins {
 					starfive,pins = <0x20>;
@@ -857,7 +845,7 @@
 			};
 
 			pcie0_clkreq_default {
-				phandle = <0x3c>;
+				phandle = <0x3b>;
[... 494 lines cut ...]
 		xrpbuffer@f0000000 {
 			reg = <0x00 0xf0000000 0x00 0x1ffffff 0x00 0xf2000000 0x00 0x1000 0x00 0xf2001000 0x00 0xfff000 0x00 0xf3000000 0x00 0x1000>;
-			phandle = <0x59>;
+			phandle = <0x57>;
 		};
 	};
 
@@ -2438,7 +2417,7 @@
 		compatible = "gpio-leds";
 
 		led-ack {
-			gpios = <0x5a 0x03 0x00>;
+			gpios = <0x58 0x03 0x00>;
 			color = <0x02>;
 			function = "heartbeat";
 			linux,default-trigger = "heartbeat";
@@ -2449,6 +2428,6 @@
 	gpio-restart {
 		compatible = "gpio-restart";
 		gpios = <0x25 0x23 0x00>;
-		priority = <0xa0>;
+		priority = <0xe0>;
 	};
 };
```

congocongo, 2023-03-17:
I tried 5.15.102 from your deb package. Still wrong ethphy settings in dtb file. btw if I'll copy starfive's dtb file it will broke pcie driver, your dtb settings for pcie is correct.

MichaIng, 2023-03-17:
Since I found this valuable information, here the upstream plan and status page: https://rvspace.org/en/project/JH7110_Upstream_Plan

All upstream mails/patches are linked, transparent to follow, including suggested changes by reviewers etc. Very interesting and looks great how upstreaming all relevant parts is driven.

MichaIng, 2023-03-21:
Wait with what? We have a board and we want to add/polish RISC-V support for DietPi, detect and reports related bugs in Debian etc, so we cannot/do not want to wait for upstream support to have fully finished 😉. For end users who just want a stable system, RISC-V SBCs in general are completely unsuitable, of course, but we are developers 😃.

As of the Ethernet issue: I fear, unless someone reports and pushes the topic, this won't be related, since it broke only for an old rarely shipped revision. Any official docs you find are about V1.3. As said, the change looks intentional, not like a mistake, and it is not a bug in the code where Linux maintainers would care about.

The only think we could do is to create a dedicated device tree, respectively an overlay, which reverts exactly the above parts. We could even enable it automatically at first boot, if the revision can be detected from within the system?

congocongo, 2023-03-21:
Well, you don’t need a dedicated device tree here. They do modifications using the u-boot scripts before the kernel.

1) https://github.com/starfive-tech/u-boot/blob/1ae835c25e3d8b7bc3c8981fb720f030c1892d57/board/starfive/visionfive2/starfive_visionfive2.c#L152

2) https://github.com/starfive-tech/u-boot/blob/1ae835c25e3d8b7bc3c8981fb720f030c1892d57/board/starfive/visionfive2/starfive_visionfive2.c#L403

3) https://github.com/starfive-tech/u-boot/blob/1539c1fb5a498fd7d20c2cffc84c187d703c8b66/configs/starfive_visionfive2_defconfig#L30

4) https://github.com/starfive-tech/u-boot/blob/1539c1fb5a498fd7d20c2cffc84c187d703c8b66/include/configs/starfive-visionfive2.h#L99

I guess the root of the problem with Ethernet and correct memory size detection is a hardcoded partitioning schema in the u-boot's scripts. They use 0 and 1 partition for OpenSBI/u-boot images, second partition for EFI/boot and third for root. Something may be wrong with u-boot scripts with one partition. 

```sudo fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 57.95 GiB, 62226694144 bytes, 121536512 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: A8D55636-C016-4B30-864A-FE61B85F33D9

Device          Start       End   Sectors  Size Type
/dev/mmcblk0p1   4096      8191      4096    2M HiFive BBL
/dev/mmcblk0p2   8192     16383      8192    4M HiFive FSBL
/dev/mmcblk0p3  16384    221183    204800  100M EFI System
/dev/mmcblk0p4 221184 121534463 121313280 57.8G Linux filesystem
```

MichaIng, 2023-03-21:
Ah good to see that. And yes, `bootcmd_distro` does not run in our case. Looks like its time to do some PRs on their U-Boot script. This hardcoded intended partitioning is very unnecessary. The U-Boot default drive scans work pretty well, as can be seen with the fallback that is done on our image. So it would be perfectly possible to just scan through all drives and partitions for `uEnv.txt` and `extlinux` and do those dynamic device tree adjustments in every case. Additionally functional default RAM addresses for a compressed kernel make sense as well, so this does not need to be set in `uEnv.txt`, making it obsolete (while it's nice to have the option for environment adjustments).

I fixed a bug which caused a wrong U-Boot environment address: https://github.com/MichaIng/linux/commit/dbd2367
With this we can write an own environment, replacing StarFive's default. I've not yet tested it, but if this works, we can do that on first boot as well to have things fixed. This also allows eMMC usage (as long as the driver in U-Boot is working) which is currently not working either because of the hardcoded device IDs in the env.

MichaIng, 2023-03-22:
Actually, while this U-Boot script commands run with common partitioning, there is also the following which should do the same before even running `bootcmd`: https://github.com/starfive-tech/u-boot/blob/JH7110_VisionFive2_devel/configs/starfive_visionfive2_defconfig#L31-L32

> CONFIG_USE_PREBOOT=y
> CONFIG_PREBOOT="run chipa_set_uboot;run mmcbootenv"

In the environment we see related `preboot` variable:
1. `preboot=run chipa_set_uboot;run mmcbootenv`
2. `chipa_set_uboot=fdt addr ${uboot_fdt_addr};run chipa_set;`
3. `chipa_set=if test ${chip_vision} = A; then run chipa_gmac_set;fi;`
4. `chipa_gmac_set=fdt set /soc/ethernet@16030000/ethernet-phy@0 tx_inverted_10 <0x0>;fdt set /soc/ethernet@16030000/ethernet-phy@0 tx_inverted_100 <0x0>;fdt set /soc/ethernet@16030000/ethernet-phy@0 tx_inverted_1000 <0x0>;fdt set /soc/ethernet@16030000/ethernet-phy@0 tx_delay_sel <0x9>;fdt set /soc/ethernet@16040000/ethernet-phy@1 tx_inverted_10 <0x0>;fdt set /soc/ethernet@16040000/ethernet-phy@1 tx_inverted_100 <0x0>;fdt set /soc/ethernet@16040000/ethernet-phy@1 tx_inverted_1000 <0x0>;fdt set /soc/ethernet@16040000/ethernet-phy@1 tx_delay_sel <0x9>`

However, obviously it does not work in your case, not sure why... Ah wait, this cannot work before the device tree is even loaded, respectively this applies it for U-Boot internal device tree, but not for the one which finally boots. This is the reason they have the device tree loading parts in the env.

Actually
```sh
set_fdt_distro=if test ${chip_vision} = A; then if test ${memory_size} = 200000000; then run chipa_gmac_set;run visionfive2_mem_set;fatwrite mmc ${fatbootpart} ${fdt_addr_r} /dtbs/${fdtfile} ${filesize};else run chipa_gmac_set;run visionfive2_mem_set;fatwrite mmc ${fatbootpart} ${fdt_addr_r} /dtbs/${fdtfile} ${filesize};fi;else run visionfive2_mem_set;fatwrite mmc ${fatbootpart} ${fdt_addr_r} /dtbs/${fdtfile} ${filesize};fi;
```
this rewrites the device tree on every boot, which I do not like at all. But reasonably this is the only way to have this parts dynamic as extlinux and common U-Boot scripts (re-)load the device tree and would hence overwrite any previous change. As long as there is no way to add commands to extlinux itself to adjust the device tree after it was loaded, or adjusting these values in userland, I'm going with the overlay route, which seems much cleaner to me than requiring write access and rewriting something essential like the device tree file on every boot. This allows vast cleanup and much simpler and complete (looping through all drives and partitions, regardless which filesystem, checking for efi>extlinux>boot.scr and booting with the first match.

MichaIng, 2023-03-22:
@congocongo 
Could you show the serial number of your VF2, respectively the first block?
```sh
cat /proc/device-tree/serial-number
```
Mine starts with `VF7110B1`, which likely means VisionFive JH7110 B1 revision, which matches the `chip_vision` of the U-Boot env. If it starts with `VF7110A` in your case, I think we found the needed identifier.

congocongo, 2023-03-23:
yes. 

`VF7110A1-2250-D008E000-00000540`

MichaIng, 2023-03-24:
Could you test this:
```sh
apt install device-tree-compiler
cat << '_EOF_' > dietpi-visionfive2-A12.dts
/dts-v1/;
/plugin/;
/ {
        compatible = "starfive,visionfive-v2", "starfive,jh7110";
        fragment@0 {
                target = <&ethernet0>;
                __overlay__ {
                        tx_delay_sel = <0x9>;
                        tx_inverted_10 = <0x0>;
                        tx_inverted_100 = <0x0>;
                        tx_inverted_1000 = <0x0>;
                };
        };
        fragment@1 {
                target = <&ethernet1>;
                __overlay__ {
                        tx_delay_sel = <0x9>;
                        tx_inverted_10 = <0x0>;
                        tx_inverted_100 = <0x0>;
                        tx_inverted_1000 = <0x0>;
                };
        };
};
_EOF_
dtc -I dts -O dtb -o /usr/lib/linux-image-visionfive2/starfive/vf2-overlay/dietpi-visionfive2-A12.dtbo dietpi-visionfive2-A12.dts
G_CONFIG_INJECT 'fdtoverlays[[:blank:]]' 'fdtoverlays /usr/lib/linux-image-visionfive2/starfive/vf2-overlay/dietpi-visionfive2-A12.dtbo' 'fdt[[:blank:]]'
```
This might not be sufficient, as the U-Boot environment lacks the [fdtoverlay_addr_r](https://github.com/starfive-tech/u-boot/issues/26) variable, but let's at least give it a shot. I'll otherwise add `fw_setenv` + config anyway to allow (and do) adding/setting some variables, also to add support for booting from eMMC, M.2 and USB.

congocongo, 2023-03-25:
does not work
```
[FAILED] File does not exist or cannot be written to by current user
Please verify the existence of the file $3
fdt[[:blank:]]
Retry with proper permissions or apply the setting manually:
fdtoverlays /boot/dietpi-visionfive2-A12.dtbo
```

MichaIng, 2023-03-25:
Whoops, I forgot the filename in the last command 😄:
```sh
G_CONFIG_INJECT 'fdtoverlays[[:blank:]]' 'fdtoverlays /boot/dietpi-visionfive2-A12.dtbo' /boot/extlinux/extlinux.conf 'fdt[[:blank:]]'
```

MichaIng, 2023-03-26:
@lgray 
Can you show your serial number as well?
```sh
cat /proc/device-tree/serial-number
```
Probably the 3rd block is about the RAM size.

@congocongo 
Does yours actually have 8 GiB RAM (while only 4 GiB are usable currently)?

congocongo, 2023-03-26:
> @congocongo
> Does yours actually have 8 GiB RAM (while only 4 GiB are usable currently)?

yes 8Gb with 4Gb visible

MichaIng, 2023-03-26:
Great, confirmed here as well: http://forum.rvspace.org/t/dietpi-os-for-visionfive-2/2356/13?u=michaing

So to enable 8 GiB, the following device tree overlay can be tested:
```sh
apt install device-tree-compiler
cat << '_EOF_' > dietpi-visionfive2-8GB.dts
/dts-v1/;
/plugin/;
/ {
	compatible = "starfive,visionfive-v2", "starfive,jh7110";
	fragment@0 {
		target-path = "/memory@40000000";
		__overlay__ {
			reg = <0x0 0x40000000 0x2 0x0>;
		};
	};
};
_EOF_
dtc -I dts -O dtb -o /boot/dietpi-visionfive2-A12.dtbo dietpi-visionfive2-8GB.dts
G_CONFIG_INJECT 'fdtoverlays[[:blank:]]' 'fdtoverlays /boot/dietpi-visionfive2-8GB.dtbo' /boot/extlinux/extlinux.conf 'fdt[[:blank:]]'
```
Same as with the Ethernet fix, probably this only works after adding an fdtoverlay memory addresses to the U-Boot environment, but lets test without this first.

And I need to find out how to add multiple device trees. Not sure whether they can be just added one after another to the same `fdtoverlays` line, or whether multiple such lines are supported.

MichaIng, 2023-04-25:
Btw, default addresses for device tree overlays as well as compressed kernel are coming with StarFive's next bootloader release:
- https://github.com/starfive-tech/u-boot/pull/44
- https://github.com/starfive-tech/u-boot/pull/45

As we already have a `dietpi-config` option to update the bootloader, we can offer this on first boot with the info that this enables support for 8 GiB and Ethernet on A1.2 revision.

I'll do some tests this weeks for own U-Boot env overrides, since e.g. eMMC and M.2 boot still do not work:
- Sadly it is not possible to set only specific variables. If the part of the drive contains any valid variable definition, it is taken as complete U-Boot environment and no other defaults are loaded.
- We could ship (offer to write) a complete U-Boot env, but this can only work if variables like the `chip_vision`, which depend on the actual hardware model, are always provided by U-Boot, even if not defined in the override env address space. Otherwise we would break the ability for U-Boot itself to fix Ethernet and RAM size for its own device tree (the one loaded by U-Boot, not the one loaded for the OS).

aragubas, 2023-05-04:
Hi, I was having issues booting the official debian image from StarFive itself, this is the only debian image that boots on my StarFive xD
here are the issues that I noticed:
- Included wifi adapter cannot connect to 5GHz SSID's (but does connect to 2.5GHz SSID's)
- Included wifi adapter doesn't scan available wifi networks
- XFCE Desktop doesn't boot, X11 crashes with error (cannot find starfive-dri)

MichaIng, 2023-05-04:
Many thanks for your feedback!

> Included wifi adapter cannot connect to 5GHz SSID's (but does connect to 2.5GHz SSID's)

That is known, that adapter, even that it is declared as Wi-Fi 6 adapter, does not support 5 GHz: https://forum.rvspace.org/t/ecr6600u-firmware/2368

> Included wifi adapter doesn't scan available wifi networks

Ah that is bad. So both does not work?
```sh
iw dev wlan0 scan
iwlist wlan0 scan
```

> XFCE Desktop doesn't boot, X11 crashes with error (cannot find starfive-dri)

Is it this error? https://forum.rvspace.org/t/failed-to-open-starfive-usr-lib-dri-starfive-dri-so/2325

However, in our case it generally worked to start a desktop. How did you try it? `startx` or one of the `dietpi-autostart` options, or starting LightDM with its service?
@StephanStS I know you're busy the next days, so just for backlog when you find time to retest a desktop with latest kernel (already installed). And if the strange colour change happens, the `xrandr` colour format change from [here](https://dietpi.com/forum/t/nanopi-r5c-hdmi-to-tv-purple-green-display-tint-after-some-time-with-tv-off/16736/4?u=michaing) could be tested.

> APT takes a looong time when "Reading packages lists..."

In `dietpi-config` > Advanced Options > APT you can (re-)enable the APT cache and disable APT list compression. Also cache, lists and the dir for downloaded DEB packages (before they are installed) can be moved to RAM. On DietPi, other than Debian, the APT cache is disabled by default and lists are left xz-compressed, which radically reduces disk writes on every `apt update` call, but also quite significantly slows it down.

> the model I have is the 8GB model with the dual ethernet ports

But only 4 GiB are currently shown, e.g. in `htop`, right? Can be currently manually solved: http://forum.rvspace.org/t/dietpi-os-for-visionfive-2/2356/15?u=michaing

But with next StarFive bootloader release, we'll implement this fix (as well as A1.2 variant Ethernet) to be applied automatically: https://github.com/starfive-tech/u-boot/pull/44

aragubas, 2023-05-04:
> Ah that is bad. So both commands do not work?
I was using the wifi settings in the dietpie-config 👀 

> Is it this error? https://forum.rvspace.org/t/failed-to-open-starfive-usr-lib-dri-starfive-dri-so/2325
Yes! this exact same error!

> However, in our case it generally worked to start a desktop. How did you try it? startx or one of the dietpi-autostart options, or starting LightDM with its service?
I tried using startx xP haven't tried the autostart options

> In dietpi-config > Advanced Options > APT you can (re-)enable the APT cache and disable APT list compression. Also cache, lists and the dir for downloaded DEB packages (before they are installed) [...]
Oky! I'll try that

> But only 4 GiB are currently shown, e.g. in htop, right? Can be currently manually solved: http://forum.rvspace.org/t/dietpi-os-for-visionfive-2/2356/15?u=michaing
for me it shows the full 8GB's

I glued a heatsink on the processor because it was overheating xP so I'll have to wait until tomorrow for the thermal glue to dry x3 I'll provide more feedback tomorrow!

MichaIng, 2023-05-04:
> I was using the wifi settings in the dietpie-config 👀

It calls `iwlist wlan0 scan`. Would be interesting whether `iw dev wlan0 scan` works. And still whether both work from console, to assure it is not something else not working in our tools.

> I tried using startx xP haven't tried the autostart options

LightDM could be also tested like this (without reboot):
```sh
apt install lightdm # probably already installed
systemctl start lighdm
```

> for me it shows the full 8GB's

Hah, interesting. @congocongo is this the case for you now as well? Probably with latest kernel?
```sh
cd /tmp
curl -O https://dietpi.com/downloads/binaries/linux-image-visionfive2.deb
dpkg -i linux-image-visionfive2.deb
rm linux-image-visionfive2.deb
```

aragubas, 2023-05-05:
[image]
also `iw dev wlan0 scan` does show the wifi connections

aragubas, 2023-05-05:
> So far so good. `iwlist wlan0 scan` shows nothing, or does it hang?

[image]
it says that the interface doesn't support scanning xP

MichaIng, 2023-05-05:
Does it return an error code, so that we could do something like that:
```sh
iwlist wlan0 scan || iw dev wlan0 scan
```

aragubas, 2023-05-05:
> ```shell
> iwlist wlan0 scan || iw dev wlan0 scan
> ```

[image]
with that command it first returns that error but then it lists the wifi interfaces owo

MichaIng, 2023-06-08:
Latest U-Boot breaks boot of our image currently: https://github.com/starfive-tech/u-boot/issues/51

Tomorrow, I will provide instructions about how to fix it.

MichaIng, 2023-06-12:
Image btw boots still fine as long as no NVMe drive is attached. I will update our firmware package the next days to provide a complete U-Boot environment override with a clean boot target/order selection and device tree overlays for 8 GiB RAM and model A1.2 support. I already tested this and it generally works fine.
___________
Golang Go currently cannot be installed since the Debian Sid riscv64 build fails: https://buildd.debian.org/status/fetch.php?pkg=golang-1.20&arch=riscv64&ver=1.20.5-1&stamp=1686533537&raw=0

darkgeek, 2023-08-10:
I see a new version of uboot is out, say v3.4.5, is it possible to resolve the 8G issue by flashing this version?

MichaIng, 2023-08-10:
The previous U-Boot version did already support it. Sorry for the delay, I did not find the time to finish the thing yet. However, here is how to do it:
1. Go to dietpi-config > Advanced Options > Update bootloader to update the SPI bootloader to the latest version. Not necessarily needed as any 3.x version supports device tree overlays already.
2. Follow the steps from here to create and apply the device tree overlay: https://github.com/MichaIng/DietPi/issues/6212#issuecomment-1484190217
3. reboot

maleadt, 2023-09-21:
Is the latest DietPi testing image compatible with bootloader [v3.6.1](https://github.com/starfive-tech/VisionFive2/releases/tag/VF2_v3.6.1)? IIUC the 1-partition layout doesn't seem supported:

```
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc1 is current device
Try booting from MMC1 ...
** Invalid partition 3 **
Couldn't find partition mmc 1:3
Can't set block device
```

MichaIng, 2023-09-21:
Do you currently have an NVMe drive attached as well? In this case there is some faulty variable setting done with the default U-Boot environment, but that can be fixed to make it universally bootable + easy way to set boot priorities.

maleadt, 2023-09-21:
No, just an SD card.

MichaIng, 2023-09-21:
Hmm, shit, I think this commit broke it: https://github.com/starfive-tech/u-boot/commit/af8ba43c10f37d4669c26406473dfb22cd6c077b

They really do everything to break any common partitioning end enforce their strange and completely unnecessary and problematic 4 partitions setup 😄. I already started working on an own clean universally compatible U-Boot environment. I just need to finish and cleanup, checking and removing all variables that are provided by bootloader depending on hardware, like serial number and such.

Need to check whether their kernel has finally a fix merged for wrong `/dev/mtd*` device addresses, which broke the possibility to R/W the U-Boot environment. But I remember I saw it, at least in the Linux 6.1 branch. If it is in their Linux 5.15 branch, then one would need to boot their Debian image first, flash the fixed U-Boot environment.

For now, from U-Boot console, it should be possible to run:
```sh
run mmc_boot
```

maleadt, 2023-09-21:
```
StarFive # env print
baudrate=115200
boot_a_script=load ${devtype} ${devnum}:${distro_bootpart} ${scriptaddr} ${prefix}${script}; source ${scriptaddr}
boot_devs=mmc nvme
boot_efi_binary=load ${devtype} ${devnum}:${distro_bootpart} ${kernel_addr_r} efi/boot/bootriscv64.efi; if fdt addr ${fdt_addr_r}; then bootefi ${kernel_addr_r} ${fdt_addr_r};else
 bootefi ${kernel_addr_r} ${fdtcontroladdr};fi
boot_efi_bootmgr=if fdt addr ${fdt_addr_r}; then bootefi bootmgr ${fdt_addr_r};else bootefi bootmgr;fi
boot_extlinux=sysboot ${devtype} ${devnum}:${distro_bootpart} any ${scriptaddr} ${prefix}${boot_syslinux_conf}
boot_net_usb_start=usb start
boot_pci_enum=pci enum
boot_prefixes=/ /boot/
boot_script_dhcp=boot.scr.uimg
boot_scripts=boot.scr.uimg boot.scr
boot_syslinux_conf=extlinux/extlinux.conf
boot_targets=mmc0 dhcp
bootargs=console=tty1 console=ttyS0,115200  debug rootwait  earlycon=sbi
bootcmd=run sdk_boot_env; run distro_boot_env
bootcmd_dhcp=devtype=dhcp; run boot_net_usb_start; run boot_pci_enum; if dhcp ${scriptaddr} ${boot_script_dhcp}; then source ${scriptaddr}; fi;setenv efi_fdtfile ${fdtfile}; seten
v efi_old_vci ${bootp_vci};setenv efi_old_arch ${bootp_arch};setenv bootp_vci PXEClient:Arch:00027:UNDI:003000;setenv bootp_arch 0x1b;if dhcp ${kernel_addr_r}; then tftpboot ${fdt
_addr_r} dtb/${efi_fdtfile};if fdt addr ${fdt_addr_r}; then bootefi ${kernel_addr_r} ${fdt_addr_r}; else bootefi ${kernel_addr_r} ${fdtcontroladdr};fi;fi;setenv bootp_vci ${efi_ol
d_vci};setenv bootp_arch ${efi_old_arch};setenv efi_fdtfile;setenv efi_old_arch;setenv efi_old_vci;
bootcmd_distro=run load_distro_uenv; run fdt_loaddtb; run fdt_sizecheck; run set_fdt_distro; sysboot ${bootdev} ${devnum}:${bootpart} fat ${scriptaddr} /${boot_syslinux_conf};
bootcmd_mmc0=devnum=0; run mmc_boot
bootdelay=2
bootdev=mmc
bootdir=/boot
bootenv=uEnv.txt
bootenv_mmc=setenv bootdev mmc;if test ${bootmode} = flash; then for mmc_devnum in ${mmc_devnum_l}; do setenv devnum ${mmc_devnum}; run mmc_test_and_boot;done;fi; if test ${bootmo
de} = sd; then setenv devnum ${sd_devnum};run mmc_test_and_boot;fi; if test ${bootmode} = emmc; then setenv devnum  ${emmc_devnum};run mmc_test_and_boot;fi;
bootenv_nvme=if test ${bootmode} = flash; then for nvme_devnum in ${nvme_devnum_l}; do setenv devnum ${nvme_devnum};if pci enum; then nvme scan; fi; if nvme dev ${devnum}; then ec
ho Try booting from NVME${devnum} ...; setenv bootdev nvme;setenv sdev_blk nvme${devnum}n1p${rootpart};run load_sdk_uenv; run boot2;fi; done; fi;
bootenv_sdk=vf2_uEnv.txt
bootfile=/extlinux/extlinux.conf
bootmode=flash
bootpart=3
chip_vision=B
chipa_gmac_set=fdt set /soc/ethernet@16030000/ethernet-phy@0 tx_inverted_10 <0x0>;fdt set /soc/ethernet@16030000/ethernet-phy@0 tx_inverted_100 <0x0>;fdt set /soc/ethernet@1603000
0/ethernet-phy@0 tx_inverted_1000 <0x0>;fdt set /soc/ethernet@16030000/ethernet-phy@0 tx_delay_sel <0x9>;fdt set /soc/ethernet@16040000/ethernet-phy@1 tx_inverted_10 <0x0>;fdt set
 /soc/ethernet@16040000/ethernet-phy@1 tx_inverted_100 <0x0>;fdt set /soc/ethernet@16040000/ethernet-phy@1 tx_inverted_1000 <0x0>;fdt set /soc/ethernet@16040000/ethernet-phy@1 tx_
delay_sel <0x9>
chipa_set=if test ${chip_vision} = A; then run chipa_gmac_set;fi;
chipa_set_linux=fdt addr ${fdt_addr_r};run visionfive2_mem_set;run chipa_set;
chipa_set_uboot=fdt addr ${uboot_fdt_addr};run chipa_set;
cma_1g=b000000
cma_2g=20000000
cma_4g=40000000
cma_8g=60000000
cma_ddr1g_set=fdt set ${cma_node} size <0x0 0x${cma_1g}>;fdt set ${cma_node} alloc-ranges <0x0 0x${cma_start} 0x0 0x${cma_1g}>;
cma_ddr2g_set=fdt set ${cma_node} size <0x0 0x${cma_2g}>;fdt set ${cma_node} alloc-ranges <0x0 0x${cma_start} 0x0 0x${cma_2g}>;
cma_ddr4g_set=fdt set ${cma_node} size <0x0 0x${cma_4g}>;fdt set ${cma_node} alloc-ranges <0x0 0x${cma_start} 0x0 0x${cma_4g}>;
cma_ddr8g_set=fdt set ${cma_node} size <0x0 0x${cma_8g}>;fdt set ${cma_node} alloc-ranges <0x0 0x${cma_start} 0x0 0x${cma_8g}>;
cma_node=/reserved-memory/linux,cma
cma_resize=if test ${memory_size} -eq 40000000; then run cma_ddr1g_set;elif test ${memory_size} -eq 80000000; then run cma_ddr2g_set;elif test ${memory_size} -eq 100000000; then r
un cma_ddr4g_set;elif test ${memory_size} -ge 200000000; then run cma_ddr8g_set;fi;
cma_start=70000000
cpu_max_vol=1040000
cpu_speed_1250_set=fdt rm /opp-table-0/opp-375000000;fdt rm /opp-table-0/opp-500000000;fdt rm /opp-table-0/opp-750000000;fdt rm /opp-table-0/opp-1500000000;
cpu_speed_1500_set=fdt rm /opp-table-0/opp-312500000;fdt rm /opp-table-0/opp-417000000;fdt rm /opp-table-0/opp-625000000;fdt rm /opp-table-0/opp-1250000000;
cpu_vol_1020_set=fdt set /opp-table-0/opp-1500000000 opp-microvolt <1020000>;
cpu_vol_1040_set=fdt set /opp-table-0/opp-1500000000 opp-microvolt <1040000>;
cpu_vol_1060_set=fdt set /opp-table-0/opp-1500000000 opp-microvolt <1060000>;
cpu_vol_set=if test ${cpu_max_vol} = 1000000; then run cpu_speed_1250_set; else run cpu_speed_1500_set; if test ${cpu_max_vol} = 1060000; then run cpu_vol_1060_set; elif test ${cp
u_max_vol} = 1020000; then run cpu_vol_1020_set; else run cpu_vol_1040_set; fi; fi;
devnum=0
distro_boot_env=echo Tring booting distro ...;for bootdev_s in ${boot_devs}; do run distro_bootenv_${bootdev_s}; done;
distro_bootcmd=setenv nvme_need_init; for target in ${boot_targets}; do run bootcmd_${target}; done
[... cut ...]
type_guid_gpt_loader2=2E54B353-1271-4842-806F-E436D6AF6985
type_guid_gpt_system=0FC63DAF-8483-4772-8E79-3D69D8477DE4
uboot_fdt_addr=0xf7fad910
usb_boot=usb start; if usb dev ${devnum}; then devtype=usb; run scan_dev_for_boot_part; fi
ver=U-Boot 2021.10 (Aug 31 2023 - 12:55:45 +0800)
visionfive2_mem_set=fdt memory ${memory_addr} ${memory_size};run cma_resize;

Environment size: 10190/65532 bytes
```

`mmc_boot` doesn't do much though:

```
StarFive # run mmc_boot
Card did not respond to voltage select! : -110
```

MichaIng, 2023-09-21:
Ah, `devnum` was set to `0` (eMMC) after SD card boot from FAT partition 3 failes. Hence:
```sh
setenv devnum "1"
run mmc_boot
```
And indeed any fallback to a generic device scan and boot has been removed from the default environment, leaving only the hardcoded possibility to boot from a FAT filesystem from partition 3. What a pain ... and so totally unnecessary + overly complex.

JamiKettunen, 2023-09-21:
At some point it might just be easier to not support stock U-Boot at all if they want to do weird downstream stuff and just enforce mainline one as the necessary pieces for it have been landing there for a while now

MichaIng, 2023-09-21:
This is the mainline U-Boot mailing list? I was/am not sure how far mainline U-Boot support actually is at this point. Mainline Linux at least still misses NVMe support, which IMO is too important to switch. So to guarantee that all hardware features are available and all storage devices supported, for now we stick with vendor kernel and bootloaers.

Another problem is that for NVMe and USB boot, we must use the SPI bootloader and cannot use an own one shipped with the image. This is actually the reason why our image does not contain any bootloader but just relies on SPI. And initially the idea was to make it easy by not requiring to reflash another bootloader or environment before being able to boot a DietPi image, but that one can stick with the one the SBC is shipped with. This however seems to be impossible now, sadly.

This device tree selection based on hardware model is actually another major problem with downstream bootloader/kernel/both: StarFive images rewrite the whole device tree file on disk on every boot (if it can be found at a hardcoded path on a hardcoded FAT filesystem on hardcoded partition 3 ...), which is IMO an absolute no go, and obviously it is the same for mainline U-Boot devs, discussing an automatic device tree selection, instead of a rewrite. The solution I am aiming for, for now with vendor U-Boot and kernel, is to have 1 overlay for 8 GiB and 1 overlay for model A1.2 Ethernet, and apply them on first boot, depending on the serial number, which contains all needed info.

JamiKettunen, 2023-09-21:
Yeah that's indeed the mainline U-Boot mailing list. Fwiw the board mostly already works on [v2023.10-rc4](https://github.com/u-boot/u-boot/releases/tag/v2023.10-rc4) when compiled from [starfive_visionfive2_defconfig](https://github.com/u-boot/u-boot/blob/master/configs/starfive_visionfive2_defconfig) but there's still some patches of interest pending like:
- [starfive: visionfive2: add mmc0 and nvme boot targets](https://patchwork.ozlabs.org/project/uboot/patch/20230918083231.20092-1-mps@arvanta.net/)
- [configs: starfive: Enable PCIE auto enum and NVME/USB stuff for Starfive Visionfive 2](https://github.com/u-boot/u-boot/commit/dccf4a843a5f0fb2befd31c47b70aecfd0a49ecb)
- [riscv: starfive: generate u-boot-spl.bin.normal.out](https://patchwork.ozlabs.org/project/uboot/cover/20230917114731.23695-1-heinrich.schuchardt@canonical.com/)
- [board: visionfive2: Select fdtfile based on revision](https://github.com/Headcrabed/u-boot/commits/ddr_fix_v1) (tree with extra patches mentioned on the RFC I already linked)
- [eeprom: starfive: set eth0 mac address properly](https://github.com/u-boot/u-boot/commit/71914337ef579bb22705d4b88fee6b1448f7a9b1)

I'm using those currently in practice on my W.I.P [Chimera Linux](https://chimera-linux.org/) packaging for this board with mainline *everything*; while especially the mainline kernel still needs some work before being comparable to downstream functionality it's certainly getting there and I'm along for the ride anyway with this being my first ever RISC-V hw :)

MichaIng, 2023-09-28:
USB boot support: #6641

rhinot, 2023-11-06:
@MichaIng a heads up that your image also works on the VisionFive 2-based [Milk-V Mars](https://milkv.io/mars) (and [CM](https://milkv.io/mars-cm)).  Thank you for the build; there is a small circle of us that is considering it the superior image to the manufacturer provided image (especially since you're on linux 6.1).

VNC, Firefox, LXDE and apt update/upgrade all confirmed to work (and not break the system).

More discussion [here](https://community.milkv.io/t/standard-debian-image-on-mars/888/12). Excited to see how 6.6 impacts the system, when you merge that in (or if you have a way to test it earlier, you have a few folks who are happy to test it out).

guitarpicva, 2023-11-20:
Not finding any kernel module ```cdc_acm`` for common-place Microchip 2221 USB chip support (USB to RS-232/UART converter).  These are frequently used in serially connected modem type devices.  Many in Amateur Radio circles.

terop, 2023-11-25:
I cannot get the https://dietpi.com/downloads/images/testing/DietPi_VisionFive2-RISC-V-Sid.img.xz image to boot on VisionFive 2. I have tried it on both SD card and eMMC with the same result. The latest (202310) official VisionFive 2 Debian image boots okay from eMMC. I have updated the bootloader as instructed in https://dietpi.com/blog/?p=2629#update-image.
Can someone advice on what to try next?

MichaIng, 2023-11-25:
Please have a look at the commits further above: The latest U-Boot (default environment) does not support out partitioning anymore. Do you have a USB-UART adapter to access the serial console?

terop, 2023-11-25:
I have a USB serial adapter and I tried it but I was unable to get it to display any output. However I am not sure whether adapter works anymore. I should try it on a known working SBC to verify this.

Yes, I tried it first on a SD card. Then I updated U-Boot and tried again on SD card. Finally I tried the image on eMMC which I know works with the official Debian image from StarFive.

terop, 2023-12-18:
@MichaIng Sorry for the delay. After some more careful reading of the instructions I got the serial console working. Below is the (very short) output I get when trying to boot the DietPi image:
```
dwmci_s: Response Timeout.
BOOT fail,Error is 0xffffffff
```

MichaIng, 2023-12-18:
Please check the boot mode jumpers and assure they are set to boot from SPI bootloader.

terop, 2023-12-18:
I am using an eMMC card but they should still be put in the QSPI boot mode?

MichaIng, 2023-12-18:
Yes. We ship our images without bootloader and rely on the SPI one instead.

[48 more replies]