MichaIng/DietPi #6214: Cant passthrough a partition to DietPi VM

Phil1988, 2023-03-05:
### Creating a bug report/issue

#### Required Information
- DietPi version:
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'

- Distro version: 
bullseye

- Kernel version:
Linux DietPi 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux

- SBC model: 
Virtual Machine (x86_64) on Proxmox VE 7.3

#### Steps to reproduce
I installed DietPi according to https://dietpi.com/docs/install/.
After finished, I attached an external USB drive to the host, formatted it via the host Shell and created 3 partitions via fdisk.

(according to 
https://forum.proxmox.com/threads/is-it-possible-to-passthrough-a-partition-of-a-drive-to-a-vm-and-not-the-entire-disk.109974/#post-473238
and
https://pve.proxmox.com/wiki/Passthrough_Physical_Disk_to_Virtual_Machine_(VM)#Hot-Plug.2FAdd_physical_device_as_new_virtual_SCSI_disk) 

Partition 1 should be passed to the DietPi VM (its ID is 101), so I passed it to it via:
`qm set 101 -scsi1 /dev/disk/by-id/usb-Mass_Storage_Device_125D20140310-0:0-part1`
Output is fine:
[image]

That looked as it worked, as I saw it it the Proxmox WebIF:
[image]

And I could see it in the console of the VM:
[image]

I then tried to mount it, but had no success (freeze).
Also the Drive Manager gets stuck:
[image]
And booting the system was also impossible:
[image]

I then detached the partition from the VM and could boot again.
After this I changed the SCSI Controller of the VM from "VirtIO SCSI single" to "VirtIO SCSI":
[image]

I attached the partition again to the VM via the Proxmox Shell, but booting stays impossible:
[image]

I can reproduce this behaviour with another fresh install.

I know I can attach the full USB drive to the VM, but I dont need the entire space for DietPi and would like to have some space for DietPi (Nextcloud data) and some for the Proxmox Backups.
Is there a way to attach a partition of an USB drive to a Proxmox DietPi VM?

Thanks a lot for your great software and that its always up to date (also for the tutorials)!

Joulinar, 2023-03-05:
Maybe @StephanStS could have a look. He has quite some systems running on Proxmox.

LOGIN-TB, 2023-03-05:
I'll try to reproduce it

LOGIN-TB, 2023-03-05:
Ok, i have checked it. This is what I did:

1. Created a DietPi VM
2. added a usb disk as passthrough (complete disk)
3. in DietPi I created a linux partition
4. removed the usb passthrough
5. added the linux portion of the usb drive to the vm via qm set 
6. in dietpi drivemanager I could format and mount the partition
7. reboot and access is working

[image]

[image]

I don't know if this relevant, but I have installed and enabled the qemu guest tools

LOGIN-TB, 2023-03-05:
I did another test. Installed a fresh vm without the qemu tools. Everything works as expected.

It seems that this is an proxmox issue and not an issue with dietpi.

What kind of partition did you create? How did you format it?

Phil1988, 2023-03-05:
Hello @LOGIN-TB and thank you for the fast testing on your side.

The interessting part is that you gave the complete disk to dietpi and did the format + partitioning there before you detached the disk and passed only one partition to it.

This is how I did it:
I went to the Proxmox node (the host) in the WebIF and there to "Disks".
There I wiped the disk first and initialized a GPT boot table:
[image]

Then I opened the Shell and listed the Partitions:
`lsblk`
[image]

Then I used fdisk to delete all partititions and create new ones:
`fdisk /dev/sda`
deleting all partiionts with "d"
created new partitions with "n"
finalized it with "w"

After this I created the file systems for each partition like this:
`mkfs.ext4 -F /dev/sda1`
[image]

Then I searched for the ID of the partition with:
`ls -l /dev/disk/by-id/`

In my case, I only want to use partition 1 => sda1
[image]

Then I passed the partition to dietpi using qmset:
`qm set 101 -scsi1 /dev/disk/by-id/usb-Mass_Storage_Device_125D20140310-0:0-part1`

In the guest DietPi VM i could see a new disk (the passed through partition sda1) as sdb:
[image]

As this didnt worked (as descriped above) I also deleted/wiped the ext4 file system from the proxmox WebIF, hoping that this was causing the issue:
[image]

Unfortunetly this didnt work.

I am a bit confused now because I dont see the mistake I did (only a different approach).
My approach was chosen, because 2 other partitions are used for Proxmox backups and I didnt want to pass the disk to a guest, that had a function to the host.

I will check your way now and report back.

Phil1988, 2023-03-05:
Well it doesnt work for me.

I did this:
1. passed the entire disk to the DietPi guest VM via WebIF -> Dietpi VM -> Hardware -> Add USB Device:[image]

2. Checked the disks + partitions in the dietpi VM:
[image]

3. Created an ext4 partition via the console:
[image]

4. Detached the USB drive from the DietPi guest VM via the Proxmos WebIF
5. Passed the Partition sda1 from the proxmox host to the DietPi guest VM via qm set: `qm set 101 -scsi1 /dev/disk/by-id/usb-Mass_Storage_Device_125D20140310-0:0-part1`
6. Checked the disks and partition in the DietPi console:[image]
I think this might already be faulty as I think it should show an ext file system on sdb in the guest, shouldn't it?

8. Tried to access the disk via dietpi drive manager, but it gets stuck:[image]
 (its not doing anything for over 10min now)

Any ideas, what I did wrong?

LOGIN-TB, 2023-03-05:
okay. I See you also have an nfs partition on this disk, right?

Maybe just try this:

remove all partitions (passthrough complete drive with usb)
create a partition (gps / linux) with fdisk command
remove passthrough 
passthrough the portion only
use drive manager to mount and format the partition (ext4)
if this works try create an nfs partition. maybe we have some side effects.

This is the way I did it

Phil1988, 2023-03-05:
Thanks @LOGIN-TB , but this doesnt work as well.

I did:
1. wipe the entire disk to this state:
2. Did forward the whole USB device to the VM
3. could see the disk and created 3 partitions in dietpi using fdisk
[image]
[image]
[image]
[image]

5. removed the usb device via Proxmox webIF and could see, that there are now 3 partitions:
[image]

6. passed the partition 1 (sda1) to the VM via qm set
7. Can see the partition in dietpi:
[image]

8. tried to start the drive manager, but its stucked:
[image]

MichaIng, 2023-03-05:
After you formatted the disk and before removing it from the VM, please unmount it, so the old `/etc/fstab` entries are removed and the `systemd.automount` units removed.

Phil1988, 2023-03-06:
Hey @MichaIng ,

Im not sure what you mean, because I only created partitions on that disk and did not set a mount point.
I did the disk wipe again, created some partitions and looked if I can see a mount point, but I cant see one for the sdb drive:
[image]

[image]

[image]

So what do I have to do before detaching the USB disk from the VM?

Thank you for the patience :)

MichaIng, 2023-03-06:
> Im not sure what you mean, because I only created partitions on that disk and did not set a mount point.

Ah okay, so you hit "Cancel" when being asked to enter the mount point? I remember some versions ago it was then still mounted on default mount point (with UUID), but now the mount is skipped.

> After changing this to "Write through" and rebooting the VM it works.

What is the difference between "No cache" and "Write through" 🤔? The kernel log even states that it assumes "write through", which is sane, works (successfully) the same way when attaching drives via USB docking station to a physical machine. And why would the cache affect whether a partition can be passed to the VM?
@LOGIN-TB how is this set in your case?

Phil1988, 2023-03-07:
No I did not hit "Cancel", because fdisk itself never asked for a mount point - it just formatted the disk/partitions.

> > Im not sure what you mean, because I only created partitions on that disk and did not set a mount point.
> 
> Ah okay, so you hit "Cancel" when being asked to enter the mount point? I remember some versions ago it was then still mounted on default mount point (with UUID), but now the mount is skipped.

No I did not hit "Cancel", because fdisk itself never asked for a mount point - it just formatted the disk/partitions.
Where should I have been asked for this?

> What is the difference between "No cache" and "Write through" 🤔? The kernel log even states that it assumes "write through", which is sane, works (successfully) the same way when attaching drives [...]

No idea on that point. 
I only know that DietPi assumes the system drive as "Write through" at the first boot and keeps it for this disk.
For the attached partition (from the VM perspective a "disk") it did not do this assessment and did not work in my case.

MichaIng, 2023-03-07:
Ah, so you didn't use `dietpi-drive_manager` for formatting, all good then.

> DietPi assumes the system drive as "Write through"

This is Linux, not DietPi, reasonably assuming "Write through" (== "no cache" as far as I can think of) if no other information can be obtained from the drive. If you actively set "Write through", the kernel message is gone, since this info can be actively obtained from the drive. I'm just not sure how it can make a difference. Probably it also depends on the configured virtual SCSI/SATA controller. Which one do you use? If not VirtIO, try VirtIO.

Phil1988, 2023-03-08:
The controller is VirtIO SCSI:
[image]

Now it would be interessting what @LOGIN-TB set as cache.
Maybe its already his habit of setting it accordingly and this didnt cause the problem on his side in.
On my side I can 100% reproduce this issue, tested with 2 different servers on the OS PVE 7.3-3 and 7.3-6 and on 2 different storages (4GB SD card and 2GB flash drive).

MichaIng, 2023-03-08:
And when setting to default (no cache), you can replicate the issue? Quite an edge case, not sure whether/where to document, and whether it is expected or a bug in Proxmox 🤔.

MichaIng, 2023-03-10:
@LOGIN-TB 
Do you have `initramfs-tools` installed or `tiny-initramfs`?

LOGIN-TB, 2023-03-10:
@MichaIng 

I am not sure what the default is. For my VM I used the DietPi x86 Bios ISO to install the system. I did not installed any software, so the initram is the default.

MichaIng, 2023-03-10:
```sh
dpkg -l | grep initramfs
```
For PCs (including the ISO), the default is `initramfs-tools`, for VMs (including the Proxmox image), the default is `tiny-initramfs`.

@Phil1988 
Try the following:
```sh
update-tirfs
reboot
```
and check whether this changes something. Our images ship with a minimal set of kernel modules in initramfs, but when this runs on the VM (is done on every kernel upgrade), it adds some more, based on actually loaded kernel modules in the running system. It affects the modules loaded in very early boot stage, which might or might not have an effect.

If it does not change something, try:
```sh
apt install initramfs-tools
apt autopurge tiny-initramfs
```
This will add a lot more kernel modules to the initramfs, regardless of which one are actually loaded on the final image. Also the initramfs binary itself has more features. Probably this has an effect on how drive caching is detected and applied.

Phil1988, 2023-03-11:
> @Phil1988 Try the following:
> 
> ```shell
> update-tirfs
> reboot
> ```
I did
```
update-tirfs
shutdown -h now
```
<details>
  <summary>Changed the cache settings to "no cache" and started the VM with no success.</summary>
  
Booting shows again issues:
[image]

After chaning the cache bacfk to "write through" in the host, it started and told me I can update one package, what I did but with no success after setting the cache back to "no". 
[image]!
  
</details>

After this, I shutdown the VM, set cache back to "write through" tried
> ```shell
> apt install initramfs-tools
> apt autopurge tiny-initramfs
> ```
shutdown the VM, changed cache to "no" again and booted.

It took a lot more time here:
[image]

Then on the "ifupdown" thing, it also took a while, but It got through (after about 2-3min), but its impossible to start the drive manager:
[image]

Phil1988, 2023-03-11:
Based on @LOGIN-TB I also installed a dietpi VM as a x86 BIOS image.
Precisely from the "Native PC for BIOS/CSM" "Installer Image" found here: https://dietpi.com/#downloadinfo

Installation worked without any issues, but after attaching the partition to that new VM via
`qm set`
I have the same issue with this partition as with the image that I used before:

MichaIng, 2023-03-11:
Not sure what the difference is then, probably the actual hardware drive below the Proxmox volumes do make a difference here.

Out of interest, which of these commands does hang?
```sh
findmnt
blkid
lsblk
```

Phil1988, 2023-03-11:
These work:
```
findmnt
lsblk
```

this hangs:
`blkid`

MichaIng, 2023-03-11:
Does `lsblk` actually list the affected partition? It probes them differently than `blkid` but generally can obtain the same information (and more).

Phil1988, 2023-03-14:
Yes, `lsblk` does list the affected partition:
[image]

MichaIng, 2023-03-14:
Can it also show the filesystem type?
```sh
lsblk -no FSTYPE /dev/sdb
```

closed 2023-04-03 by MichaIng