MichaIng/DietPi #6216: RPi Zero no wifi after reboot

Xolgrimm, 2023-03-05:
#### Required Information
- DietPi version (8.15 beta)
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=15
G_DIETPI_VERSION_RC=0
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

- DietPi version (8.14.2)
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'

- Distro version | bullseye 1

- Kernel version | Linux PiZero 5.15.84+ #1613 Thu Jan 5 11:58:09 GMT 2023 armv6l GNU/Linux

- SBC model | RPi Zero W (armv6l) & RPi Zero WH (armv6l)

- Power supply used | 5V 1A PowerSupply / 5V 2.5A PowerSupply / Mainboard-USB-Power
- SD card used | 2x SanDisk ultra 32GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

Used a fresh installtion from 8.14.2 and 8.15.0 (beta) both had the same Problem.
Used with and without additionl software from your list. VIM & OpenSSH-Server
Used downloaded Image is "DietPi_RPi-ARMv6-Bullseye.img"

#### Steps to reproduce

1. Flashed SD Card with "DietPi_RPi-ARMv6-Bullseye.img" and the tool "Rufus"/ "Balea Etcher" 
2. Edit the "dietpi-wifi.txt" with my SSID and KEY in Entry 0
3. Edit the "dietpi.txt"
	AUTO_SETUP_LOCALE=de_DE.UTF-8
	AUTO_SETUP_KEYBOARD_LAYOUT=de
	AUTO_SETUP_TIMEZONE=Europe/Berlin
	AUTO_SETUP_NET_ETHERNET_ENABLED=0
	AUTO_SETUP_NET_WIFI_ENABLED=1
	AUTO_SETUP_NET_WIFI_COUNTRY_CODE=DE
	AUTO_SETUP_DHCP_TO_STATIC=0    | and 1 tested with static IP and DHCP.
	AUTO_SETUP_NET_HOSTNAME=PiZero
	AUTO_SETUP_HEADLESS=0 | and 1
	AUTO_SETUP_SSH_SERVER_INDEX=-2  | OpenSSH i used it normaly
	AUTO_SETUP_SSH_PUBKEY= xxxx
	AUTO_SETUP_BROWSER_INDEX=0
	AUTO_SETUP_AUTOMATED=1
	AUTO_SETUP_GLOBAL_PASSWORD= xxx
	AUTO_SETUP_INSTALL_SOFTWARE_ID=20    | tested with and without that
	AUTO_SETUP_INSTALL_SOFTWARE_ID=105   | tested with and without that
	DEV_GITBRANCH=master 	| and with beta
4. Extracted SD-Card from Windows, then from the PC
5. Start the RPi Zero with the SD-Card (first installtion)
6. Login in with Putty  | this and Point 7 is not nessasary, but it shows, that the wifi runs.
7. dietpi-config 	| or don't that | with changes | without changes | bluetooth on or off | autoreconnect on or off
8. reboot        	| Power-lost, command or "automatic" from the dietpi-config

9. RPi Zeros don't connect to wifi 

#### Expected behaviour
 The RPi Zeros shoud be connect to the wifi after reboot.

#### Actual behaviour
 It don't connect to the wifi after a reboot.

#### Extra details
 I don't have a connector for a keyboard. I don't get any other informations from the SD-Card.

When i set the AUTO_SETUP_AUTOMATED to 0
After first start (on Display "Please hit <return> to login" ) -> reboot -> nowifi but on Display "Please hit <return> to login" 

"RPi Zero 2" armv7l works

MichaIng, 2023-03-05:
Without debugging via keyboard this is hard to investigate.

When accessing `dietpi-config` network adapter settings, assure that onboard WiFi is/stays enabled. Also, before reboot, check `/boot/dietpi.txt` that it is enabled there, i.e. not `dtoverlay=disable-wifi` line. You can loop through relevant files:
- `/etc/wpa_supplicant/wpa_supplicant.conf`
- `/etc/network/interfaces`

and assure that WPA2 credentials are correct (ah, the key is entered as 64-bit hex digit value) and `allow-hotplug wlan0` uncommented + interface values set correctly, DHCP or correct static IP info.

If this works before reboot, I see no reason why it should fail after reboot, given that onboard WiFi remains enabled:
```sh
ifdown wlan0 # kills SSH!
ifup wlan0
```

Probably the issue is of different kind, e.g. broken SD card, insufficient power, or any such which causes kernel errors and related failures.

Xolgrimm, 2023-03-05:
Adapter for usb a to usb micro are ordered (delivered Wednesday)
I use 2 different SD-Cards both SanDisk 32 GB one 2 Years old the other new

all above was ok. now Crontab

crontab with ifdown and ifup  (with static and DHCP tested)
@reboot    /usr/sbin/ifdown wlan0 && /usr/sbin/ifup wlan0

after reboot this information at login-screen.
- LAN IP : Use dietpi-config to setup a connection (wlan0)

[after_change_to_DHCP.txt](https://github.com/MichaIng/DietPi/files/10892602/after_change_to_DHCP.txt)

MichaIng, 2023-03-05:
Looks all good. The missing connection on login is due to `ifdown` as `@reboot` cron job. Generally `journalctl -u ifup@wlan0` gives you the relevant logs about network bringup at boot.

Xolgrimm, 2023-03-05:
ok wendsday

Xolgrimm, 2023-03-06:
We have found the issue

The german Wifi-Channel 13 is bad
switched to Channel 6 or 10

now connection is fine.

Thanks for help

closed 2023-03-06 by MichaIng