MichaIng/DietPi #6218: Gitea service fails to start

anselal, 2023-03-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'

- Distro version | buster 1
- Kernel version | `uname -a`
- SBC model | RPi B+ (armv6l)
- Power supply used | 5V 1.5A
- SD card used | SanDisk

#### Additional Information (if applicable)
- Software title | Gitea
- Was the software title installed freshly or updated/migrated? Tried both
- Can this issue be replicated on a fresh installation of DietPi? No clue
- Bug report ID | ecabba6b-7321-48b2-860d-68de40cdb4c4

#### Steps to reproduce
1. `dietpi-software reinstall 165`
2. reboot

#### Expected behaviour
- Gitea service running and able to login via web

#### Actual behaviour
- Gitea service failed to start

#### Extra details
[image]

MichaIng, 2023-03-07:
You seem to have a deprecated setting in your config, see first log entry of your screenshot and adjust it accordingly in
```
/mnt/dietpi_userdata/gitea/custom/conf/app.ini
```
However, this sounds like the fallback is still done. Show the full log, please:
```sh
journalctl -u gitea
```

anselal, 2023-03-07:
```bash
-- Logs begin at Sun 2023-03-05 09:47:18 EET, end at Tue 2023-03-07 14:42:55 EET. --
Mar 05 09:47:51 DietPi systemd[1]: Started Gitea (DietPi).
Mar 05 09:47:57 DietPi gitea[545]: 2023/03/05 09:47:57 [T] AppPath: /mnt/dietpi_userdata/gitea/gitea
Mar 05 09:47:57 DietPi gitea[545]: 2023/03/05 09:47:57 [T] AppWorkPath: /mnt/dietpi_userdata/gitea
Mar 05 09:47:57 DietPi gitea[545]: 2023/03/05 09:47:57 [T] Custom path: /mnt/dietpi_userdata/gitea/custom
Mar 05 09:47:57 DietPi gitea[545]: 2023/03/05 09:47:57 [T] Log path: /mnt/samba/nas/workspace/gitea/log
Mar 05 09:48:40 DietPi systemd[1]: gitea.service: Main process exited, code=exited, status=1/FAILURE
Mar 05 09:48:40 DietPi systemd[1]: gitea.service: Failed with result 'exit-code'.
Mar 05 09:50:41 DietPi systemd[1]: Started Gitea (DietPi).
Mar 05 09:50:43 DietPi gitea[921]: 2023/03/05 09:50:43 [T] AppPath: /mnt/dietpi_userdata/gitea/gitea
Mar 05 09:50:43 DietPi gitea[921]: 2023/03/05 09:50:43 [T] AppWorkPath: /mnt/dietpi_userdata/gitea
Mar 05 09:50:43 DietPi gitea[921]: 2023/03/05 09:50:43 [T] Custom path: /mnt/dietpi_userdata/gitea/custom
Mar 05 09:50:43 DietPi gitea[921]: 2023/03/05 09:50:43 [T] Log path: /mnt/samba/nas/workspace/gitea/log
Mar 05 09:50:57 DietPi systemd[1]: gitea.service: Main process exited, code=exited, status=1/FAILURE
Mar 05 09:50:57 DietPi systemd[1]: gitea.service: Failed with result 'exit-code'.
Mar 05 09:58:03 DietPi systemd[1]: Started Gitea (DietPi).
Mar 05 09:58:03 DietPi gitea[1268]: 2023/03/05 09:58:03 [T] AppPath: /mnt/dietpi_userdata/gitea/gitea
Mar 05 09:58:03 DietPi gitea[1268]: 2023/03/05 09:58:03 [T] AppWorkPath: /mnt/dietpi_userdata/gitea
Mar 05 09:58:03 DietPi gitea[1268]: 2023/03/05 09:58:03 [T] Custom path: /mnt/dietpi_userdata/gitea/custom
Mar 05 09:58:03 DietPi gitea[1268]: 2023/03/05 09:58:03 [T] Log path: /mnt/samba/nas/workspace/gitea/log
Mar 05 09:58:11 DietPi systemd[1]: gitea.service: Main process exited, code=exited, status=1/FAILURE
Mar 05 09:58:11 DietPi systemd[1]: gitea.service: Failed with result 'exit-code'.
Mar 05 09:58:30 DietPi systemd[1]: Started Gitea (DietPi).
Mar 05 09:58:30 DietPi gitea[1353]: 2023/03/05 09:58:30 [T] AppPath: /mnt/dietpi_userdata/gitea/gitea
Mar 05 09:58:30 DietPi gitea[1353]: 2023/03/05 09:58:30 [T] AppWorkPath: /mnt/dietpi_userdata/gitea
Mar 05 09:58:30 DietPi gitea[1353]: 2023/03/05 09:58:30 [T] Custom path: /mnt/dietpi_userdata/gitea/custom
Mar 05 09:58:30 DietPi gitea[1353]: 2023/03/05 09:58:30 [T] Log path: /mnt/samba/nas/workspace/gitea/log
Mar 05 09:58:39 DietPi systemd[1]: gitea.service: Main process exited, code=exited, status=1/FAILURE
Mar 05 09:58:39 DietPi systemd[1]: gitea.service: Failed with result 'exit-code'.
Mar 05 10:18:56 DietPi systemd[1]: Started Gitea (DietPi).
Mar 05 10:18:59 DietPi gitea[1841]: 2023/03/05 10:18:59 [T] AppPath: /mnt/dietpi_userdata/gitea/gitea
Mar 05 10:18:59 DietPi gitea[1841]: 2023/03/05 10:18:59 [T] AppWorkPath: /mnt/dietpi_userdata/gitea
Mar 05 10:18:59 DietPi gitea[1841]: 2023/03/05 10:18:59 [T] Custom path: /mnt/dietpi_userdata/gitea/custom
Mar 05 10:18:59 DietPi gitea[1841]: 2023/03/05 10:18:59 [T] Log path: /mnt/samba/nas/workspace/gitea/log
Mar 05 10:19:11 DietPi systemd[1]: gitea.service: Main process exited, code=exited, status=1/FAILURE
Mar 05 10:19:11 DietPi systemd[1]: gitea.service: Failed with result 'exit-code'.
Mar 05 13:05:07 DietPi systemd[1]: Started Gitea (DietPi).
Mar 05 13:05:13 DietPi gitea[3633]: 2023/03/05 13:05:13 cmd/web.go:106:runWeb() [I] Starting Gitea on PID: 3633
Mar 05 13:05:13 DietPi gitea[3633]: 2023/03/05 13:05:13 ...s/setting/setting.go:614:deprecatedSetting() [E] Deprecated fallback `[server]` `LFS_CONTENT_PATH` present. Use `[lfs]` `PATH` instead. This fallback will be removed in v1.19.0
Mar 05 13:05:13 DietPi gitea[3633]: 2023/03/05 13:05:13 cmd/web.go:160:runWeb() [I] Global init
Mar 05 13:05:13 DietPi gitea[3633]: 2023/03/05 13:05:13 ...s/setting/setting.go:614:deprecatedSetting() [E] Deprecated fallback `[server]` `LFS_CONTENT_PATH` present. Use `[lfs]` `PATH` instead. This fallback will be removed in v1.19.0
Mar 05 13:05:14 DietPi gitea[3633]: 2023/03/05 13:05:14 routers/init.go:116:GlobalInitInstalled() [I] Git Version: 2.20.1, Wire Protocol Version 2 Enabled (home: /mnt/dietpi_userdata/gitea/data/home)
Mar 05 13:05:14 DietPi gitea[3633]: 2023/03/05 13:05:14 routers/init.go:117:GlobalInitInstalled() [I] AppPath: /mnt/dietpi_userdata/gitea/gitea
Mar 05 13:05:14 DietPi gitea[3633]: 2023/03/05 13:05:14 routers/init.go:118:GlobalInitInstalled() [I] AppWorkPath: /mnt/dietpi_userdata/gitea
Mar 05 13:05:14 DietPi gitea[3633]: 2023/03/05 13:05:14 routers/init.go:119:GlobalInitInstalled() [I] Custom path: /mnt/dietpi_userdata/gitea/custom
Mar 05 13:05:14 DietPi gitea[3633]: 2023/03/05 13:05:14 routers/init.go:120:GlobalInitInstalled() [I] Log path: /mnt/samba/nas/workspace/gitea/log
Mar 05 13:05:14 DietPi gitea[3633]: 2023/03/05 13:05:14 routers/init.go:121:GlobalInitInstalled() [I] Configuration file: /mnt/dietpi_userdata/gitea/custom/conf/app.ini
Mar 05 13:05:14 DietPi gitea[3633]: 2023/03/05 13:05:14 routers/init.go:122:GlobalInitInstalled() [I] Run Mode: Prod
Mar 05 13:05:20 DietPi systemd[1]: gitea.service: Main process exited, code=exited, status=1/FAILURE
Mar 05 13:05:20 DietPi systemd[1]: gitea.service: Failed with result 'exit-code'.
Mar 05 13:10:15 DietPi systemd[1]: Started Gitea (DietPi).
[... cut ...]
Mar 07 00:49:09 DietPi gitea[14628]: 2023/03/07 00:49:09 ...s/setting/setting.go:614:deprecatedSetting() [E] Deprecated fallback `[server]` `LFS_CONTENT_PATH` present. Use `[lfs]` `PATH` instead. This fallback will be removed in v1.19.0
Mar 07 00:49:09 DietPi gitea[14628]: 2023/03/07 00:49:09 ...s/setting/setting.go:1059:loadFromConf() [F] Expect user 'dietpi' but current user is: gitea
Mar 07 00:49:10 DietPi systemd[1]: gitea.service: Main process exited, code=exited, status=1/FAILURE
Mar 07 00:49:10 DietPi systemd[1]: gitea.service: Failed with result 'exit-code'.
```

MichaIng, 2023-03-07:
> Expect user 'dietpi' but current user is: gitea

This should have been adjusted on reinstall, or did you restore an old `app.ini` manually afterwards?
```sh
G_CONFIG_INJECT 'RUN_USER[[:blank:]]' 'RUN_USER = gitea' /mnt/dietpi_userdata/gitea/custom/conf/app.ini
```

anselal, 2023-03-07:
No, I did not restore an old `app.ini`.
It still won't start after this. Also the user was already set to `gitea` in `app.ini`
Shall I change `LFS_CONTENT_PATH` to something else ???

MichaIng, 2023-03-07:
> `[server]` `LFS_CONTENT_PATH` present. Use `[lfs]` `PATH` instead

1. Remove `LFS_CONTENT_PATH` from the `[server]` block.
2. Add `PATH` with same value to `[lfs]` block.

Joulinar, 2023-03-07:
Maybe system is running for quite some time as it's Buster still. Therefore an old config is used?

anselal, 2023-03-07:
No I am not getting any "errors" but the process still won't start.
Maybe I need to format the rpi and start all over ?

Joulinar, 2023-03-07:
pls reboot your system and share `journalctl -u gitea` again

anselal, 2023-03-08:
I reboot and here is the output of `journalctl -u gitea` and `service gitea status`

```sh
-- Logs begin at Wed 2023-03-08 08:56:11 EET, end at Wed 2023-03-08 08:57:29 EET. --
Mar 08 08:56:45 DietPi systemd[1]: Started Gitea (DietPi).
Mar 08 08:57:13 DietPi gitea[548]: 2023/03/08 08:57:13 cmd/web.go:106:runWeb() [I] Starting Gitea on PID: 548
Mar 08 08:57:13 DietPi gitea[548]: 2023/03/08 08:57:13 cmd/web.go:160:runWeb() [I] Global init
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:116:GlobalInitInstalled() [I] Git Version: 2.20.1, Wire Protocol Version 2 Enabled (home: /mnt/dietpi_userdata/g
itea/data/home)
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:117:GlobalInitInstalled() [I] AppPath: /mnt/dietpi_userdata/gitea/gitea
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:118:GlobalInitInstalled() [I] AppWorkPath: /mnt/dietpi_userdata/gitea
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:119:GlobalInitInstalled() [I] Custom path: /mnt/dietpi_userdata/gitea/custom
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:120:GlobalInitInstalled() [I] Log path: /mnt/samba/nas/workspace/gitea/log
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:121:GlobalInitInstalled() [I] Configuration file: /mnt/dietpi_userdata/gitea/custom/conf/app.ini
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:122:GlobalInitInstalled() [I] Run Mode: Prod
```

```sh
root@DietPi:~# service gitea status
● gitea.service - Gitea (DietPi)
   Loaded: loaded (/etc/systemd/system/gitea.service; disabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/gitea.service.d
           └─dietpi-services_edit.conf
   Active: failed (Result: exit-code) since Wed 2023-03-08 08:58:03 EET; 2min 18s ago
  Process: 548 ExecStart=/mnt/dietpi_userdata/gitea/gitea web (code=exited, status=1/FAILURE)
 Main PID: 548 (code=exited, status=1/FAILURE)

Mar 08 08:57:13 DietPi gitea[548]: 2023/03/08 08:57:13 cmd/web.go:160:runWeb() [I] Global init
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:116:GlobalInitInstalled() [I] Git Version: 2.20.1, Wire Protocol Version 2 Enabled (home: /mnt/dietpi_userdata/g
itea/data/home)
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:117:GlobalInitInstalled() [I] AppPath: /mnt/dietpi_userdata/gitea/gitea
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:118:GlobalInitInstalled() [I] AppWorkPath: /mnt/dietpi_userdata/gitea
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:119:GlobalInitInstalled() [I] Custom path: /mnt/dietpi_userdata/gitea/custom
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:120:GlobalInitInstalled() [I] Log path: /mnt/samba/nas/workspace/gitea/log
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:121:GlobalInitInstalled() [I] Configuration file: /mnt/dietpi_userdata/gitea/custom/conf/app.ini
Mar 08 08:57:14 DietPi gitea[548]: 2023/03/08 08:57:14 routers/init.go:122:GlobalInitInstalled() [I] Run Mode: Prod
Mar 08 08:58:03 DietPi systemd[1]: gitea.service: Main process exited, code=exited, status=1/FAILURE
Mar 08 08:58:03 DietPi systemd[1]: gitea.service: Failed with result 'exit-code'.
```

MichaIng, 2023-03-08:
It starts up without any error, but after one minute without any further log just exists 🤔. In my case it goes on:
```sh
Mar 08 19:51:23 VM-Bullseye systemd[1]: Started Gitea (DietPi).
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 cmd/web.go:106:runWeb() [I] Starting Gitea on PID: 2290
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 cmd/web.go:160:runWeb() [I] Global init
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 routers/init.go:116:GlobalInitInstalled() [I] Git Version: 2.30.2, Wire Protocol Version 2 Enabled (home: /mnt/dietpi_userdata/gitea/data/home)
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 routers/init.go:117:GlobalInitInstalled() [I] AppPath: /mnt/dietpi_userdata/gitea/gitea
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 routers/init.go:118:GlobalInitInstalled() [I] AppWorkPath: /mnt/dietpi_userdata/gitea
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 routers/init.go:119:GlobalInitInstalled() [I] Custom path: /mnt/dietpi_userdata/gitea/custom
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 routers/init.go:120:GlobalInitInstalled() [I] Log path: /var/log/gitea
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 routers/init.go:121:GlobalInitInstalled() [I] Configuration file: /mnt/dietpi_userdata/gitea/custom/conf/app.ini
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 routers/init.go:122:GlobalInitInstalled() [I] Run Mode: Prod
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 ...dules/setting/log.go:288:newLogService() [I] Gitea v1.18.5 built with GNU Make 4.1, go1.19.6 : bindata, sqlite, sqlite_unlock_notify
Mar 08 19:51:24 VM-Bullseye gitea[2290]: 2023/03/08 19:51:24 ...dules/setting/log.go:335:newLogService() [I] Gitea Log Mode: Console(Console:info)
...
Mar 08 19:51:25 VM-Bullseye gitea[2290]: 2023/03/08 19:51:25 cmd/web.go:220:listen() [I] [6408d92d-24] Listen: http://0.0.0.0:3000
```
Can you try to raise your log level? In my case debug logs start only after the service crashed in your case, but let's try:
```sh
G_CONFIG_INJECT '^LEVEL[[:blank:]]*=' 'LEVEL = debug' /mnt/dietpi_userdata/gitea/custom/conf/app.ini '\[log\]'
```
And could you post your whole `/mnt/dietpi_userdata/gitea/custom/conf/app.ini` after removing all passwords, secrets and tokens?

anselal, 2023-03-08:
I run your command to raise the log level, then rebooted.
This is the log of `journalctl -u gitea` after couple of minutes:

```sh
root@DietPi:~# journalctl -u gitea
-- Logs begin at Wed 2023-03-08 22:05:22 EET, end at Wed 2023-03-08 22:07:20 EET. --
Mar 08 22:05:57 DietPi systemd[1]: Started Gitea (DietPi).
Mar 08 22:06:22 DietPi gitea[580]: 2023/03/08 22:06:22 cmd/web.go:106:runWeb() [I] Starting Gitea on PID: 580
Mar 08 22:06:22 DietPi gitea[580]: 2023/03/08 22:06:22 cmd/web.go:160:runWeb() [I] Global init
Mar 08 22:06:23 DietPi gitea[580]: 2023/03/08 22:06:23 routers/init.go:116:GlobalInitInstalled() [I] Git Version: 2.20.1, Wire Protocol Version 2 Enabled (home: /mnt/dietpi_userdata/g
itea/data/home)
Mar 08 22:06:23 DietPi gitea[580]: 2023/03/08 22:06:23 routers/init.go:117:GlobalInitInstalled() [I] AppPath: /mnt/dietpi_userdata/gitea/gitea
Mar 08 22:06:23 DietPi gitea[580]: 2023/03/08 22:06:23 routers/init.go:118:GlobalInitInstalled() [I] AppWorkPath: /mnt/dietpi_userdata/gitea
Mar 08 22:06:23 DietPi gitea[580]: 2023/03/08 22:06:23 routers/init.go:119:GlobalInitInstalled() [I] Custom path: /mnt/dietpi_userdata/gitea/custom
Mar 08 22:06:23 DietPi gitea[580]: 2023/03/08 22:06:23 routers/init.go:120:GlobalInitInstalled() [I] Log path: /mnt/samba/nas/workspace/gitea/log
Mar 08 22:06:23 DietPi gitea[580]: 2023/03/08 22:06:23 routers/init.go:121:GlobalInitInstalled() [I] Configuration file: /mnt/dietpi_userdata/gitea/custom/conf/app.ini
Mar 08 22:06:23 DietPi gitea[580]: 2023/03/08 22:06:23 routers/init.go:122:GlobalInitInstalled() [I] Run Mode: Prod
Mar 08 22:07:09 DietPi systemd[1]: gitea.service: Main process exited, code=exited, status=1/FAILURE
Mar 08 22:07:09 DietPi systemd[1]: gitea.service: Failed with result 'exit-code'.
```

and this is my `app.ini`:

```ini
APP_NAME = Gitea: Git with a cup of tea
RUN_USER = gitea
RUN_MODE = prod

[security]
INTERNAL_TOKEN = secret
INSTALL_LOCK   = true
SECRET_KEY     = secret

[database]
DB_TYPE  = mysql
HOST     = 127.0.0.1:3306
NAME     = gitea
USER     = secret
PASSWD   = secret
SSL_MODE = disable
PATH     = data/gitea.db

[repository]
ROOT = /mnt/samba/nas/workspace/gitea/gitea-repositories

[server]
SSH_DOMAIN       = localhost
DOMAIN           = 192.168.1.9
HTTP_PORT        = 3000
ROOT_URL         = http://192.168.1.9:3000/
DISABLE_SSH      = false
SSH_PORT         = 22
LFS_START_SERVER = true
LFS_JWT_SECRET   = secret
OFFLINE_MODE     = false

[lfs]
PATH = /mnt/samba/nas/workspace/gitea/lfs

[mailer]
ENABLED = true
HOST    = secret
FROM    = secret
USER    = secret
PASSWD  = secret

[service]
REGISTER_EMAIL_CONFIRM            = true
ENABLE_NOTIFY_MAIL                = true
DISABLE_REGISTRATION              = true
ALLOW_ONLY_EXTERNAL_REGISTRATION  = false
ENABLE_CAPTCHA                    = false
REQUIRE_SIGNIN_VIEW               = true
DEFAULT_KEEP_EMAIL_PRIVATE        = true
DEFAULT_ALLOW_CREATE_ORGANIZATION = true
DEFAULT_ENABLE_TIMETRACKING       = true
NO_REPLY_ADDRESS                  = noreply.example.org

[picture]
DISABLE_GRAVATAR        = false
ENABLE_FEDERATED_AVATAR = false

[openid]
ENABLE_OPENID_SIGNIN = false
ENABLE_OPENID_SIGNUP = false

[session]
PROVIDER = file

[log]
MODE      = file
LEVEL = debug
ROOT_PATH = /mnt/samba/nas/workspace/gitea/log

[cron.update_mirrors]
SCHEDULE = @every 12h

[oauth2]
JWT_SECRET = secret
```

Joulinar, 2023-03-08:
Did you change path to a SAMBA share `ROOT = /mnt/samba/nas/`?

MichaIng, 2023-03-08:
Assure that the new service user `gitea` has write access to the data directory, i.e. the Samba share. Other than that, I do not have `[cron.update_mirrors]` and `[oauth2]`, but this high likely depends on added/enabled features, and I cannot imagine that invalid entries produce such a sudden crash without any log entry.

Talking about logs, `/mnt/samba/nas/workspace/gitea/log` does not contain more, does it? On fresh installs, we have `MODE = console`, hence it logs to journal only, not any file.

Unrelated, but a little enhancement you could do:
```sh
G_CONFIG_INJECT 'HOST[[:blank:]]' 'HOST = /run/mysqld/mysqld.sock' /mnt/dietpi_userdata/gitea/custom/conf/app.ini
```
I.e. let it use the UNIX socket to connect to MariaDB instead of TCP connection.

anselal, 2023-03-08:
> Did you change path to a SAMBA share `ROOT = /mnt/samba/nas/`?

It was there from the beginning. I use my nas to save the repos.
I have this setup for years.
The nas is mounted and the user has access to the folder

MichaIng, 2023-03-08:
> The nas is mounted and the user has access to the folder

The `gitea` user? Previously the service ran as `dietpi` user, so this has changed and hence permission needs to be verified.

anselal, 2023-03-08:
```sh
root@DietPi:/mnt/samba/nas/workspace# ls -al
total 8
drwxr-xr-x 3 root  root  4096 Mar  8 08:43 .
drwxr-xr-x 3 root  root  4096 Jun 10  2019 ..
drwxrwx--- 2 gitea gitea    0 Mar  8 08:35 gitea
root@DietPi:/mnt/samba/nas/workspace# 
```

anselal, 2023-03-08:
> Assure that the new service user `gitea` has write access to the data directory, i.e. the Samba share. Other than that, I do not have `[cron.update_mirrors]` and `[oauth2]`, but this high likely depends on added/enabled features, and I cannot imagine that invalid entries produce such a sudden crash without any log entry.
> 
> Talking about logs, `/mnt/samba/nas/workspace/gitea/log` does not contain more, does it? On fresh installs, we have `MODE = console`, hence it logs to journal only, not any file.
> 
> Unrelated, but a little enhancement you could do:
> 
> ```shell
> G_CONFIG_INJECT 'HOST[[:blank:]]' 'HOST = /run/mysqld/mysqld.sock' /mnt/dietpi_userdata/gitea/custom/conf/app.ini
> ```
> 
> I.e. let it use the UNIX socket to connect to MariaDB instead of TCP connection.

I just saw the logs in the git repo.

```sh
2023/03/08 22:07:08 routers/common/db.go:27:InitDBEngine() [I] ORM engine initialization attempt #10/10...
2023/03/08 22:07:08 cmd/web.go:163:runWeb() [I] PING DATABASE mysql
2023/03/08 22:07:08 models/db/engine.go:191:InitEngineWithMigration() [I] [SQL] SELECT `TABLE_NAME`, `ENGINE`, `AUTO_INCREMENT`, `TABLE_COMMENT` from `INFORMATION_SCHEMA`.`TABLES` WHERE `TABLE_SCHEMA`=? AND (`ENGINE`='MyISAM' OR `ENGINE` = 'InnoDB' OR `ENGINE` = 'TokuDB') [gitea] - 76.451ms
2023/03/08 22:07:08 ...ations/migrations.go:479:Migrate() [I] [SQL] SELECT `COLUMN_NAME`, `IS_NULLABLE`, `COLUMN_DEFAULT`, `COLUMN_TYPE`, `COLUMN_KEY`, `EXTRA`, `COLUMN_COMMENT`, `CHARACTER_MAXIMUM_LENGTH`, (INSTR(VERSION(), 'maria') > 0 && (SUBSTRING_INDEX(VERSION(), '.', 1) > 10 || (SUBSTRING_INDEX(VERSION(), '.', 1) = 10 && (SUBSTRING_INDEX(SUBSTRING(VERSION(), 4), '.', 1) > 2 || (SUBSTRING_INDEX(SUBSTRING(VERSION(), 4), '.', 1) = 2 && SUBSTRING_INDEX(SUBSTRING(VERSION(), 6), '-', 1) >= 7))))) AS NEEDS_QUOTE FROM `INFORMATION_SCHEMA`.`COLUMNS` WHERE `TABLE_SCHEMA` = ? AND `TABLE_NAME` = ? ORDER BY `COLUMNS`.ORDINAL_POSITION ASC [gitea version] - 30.675ms
2023/03/08 22:07:08 ...ations/migrations.go:479:Migrate() [I] [SQL] SELECT `INDEX_NAME`, `NON_UNIQUE`, `COLUMN_NAME` FROM `INFORMATION_SCHEMA`.`STATISTICS` WHERE `TABLE_SCHEMA` = ? AND `TABLE_NAME` = ? ORDER BY `SEQ_IN_INDEX` [gitea version] - 8.048ms
2023/03/08 22:07:09 ...bce556200f/engine.go:1244:Get() [I] [SQL] SELECT `id`, `version` FROM `version` WHERE `id`=? LIMIT 1 [1] - 16.521ms
2023/03/08 22:07:09 ...ations/migrations.go:526:Migrate() [I] Migration[93]: add email notification enabled preference to user
2023/03/08 22:07:09 ...ls/migrations/v93.go:15:addEmailNotificationEnabledToUser() [I] [SQL] SELECT `TABLE_NAME`, `ENGINE`, `AUTO_INCREMENT`, `TABLE_COMMENT` from `INFORMATION_SCHEMA`.`TABLES` WHERE `TABLE_SCHEMA`=? AND (`ENGINE`='MyISAM' OR `ENGINE` = 'InnoDB' OR `ENGINE` = 'TokuDB') [gitea] - 68.858ms
2023/03/08 22:07:09 ...bce556200f/engine.go:1128:Sync2() [I] [SQL] SELECT `COLUMN_NAME`, `IS_NULLABLE`, `COLUMN_DEFAULT`, `COLUMN_TYPE`, `COLUMN_KEY`, `EXTRA`, `COLUMN_COMMENT`, `CHARACTER_MAXIMUM_LENGTH`, (INSTR(VERSION(), 'maria') > 0 && (SUBSTRING_INDEX(VERSION(), '.', 1) > 10 || (SUBSTRING_INDEX(VERSION(), '.', 1) = 10 && (SUBSTRING_INDEX(SUBSTRING(VERSION(), 4), '.', 1) > 2 || (SUBSTRING_INDEX(SUBSTRING(VERSION(), 4), '.', 1) = 2 && SUBSTRING_INDEX(SUBSTRING(VERSION(), 6), '-', 1) >= 7))))) AS NEEDS_QUOTE FROM `INFORMATION_SCHEMA`.`COLUMNS` WHERE `TABLE_SCHEMA` = ? AND `TABLE_NAME` = ? ORDER BY `COLUMNS`.ORDINAL_POSITION ASC [gitea user] - 31.682ms
2023/03/08 22:07:09 ...bce556200f/engine.go:1128:Sync2() [I] [SQL] SELECT `INDEX_NAME`, `NON_UNIQUE`, `COLUMN_NAME` FROM `INFORMATION_SCHEMA`.`STATISTICS` WHERE `TABLE_SCHEMA` = ? AND `TABLE_NAME` = ? ORDER BY `SEQ_IN_INDEX` [gitea user] - 7.911ms
2023/03/08 22:07:09 ...bce556200f/engine.go:1128:Sync2() [I] [SQL] ALTER TABLE `user` ADD `email_notifications_preference` VARCHAR(20) DEFAULT 'enabled' NOT NULL [] - 22.931ms
2023/03/08 22:07:09 routers/init.go:68:mustInitCtx() [F] code.gitea.io/gitea/routers/common.InitDBEngine(ctx) failed: migrate: migration[93]: add email notification enabled preference to user failed: Error 1118: Row size too large. The maximum row size for the used table type, not counting BLOBs, is 8126. This includes storage overhead, check the manual. You have to change some columns to TEXT or BLOBs
```

Joulinar, 2023-03-08:
> ` failed: migrate: migration[93]:`

looks like Gitea is trying to migrate some database table, which failed. Maybe something to report to Gitea developer?

MichaIng, 2023-03-08:
Indeed:
```sh
2023/03/08 22:07:09 ...bce556200f/engine.go:1128:Sync2() [I] [SQL] ALTER TABLE `user` ADD `email_notifications_preference` VARCHAR(20) DEFAULT 'enabled' NOT NULL [] - 22.931ms
2023/03/08 22:07:09 routers/init.go:68:mustInitCtx() [F] code.gitea.io/gitea/routers/common.InitDBEngine(ctx) failed: migrate: migration[93]: add email notification enabled preference to user failed: Error 1118: Row size too large. The maximum row size for the used table type, not counting BLOBs, is 8126. This includes storage overhead, check the manual. You have to change some columns to TEXT or BLOBs
```
I'm not sure whether I understand it correctly, since the size of the added row is much smaller than 8126 with a key `email_notifications_preference` and a value `enabled` of type `VARCHAR(20)` (max 20 characters). Probably other entries of this table are too large, however this was possible in the first place. To have a look into the table:
```sh
mysql -e 'select * from gitea.user'
```

Joulinar, 2023-03-08:
> I found an issue

can you share this one? Just for reference. :)

MichaIng, 2023-03-08:
Jep, just found the same: https://github.com/go-gitea/gitea/issues/12777

I though Gitea would do such automatically, every other software does 🤔. We may want to add this to our reinstall code.

anselal, 2023-03-08:
> Jep, just found the same: [go-gitea/gitea#12777](https://github.com/go-gitea/gitea/issues/12777)
> 
> I though Gitea would do such automatically, every other software does thinking. We may want to add this to our reinstall code.
> 
> ... ah, I think it is a one time step needed for old installs on old Debian versions, where the MariaDB table defaults were different. I.e. try this:
> 
> ```shell
> sudo -u gitea /mnt/dietpi_userdata/gitea/gitea convert
> ```

I am getting this `This account is currently not available.` so I edited `app.ini` changed the user to dietpi and run the command with dietpi instead. I guess this happens cause the user gitea has nologin ?

anselal, 2023-03-08:
Nvm, ignore the last post, I run the command using `su - -c "/mnt/dietpi_userdata/gitea/gitea convert" gitea`.
Using sudo it run correct.
It took some time to finish, after that I got a log of logs in `/mnt/samba/nas/workspace/gitea/log/gitea.log`
When they finished I tried again to access `gitea` but the service was still not running...

anselal, 2023-03-08:
The result of `gitea convert` was, besides `ALTER TABLE` statements, `Converted successfully, please confirm your database's character set is now utf8mb4`

MichaIng, 2023-03-08:
Do not use `su`, use `sudo`. The user cannot be used for a login shell, but for a non-interactive shell it can.

anselal, 2023-03-08:
> The result of `gitea convert` was, besides `ALTER TABLE` statements, `Converted successfully, please confirm your database's character set is now utf8mb4`

I checked and the character set is `utf8mb4`

MichaIng, 2023-03-08:
And the user changed back to `gitea`? If it still fails, check again `journalctl -u gitea` and log file entries. Probably its easier to move everything to journal:
```sh
G_CONFIG_INJECT 'MODE[[:blank:]]' 'MODE = console' /mnt/dietpi_userdata/gitea/custom/conf/app.ini
systemctl restart gitea
journalctl -u gitea
```

anselal, 2023-03-08:
Now it started.. :shrug:

anselal, 2023-03-08:
Thank you very much for your time and support.
Hope it doesn't break again any time soon :sweat_smile:

MichaIng, 2023-03-10:
From MariaDB 10.1 to 10.3 (Debian Stretch to Buster), there were quite some significant changes around the defaults, so for now this was a one-time requirement. Makes sense to keep Gitea updated more regularly and read the release notes if it is a major or minor upgrade, where such things are usually mentioned. E.g. next is v1.19 with some breaking changes: https://github.com/go-gitea/gitea/releases

closed 2023-03-10 by MichaIng