MichaIng/DietPi #6219: Error installing portainer

lmerega, 2023-03-07:
#### Details:
- Date           | Tue Mar  7 12:50:26 CET 2023
- Bug report     | f19368ee-888d-42b0-97f5-dab4a770c5e9
- DietPi version | v8.14.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux DietWyse 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `docker run -d -p 9002:9000 --name=portainer --restart=always -v /run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer-ce`
- Exit code      | 127
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Unable to find image 'portainer/portainer-ce:latest' locally
latest: Pulling from portainer/portainer-ce
772227786281: Pulling fs layer
96fd13befc87: Pulling fs layer
b733663f020c: Pulling fs layer
9fbfa87be55d: Pulling fs layer
9fbfa87be55d: Waiting
772227786281: Download complete
96fd13befc87: Verifying Checksum
96fd13befc87: Download complete
772227786281: Pull complete
9fbfa87be55d: Verifying Checksum
9fbfa87be55d: Download complete
96fd13befc87: Pull complete
b733663f020c: Download complete
b733663f020c: Pull complete
9fbfa87be55d: Pull complete
Digest: sha256:9fa1ec78b4e29d83593cf9720674b72829c9cdc0db7083a962bc30e64e27f64e
Status: Downloaded newer image for portainer/portainer-ce:latest
9b3c2b14ec6daa3b941b7c2c7405d12f2cf5d70653c8e58167421fa8d8f8f026
docker: Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: unable to apply apparmor profile: apparmor failed to apply profile: write /proc/self/attr/apparmor/exec: no such file or directory: unknown.
```

Joulinar, 2023-03-07:
known behavior of Docker software https://github.com/MichaIng/DietPi/issues/6126

closed 2023-03-07 by MichaIng

lmerega, 2023-03-08:
Thank you very much!