MichaIng/DietPi #6223: Docker + Portainer: Macvlan support possibly flawed?

holocronology, 2023-03-07:
> If the DietPi (which I greatly approve of mind you) implementation of Docker does not support

I meant the kernel, not Docker itself, which may not support all features. However, also current Quartz64 images do fully support all this. Only the highly experimental RISC-V VisionFive 2 does not, but Docker doesn't start up there at all.

Did you check whether anything is listening on port 80 on the host? How did you setup the redirects? Are the VLANs up as expected (`ip a`)?

**EDIT: Ah, let's not spam this software request. Feel free to open a new issue for investigating this case, instead.**

_Originally posted by @MichaIng in https://github.com/MichaIng/DietPi/issues/4417#issuecomment-1458858738_

Joulinar, 2023-03-07:
ok. Just played with `macvlan `feature and it is working without issues. NPM container is starting as dedicated system and is reachable within my network on its individual IP address. There is even no port conflict between NPM conatiner and PiHole. Both use port 80. From inside the container, I can reach other systems 

```
[root@docker-8a35d78599c2:/app]# ping 192.168.0.1
PING 192.168.0.1 (192.168.0.1) 56(84) bytes of data.
64 bytes from 192.168.0.1: icmp_seq=1 ttl=64 time=0.922 ms
64 bytes from 192.168.0.1: icmp_seq=2 ttl=64 time=0.824 ms
64 bytes from 192.168.0.1: icmp_seq=3 ttl=64 time=0.756 ms
^C
--- 192.168.0.1 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 4ms
rtt min/avg/max/mdev = 0.756/0.834/0.922/0.068 ms
```

and

```
[root@docker-8a35d78599c2:/app]# ping 192.168.0.2 
PING 192.168.0.2 (192.168.0.2) 56(84) bytes of data.
64 bytes from 192.168.0.2: icmp_seq=1 ttl=64 time=2.07 ms
64 bytes from 192.168.0.2: icmp_seq=2 ttl=64 time=0.573 ms
^C
--- 192.168.0.2 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 2ms
rtt min/avg/max/mdev = 0.573/1.323/2.074/0.751 ms
```

BUT, there is always a BUT, NPM container is not able to reach the host system (IP 192.168.0.17) where Docker is running on

```
[root@docker-8a35d78599c2:/app]# ping 192.168.0.17
PING 192.168.0.17 (192.168.0.17) 56(84) bytes of data.
From 192.168.0.253 icmp_seq=1 Destination Host Unreachable
From 192.168.0.253 icmp_seq=2 Destination Host Unreachable
From 192.168.0.253 icmp_seq=3 Destination Host Unreachable
From 192.168.0.253 icmp_seq=4 Destination Host Unreachable
From 192.168.0.253 icmp_seq=5 Destination Host Unreachable
From 192.168.0.253 icmp_seq=6 Destination Host Unreachable
^C
--- 192.168.0.17 ping statistics ---
8 packets transmitted, 0 received, +6 errors, 100% packet loss, time 152ms
pipe 3
```

And this is your main issue. The container is not able to comunicate with the host. And there we go https://forums.docker.com/t/host-and-containers-cannot-communicate-macvlan/112968/2

So it's working as expected and further Docker configuration needed to allow communication between Docker host and Conatiner connected to macvlan.

holocronology, 2023-03-08:
I'm not trying to communicate with the host, I was trying to communicate with another device on the host nework.

The same issue appears to be happening even without macvlan. So, it's likely not a macvlan issue, but something else.

holocronology, 2023-03-08:
> The container is not able to communicate with the host. And there we go https://forums.docker.com/t/host-and-containers-cannot-communicate-macvlan/112968/2

I believe that this does not apply when creating the macvlan within Portainer, as it is a two step process where you create a configuration and then create the vlan. It effectively creates the 2nd virt interface reference in this posting above.

They also point out the reason for the need for the macvlan to set up port forwarding on a firewall/router/gateway.

Joulinar, 2023-03-08:
Yesterday I tried `macvlan `without issue. I used the official Docker docs to create and setup `macvlan` interface. As well no issues to assign the new interface to NPM container. But I used a STATIC IP for NPM container. Because without STATIC IP, Docker assigned a random IP from my network that was already in use, causing some address conflicts. This way I was able to setup NPM to proxy request to other web apps running on different server inside my LAN.

holocronology, 2023-03-08:
> Because without STATIC IP, Docker assigned a random IP from my network that was already in use, causing some address conflicts.

I believe this is why the documentation for macvlan specifically states to set an IP range that is not currently in use by the network, as the first available in IP in the defined range will be assigned.
Within Portainer, it is fairly easy to change to a static IP, if not assigning a static IP by mac address in your router.

Joulinar, 2023-03-08:
Don't think that the IP is assigned by the DHCP server. It's done by Docker itself. 

Anyway STATIC IP is highly recommended for the container to ensure same IP address all the time. Otherwise no guarantee that your port forwarding stay valid if IP of container is changing.

holocronology, 2023-03-08:
> 

Yes, read my comment above.

MichaIng, 2023-04-03:
Anything left here? Does it work in the meantime? Also, which SBC is it actually?

holocronology, 2023-04-04:
I haven't come back to this, as I basically gave up trying to make it work. I move NPM to an Unraid server and it works flawlessly.

Joulinar, 2023-04-04:
ok I'm going to close this one. Feel free to reopen if needed.

closed 2023-04-04 by Joulinar