MichaIng/DietPi #6224: Error trying to install HotSpot with Wifi Stick

as-shura, 2023-03-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
bullseye
```
- Kernel version | `uname -a`
```
Linux DietPi 6.1.11-meson64 #23.02.2 SMP PREEMPT Sat Feb 18 00:07:55 UTC 2023 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
Odroid C2 (aarch64)
```
- Power supply used | (EG: 5V 1A RAVpower)

https://sha256systems.eu/PSU-ODROID-C1-C0-C2

- SD card used | (EG: SanDisk ultra)

```
SanDisk ultra 32 GB https://www.amazon.de/SanDisk-microSDHC-Speicherkarte-SD-Adapter-App-Leistung/dp/B08GY9NYRM/ref=asc_df_B08GY9NYRM/?tag=googshopde-21&linkCode=df0&hvadid=474234094943&hvpos=&hvnetw=g&hvrand=17165416810116209753&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1004607&hvtargid=pla-1067844298997&psc=1&th=1&psc=1
```

#### Steps to reproduce

1. Use dietpi-software to install HotSpot

#### Expected behaviour

- DHCP server should work

#### Actual behaviour

```
● isc-dhcp-server.service - LSB: DHCP server
     Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
     Active: failed (Result: exit-code) since Tue 2023-03-07 22:23:58 CET; 49ms ago
       Docs: man:systemd-sysv-generator(8)
    Process: 13361 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)
        CPU: 131ms

Mar 07 22:23:56 DietPi dhcpd[13376]: before submitting a bug.  These pages explain the proper
Mar 07 22:23:56 DietPi dhcpd[13376]: process and the information we find helpful for debugging.
Mar 07 22:23:56 DietPi dhcpd[13376]:
Mar 07 22:23:56 DietPi dhcpd[13376]: exiting.
Mar 07 22:23:58 DietPi isc-dhcp-server[13361]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ...
Mar 07 22:23:58 DietPi isc-dhcp-server[13381]:  failed!
Mar 07 22:23:58 DietPi isc-dhcp-server[13382]:  failed!
Mar 07 22:23:58 DietPi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
Mar 07 22:23:58 DietPi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
Mar 07 22:23:58 DietPi systemd[1]: Failed to start LSB: DHCP server.
Processing triggers for libc-bin (2.31-13+deb11u5) ...
```

as-shura, 2023-03-07:
I am able to see the SSID from the Network and connect to it but there is no IP address assigned to my mobile phone due to this error.

MichaIng, 2023-03-07:
Please show the full log of the DHCP server:
```sh
journalctl -u isc-dhcp-server
```

Joulinar, 2023-03-07:
has the system been rebooted after hotspot installation?

as-shura, 2023-03-08:
```
-- Journal begins at Tue 2023-03-07 22:48:54 CET, ends at Wed 2023-03-08 01:30:07 CET. --
Mar 07 22:49:02 DietPi systemd[1]: Starting LSB: DHCP server...
Mar 07 22:49:02 DietPi isc-dhcp-server[1071]: Launching IPv4 server only.
Mar 07 22:49:02 DietPi dhcpd[1080]: Internet Systems Consortium DHCP Server 4.4.1
Mar 07 22:49:02 DietPi dhcpd[1080]: Copyright 2004-2018 Internet Systems Consortium.
Mar 07 22:49:02 DietPi dhcpd[1080]: All rights reserved.
Mar 07 22:49:02 DietPi dhcpd[1080]: For info, please visit https://www.isc.org/software/dhcp/
Mar 07 22:49:02 DietPi dhcpd[1086]: Internet Systems Consortium DHCP Server 4.4.1
Mar 07 22:49:02 DietPi dhcpd[1086]: Copyright 2004-2018 Internet Systems Consortium.
Mar 07 22:49:02 DietPi dhcpd[1086]: All rights reserved.
Mar 07 22:49:02 DietPi dhcpd[1086]: For info, please visit https://www.isc.org/software/dhcp/
Mar 07 22:49:02 DietPi dhcpd[1086]: Wrote 0 leases to leases file.
Mar 07 22:49:02 DietPi dhcpd[1086]: Server starting service.
Mar 07 22:49:04 DietPi isc-dhcp-server[1071]: Starting ISC DHCPv4 server: dhcpd.
Mar 07 22:49:04 DietPi systemd[1]: Started LSB: DHCP server.
Mar 07 22:55:17 DietPi dhcpd[1086]: receive_packet failed on wlan0: Network is down
```

as-shura, 2023-03-08:
> has the system been rebooted after hotspot installation?

No no restart just quit

as-shura, 2023-03-08:
Retried and got this

```
[FAILED] DietPi-Config | Unable to continue, DietPi-Config will now terminate.

root@DietPi:~# systemctl status isc-dhcp-server.service
● isc-dhcp-server.service - LSB: DHCP server
     Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
     Active: failed (Result: exit-code) since Wed 2023-03-08 03:01:40 CET; 16s ago
       Docs: man:systemd-sysv-generator(8)
    Process: 15324 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)
        CPU: 159ms

Mar 08 03:01:38 DietPi dhcpd[15339]: before submitting a bug.  These pages explain the proper
Mar 08 03:01:38 DietPi dhcpd[15339]: process and the information we find helpful for debugging.
Mar 08 03:01:38 DietPi dhcpd[15339]:
Mar 08 03:01:38 DietPi dhcpd[15339]: exiting.
Mar 08 03:01:40 DietPi isc-dhcp-server[15324]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ...
Mar 08 03:01:40 DietPi isc-dhcp-server[15345]:  failed!
Mar 08 03:01:40 DietPi isc-dhcp-server[15346]:  failed!
Mar 08 03:01:40 DietPi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
Mar 08 03:01:40 DietPi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
Mar 08 03:01:40 DietPi systemd[1]: Failed to start LSB: DHCP server.
```

as-shura, 2023-03-08:
```
Mar 08 03:03:22 DietPi isc-dhcp-server[15457]: Launching IPv4 server only.
Mar 08 03:03:22 DietPi dhcpd[15466]: Internet Systems Consortium DHCP Server 4.4.1
Mar 08 03:03:22 DietPi dhcpd[15466]: Copyright 2004-2018 Internet Systems Consortium.
Mar 08 03:03:22 DietPi dhcpd[15466]: All rights reserved.
Mar 08 03:03:22 DietPi dhcpd[15466]: For info, please visit https://www.isc.org/software/dhcp/
Mar 08 03:03:22 DietPi dhcpd[15472]: Internet Systems Consortium DHCP Server 4.4.1
Mar 08 03:03:22 DietPi dhcpd[15472]: Copyright 2004-2018 Internet Systems Consortium.
Mar 08 03:03:22 DietPi dhcpd[15472]: All rights reserved.
Mar 08 03:03:22 DietPi dhcpd[15472]: For info, please visit https://www.isc.org/software/dhcp/
Mar 08 03:03:22 DietPi dhcpd[15472]: Wrote 0 leases to leases file.
Mar 08 03:03:22 DietPi dhcpd[15472]:
Mar 08 03:03:22 DietPi dhcpd[15472]: No subnet declaration for wlan0 (no IPv4 addresses).
Mar 08 03:03:22 DietPi dhcpd[15472]: ** Ignoring requests on wlan0.  If this is not what
Mar 08 03:03:22 DietPi dhcpd[15472]:    you want, please write a subnet declaration
Mar 08 03:03:22 DietPi dhcpd[15472]:    in your dhcpd.conf file for the network segment
Mar 08 03:03:22 DietPi dhcpd[15472]:    to which interface wlan0 is attached. **
Mar 08 03:03:22 DietPi dhcpd[15472]:
Mar 08 03:03:22 DietPi dhcpd[15472]:
Mar 08 03:03:22 DietPi dhcpd[15472]: Not configured to listen on any interfaces!
Mar 08 03:03:22 DietPi dhcpd[15472]:
Mar 08 03:03:22 DietPi dhcpd[15472]: If you think you have received this message due to a bug rather
Mar 08 03:03:22 DietPi dhcpd[15472]: than a configuration issue please read the section on submitting
Mar 08 03:03:22 DietPi dhcpd[15472]: bugs on either our web page at www.isc.org or in the README file
Mar 08 03:03:22 DietPi dhcpd[15472]: before submitting a bug.  These pages explain the proper
Mar 08 03:03:22 DietPi dhcpd[15472]: process and the information we find helpful for debugging.
Mar 08 03:03:22 DietPi dhcpd[15472]:
Mar 08 03:03:22 DietPi dhcpd[15472]: exiting.
Mar 08 03:03:24 DietPi isc-dhcp-server[15457]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ...
Mar 08 03:03:24 DietPi isc-dhcp-server[15477]:  failed!
Mar 08 03:03:24 DietPi isc-dhcp-server[15478]:  failed!
Mar 08 03:03:24 DietPi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit isc-dhcp-server.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 1.
Mar 08 03:03:24 DietPi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit isc-dhcp-server.service has entered the 'failed' state with result 'exit-code'.
Mar 08 03:03:24 DietPi systemd[1]: Failed to start LSB: DHCP server.
░░ Subject: A start job for unit isc-dhcp-server.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit isc-dhcp-server.service has finished with a failure.
░░
░░ The job identifier is 687 and the job result is failed.
```

Joulinar, 2023-03-08:
Did you reboot the system in meantime?

as-shura, 2023-03-08:
Yes tried many time and the error still persists.

Joulinar, 2023-03-08:
> `Mar 08 03:03:22 DietPi dhcpd[15472]: No subnet declaration for wlan0 (no IPv4 addresses).`

That seems to be the issue. Can you share following

```
ip a
cat /etc/hostapd/hostapd.conf
cat /etc/dhcp/dhcpd.conf
cat /etc/network/interfaces
```

MichaIng, 2023-03-08:
It seems it cannot find the configured interface. Please check:
```sh
cat /etc/default/isc-dhcp-server
```
which must match the intended WiFi interface shown in `ip a` output.

as-shura, 2023-03-08:
> ```shell
> cat /etc/default/isc-dhcp-server
> ```

```
INTERFACESv4="wlan0"
```

as-shura, 2023-03-08:
> > `Mar 08 03:03:22 DietPi dhcpd[15472]: No subnet declaration for wlan0 (no IPv4 addresses).`
> 
> That seems to be the issue. Can you share following
> 
> ```
> ip a
> cat /etc/hostapd/hostapd.conf
> cat /etc/dhcp/dhcpd.conf
> cat /etc/network/interfaces
> ```

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 00:1e:06:34:4d:61 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.4/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 54:af:97:d6:23:f7 brd ff:ff:ff:ff:ff:ff
    inet 192.168.42.1/24 brd 192.168.42.255 scope global wlan0
       valid_lft forever preferred_lft forever
       ```

as-shura, 2023-03-08:
Isn't this supposed to be dhcp instead of static ?

```
iface wlan0 inet static
```

as-shura, 2023-03-09:
This is really interesting !!!!!

I just flashed another microSD card and ran the same OS with another odroid-c2 board with the USB Wifi stick and it worked. I am wondering what is wrong with my original installation.

Joulinar, 2023-03-09:
> ```
> pre-up iw dev wlan0 set power_save off
> post-down iw dev "WIFI_DEV_IFACE set power_save on
> ```

But these lines are not added by our tools. At least I'm not aware of.

as-shura, 2023-03-10:
> But these lines are not added by our tools. At least I'm not aware of.

It is actually in fact because I never touch those line personally. The dietpi-software made it and initialized the settings.

closed 2023-03-10 by as-shura

MichaIng, 2023-03-10:
A bug/typo our end. Fixed with: https://github.com/MichaIng/DietPi/commit/b47d218

Many thanks for reporting!