MichaIng/DietPi #6226: Koel installation fails

bartolus39, 2023-03-08:
#### Details:
- Date           | Wed Mar  8 12:33:22 CET 2023
- DietPi version | v8.14.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | `sudo -u koel php7.3 artisan koel:init -n --no-assets`
- Exit code      | 255
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Composer detected issues in your platform:

Your Composer dependencies require a PHP version ">= 7.4.0". You are running 7.3.31-1~deb10u3.

PHP Fatal error:  Composer detected issues in your platform: Your Composer dependencies require a PHP version ">= 7.4.0". You are running 7.3.31-1~deb10u3. in /mnt/dietpi_userdata/koel/vendor/composer/platform_check.php on line 24
```

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2023-03-08:
Basically you are running old Debian Buster and it would be needed to upgrade your Debian version to Bullseye. https://dietpi.com/blog/?p=811

@MichaIng maybe we would need to block Koel on old Buster systems.

MichaIng, 2023-03-08:
We already block Koel 6 on Bullseye and Buster, but I wasn't aware that Koel 5 doesn't run on Buster either? Will verify and in case limit the version to latest 4 there.

MichaIng, 2023-03-08:
Koel 5 => Laravel 8 => https://laravel.com/docs/8.x/deployment#server-requirements => PHP 7.3 🤔.

So it must be some other dependency, likely unintended as this would usually have caused a new major version release.

... v5.1.12 introduced this version check, hence v5.1.11 should be the last which works on Buster.

MichaIng, 2023-03-08:
With v5.1.x I get
```
In AppServiceProvider.php line 28:

  syntax error, unexpected 'fn' (T_STRING), expecting :: (T_PAAMAYIM_NEKUDOTAYIM)
```
which is interesting, since this kind of error usually happens after an upgrade to PHP 7.4, while downgrading to PHP 7.3 is a common workaround. Probably the error is the exact opposite way around here, not sure.

However, hence v5.0.2 is the latest functional one with PHP 7.3.

MichaIng, 2023-03-08:
Fixed with: https://github.com/MichaIng/DietPi/commit/f8db516

closed 2023-03-10 by MichaIng