MichaIng/DietPi #6227: Debian Bookworm testing and upgrade script feedback

MichaIng, 2023-03-08:
This issue aims to collect feedback, testing results and info around the upcoming Debian Bookworm release.

Please read our article to get an overview: T.b.d.

Additional info can be found on our wiki: https://github.com/MichaIng/DietPi/wiki/Debian-Bookworm-testing

manilx, 2023-03-15:
Used the script to update 6 VM's. On 4 it worked without issue on 2 it errored out.

grub-pc: Running grub-install ...
/dev/sda does not exist, so cannot grub-install to it!
You must correct your GRUB install devices before proceeding:

  DEBIAN_FRONTEND=dialog dpkg --configure grub-pc
  dpkg --configure -a
dpkg: error processing package grub-pc (--configure):
 installed grub-pc package post-installation script subprocess returned error exit status 1
Setting up python3-six (1.16.0-4) ...

Strange thing is they all had a base template......

manilx, 2023-03-15:
`Setting up grub-pc (2.06-8) ...
Replacing config file /etc/default/grub with new version
grub-pc: Running grub-install ...
/dev/sda does not exist, so cannot grub-install to it!
You must correct your GRUB install devices before proceeding:

  DEBIAN_FRONTEND=dialog dpkg --configure grub-pc
  dpkg --configure -a
dpkg: error processing package grub-pc (--configure):
 installed grub-pc package post-installation script subprocess returned error exit status 1`

Joulinar, 2023-03-15:
can you share following from the 2 VM's having an issue and another example from a system working

```
lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
```

manilx, 2023-03-15:
Working: 
NAME    FSTYPE LABEL  SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
xvda                    8G  0 disk
|-xvda1 ext4            7G  0 part /          42e48e80-01                          7e5234cc-eb02-4544-b50b-9b14cda36655
|-xvda2                 1K  0 part            42e48e80-02
`-xvda5 swap          975M  0 part            42e48e80-05                          b77c54c2-e515-4c40-b2a5-6297b1a2a944

Not working:
NAME    FSTYPE LABEL  SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
xvda                    8G  0 disk
├─xvda1 ext4            7G  0 part /          42e48e80-01                          7e5234cc-eb02-4544-b50b-9b14cda36655
├─xvda2                 1K  0 part            42e48e80-02
└─xvda5 swap          975M  0 part            42e48e80-05                          b77c54c2-e515-4c40-b2a5-6297b1a2a944

manilx, 2023-03-15:
Sorry. This is the non working.

NAME    FSTYPE LABEL  SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
xvda                    8G  0 disk
|-xvda1 ext4            7G  0 part /          42e48e80-01                          7e5234cc-eb02-4544-b50b-9b14cda36655
|-xvda2                 1K  0 part            42e48e80-02
`-xvda5 swap          975M  0 part            42e48e80-05                          b77c54c2-e515-4c40-b2a5-6297b1a2a944

And working:
NAME    FSTYPE LABEL  SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
xvda                    8G  0 disk
├─xvda1 ext4            7G  0 part /          42e48e80-01                          7e5234cc-eb02-4544-b50b-9b14cda36655
├─xvda2                 1K  0 part            42e48e80-02
└─xvda5 swap          975M  0 part            42e48e80-05                          b77c54c2-e515-4c40-b2a5-6297b1a2a944

There's a difference in xvda5

manilx, 2023-03-15:
Actually that swap is not needed... Guess it's a leftover off debian 11 install (installed it from debian 11 iso) and then converted via dietpi script.

MichaIng, 2023-03-15:
The GRUB package needs to know where it can flash GRUB to. Since we cannot know the device name you flash the image to, we use ship them with `/dev/sda` as most common root device name. On first GRUB update, you however needed to change this manually, if it was wrong. Looks like you did this on 4 VMs already, but not yet on 2 of them. Run manually once:
```sh
dpkg --configure grub-pc
```
and select the correct device name, which seems to be `/dev/xvda` in your case.

Since DietPi v7.5, it is adjusted automatically on first boot: https://github.com/MichaIng/DietPi/commit/d430094
So probably your two affected VMs are the only older ones. Or if the others are older as well, probably the root device name changed after first boot, e.g. if you changed some settings on the hypervisor or moved the drive to another VM.

MichaIng, 2023-03-15:
Try
```sh
dpkg --configure -a
```
of
```sh
apt -f install
```
or
```sh
apt update
apt install grub-pc
```
instead. Not sure which state your packages are exactly on each VM.

manilx, 2023-03-15:
Doesn't help.

┌──(root💀Wireguard)-[~]
└─# apt -f install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
┌──(root💀Wireguard)-[~]
└─# apt update
apt install grub-pc
Hit:1 https://deb.debian.org/debian bullseye InRelease
Get:2 https://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Get:4 https://deb.debian.org/debian bullseye-backports InRelease [49.0 kB]
Fetched 93.0 kB in 1s (97.8 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
grub-pc is already the newest version (2.06-3~deb11u5).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
┌──(root💀Wireguard)-[~]
└─# dpkg --configure -a
┌──(root💀Wireguard)-[~]
└─# dpkg --configure grub-pc
dpkg: error processing package grub-pc (--configure):
 package grub-pc is already installed and configured
Errors were encountered while processing:
 grub-pc

MichaIng, 2023-03-15:
This is on a system where the Bookworm upgrade script just failed? Seems all fine. You can force the configuration with
```sh
dpkg-reconfigure grub-pc
```

manilx, 2023-03-15:
This a bullseye system before the upgrade script (rolled back) because the update script was failing (this is why I reported here as asked on the announcement).

Joulinar, 2023-03-15:
But you have the issue with Grub already on the Bullseye system?

manilx, 2023-03-15:
No issues whatsoever! Runs fine.

manilx, 2023-03-15:
Tried the upgrade script again. Errored out.
Entered "dpkg-reconfigure grub-pc" and got 

/usr/sbin/dpkg-reconfigure: grub-pc is broken or not fully installed

manilx, 2023-03-15:
Now I entered "dpkg --configure -a" and I could choose the device.

Seems OK now.

Joulinar, 2023-03-15:
But as shown above, `dpkg --configure grub-pc` was failing on Bullseye?

Joulinar, 2023-03-15:
This would mean the issue was already there on Bullseye and just popped up due to the Bookworm upgrade. Right?

manilx, 2023-03-15:
Bullseye was running fine and I had no issue booting..... So I can't tell if there was an issue.
Tried the upgrade script and it failed.

MichaIng, 2023-03-15:
Makes all sense then. You could have done `dpkg-reconfigure grub-pc` already before running the script, to fix the debconf entry in the first place.

manilx, 2023-03-15:
OK.... Doesn't make to me but then. As I was not experiencing any issue booting.
Anyway fixed now. Can't tell what /when started it.

Joulinar, 2023-03-15:
If you find time, try to restore your backup und fix grub on Bullseye before running the update. This would be helpful and appreciated.

manilx, 2023-03-15:
I would of I could!! Backup is gone and new one created.

MichaIng, 2023-03-15:
> As I was not experiencing any issue booting.

Of course not, because GRUB is shipped with our image. It is only the GRUB package, on upgrade or reinstall, which does not know where to flash GRUB to, if the pre-configured device `/dev/sda` does not exist. And that package is very rarely upgraded on a stable Debian release.

Joulinar, 2023-03-26:
PHP8.2 is not working on ARMv6

```
root@DietPi1:~# /usr/sbin/php-fpm8.2 --nodaemonize --fpm-config /etc/php/8.2/fpm/php-fpm.conf
Illegal instruction
root@DietPi1:~#

```

Joulinar, 2023-04-04:
Domoticz is not working on Bookworm because it requires `ibssl1.1`. But it is not available on Bookworm 

```
root@DietPiOr5:~# journalctl -u domoticz.service
Apr 05 00:46:38 DietPiOr5 systemd[1]: Started domoticz.service - Domoticz (DietPi).
Apr 05 00:46:38 DietPiOr5 domoticz[2893]: /opt/domoticz/domoticz: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
Apr 05 00:46:38 DietPiOr5 systemd[1]: domoticz.service: Main process exited, code=exited, status=127/n/a
Apr 05 00:46:38 DietPiOr5 systemd[1]: domoticz.service: Failed with result 'exit-code'.
root@DietPiOr5:~#
```

Joulinar, 2023-04-18:
It doesn't make sense to post your issue twice. Pls try to stay with your individual issue for now. #6322

rickcrash, 2023-04-21:
Hi,

updated today via the script. Almost everthing worked fine. I just had a little hassle with my old ubound which I installed long time ago by hand.

After the final reboot I found some services masked.

```
systemctl list-unit-files --type=service | grep mask

alsa-utils.service                     masked          enabled
apt-daily-upgrade.service              masked          enabled
apt-daily.service                      masked          enabled
cloudprintd.service                    masked          enabled
collectd.service                       masked          enabled
containerd.service                     masked          enabled
cryptdisks-early.service               masked          enabled
cryptdisks.service                     masked          enabled
docker.service                         masked          enabled
getty-static.service                   masked          enabled
hwclock.service                        masked          enabled
mdadm-waitidle.service                 masked          enabled
mdadm.service                          masked          enabled
mosquitto.service                      masked          enabled
nfs-common.service                     masked          enabled
ntp.service                            masked          enabled
rc.service                             masked          enabled
rcS.service                            masked          enabled
screen-cleanup.service                 masked          enabled
serial-getty@ttyS0.service             masked          enabled
serial-getty@ttyS1.service             masked          enabled
serial-getty@ttyS2.service             masked          enabled
serial-getty@ttyS3.service             masked          enabled
sudo.service                           masked          enabled
ups-monitor.service                    masked          enabled
x11-common.service                     masked          enabled

```
What's the reason for this masking?

Can I unmask some services? I already tried to unmask mdadm, without succes.

btw: Thanks a lot!
```
```

MichaIng, 2023-04-21:
- Most of them are masked automatically by systemd to prevent the obsolete SysV/init.d services from starting.
- The `apt-daily` ones are masked on DietPi by default since we have our own daily APT update check implemented.
- `cloudprintd` is this old Google printing service which does not exist anymore. You should be able to unmask and in case remove it: `systemctl unmask cloudprintd; apt autopurge cloudprint-service`
- `collectd` must have been manually installed by you: https://github.com/collectd/collectd
- Not sure why you have Docker masked? If you do not currently run any containers but want to keep Docker installed, that is probably a good way to prevent it from unintentionally starting up.
- Same with Mosquitto.
- If you do not use NTP, better just remove it: `systemctl unmask ntp; apt autopurge ntp`
- Not sure what `ups-monitor` is.
- The `serial-getty` services are masked to prevent login prompts on those UART devices. This has no effect if there are anyway no UART adapters attached, but generally it makes sense to not allow logins and have some waiting service for this on UART devices which you never intend to use.

whyisthisbroken, 2023-04-23:
everything works... thanks for this easy script. very cool.
stupid question, why wasn't python 3.10 updated to 3.11?
the home assistant is completely reinstalled but in 3.10.11... is this normal ?

the only thing I have is as described here -> https://github.com/MichaIng/DietPi/pull/6103#issuecomment-1507296719

Joulinar, 2023-04-23:
HA is running within its own `pyenv`

MichaIng, 2023-04-23:
Jep, I think we'll switch it to 3.11 once Bookworm is released. For now there are still some issues and the official docs and also HA's own images/containers all use Python 3.10: https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+Python+3.11

rickcrash, 2023-04-24:
> * Most of them are masked automatically by systemd to prevent the obsolete SysV/init.d services from starting.
> 
>  * The `apt-daily` ones are masked on DietPi by default since we have our own daily APT update check implemented.
> 
>  * `cloudprintd` is this old Google printing service which does not exist anymore. You should be able to unmask and in case remove it: `systemctl unmask cloudprintd; apt autopurge cloudprint-service`: [DietPi-Software | CloudPrint: Remove from software options #3860](https://github.com/MichaIng/DietPi/pull/3860)
> ...

Thank you for your fast reply. I followed all your advices.

One service I really need is mdadm. my raid5-device was unavailable when mdadm was masked. Is there another way to use the raid5-device without unmasking mdadm?

ups-monitor is a program for monitoring and controlling a UPS. I have unmasked this, but it does not run yet.

MichaIng, 2023-04-26:
I don't know much about `mdadm`, I think @StephanStS can verify whether this service is required to run a RAID generated with the `mdadm` command.

However, if it is needed for your RAID, unmask it. It is not one of the services masked by systemd and it is neither one masked by the DietPi installer. So I don't know why it is masked on your system (as with Docker and Mosquitto).

StephanStS, 2023-04-27:
On my Bullseye RAID 5 system, 
- no mdadm service is shown via `dietpi-services`
- the mdadm.service is not active
- in htop it is shown as a process

[image]

```
root@raid:~# systemctl status mdadm
● mdadm.service
     Loaded: masked (Reason: Unit mdadm.service is masked.)
     Active: inactive (dead)
```

[image]

MichaIng, 2023-04-27:
Just checked it, the `mdadm` package itself ships the mask, in a way that it cannot even be unmasked, and an init.d service. Hence this service is only needed when using SysV as init system (resp. not systemd) but explicitly not needed when using systemd, hence the mask is fully intended.

rickcrash, 2023-04-27:
Thank you for your help, i'll check the assembly of my raid5 at this weekend.

https://github.com/MichaIng/DietPi/issues/6227#issuecomment-1526247269

Diese Nachricht wurde von meinem GrapheneOS-Gerät mit K-9 Mail gesendet.

rickcrash, 2023-05-07:
I have restarted the server with the `mdadm` package masked. The raid device was not assembled. If `mdadm` is unmasked, it is not assembled either. 

Manually running `mdadm --assemble --scan` is no problem though, I don't start the server that often.

Would an `update-initramfs -u` help?

Joulinar, 2023-05-08:
why not unmasking `mdadm.service`?

MichaIng, 2023-05-10:
The RAID topic has actually nothing to do with Bookworm, does it? @rickcrash please open a new issue about this so we do not further spam it here.

Some input:

I need to setup a RAID myself to efficiently check how this `mdadm` process is supposed to be started. At least when following [our guide](https://dietpi.com/blog/?p=2239), the `mdadm.service` is not the intended way, or more precisely: With systemd, the `/etc/init.d/mdadm` SysV service is not intended to do so (otherwise the package wouldnt' explicitly ship the mask), but some other mechanism, probably even systemd itself if a `/dev/mdX` fstab entry is present.

> Would an `update-initramfs -u` help?

You can try it, but I cannot imagine what the initramfs should have to do with the RAID, which is assembled long after the initramfs has finished its task, as long as you did not create one yourself which explicitly aims to mount the RAID for whatever reason.

@Joulinar 
The package itself does not ship a systemd service but only the mentioned SysV service and explicitly a mask (symlink to `/dev/null`) at `/lib/systemd/system/mdadm.service` to mask that SysV service if systemd is used. This cannot be "unmasked" via `systemctl unmask mdadm`. Of course the mask could be manually removed, but this would explicitly break the intended behaviour of the package, so is probably a bad idea.

MichaIng, 2023-05-11:
Hmm, if `update-initramfs -u` made a difference, then the kernel module was not enabled. Interesting, usually these are enabled on demand. However, good that it works when you have the module loaded in initramfs.

tmznix, 2023-06-12:
Hi!

Pi-hole is failing to install due to PHP 8.2 apparently not working on ARMv6.

Tested on a Pi Zero W.

Thank you!

aszurnasirpal, 2023-06-25:
After I moved to Bookworm via script I couldn't run any docker-compose

I got the following error:

`dietpi@DietPi:/opt/homepage$ sudo docker-compose restart
Traceback (most recent call last):
  File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 563, in from_name
    return next(cls.discover(name=name))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/docker-compose", line 33, in <module>
    sys.exit(load_entry_point('docker-compose==1.29.2', 'console_scripts', 'docker-compose')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/docker-compose", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
                       ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 981, in distribution
    return Distribution.from_name(distribution_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 565, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: No package metadata was found for docker-compose`

Joulinar, 2023-06-25:
How does Docker Compose has been installed? Because Docker developer changed the why how Docker Compose should be used. The switched the command from `docker-compose` to `docker compose` (without dash). Using `dietpi-software` will install docker plugin package `docker-compose-plugin` offering the new method. Therefore `docker-compose` (with dash) should not be working anymore.

aszurnasirpal, 2023-06-29:
> How does Docker Compose has been installed? Because Docker developer changed the why how Docker Compose should be used. They switched the command from `docker-compose` to `docker compose` (without dash). Using `dietpi-software` will install docker plugin package `docker-compose-plugin` offering the new method. Therefore `docker-compose` (with dash) should not be working anymore.
> 
> ```
> root@DietPi4:~# echo $G_DISTRO_NAME
> bookworm
> root@DietPi4:~# docker compose version
> Docker Compose version v2.18.1
> root@DietPi4:~#
> ```
Thanks for the explanation. This explains a lot. On my system, both are present (with and without a dash).

Joulinar, 2023-06-29:
you should be able to remove executable `docker-compose` (with dash) manually.

MichaIng, 2023-06-29:
> Thanks for the explanation. This explains a lot. On my system, both are present (with and without a dash).

Hmm strange, a reinstall should actually remove the old standalone binary. This is the code that runs on reinstall:
```sh
[[ -f '/usr/local/bin/docker-compose' ]] && G_EXEC rm /usr/local/bin/docker-compose
```

The standalone binary is btw still a legit option: https://github.com/docker/compose/releases
We migrated to the module variant as it will be automatically updated via `apt upgrade`. However, quite possible that older `docker-compose` binaries are not compatible yet with Python 3.11. Should we reinstall Docker Compose as part of the Bookworm upgrade script and in the same turn inform about the new `docker compose` command?

Joulinar, 2023-06-30:
> Should we reinstall Docker Compose as part of the Bookworm upgrade script

We already do (ID 134)

https://github.com/MichaIng/DietPi/blob/8e8408834eceecee0c9b063acbf755db8a7a33ae/.meta/dietpi-bookworm-upgrade#L85C107-L85C111

droogi, 2023-06-30:
MHerger seems to have patched Logitech Media Server to be compatible to perl 5.3.6 https://forums.slimdevices.com/forum/user-forums/logitech-media-server/1638213-lms-on-debian-bookworm https://github.com/Logitech/slimserver/actions/runs/5281501230 https://www.dropbox.com/s/oxeuhc4jtb5sg14/536-aarch64-linux-thread-multi.tar.gz?dl=0

MichaIng, 2023-06-30:
Many thanks for the info. Then I hope there will be a release of LMS 8.4 soon.

HyperCriSiS, 2023-07-08:
Hi,

the update went fine. :+1: :grinning:
The only issue I have seen so far is HA still showing errors:

```
Logger: homeassistant.setup
Source: setup.py:209
First occurred: 03:53:27 (2 occurrences)
Last logged: 03:53:35

Setup failed for stream: Requirements for stream not found: ['ha-av==10.1.0'].
Setup failed for default_config: (DependencyError(...), 'Could not setup dependencies: stream')
```
```
Logger: homeassistant.setup
Source: setup.py:207
First occurred: 03:53:35 (1 occurrences)
Last logged: 03:53:35

Unable to set up dependencies of default_config. Setup failed for dependencies: stream
```
`Logger: homeassistant.util.package
Source: util/package.py:107
First occurred: 03:44:50 (3 occurrences)
Last logged: 03:53:27

Unable to install package ha-av==10.1.0: error: subprocess-exited-with-error × Building wheel for ha-av (pyproject.toml) did not run successfully. │ exit code: 1 ╰─> [173 lines of output] running bdist_wheel running build running build_py creating build creating build/lib.linux-aarch64-cpython-311 creating build/lib.linux-aarch64-cpython-311/av copying av/deprecation.py -> build/lib.linux-aarch64-cpython-311/av copying av/datasets.py -> build/lib.linux-aarch64-cpython-311/av copying av/about.py -> build/lib.linux-aarch64-cpython-311/av copying av/__main__.py -> build/lib.linux-aarch64-cpython-311/av copying av/__init__.py -> build/lib.linux-aarch64-cpython-311/av creating build/lib.linux-aarch64-cpython-311/av/video copying av/video/__init__.py -> build/lib.linux-aarch64-cpython-311/av/video creating build/lib.linux-aarch64-cpython-311/av/subtitles copying av/subtitles/__init__.py -> build/lib.linux-aarch64-cpython-311/av/subtitles creating build/lib.linux-aarch64-cpython-311/av/sidedata copying av/sidedata/__init__.py -> build/lib.linux-aarch64-cpython-311/av/sidedata creating build/lib.linux-aarch64-cpython-311/av/filter copying av/filter/__init__.py -> build/lib.linux-aarch64-cpython-311/av/filter creating build/lib.linux-aarch64-cpython-311/av/data copying av/data/__init__.py -> build/lib.linux-aarch64-cpython-311/av/data creating build/lib.linux-aarch64-cpython-311/av/container copying av/container/__init__.py -> build/lib.linux-aarch64-cpython-311/av/container creating build/lib.linux-aarch64-cpython-311/av/codec copying av/codec/__init__.py -> build/lib.linux-aarch64-cpython-311/av/codec creating build/lib.linux-aarch64-cpython-311/av/bitstream copying av/bitstream/__init__.py -> build/lib.linux-aarch64-cpython-311/av/bitstream creating build/lib.linux-aarch64-cpython-311/av/audio copying av/audio/__init__.py -> build/lib.linux-aarch64-cpython-311/av/audio copying av/utils.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/stream.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/plane.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/packet.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/option.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/logging.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/frame.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/format.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/error.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/enum.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/dictionary.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/descriptor.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/bytesource.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/buffer.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/__init__.pxd -> build/lib.linux-aarch64-cpython-311/av copying av/video/stream.pxd -> build/lib.linux-aarch64-cpython-311/av/video copying av/video/reformatter.pxd -> build/lib.linux-aarch64-cpython-311/av/video copying av/video/plane.pxd -> build/lib.linux-aarch64-cpython-311/av/video copying av/video/frame.pxd -> build/lib.linux-aarch64-cpython-311/av/video copying av/video/format.pxd -> build/lib.linux-aarch64-cpython-311/av/video copying av/video/codeccontext.pxd -> build/lib.linux-aarch64-cpython-311/av/video copying av/video/__init__.pxd -> build/lib.linux-aarch64-cpython-311/av/video copying av/subtitles/subtitle.pxd -> build/lib.linux-aarch64-cpython-311/av/subtitles copying av/subtitles/stream.pxd -> build/lib.linux-aarch64-cpython-311/av/subtitles copying av/subtitles/codeccontext.pxd -> build/lib.linux-aarch64-cpython-311/av/subtitles copying av/subtitles/__init__.pxd -> build/lib.linux-aarch64-cpython-311/av/subtitles copying av/sidedata/sidedata.pxd -> build/lib.linux-aarch64-cpython-311/av/sidedata copying av/sidedata/motionvectors.pxd -> build/lib.linux-aarch64-cpython-311/av/sidedata copying av/sidedata/__init__.pxd -> build/lib.linux-aarch64-cpython-311/av/sidedata copying av/filter/pad.pxd -> build/lib.linux-aarch64-cpython-311/av/filter copying av/filter/link.pxd -> build/lib.linux-aarch64-cpython-311/av/filter copying av/filter/graph.pxd -> build/lib.linux-aarch64-cpython-311/av/filter copying av/filter/filter.pxd -> build/lib.linux-aarch64-cpython-311/av/filter copying av/filter/context.pxd -> build/lib.linux-aarch64-cpython-311/av/filter copying av/filter/__init__.pxd -> build/lib.linux-aarch64-cpython-311/av/filter copying av/data/stream.pxd -> build/lib.linux-aarch64-cpython-311/av/data copying av/data/__init__.pxd -> build/lib.linux-aarch64-cpython-311/av/data copying av/container/streams.pxd -> build/lib.linux-aarch64-cpython-311/av/container copying av/container/pyio.pxd -> build/lib.linux-aa [... cut ...] (((AV_FRAME_DATA_SEI_UNREGISTERED != -1L) != 0)) { | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ src/av/sidedata/sidedata.c:5664:9: note: each undeclared identifier is reported only once for each function it appears in error: command '/usr/bin/gcc' failed with exit code 1 [end of output] note: This error originates from a subprocess, and is likely not a problem with pip. ERROR: Failed building wheel for ha-av ERROR: Could not build wheels for ha-av, which is required to install pyproject.toml-based projects`

MichaIng, 2023-07-10:
Why the hack is it trying to build a wheel for `ha-av` on an `aarch64` system? PyPI ships pre-compiled feels for it: https://pypi.org/project/ha-av/#files

Let's see whether it does the same here: https://github.com/MichaIng/DietPi/actions/runs/5511821033

Restarting the service does not help, does it?
```sh
systemctl restart home-assistant
```

HyperCriSiS, 2023-07-10:
> Why the hack is it trying to build a wheel for `ha-av` on an `aarch64` system? PyPI ships pre-compiled feels for it: https://pypi.org/project/ha-av/#files
> 
> Let's see whether it does the same here: https://github.com/MichaIng/DietPi/actions/runs/5511821033
> 
> Restarting the service does not help, does it?
> 
> ```shell
> systemctl restart home-assistant
> ```

I have a 32bit system. When I do this, HA does not even start anymore. I had to restart the system.
Strange thing is because of the systemctl command the errors are not shown in HA anymore but still many integrations are not loaded, so the problem still persists.

HyperCriSiS, 2023-07-11:
Now it is again as before. Integrations are working again and the Invalid config shows also up again.
I had switched from Pi-Hole to Adguard and not back to Pi-Hole again.

MichaIng, 2023-07-12:
> I have a 32bit system.

That is interesting, as in your logs it tries to build 64-bit `aarch64` wheels:
```
  × Building wheel for ha-av (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [173 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/av
      copying av/deprecation.py -> build/lib.linux-aarch64-cpython-311/av
      copying av/datasets.py -> build/lib.linux-aarch64-cpython-311/av
      copying av/about.py -> build/lib.linux-aarch64-cpython-311/av
      copying av/__main__.py -> build/lib.linux-aarch64-cpython-311/av
      copying av/__init__.py -> build/lib.linux-aarch64-cpython-311/av
```
Can you verify:
```sh
uname -m
dpkg --print-architecture
```
However, great that it works again.

HyperCriSiS, 2023-07-12:
uname -m
aarch64

dpkg
armhf

getconf LONG_BIT
32 

I  am confused 😅

Joulinar, 2023-07-13:
You are running RPi4?

MichaIng, 2023-07-13:
So it is a 32-bit image, but RPi 4 loads the 64-bit kernel regardless since its Linux 6.1 update. Indeed `pip` somehow labels wheels/paths falsely in this situation, as it cannot possibly be actual `aarch64` wheels. However, they work all fine. Also it makes sense then that the `ha-av` wheel is compiled, as PyPI serves `aarch64` ones only and piwheels does not support Python 3.11 yet.

HyperCriSiS, 2023-07-13:
> So it is a 32-bit image, but RPi 4 loads the 64-bit kernel regardless since its Linux 6.1 update. Indeed `pip` somehow labels wheels/paths falsely in this situation, as it cannot possibly be actual [...]

What is the solution then to get rid of the error notifications :)?

MichaIng, 2023-07-13:
I thought it works now? Or does it, but the `ha-av` compiling error keeps showing up in
```sh
journalctl -u home-assistant
```
?

HyperCriSiS, 2023-07-13:
> I thought it works now? Or does it, but the `ha-av` compiling error keeps showing up in
> 
> ```shell
> journalctl -u home-assistant
> ```
> 
> ?

Integrations working, yes, but errors still show up.

Building wheel for ha-av (pyproject.toml) did not run successfully.

The companion app still shows the same errors in stream and default config.

[83 more replies]

closed 2024-05-13 by MichaIng