MichaIng/DietPi #6228: Raspberry Pi 1 / Zero Image not booting - kernel.img missing

ikkerus, 2023-03-08:
Hi,

I have an old Raspberry Pi Zero W which I tried to boot with the latest ARMv6 image. LED flashes 7 times which indicates that it couldn't find the kernel.img file. I have used two different brand new Samsung EVO microSD cards and also veryfied the SHA256 checksum of the image.

#### Required Information
- latest DietPi ARMv6 image for Raspberry Pi Zero W

#### Additional Information (if applicable)

#### Steps to reproduce
Download ARMv6 image and try to boot on rpi Zero

#### Expected behaviour
The system boots up.

#### Actual behaviour
I have a signal but black screen. LED flashes 7 times

Please take a look. Or am I missing something? 

Thanks!

Joulinar, 2023-03-09:
Can you have a look to your SD card boot partition if all files have been written correctly?

ikkerus, 2023-03-10:
Sure.
I used BalenaEtcher to write the SD-Card and it was successfully verified.
When I now take a look at the contents, then I can only see a kernel8.img

[image]

MichaIng, 2023-03-10:
Looks like the ARMv8 image, as also the device trees for RPI 1 and Zero are missing. Please verify you downloaded the ARMv6 image.

MichaIng, 2023-04-03:
Marking as closed due to outstanding reply. Based on what we see, the wrong image was downloaded.