MichaIng/DietPi #6230: Radxa Zero image doesn't have boot partition

htjain, 2023-03-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | DietPi_RadxaZero-ARMv8-Bullseye.img
- SBC model | `Radxa Zero `

#### Image details
```
FILE:   DietPi_RadxaZero-ARMv8-Bullseye.img
DATE:   Mon Feb 13 22:39:42 UTC 2023
MD5:    e72f629bbffc95159a751bf8838fb4a8
SHA1:   48e57d9df2101d998ba87ae6a1be52500b4719ad
SHA256: 636a3d3453a0ae9eb90378d546b73c18b3e13fbeff2e74d0c6988c80e5f78dbe

```
#### Additional Information (if applicable)
```
$ fdisk -l  DietPi_RadxaZero-ARMv8-Bullseye.img
Disk DietPi_RadxaZero-ARMv8-Bullseye.img: 493.54 MiB, 517517312 bytes, 1010776 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xf92b5168

Device                               Boot Start     End Sectors   Size Id Type
DietPi_RadxaZero-ARMv8-Bullseye.img1 *     8192 1010775 1002584 489.5M 83 Linux

```
#### Expected behaviour
Image should have both boot and root partition. 
#### Actual behaviour
Missing root partition.

htjain, 2023-03-10:
So how does I boot it ?  while booting I get only blank screen. Is there any guide for the same?

Joulinar, 2023-03-10:
I guess serial console access would be needed to be able to check what happen.

MichaIng, 2023-03-10:
Have you checked whether it appears at the network? Probably it does boot and only HDMI output is for some reason broken. Assure PSU is sufficient.

jilhakumi, 2023-03-13:
@MichaIng I have the same problem and it does not appear on the network.

The biggest problem is that without a boot partition then it doesn't seem possible to encrypt the root partition.

So should I raise a new ticket asking to have a separate boot partition for the Radxa Zero?

MichaIng, 2023-03-13:
> The biggest problem is that without a boot partition then it doesn't seem possible to encrypt the root partition.

A very different topic, discussed e.g. here: #4226
Note that a dedicated boot partition is actually very rare and uncommon among Linux distros. In case one has an EFI partition, but the boot partition for some SBCs found its way as a necessity on Raspberry Pi, which does not support booting from proper Linux partitions. Most modern SBCs however do, which makes the dedicated boot partition obsolete. Of course rootfs encryption can be a reason for this, but can be achieved in different ways. We'll however not ship our images with dedicated boot partition unless essentially required (like on RPi), but looking for a solution to have this as an option to apply afterwards, similar to e.g. rootfs compression.

MichaIng, 2023-03-13:
Can you guys also test the new image I just moved into place?

jilhakumi, 2023-03-13:
> A very different topic, discussed e.g. here: #4226

I don't think it's the same topic as OP didn't clarify what he meant by encrypted rootfs.

Thought, I'd suppose that means having a separate partition which boot an initramfs that has the purpose of asking for password, decrypting the encrypted rootfs and booting it.

But my linux knowledge is close to zero when it comes to such things and the will to climb such steep learning curve is definitely not there now. So I guess I bought a Radxa Zero for nothing and have to stick to a Pi for the time being, feels bad.

MichaIng, 2023-03-13:
If rootfs encryption is essential to you, then for now yes. I'm not sure about your use case, as the need for an encrypted rootfs seems extremely rare to me, but I suggest to have the actual sensitive data on an external drive and encrypt that.

jilhakumi, 2023-03-13:
> If rootfs encryption is essential to you, then for now yes. I'm not sure about your use case, as the need for an encrypted rootfs seems extremely rare to me, but I suggest to have the actual sensitive data on an external drive and encrypt that.

Thanks for wanting to help but it's completely for personal satisfaction. I'll give a check to the new image but I have doubts the Radxa Zero came with the right bootloaders (if any at all) to load from the sdcard/eMMC.

MichaIng, 2023-03-14:
> but I have doubts the Radxa Zero came with the right bootloaders (if any at all) to load from the sdcard/eMMC.

Of course it comes with the bootloader that always worked. That has no changed since October last year. The new image finally has a boot loader update from end of February, but I'm not sure what changed, aside of possibly a rebase on newer upstream U-Boot sources.

jilhakumi, 2023-03-17:
> Of course it comes with the bootloader that always worked. That has no changed since October last year. The new image finally has a boot loader update from end of February, but I'm not sure what [...]

Sorry I meant about the eMMC bootloader.

I gave it another try using the previous DietPi image which was causing issue (not the new latest one) and it appeared on the network this time.
So it was a misconfiguration on my side. For some reasons it probably didn't unmount correctly after I edited the dietpi-wifi.txt.

All seems good now.

> The new image finally has a boot loader update from end of February, but I'm not sure what changed, aside of possibly a rebase on newer upstream U-Boot sources.

Ok I suppose it'll be fine anyway.

I'm currently working on moving the /boot to a dedicated /boot partition.

From what I've seen it seems the U-boot bootloader is pretty straightforward and shouldn't require any configuration changes as long as the rootenv UUID stays the same. Might deserve its own ticket.

MichaIng, 2023-03-17:
Okay great that it works now.

> Sorry I meant about the eMMC bootloader.

Ah, the preinstalled one you mean. Yeah, most likely that wouldn't work. But you can flash DietPi to the eMMC as well, including the bootloader.

> I'm currently working on moving the /boot to a dedicated /boot partition.

Why you want to do that? We aim to eliminate dedicated boot partitions wherever we can.

> From what I've seen it seems the U-boot bootloader is pretty straightforward and shouldn't require any configuration changes as long as the rootenv UUID stays the same. Might deserve its own ticket.

The UUID in U-Boot script/environment as well as the `/etc/fstab` on the rootfs need to be adjusted so that the boot partition is mounted to `/boot`. But otherwise yes, that U-Boot should be able to load the script from any partition with FAT or ext4 filesystem from `/boot` sub directory and the root of the filesystem..

MichaIng, 2023-04-03:
Marking as closed as the original issue has been sorted.

closed 2023-04-03 by MichaIng