MichaIng/DietPi #6233: Setting the hostname with dietpi-config doesn't get recognized when pinging from another device.

as-shura, 2023-03-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
bullseye
```
- Kernel version | `uname -a`
```
Linux DietPi 6.1.11-meson64 #23.02.2 SMP PREEMPT Sat Feb 18 00:07:55 UTC 2023 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
Odroid C2 (aarch64)
```
- Power supply used | (EG: 5V 1A RAVpower)

https://sha256systems.eu/PSU-ODROID-C1-C0-C2

- SD card used | (EG: SanDisk ultra)

```
SanDisk ultra 32 GB https://www.amazon.de/SanDisk-microSDHC-Speicherkarte-SD-Adapter-App-Leistung/dp/B08GY9NYRM/ref=asc_df_B08GY9NYRM/?tag=googshopde-21&linkCode=df0&hvadid=474234094943&hvpos=&hvnetw=g&hvrand=17165416810116209753&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1004607&hvtargid=pla-1067844298997&psc=1&th=1&psc=1
```

#### Steps to reproduce
Running the dietpi-config to set the hostname works but the hostname is not recognize in an internal network when multiple device ping each other with their hostname.

#### Expected behaviour

ping should work

#### Actual behaviour

(ping: server1: Name or service not known)

I also installed avahi-daemon and it still doesn't work. But when I ping each server to itself it works

as-shura, 2023-03-11:
The IP addresses of two devices are static and the router is an OpenWrt router on which I set the ignore DHCP.

MichaIng, 2023-03-11:
Then it makes sense that the router does not resolve their hostname, but Avahi/mDNS should work. Note that it broadcasts hostname with `.local` appended, so it would be `DietPi.local` with default hostname.