MichaIng/DietPi #6237: System Failure when navigating to docker volumes directory

holocronology, 2023-03-11:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.15.1 (beta) -- also experienced this on current release version
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
- SBC model | RPi 3 Model B+ (aarch64)
- Power supply used | 5V 5A
- SD card used | Samsung Evo Pro

#### Additional Information (if applicable)
- Software title | dietpi-dashboard, docker
- Was the software title installed freshly or updated/migrated? fresh
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | 0d4cdd01-2b81-40d6-884e-b157253dfbb8

#### Steps to reproduce
1. Access dietpi-dashboard
2. Select "file manager"
3. navigate to /mnt/dietpi_userdata/docker-data
4. click on "volumes" folder
5. system hangs, physical power off/reboot required

#### Expected behaviour

- It should open the "volumes" folder the same as other folders and not crash/hang system

#### Actual behaviour

- when you click on the "volumes' folder the system crashes/hangs/is extremely slow

#### Extra details

- when clicking on volumes folder is does not even open:
[image]

- side note, say this while running the bug report:
[image]

MichaIng, 2023-03-11:
This again is a DietPi-Dashboard specific issue. Could you recreate (just copy&paste content) over there? Or do you face the same issue when trying to access the volumes form SSH or local console?

MichaIng, 2023-03-11:
> side note, say this while running the bug report:

All expected. It runs when audio is enabled, else it is good to know for us in the bug report that it is not there.

holocronology, 2023-03-11:
> This again is a DietPi-Dashboard specific issue. Could you recreate (just copy&paste content) over there? Or do you face the same issue when trying to access the volumes form SSH or local console?

I will move over...as noted, it seems to work OK via terminal.

MichaIng, 2023-03-11:
The bug report "command not found" error is now also passed into the upload file, hence hidden from console: https://github.com/MichaIng/DietPi/commit/cabcb48

MichaIng, 2023-03-11:
Thanks, marking it as closed here.

closed 2023-03-11 by MichaIng