MichaIng/DietPi #6238: Shairport Sync not installing

guussie, 2023-03-11:
#### Details:
- Date           | Sat Mar 11 20:37:38 CET 2023
- DietPi version | v8.14.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero 2 W (armv7l) (ID=3)
- Kernel version | `Linux EP2 5.15.84-v7+ #1613 SMP Thu Jan 5 11:59:48 GMT 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl stop shairport-sync`
- Exit code      | 5
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to stop shairport-sync.service: Unit shairport-sync.service not loaded.
```

Joulinar, 2023-03-11:
just tested on my RPi3B+ 32bit without issues. Can you try again and share the entire log pls.

MichaIng, 2023-04-03:
Marking as closed due to outstanding reply. Feel free to reopen if issue persists.

closed 2023-04-03 by MichaIng