MichaIng/DietPi #6244: Pine64 Quartz64 Model B Poor performance with Kodi

pswilde, 2023-03-13:
### Required Info
- DietPi version: 8.15.2 - master
- Distro version: bullseye`
- Kernel version: 6.1.14
- SBC model: Quartz64 (aarch64)
- Power supply used: 
- SD card used: no SD, using eMMC 32GB module

#### Additional Information (if applicable)
- Software title: Kodi
- Fresh install
- Haven't tried a fresh install yet, I am getting a set of SDcards tomorrow where I will try with them

#### Steps to reproduce
1. Start DietPi
2. Auto load or manual start Kodi

#### Expected behaviour
Menu should be quick to respond, videos should not stutter

#### Actual behaviour
Menu navigation is delayed (0.5 - 1s after keypress) and video performance is very slow. Audio seems to be OK, but Video is less than 1 frame per second. When pausing, audio pauses immediately, video still stutters for a few seconds.

#### Extra details
I have looked at other "Slow Kodi Performance" issues for various other SBCs which talk about checking Xorg logs, my system is not generating these to look at. I have adjusted my swap file to use 4GB as one issue requests to do, but this has made no difference.
The Kodi logs seem to be showing errors such as "WARNING <general>: OutputPicture - timeout waiting for buffer" and when kodi is running, cpu usage is at 100%.

I appreciate it's still early days for the Quartz64, so I'd be happy to help test any changes on it if you need!

MichaIng, 2023-03-13:
Many thanks for reporting. I read that some GPU related features were added for Quartz64/RK3566 with Linux 6.2. I already created a build. Can you test, please (but at best do a system/drive backup first):
```sh
cd /tmp
curl -fO 'https://dietpi.com/downloads/binaries/firmware-quartz64b.deb'
dpkg -i firmware-quartz64b.deb
rm firmware-quartz64b.deb
```
More RAM isn't be needed, also if swapping is a performance killer. It's all about GPU acceleration/decoding support, as the CPU of the RK3566 isn't that powerful.

If this alone does not help, you may test our Bookworm images or upgrading your system to Debian Bookworm basis, as this brings new Mesa GPU drivers, possibly making a difference: https://dietpi.com/blog/?p=2809

pswilde, 2023-03-13:
Thanks for getting back so quickly!
Sure, I'll give the build a test in a bit and if that doesn't help I'll try the bookworm images and report back to you!

MichaIng, 2023-03-13:
Images with Linux v6.2.5 are here, including Bookworm ones: https://dietpi.com/downloads/images/testing/

pswilde, 2023-03-13:
just a quick update from me. updating the build didn't seem to have any impact unfortunately, but upgrading to bookworm has made things run really well!
I may try with bullseye on the 6.2.5 kernel tomorrow to see if there's any difference there, but for now bookworm with kernel 6.1.14 kernel is running nicely (tested with videos from BBC iPlayer WWW and Jellyfin)

MichaIng, 2023-03-14:
Great, so it is the Mesa drivers and/or Kodi 19.1 vs 20.0. The new kernel won't hurt, of course, probably add other minor enhancements for other GUI applications.

MichaIng, 2023-03-28:
Marking this as closed. I'm pretty sure its Mesa 22 which makes the difference here, adding/enhancing support for RK3566/Mali G52 GPU.

closed 2023-03-28 by MichaIng