MichaIng/DietPi #6245: Torrent clients don't work, like no internet connection

davigar1391, 2023-03-14:
Hello everyone, I bought a new Raspberry Pi 4 to replace my old 2.

I have put the latest Dietpi image on a SSD and I don’t use MicroSD.

I have configured everything and as torrent client I have installed qbittorrent from Dietpi-Software.
Everything works perfect except the torrent client. It doesn’t start the downloads and doesn’t connect to the seeds.

I have tried Transmission, Deluge, rTorrent, I have opened ports on the router and nothing. I have also tried with a MicroSD and a clean install of Dietpi with nothing else installed except the torrent client and none of them work either.

I don’t know where to go next to try to solve it but on the Raspberry Pi 2 it works perfectly.

Any ideas?

Joulinar, 2023-03-14:
basically a duplicate of https://dietpi.com/forum/t/torrent-clients-dont-work/16225

@davigar1391 if possible, try to stick with one single platform and avoid reporting your issue twice.

davigar1391, 2023-03-14:
Ok sorry, I have deleted the thread from the forum.

Today I tried to run with the ARMv7 image and it worked perfectly so it is due to some problem with the ARMv8 image that I can't figure out under any circumstances. 

On the ARMv8 image even installing docker and putting qbittorrent-nox in a container I still have the same problem, 0 connections.

Joulinar, 2023-03-14:
ok I tested Transmission on 3 different device RPi 4 Model B, NanoPi R6S and Orange Pi 5. All of them are aarch64 systems and I didn't do any configuration or port forwarding. There were no issues, Transmission was connecting to max number of peers (30).

MichaIng, 2023-04-03:
Does the issue persist? Assure that port 6881 is open and forwarded for DHT. If possible, try another tracker and to rule it out test with different torrents. There are several possible reasons why no connections are done or only slowly, so those would need to be ruled out one by one.

Joulinar, 2024-01-01:
going to close this issue as we don't have any feedback from user.

closed 2024-01-01 by Joulinar