MichaIng/DietPi #6248: systemd-timesyncd.service log spam when CIFS share goes offline

aarste, 2023-03-15:
- G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=15
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- bullseye
- Linux DietPi-proxmox 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux
- Virtual Machine (x86_64)

#### Additional Information (if applicable)
- Software title | systemd-timesyncd.service
- Was the software title installed freshly or updated/migrated? N/A
- Can this issue be replicated on a fresh installation of DietPi? Not sure. Did successfully replicate it in another VM in Proxmox but I may have done changes to that test VM in the past.

#### Steps to reproduce
Add a network drive to a Windows share in Drive Manager, turn Windows 10 Desktop PC off overnight

#### Expected behaviour
Functioning timesync?

#### Actual behaviour
Journalctl logs are being spammed overnight with the following, this is the final entry as my desktop PC was booted back up this morning:

> Mar 15 09:36:12 dietpitest systemd[1]: Starting Network Time Synchronization...
Mar 15 09:36:21 dietpitest systemd[24852]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Host is down
Mar 15 09:36:21 dietpitest systemd[24852]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Host is down
Mar 15 09:36:21 dietpitest systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Mar 15 09:36:21 dietpitest systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Mar 15 09:36:21 dietpitest systemd[1]: Failed to start Network Time Synchronization.
Mar 15 09:36:21 dietpitest systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 2879.
Mar 15 09:36:21 dietpitest systemd[1]: Stopped Network Time Synchronization.
Mar 15 09:36:21 dietpitest kernel: CIFS: VFS: \\192.168.1.10 has not responded in 180 seconds. Reconnecting...
Mar 15 09:36:21 dietpitest systemd[1]: Starting Network Time Synchronization...
Mar 15 09:36:21 dietpitest systemd[1]: Started Network Time Synchronization.
Mar 15 09:36:21 dietpitest systemd-timesyncd[24855]: Initial synchronization to time server 80.87.128.222:123 (0.debian.pool.ntp.org).

#### Extra details
Nothing appears to be broken as far as I know, I'm just concerned with the amount of logging this does and doesn't appear to be normal? As you can see it's attempted to run 2879 times overnight. This spam stops if I remove the windows shares I've set up in Drive Manager, then check the logs the next morning.

Joulinar, 2023-03-15:
Did you power of the SAMBA server?

Joulinar, 2023-03-15:
looks like `systemd` don't like a samba share being offline and not reachable (e.g. your Windows box is off)

MichaIng, 2023-03-15:
The systemd automount option generally does not like when a drive is not available, regardless whether it is a physical drive, that was detached or a network drive, with its server shut down. When you access the mount point then, the whole system hangs for 90 seconds (timeout) until the automount feature gives up.

Regardless of whether systemd automount is used or not or even another init system, any UNIX system does not like when you shutdown the remote network share while it is mounted. AFAIK you'll see failures for the failing polls then. So in such case, when this is not a permanently running server, always unmount the network drive before shutting the SMB server system down. To not lose the entered credentials, you can just use `umount /mnt/whatever`. On reboot, the automount unit is active by default, but as long as nothing accesses the mount point unless the remote share system is running, it is no issue.

However, I've no idea how this would affect systemd-timesyncd 🤔.

aarste, 2023-03-15:
I'm unsure when it started really, I only noticed it when I decided to run journalctl -e to check everything's okay as I'm trying to familiarise myself with using dmesg and journalctl commands to check for issues. 

I was testing this for a couple of days so I can confirm removing the network drives in Drive Manager stops the log spam. 

I did install autofs a few weeks ago if that matters. I wanted to see if I could force my docker container of File Browser to start with an offline network mount (because Docker is stopped and restarted for dietpi-backup at 3am). Autofs messed with my current mounts so I uninstalled it, and rebooted.

MichaIng, 2023-03-15:
I guess `autofs` conflicts with the `x-systemd.automount` feature, as both seem to do the same, using the same kernel module.

Can you show the output of:
```sh
df -a
cat /etc/fstab
```

aarste, 2023-03-15:
```
Filesystem     1K-blocks     Used Available Use% Mounted on
proc                   0        0         0    - /proc
udev               10240        0     10240   0% /dev
/dev/sda1       24747988 11997320  11722020  51% /
sysfs                  0        0         0    - /sys
securityfs             0        0         0    - /sys/kernel/security
tmpfs            3041756     2032   3039724   1% /dev/shm
devpts                 0        0         0    - /dev/pts
tmpfs            1216704    27064   1189640   3% /run
tmpfs               5120        0      5120   0% /run/lock
cgroup2                0        0         0    - /sys/fs/cgroup
pstore                 0        0         0    - /sys/fs/pstore
none                   0        0         0    - /sys/fs/bpf
systemd-1              -        -         -    - /proc/sys/fs/binfmt_misc
mqueue                 0        0         0    - /dev/mqueue
hugetlbfs              0        0         0    - /dev/hugepages
debugfs                0        0         0    - /sys/kernel/debug
tracefs                0        0         0    - /sys/kernel/tracing
configfs               0        0         0    - /sys/kernel/config
fusectl                0        0         0    - /sys/fs/fuse/connections
sunrpc                 0        0         0    - /run/rpc_pipefs
systemd-1              -        -         -    - /mnt/samsung256
tmpfs            2012160        0   2012160   0% /tmp
/dev/sdb1      244981300 26270228 218694688  11% /mnt/samsung256
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/413bd804a0ee08b03757ba86856f687bd581224e6eeb8ee8a3f7d428c911f579/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/c507160747083bbdfa5865a9a39a51be2bef41a0193a1fa2d33a0986cddd5ff8/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/93d93f21a696bb10b32f68b8a6d427707ec0d25b039a3f0791b124f98799a5e8/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/b4006e1fdc5b68ad62b50c0a9057a2de2a1f6572bc3beff6ea713ec33ea7e758/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/2d71105ff0109473995a0bc8bc6ab2ec47e9782fec536d133d0a28bdb3310b2c/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/b24005c266d0648349990b4c0e383709b2bf3b238b18dd61227f11a1255af216/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/f27eec8f3fa1acc4e073ef9cc9a0e99de6c014ff00b62c6cda88fcd0d2eeb062/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/29d31ee4170e428408234bd5cc3793bbc8ded352e7ee8872ae305a88a977cdee/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/6aef3b7be060f78ed5917302774d2ee843eae53233ebe03fe8a3d743ea38015a/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/00607ee5265902b8761db6ec180e14793f8b7f11745a8d918df55732ca0003d3/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/5be1096873d7ea2c1d4d5a668ded7030dd4503560528846dd7ae2c9dbc595c89/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/a8ebb7400a8a30e7b23d20a44614a1b3af934328c392b1e0f1dd0f6a0ba8bade/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/03efb6390dc2bdc5fa9fb55573f61649a6b4cf242f481ea7f9b2c33f3a78bf58/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/1d2f51b0251b4ce8229e20ac61dedfa418aaa9941bb2fdaa8672042caf656d66/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/45a9a582273b994a2b2bef372a86bbd31ed625f533a91b8af6622af79b3b57dc/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/8eb34c52b5226a83f630de3bca994e9a85f3830db59916a7ea64a67d1e5504f3/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/c61a2ea28726ed2133154d6592f971150af84fe34e431d9b048a90d3145835e5/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/da3f075b99f139e768d96a2917d4c2b2bc43ff9c7a2120a9db87b40fce3ae9ae/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/4e91ac82e6914f20f2313c305da71ad43d708d8f9a41c4a47bc62ceeb8f0c451/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/a35c0af717ab24a9ae5f776e759306f04c416597d83e35f6cd93ff7eec10c0fa/merged
overlay                -        -         -    - /mnt/dietpi_userdata/docker-data/overlay2/ba23cc9f30b18054493acf897dc18c280c547ffe1bf8852a556d6a364b76d6e6/merged
nsfs                   -        -         -    - /run/docker/netns/b49f1cfd52ec
nsfs                   -        -         -    - /run/docker/netns/d9f0cf6f8ede
nsfs                   -        -         -    - /run/docker/netns/31ea263c22c8
nsfs                   -        -         -    - /run/docker/netns/fea2c872368a
nsfs                   -        -         -    - /run/docker/netns/c22492a66462
nsfs                   -        -         -    - /run/docker/netns/4964cdd5a042
nsfs                   -        -         -    - /run/docker/netns/0056b3634eea
[... cut ...]
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
UUID=2157c567-327a-4fc2-ad7a-6a362e9b1579 / ext4 noatime,lazytime,rw 0 1
UUID=bfb6bc5f-b187-478c-8a9c-c160a812df77 /mnt/samsung256 ext4 noatime,lazytime,rw,nofail,noauto,x-systemd.automount
```

MichaIng, 2023-03-15:
Can you do the same after adding your CIFS mounts?

aarste, 2023-03-15:
```
Filesystem                                                     1K-blocks       Used Available Use% Mounted on
proc                                                                   0          0         0    - /proc
udev                                                               10240          0     10240   0% /dev
/dev/sda1                                                       24747988   11997376  11721964  51% /
sysfs                                                                  0          0         0    - /sys
securityfs                                                             0          0         0    - /sys/kernel/security
tmpfs                                                            3041756       2032   3039724   1% /dev/shm
devpts                                                                 0          0         0    - /dev/pts
tmpfs                                                            1216704      27080   1189624   3% /run
tmpfs                                                               5120          0      5120   0% /run/lock
cgroup2                                                                0          0         0    - /sys/fs/cgroup
pstore                                                                 0          0         0    - /sys/fs/pstore
none                                                                   0          0         0    - /sys/fs/bpf
systemd-1                                                              -          -         -    - /proc/sys/fs/binfmt_misc
mqueue                                                                 0          0         0    - /dev/mqueue
hugetlbfs                                                              0          0         0    - /dev/hugepages
debugfs                                                                0          0         0    - /sys/kernel/debug
tracefs                                                                0          0         0    - /sys/kernel/tracing
configfs                                                               0          0         0    - /sys/kernel/config
fusectl                                                                0          0         0    - /sys/fs/fuse/connections
sunrpc                                                                 0          0         0    - /run/rpc_pipefs
systemd-1                                                              -          -         -    - /mnt/samsung256
tmpfs                                                            2012160          0   2012160   0% /tmp
/dev/sdb1                                                      244981300   26270228 218694688  11% /mnt/samsung256
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/413bd804a0ee08b03757ba86856f687bd581224e6eeb8ee8a3f7d428c911f579/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/c507160747083bbdfa5865a9a39a51be2bef41a0193a1fa2d33a0986cddd5ff8/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/93d93f21a696bb10b32f68b8a6d427707ec0d25b039a3f0791b124f98799a5e8/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/b4006e1fdc5b68ad62b50c0a9057a2de2a1f6572bc3beff6ea713ec33ea7e758/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/2d71105ff0109473995a0bc8bc6ab2ec47e9782fec536d133d0a28bdb3310b2c/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/b24005c266d0648349990b4c0e383709b2bf3b238b18dd61227f11a1255af216/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/f27eec8f3fa1acc4e073ef9cc9a0e99de6c014ff00b62c6cda88fcd0d2eeb062/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/29d31ee4170e428408234bd5cc3793bbc8ded352e7ee8872ae305a88a977cdee/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/6aef3b7be060f78ed5917302774d2ee843eae53233ebe03fe8a3d743ea38015a/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/00607ee5265902b8761db6ec180e14793f8b7f11745a8d918df55732ca0003d3/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/5be1096873d7ea2c1d4d5a668ded7030dd4503560528846dd7ae2c9dbc595c89/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/a8ebb7400a8a30e7b23d20a44614a1b3af934328c392b1e0f1dd0f6a0ba8bade/merged
overlay                                                                -          -         -    - /mnt/dietpi_userdata/docker-data/overlay2/03efb6390dc2bdc5fa9fb55573f61649a6b4cf242f481ea7f9b2c33f3a78bf58/merged
[... cut ...]
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
UUID=2157c567-327a-4fc2-ad7a-6a362e9b1579 / ext4 noatime,lazytime,rw 0 1
UUID=bfb6bc5f-b187-478c-8a9c-c160a812df77 /mnt/samsung256 ext4 noatime,lazytime,rw,nofail,noauto,x-systemd.automount

MichaIng, 2023-03-15:
Okay, the automount units for the SMB drives are not active until reboot, I guess. systemd-timesyncd is happy right now, e.g. when doing
```sh
systemctl start systemd-timesyncd
```
?

aarste, 2023-03-15:
It seems fine, but that's because my Windows PC is online I assume. I will see what it logs show tomorrow morning.

The mounts are accessible, did you wanted me to reboot DietPi?

```
dietpi@DietPi-proxmox:~$ sudo systemctl start systemd-timesyncd
dietpi@DietPi-proxmox:~$ sudo systemctl status systemd-timesyncd
● systemd-timesyncd.service - Network Time Synchronization
     Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-03-15 19:31:01 GMT; 3h 17min ago
       Docs: man:systemd-timesyncd.service(8)
   Main PID: 2157478 (systemd-timesyn)
     Status: "Initial synchronization to time server 193.47.147.20:123 (0.debian.pool.ntp.org)."
      Tasks: 2 (limit: 7119)
     Memory: 1.0M
        CPU: 137ms
     CGroup: /system.slice/systemd-timesyncd.service
             └─2157478 /lib/systemd/systemd-timesyncd

Mar 15 19:31:01 DietPi-proxmox systemd[1]: Starting Network Time Synchronization...
Mar 15 19:31:01 DietPi-proxmox systemd[1]: Started Network Time Synchronization.
Mar 15 19:30:31 DietPi-proxmox systemd-timesyncd[2157478]: Initial synchronization to time server 193.47.147.20:123 (0.debian.pool.ntp.org).
```

aarste, 2023-03-16:
This morning:
```
Mar 16 08:30:15 DietPi-proxmox systemd[1]: Starting Network Time Synchronization...
Mar 16 08:30:25 DietPi-proxmox systemd[2490287]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Host is down
Mar 16 08:30:25 DietPi-proxmox systemd[2490287]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Host is down
Mar 16 08:30:25 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Mar 16 08:30:25 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Mar 16 08:30:25 DietPi-proxmox systemd[1]: Failed to start Network Time Synchronization.
Mar 16 08:30:25 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 1936.
Mar 16 08:30:25 DietPi-proxmox systemd[1]: Stopped Network Time Synchronization.
Mar 16 08:30:25 DietPi-proxmox systemd[1]: Starting Network Time Synchronization...
Mar 16 08:30:35 DietPi-proxmox systemd[2490324]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Host is down
Mar 16 08:30:35 DietPi-proxmox systemd[2490324]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Host is down
Mar 16 08:30:35 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Mar 16 08:30:35 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Mar 16 08:30:35 DietPi-proxmox systemd[1]: Failed to start Network Time Synchronization.
Mar 16 08:30:35 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 1937.
Mar 16 08:30:35 DietPi-proxmox systemd[1]: Stopped Network Time Synchronization.
Mar 16 08:30:35 DietPi-proxmox systemd[1]: Starting Network Time Synchronization...
Mar 16 08:30:42 DietPi-proxmox systemd[2490387]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Host is down
Mar 16 08:30:42 DietPi-proxmox systemd[2490387]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Host is down
Mar 16 08:30:42 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Mar 16 08:30:42 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Mar 16 08:30:42 DietPi-proxmox systemd[1]: Failed to start Network Time Synchronization.
Mar 16 08:30:42 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 1938.
Mar 16 08:30:42 DietPi-proxmox systemd[1]: Stopped Network Time Synchronization.
Mar 16 08:30:42 DietPi-proxmox kernel: CIFS: VFS: \\192.168.1.10 has not responded in 180 seconds. Reconnecting...
Mar 16 08:30:42 DietPi-proxmox systemd[1]: Starting Network Time Synchronization...
Mar 16 08:30:42 DietPi-proxmox systemd[1]: Started Network Time Synchronization.
Mar 16 08:30:42 DietPi-proxmox systemd-timesyncd[2490447]: Initial synchronization to time server 194.54.80.29:123 (0.debian.pool.ntp.org).
```

I will set up a new DietPi install in Proxmox to see if I can reproduce it there

Joulinar, 2023-03-16:
can you try to unmount the CIFS share before going to switch of your Windows box? This should stop the spam over night.

aarste, 2023-03-16:
Yes, that would stop the log spam but I would have to automate the mount & unmount somehow when it detects my Windows box is offline/online

This isn't hugely important for me to fix though, I can always use Syncthing instead. I should invest in a 24/7 NAS box ideally!

aarste, 2023-03-17:
I took out my laptop to ssh in, and then disabled the NIC on my Windows PC (that should have the same effect as an offline machine shouldn't it?)

These are the results while CIFS shares are mounted, there was a hang when running systemctl restart systemd-timesyncd:
```
Mar 16 23:50:26 DietPi-proxmox systemd[1]: Stopping Network Time Synchronization...
Mar 16 23:50:26 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Succeeded.
Mar 16 23:50:26 DietPi-proxmox systemd[1]: Stopped Network Time Synchronization.
Mar 16 23:50:26 DietPi-proxmox systemd[1]: Starting Network Time Synchronization...
Mar 16 23:51:56 DietPi-proxmox systemd[1]: systemd-timesyncd.service: start operation timed out. Terminating.
Mar 16 23:51:56 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Failed with result 'timeout'.
Mar 16 23:51:56 DietPi-proxmox systemd[1]: Failed to start Network Time Synchronization.
Mar 16 23:51:56 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 1.
Mar 16 23:51:56 DietPi-proxmox systemd[1]: Stopped Network Time Synchronization.
Mar 16 23:51:56 DietPi-proxmox systemd[1]: Starting Network Time Synchronization...
Mar 16 23:52:59 DietPi-proxmox kernel: CIFS: VFS: \\192.168.1.10 has not responded in 180 seconds. Reconnecting...
Mar 16 23:53:09 DietPi-proxmox systemd[2768949]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Host is down
Mar 16 23:53:09 DietPi-proxmox systemd[2768949]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Host is down
Mar 16 23:53:09 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Mar 16 23:53:09 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Mar 16 23:53:09 DietPi-proxmox systemd[1]: Failed to start Network Time Synchronization.
Mar 16 23:53:09 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 2.
Mar 16 23:53:09 DietPi-proxmox systemd[1]: Stopped Network Time Synchronization.
Mar 16 23:53:09 DietPi-proxmox systemd[1]: Starting Network Time Synchronization...
Mar 16 23:53:20 DietPi-proxmox systemd[2769286]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Host is down
Mar 16 23:53:20 DietPi-proxmox systemd[2769286]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Host is down
Mar 16 23:53:20 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Mar 16 23:53:20 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Mar 16 23:53:20 DietPi-proxmox systemd[1]: Failed to start Network Time Synchronization.
Mar 16 23:53:20 DietPi-proxmox systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 3.
Mar 16 23:53:20 DietPi-proxmox systemd[1]: Stopped Network Time Synchronization.
```

For your second instruction I ran umount /mnt/filebrowser (the NIC is still disabled), this took about 30 seconds to complete. systemctl restart systemd-timesyncd completes successfully. (/mnt/backups is still mounted however)

So it looks like it doesn't like my filebrowser mount, I suspect it is due to the spacing or possibly hash sign I have in the windows share of this folder?

Joulinar, 2023-03-17:
yeah `systemd` don't like to share being offline. Not sure if there is a way to avoid these error messages in this case.

MichaIng, 2023-03-17:
I was able to replicate it. I was wondering why restarting other services does not fail, and the relevant difference is the `ProtectSystem=strict` sandboxing, which remounts the whole filesystem tree read-only for the particular service. I guess this implies mounting the SMB share as well into the service environment, which is the reason it hangs and fails.

However, other things hang as well, like `systemctl daemon-reload`, `df` and others, basically anything which directly or indirectly probes that mount. And systemd.automount btw is unrelated, but would cause additional hangs on mount point access.

Long story short, umount that share before shutting down the system. GNU/Linux seems to not intend handle offline network mounts gracefully, so this is not related to DietPi or even systemd (while systemd makes often use of mount point probes for its sandboxing and automount feature).

aarste, 2023-03-18:
Thanks for looking into this. I had to do something similar in Proxmox a few months back when I had a Windows share set up as a storage in Promox for VM backups, a cron job to unmount my Windows

closed 2023-03-18 by aarste

Joulinar, 2023-03-18:
Better to think of a solution that is available 24/7 if the share is mounted network wide on various systems. Could be a Proxmox VM or SBC. 

We have a series of blog post describing on how to create your own raid system, which could be used as base for an own SAMBA server https://dietpi.com/blog/?p=2239 

Or as alternat some ready to use solutions from Qnap, Synology or WD. 😄

MichaIng, 2023-03-18:
Or the other way round: Instead of mounting an SMB share from the PC on DietPi, mount a share from DietPi on the PC, to exchange files. As DietPi runs 24/7, it is much suited to be a file server.