MichaIng/DietPi #6249: ODROID-C1 Doesn't boot with latest images

as-shura, 2023-03-15:
Flash the image with Rufus and stick it into the sd slot on the ODROID-C1.

Serial Console logs:

```
QA5:A;SVN:B72;POC:17F;STS:0;BOOT:0;INIT:10;BOOT:1;INIT:0;READ:0;CHECK:0;PASS:1;
-----------------------------------------------------------------------
* Welcome to Hardkernel's ODROID-C... (Built at 19:33:00 Dec  8 2014) *
-----------------------------------------------------------------------
CPU : AMLogic S805
MEM : 1024MB (DDR3@792MHz)
BID : HKC1310001
S/N : HKC11122F37DE839
0x0000009f
check SD_boot_type:0x1   card_type:0x1
Loading U-boot...success.

U-boot-00000-gb7b8dc2-dirty(odroidc@b7b8dc21) (Mar 08 2021 - 18:45:29)

I2C:   clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[20]=0
set output val 0xc1108058[20]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
out reg=c1108058,value=ffccfa00
set output en 0xc1108054[20]=0
set output val 0xc1108058[20]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffecfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
ready
DRAM:  1 GiB
relocation Offset is: 2ff18000
MMC:   SDCARD: 0, eMMC: 1
IR init is done!
vpu clk_level = 3
set vpu clk: 182150000Hz, readback: 182150000Hz(0x701)
mode = 6  vic = 4
set HDMI vic: 4
mode is: 6
viu chan = 1
config HPLL
config HPLL done
reconfig packet setting done
MMC read: dev # 0, block # 33984, count 12288 ... 12288 blocks read: OK
Error: Bad gzipped data
There is no valid bmp file at the given address
============================================================
Vendor: Man 035344 Snr dcee2413 Rev: 8.0 Prod: SC32G
            Type: Removable Hard Disk
            Capacity: 30436.5 MB = 29.7 GB (62333952 x 512)
------------------------------------------------------------
Partition     Start Sector     Num Sectors     Type
    1                 8192          262144       c
    2               270336          760088      83
============================================================
Net:   Meson_Ethernet
init suspend firmware done. (ret:0)
Hit Enter key to stop autoboot -- :  0
exit abortboot: 0
reading boot.ini

3236 bytes read
Loading boot.ini from mmc0:1 (vfat)
Executing the script...
setenv m "1080p"                # 1080P@60Hz
setenv vout "hdmi"
setenv m_bpp "32"
setenv monitor_onoff "true"
setenv condev "console=ttyAML0,115200n8 console=tty1"
setenv max_freq "1536"
if test "${hpd}" = "0"; then setenv hdmi_hpd "disablehpd=true"; fi
if test "${cec}" = "1"; then setenv hdmi_cec "hdmitx=cecf"; fi
if test "${disable_vu7}" = "false"; then setenv hid_quirks "usbhid.quirks=0x0eef:0x0005:0x0004"; fi
setenv bootargs "root=UUID=b83304f6-e29e-411d-ae85-421876dfcce1 rootfstype=ext4 rootwait rw ${condev} loglevel=4 no_console_suspend consoleblank=0 vdaccfg=0xa000 dmfc=3 cvbsmode=576cvbs hdmimode=${m} m_bpp=${m_bpp} vout=${vout} ${disableuhs} ${hdmi_hpd} ${hdmi_cec} ${enabledac} monitor_onoff=${monitor_onoff} max_freq=${max_freq} ${hid_quirks} ${extraargs}"
fatload mmc 0:1 0x20800000 uImage
reading uImage
Invalid FAT entry

** Unable to read "uImage" from mmc 0:1 **
fatload mmc 0:1 0x22000000 uInitrd
reading uInitrd
Invalid FAT entry

** Unable to read "uInitrd" from mmc 0:1 **
fatload mmc 0:1 0x21800000 dtb/meson8b-odroidc1.dtb
reading dtb/meson8b-odroidc1.dtb
Invalid FAT entry

** Unable to read "dtb/meson8b-odroidc1.dtb" from mmc 0:1 **
fdt addr 21800000
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
if test "${vpu}" = "0"; then fdt rm /mesonstream; fdt rm /vdec; fdt rm /ppmgr; fi
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
if test "${hdmioutput}" = "0"; then fdt rm /mesonfb; fi
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
bootm 0x20800000 0x22000000 0x21800000
Wrong Image Format for bootm command
ERROR: can't get kernel image!
▒
Unknown command '▒' - try 'help'
MMC read: dev # 0, block # 1216, count 16384 ... 16384 blocks read: OK
MMC read: dev # 0, block # 1088, count 128 ... 128 blocks read: OK
Wrong Image Format for bootm command
ERROR: can't get kernel image!
```

MichaIng, 2023-03-15:
Many thanks for your report.

It loads the `boot.ini` script correctly but it fails to load kernel image, initramfs and device tree. Those three files are however present in our image. Can you verify this when attaching the SD card to another system and accessing the first/FAT partition there?

Thomas-O, 2023-03-15:
I've also been unable to boot the latest images. Though it is possible to boot Dietpi on the Odroid C1 if I use the conversion script to rework Armbian Buster 5.10.12.

However, if I mount a usb flash drive that version will also fail to boot.

MichaIng, 2023-03-15:
Could you answer the question form my previous post, please?

> However, if I mount a usb flash drive that version will also fail to boot.

You mean any image fails to boot from SD card as long as you also have a USB flash drive attached? Sounds like a power/voltage issue.

as-shura, 2023-03-15:
> Many thanks for your report.
> 
> It loads the `boot.ini` script correctly but it fails to load kernel image, initramfs and device tree. Those three files are however present in our image. Can you verify this when attaching the SD card to another system and accessing the first/FAT partition there?

Let me reflash an SD card and check for validation.

as-shura, 2023-03-15:
Content of the boot.ini file:

```
ODROIDC-UBOOT-CONFIG

# HDMI resolution
# - Exactly one line needs to be uncommented!
#setenv m "vga"          	# 640x480
#setenv m "480p"         	# 720x480
#setenv m "576p"         	# 720x576
#setenv m "800x480p60hz" 	# 800x480
#setenv m "800x600p60hz" 	# 800x600
#setenv m "1024x600p60hz"	# 1024x600
#setenv m "1024x768p60hz"	# 1024x768
#setenv m "1360x768p60hz"	# 1360x768
#setenv m "1440x900p60hz"	# 1440x900
#setenv m "1600x900p60hz"	# 1600x900
#setenv m "1680x1050p60hz"	# 1680x1050
#setenv m "720p"         	# 720p 1280x720
#setenv m "800p"         	# 1280x800
#setenv m "sxga"         	# 1280x1024
#setenv m "1080i50hz"    	# 1080I@50Hz
#setenv m "1080p24hz"    	# 1080P@24Hz
#setenv m "1080p50hz"    	# 1080P@50Hz
setenv m "1080p"         	# 1080P@60Hz
#setenv m "1920x1200"    	# 1920x1200

# HDMI/DVI selection: "hdmi" or "dvi"
# - DVI mode disables HDMI sound
setenv vout "hdmi"

# HDMI BPP Mode: "32", "24" or "12"
setenv m_bpp "32"

# Monitor output: "true" or "false"
# - Controls if HDMI PHY should output anything to the monitor
setenv monitor_onoff "true"

# HDMI Hot Plug detection
# - "0" disables auto-detection and forces HDMI output.
# - "1" enables HDMI detection based on cable connection (default).
#setenv hpd "0"

# CEC (requires hardware modification)
# - "0" disables HDMI CEC (default).
# - "1" enables HDMI CEC.
#setenv cec "0"

# PCM5102 I2S Audio DAC
# - PCM5102 is an I2S Audio DAC addon board for ODROID-C1+
# - Uncomment the line below to __ENABLE__ support for this addon board.
#setenv enabledac "enabledac"

# UHS Card Configuration
# - Uncomment the line below to __DISABLE__ UHS-1 Micro SD support
# - This might break boot for some brand models of cards.
#setenv disableuhs "disableuhs"

# Disable VPU (Video decoding engine, saves RAM!!!)
# - 0 = disabled
# - 1 = enabled (default)
#setenv vpu "1"

# Disable HDMI Output (Again, saves RAM!)
# - 0 = disabled
# - 1 = enabled (default)
#setenv hdmioutput "1"

# Default Console Device Setting
setenv condev "console=ttyAML0,115200n8 console=tty1"

# ODROID-VU7 touchscreen: "false" or "true", defaults to "true"
#setenv disable_vu7 "false"

# CPU Max Frequency: 96 192 312 408 504 600 720 816 1008 1200 1320 1488 1536 1632 1728 or 1824
setenv max_freq "1536"

### DO NOT EDIT ANYTHING BELOW THIS LINE ###

if test "${hpd}" = "0"; then setenv hdmi_hpd "disablehpd=true"; fi
if test "${cec}" = "1"; then setenv hdmi_cec "hdmitx=cecf"; fi
if test "${disable_vu7}" = "false"; then setenv hid_quirks "usbhid.quirks=0x0eef:0x0005:0x0004"; fi

# Boot arguments
setenv bootargs "root=UUID=b83304f6-e29e-411d-ae85-421876dfcce1 rootfstype=ext4 rootwait rw ${condev} loglevel=4 no_console_suspend consoleblank=0 vdaccfg=0xa000 dmfc=3 cvbsmode=576cvbs hdmimode=${m} m_bpp=${m_bpp} vout=${vout} ${disableuhs} ${hdmi_hpd} ${hdmi_cec} ${enabledac} monitor_onoff=${monitor_onoff} max_freq=${max_freq} ${hid_quirks} ${extraargs}"

# Booting
fatload mmc 0:1 0x20800000 uImage
fatload mmc 0:1 0x22000000 uInitrd
fatload mmc 0:1 0x21800000 dtb/meson8b-odroidc1.dtb

fdt addr 21800000

if test "${vpu}" = "0"; then fdt rm /mesonstream; fdt rm /vdec; fdt rm /ppmgr; fi
if test "${hdmioutput}" = "0"; then fdt rm /mesonfb; fi

bootm 0x20800000 0x22000000 0x21800000
```

Folder structures :

[image]

[image]

[image]

Thomas-O, 2023-03-15:
> Could you answer the question form my previous post, please?
> 
> > However, if I mount a usb flash drive that version will also fail to boot.
> 
> You mean any image fails to boot from SD card as long as you also have a USB flash drive attached? Sounds like a power/voltage issue.

I don't think it's a power issue. The Armbian Buster image will boot with two usb drives attached. But it's not possible when I convert Armbian to DietPi.

as-shura, 2023-03-15:
@Thomas-O I don't know what you are talking about but before it gets to any of the things you are talking about it doesn't enter inside of the kernel. After we achieve that can we validate your hypothesis.

MichaIng, 2023-03-15:
Did you upgrade to latest kernel version 5.15.93 on the Armbian image? The DietPi conversion cannot really have any impact on booting, but the kernel and U-Boot upgrades can have (which are implied by the DietPi update). And it can have effect on peak power consumption.

All files are there. The U-Boot package is and was very old and the same, as is `boot.ini`. Only the kernel changed, upgraded in mid February, but this cannot explain the U-Boot failures.

as-shura, 2023-03-15:
> Did you upgrade to latest kernel version 5.15.93 on the Armbian image? The DietPi conversion cannot really have any impact on booting, but the kernel and U-Boot upgrades can have (which are implied [...]

@MichaIng who are you addressing yourself to?

as-shura, 2023-03-15:
Why can't the device get to the kernel image if it is there ?

MichaIng, 2023-03-15:
@as-shura 
My last post was addressing @Thomas-O as he was testing the Armbian image and conversion.

> Why can't the device get to the kernel image if it is there ?

I have no idea. I rechecked and compared the image and couldn't find any issue. I just triggered a new build, probably a quirk on the particular run.

To both of you: In the meantime you could also test the Bookworm image to rule out an issue with the particular build.

And a question: Do you have any need for the dedicated boot FAT partition? Based on the Armbian `boot.ini`, I think the bootloader actually supports booting from a single ext4 partition. What would simplify clean things up. Only argument to keep the FAT partition is accessing it easier from Windows and macOS to pre-configure `dietpi.txt` and/or `boot.ini`.

as-shura, 2023-03-16:
> And a question: Do you have any need for the dedicated boot FAT partition? Based on the Armbian `boot.ini`, I think the bootloader actually supports booting from a single ext4 partition. What would [...]

I think the old way of booting is not necessary anymore. It will be a good think to make it a single partition. Good call on this one.

I will test Bookworm in the mean time and keep you updated.

as-shura, 2023-03-16:
> Only argument to keep the FAT partition is accessing it easier from Windows and macOS to pre-configure `dietpi.txt` and/or `boot.ini`.

You can always use this https://github.com/matt-wu/Ext3Fsd to explore on Windows ext4 partitions.

as-shura, 2023-03-16:
> I have no idea. I rechecked and compared the image and couldn't find any issue. I just triggered a new build, probably a quirk on the particular run: https://github.com/MichaIng/DietPi/actions/runs/4431965962
> Ready in about 20 minutes.

Where can I download the newly generated image if there is something to test?

as-shura, 2023-03-16:
Bookword console log:

```odroidc#QA5:A;SVN:B72;POC:17F;STS:0;BOOT:0;INIT:10;BOOT:1;INIT:0;READ:0;CHECK:0;PASS:1;
-----------------------------------------------------------------------
* Welcome to Hardkernel's ODROID-C... (Built at 19:33:00 Dec  8 2014) *
-----------------------------------------------------------------------
CPU : AMLogic S805
MEM : 1024MB (DDR3@792MHz)
BID : HKC1310001
S/N : HKC11122F37DE839
0x0000009f
check SD_boot_type:0x1   card_type:0x1
Loading U-boot...success.

U-boot-00000-gb7b8dc2-dirty(odroidc@b7b8dc21) (Mar 08 2021 - 18:45:29)

I2C:   clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[20]=0
set output val 0xc1108058[20]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
[... 152 lines cut ...]
** Unable to read "uInitrd" from mmc 0:1 **
fatload mmc 0:1 0x21800000 dtb/meson8b-odroidc1.dtb
reading dtb/meson8b-odroidc1.dtb
Invalid FAT entry

** Unable to read "dtb/meson8b-odroidc1.dtb" from mmc 0:1 **
fdt addr 21800000
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
if test "${vpu}" = "0"; then fdt rm /mesonstream; fdt rm /vdec; fdt rm /ppmgr; fi
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
if test "${hdmioutput}" = "0"; then fdt rm /mesonfb; fi
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
bootm 0x20800000 0x22000000 0x21800000
Wrong Image Format for bootm command
ERROR: can't get kernel image!
▒
Unknown command '▒' - try 'help'
MMC read: dev # 0, block # 1216, count 16384 ... 16384 blocks read: OK
MMC read: dev # 0, block # 1088, count 128 ... 128 blocks read: OK
Wrong Image Format for bootm command
ERROR: can't get kernel image!
odroidc#
```

MichaIng, 2023-03-16:
But that Windows ext4 driver is read-only, isn't it? Reading the content of the image file can be done with 7-Zip as well. Writing from Windows and macOS is the challenge. There are drivers which support it, but all I tested destroyed the filesystem earlier or later, so this is extremely dangerous. A Linux VM is best to do it safely. Sadly WSL2 `wsl --mount` does not support USB devices OOTB and with some TCP based pass through method it does not support MMC/SD devices 😞.

No need to spam the issue with long U-Boot log, if it is just the same 😄.

Ah sorry, new image is ready here: https://dietpi.com/downloads/images/testing/

as-shura, 2023-03-16:
It is both read/write I use it everyday.

as-shura, 2023-03-16:
Same problem with the new image:

```
setenv bootargs "root=UUID=518088d3-8e71-4da2-85bf-00bdd1671f22 rootfstype=ext4 rootwait rw ${condev} loglevel=4 no_console_suspend consoleblank=0 vdaccfg=0xa000 dmfc=3 cvbsmode=576cvbs hdmimode=${m} m_bpp=${m_bpp} vout=${vout} ${disableuhs} ${hdmi_hpd} ${hdmi_cec} ${enabledac} monitor_onoff=${monitor_onoff} max_freq=${max_freq} ${hid_quirks} ${extraargs}"
fatload mmc 0:1 0x20800000 uImage
reading uImage
Invalid FAT entry

** Unable to read "uImage" from mmc 0:1 **
fatload mmc 0:1 0x22000000 uInitrd
reading uInitrd
Invalid FAT entry

** Unable to read "uInitrd" from mmc 0:1 **
fatload mmc 0:1 0x21800000 dtb/meson8b-odroidc1.dtb
reading dtb/meson8b-odroidc1.dtb
Invalid FAT entry

** Unable to read "dtb/meson8b-odroidc1.dtb" from mmc 0:1 **
fdt addr 21800000
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
if test "${vpu}" = "0"; then fdt rm /mesonstream; fdt rm /vdec; fdt rm /ppmgr; fi
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
if test "${hdmioutput}" = "0"; then fdt rm /mesonfb; fi
libfdt fdt_path_offset() returned FDT_ERR_BADMAGIC
bootm 0x20800000 0x22000000 0x21800000
Wrong Image Format for bootm command
ERROR: can't get kernel image!
▒
Unknown command '▒' - try 'help'
MMC read: dev # 0, block # 1216, count 16384 ... 16384 blocks read: OK
MMC read: dev # 0, block # 1088, count 128 ... 128 blocks read: OK
Wrong Image Format for bootm command
ERROR: can't get kernel image!

```

as-shura, 2023-03-16:
Why isn't Dietpi using the latest mainline kernel with odroid-c1?

https://kernel.ubuntu.com/~kernel-ppa/mainline/v6.3-rc2/

MichaIng, 2023-03-16:
Okay I have no idea what changed, looked through all commits and checked the image, all seems identical. Of course in theory some defaults on latest GitHub Actions runners parted or dosfstools could have changed, or a bug introduced or such, but this is extremely unlikely for fundamental tools like this on the most settled filesystem that exists. And neither Windows, 7-Zip or RPis have any issue accessing this FAT filesystem 🤔.

However, lets skip that and test whether a single ext4 partition works. Build is running and image available in about 30 minutes as linked above.

 > Why isn't Dietpi using the latest mainline kernel with odroid-c1?

Because we use the Armbian meson kernel which is currently at 5.15.93. However, the kernel is not the issue here, but the combination of bootloader and this FAT filesystem and/or partition table.

We just started to do own kernel builds with Quartz64 and RISC-V VisionFive 2 SBCs. When the CI development for this settled, I see no problem to do own kernel and in case bootloader builds as well for other SBCs. Upstream Linux has an Odroid C1 device tree, so theoretically support is there. And since Armbian kernels have a bunch of issues as well, sometimes caused by their large amount of patches causing conflicts with upstream changes, it might even make things better. E.g. HDMI is not working on C1, right? Probably with unmodified current upstream Linux is does, who knows.

as-shura, 2023-03-16:
> Because we use the Armbian meson kernel which is currently at 5.15.93. However, the kernel is not the issue here, but the combination of bootloader and this FAT filesystem and/or partition table.

Great news waiting for the build to finish and will test it ASAP.

> We just started to do own kernel builds with Quartz64 and RISC-V VisionFive 2 SBCs (never did this before). When the CI development for this settled, I see no problem to do own kernel and in case bootloader builds as well for other SBCs.

I think that we should have parallel kernels just to se the upcoming challenges while we are running working kernel. Good move for Future.

> Upstream Linux has an Odroid C1 device tree, so theoretically support is there. And since Armbian kernels have a bunch of issues as well, sometimes caused by their large amount of patches causing conflicts with upstream changes, it might even make things better.

I tested all **armbian** mainline images with 5.1* and they are all corrupting the SD card at some point and the system is not usable at all. I am getting all these errors:

```
[  398.788954] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #2: comm systemd-udevd: reading directory lblock 0
[  398.799372] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #10930: comm systemd-udevd: reading directory lblock 0
[  398.831376] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #10930: comm systemd-udevd: reading directory lblock 0
[  398.841743] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #5352: comm systemd-udevd: reading directory lblock 0
[  398.857158] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #2: comm systemd-udevd: reading directory lblock 0
[  399.019189] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #10930: comm systemd-udevd: reading directory lblock 0
[  399.069134] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #76498: comm systemd-udevd: reading directory lblock 0
[  399.093787] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #5352: comm systemd-udevd: reading directory lblock 0
[  399.104768] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #2: comm systemd-udevd: reading directory lblock 0
[  399.119149] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #10930: comm systemd-udevd: reading directory lblock 0
[  399.129577] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #76498: comm systemd-udevd: reading directory lblock 0
[  399.140373] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1660: inode #5361: comm systemd-udevd: reading directory lblock 0
```

> E.g. HDMI is not working on C1, right? Probably with unmodified current upstream Linux is does, who knows.

IMHO the best use case for ODROID-C1 series is only serverless with docker for home server setup. Only needed is ssh+network+external-usb-HD

MichaIng, 2023-03-16:
> I think that we should have parallel kernels just to se the upcoming challenges while we are running working kernel. Good move for Future.

It is always possible to install the "edge" kernel builds:
```sh
apt install linux-{image,dtb}-meson-edge linux-u-boot-odroidc1-edge
```
I think the "edge" U-Boot does not exist (for Odroid C1, as it is not officially supported anymore by Armbian), or is practically the same as the "current" one. In case skip that.

However, the benefit is probably a more reliable kernel basis. Time will tell if upstream Linux is more reliable (not suddenly breaking features with upgrades) for Odroid C1 or in general, compared to Armbian.

Even if HDMI is not seen important, it would be just nice to have all basic hardware features supported. I remember it broke for Odroid C1 when Armbian moved from the ancient Linux 3.x kernel from Hardkernel sources to mainline Linux. But possible mainline Linux did get support in the meantime, and there is only one of the huge number of Armbian kernel patches breaking it.

MichaIng, 2023-03-16:
And, does the new image boot? https://dietpi.com/downloads/images/testing/

as-shura, 2023-03-16:
> And, does the new image boot? https://dietpi.com/downloads/images/testing/

I just arrived now and will test it right away.

as-shura, 2023-03-16:
I can confirm this because I tested on 2 different SD card and still same looping.

as-shura, 2023-03-16:
If you make this happen you will save 2 ODROID-C1 boards and their future. We will employ them for their lifetime as docker host and few services. Otherwise I am going to crush the boards and throw them in garbage by saying "what a bad investment choosing you for the future" 😄

MichaIng, 2023-03-17:
Ah my mistake. New image builds: https://github.com/MichaIng/DietPi/actions/runs/4442800002

as-shura, 2023-03-17:
Testing soon still download and will be flashing afterwards.

Thomas-O, 2023-03-17:
I've just tried the latest (2023-03-17) Odroid C1 Bullseye image and I'm sorry to report that there's still nothing happening at my end.

As a semi-technical user for whom DietPi is a godsend I won't be able to provide a sophisticated analysis of any shortcomings, but I was a little puzzled by the missing boot partition. When I flashed the image all I could see was the 400mb ext4 filesystem. Sorry if this is a doltish observation but installing and clumsily configuring DietPi already stretches the limits of my comprehension and capacity.

as-shura, 2023-03-17:
> One last attempt: https://github.com/MichaIng/DietPi/actions/runs/4447536627

I think we need more people on this task if you are at the end of your roll of ideas.

```
QA5:A;SVN:B72;POC:17F;STS:0;BOOT:0;INIT:10;BOOT:1;INIT:0;READ:0;CHECK:0;PASS:1;
-----------------------------------------------------------------------
* Welcome to Hardkernel's ODROID-C... (Built at 19:33:00 Dec  8 2014) *
-----------------------------------------------------------------------
CPU : AMLogic S805
MEM : 1024MB (DDR3@792MHz)
BID : HKC1310001
S/N : HKC11122F37DE839
0x0000009f
check SD_boot_type:0x1   card_type:0x1
Loading U-boot...success.

U-boot-00000-gb7b8dc2-dirty(odroidc@b7b8dc21) (Mar 08 2021 - 18:45:29)

I2C:   clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[20]=0
set output val 0xc1108058[20]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
[... 116 lines cut ...]
Vendor: Man 035344 Snr bcd3427f Rev: 8.0 Prod: ACLCD
            Type: Removable Hard Disk
            Capacity: 30436.5 MB = 29.7 GB (62333952 x 512)
------------------------------------------------------------
Partition     Start Sector     Num Sectors     Type
    1                 8192          843696      83
============================================================
Net:   Meson_Ethernet
init suspend firmware done. (ret:0)
Hit Enter key to stop autoboot -- :  0
exit abortboot: 0

** Unable to use mmc 0:1 for fatload **
Loading file "/boot/boot.ini" from mmc device 0:1 xxa1
** File not found /boot/boot.ini
Loading file "boot.ini" from mmc device 0:1 xxa1
** File not found boot.ini
Failed to load boot.init, maybe empty file?
MMC read: dev # 0, block # 1216, count 16384 ... 16384 blocks read: OK
MMC read: dev # 0, block # 1088, count 128 ... 128 blocks read: OK
Wrong Image Format for bootm command
ERROR: can't get kernel image!
odroidc#

```

MichaIng, 2023-03-17:
More people won't help. That U-Boot supports FAT only, that's it 😢. I'm regenerating a FAT image now. If it does not boot like the previous one, I have two ideas to test.

as-shura, 2023-03-17:
I am behind you I will test as much as I can to support. I got your back 😄

Joulinar, 2023-03-18:
> I think we need more people on this task

How much people do you expect working on DietPi project? We are an extremely small group of people, where @MichaIng is the only developer we have. 😉

as-shura, 2023-03-18:
> > I think we need more people on this task
> 
> How much people do you expect working on DietPi project? We are an extremely small group of people, where @MichaIng is the only developer we have. 😉

I am a developer also but never worked with Linux programming.

as-shura, 2023-03-18:
> More people won't help. That U-Boot supports FAT only, that's it 😢. I'm regenerating a FAT image now. If it does not boot like the previous one, I have two ideas to test.

When ever you have a new build let me knowI will be outside but close to home and will come back to test on the board.

as-shura, 2023-03-19:
This is the last release on the 16th.

[image]

MichaIng, 2023-03-19:
This just looks like a wrong baudrate. You do use `115200`?

as-shura, 2023-03-19:
> This just looks like a wrong baudrate. You do use `115200`?

[image]

MichaIng, 2023-03-21:
It is strange, according to Hardkernel's U-Boot sources as well as a matching patch from Armbian, it _should_ support an ext4 partition:
- https://github.com/hardkernel/u-boot/blob/odroidc-v2011.03/board/hardkernel/odroidc/cfgload.c#L32-L38
- https://github.com/armbian/build/blob/main/patch/u-boot/legacy/board_odroidc1/cfgload-boot-ext4-improvements.patch

Those two messages are `ext4load` attempts, after `fatload` failed:
```sh
Loading file "/boot/boot.ini" from mmc device 0:1 xxa1
** File not found /boot/boot.ini
Loading file "boot.ini" from mmc device 0:1 xxa1
** File not found boot.ini
```
So it is a more fundamental issue with the image or SD card or socket or power of whatever that loading something from any partition/filesystem fails. Strange also that it happens at different stages:
- At first it did load `boot.ini` from the FAT partition, reporting `Loading boot.ini from mmc0:1 (vfat)`, but then failed to load the kernel, initramfs and dtb.
- With the ext4 image it wasn't even able to load the `boot.ini`, while it did try it with `ext4load`.

Just to rule it out: Does the [latest Armbian image](https://archive.armbian.com/odroidc1/archive/Armbian_21.02.3_Odroidc1_buster_current_5.10.21.img.xz) work for you @as-shura?

as-shura, 2023-03-21:
Interesting enough good news @MichaIng !!!!!

There is a heart beat and the blue led is blinking normal but the kernel is not loading. I will try to flash another SD card and see if the problem is at that level.

```
QA5:A;SVN:B72;POC:17F;STS:0;BOOT:0;INIT:10;BOOT:1;INIT:0;READ:0;CHECK:0;PASS:1;
-----------------------------------------------------------------------
* Welcome to Hardkernel's ODROID-C... (Built at 19:33:00 Dec  8 2014) *
-----------------------------------------------------------------------
CPU : AMLogic S805
MEM : 1024MB (DDR3@792MHz)
BID : HKC1310001
S/N : HKC11122F37DE839
0x0000009f
check SD_boot_type:0x1   card_type:0x1
Loading U-boot...success.

U-boot-00000-gb7b8dc2-dirty(odroidc@b7b8dc21) (Mar 08 2021 - 18:45:29)

I2C:   clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[20]=0
set output val 0xc1108058[20]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
[... 167 lines cut ...]
   Load Address: 02080000
   Entry Point:  02080000
   Verifying Checksum ... OK
## Loading init Ramdisk from Legacy Image at 22000000 ...
   Image Name:   uInitrd
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    5280705 Bytes = 5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 21800000
   Booting using the fdt blob at 0x21800000
   Loading Kernel Image ... OK
OK
uboot time: 6216223 us.
Using machid 0xf81 from environment
faild to get aml_reserved_end address
the default relocate ramdisk and fdt address-relocate_addr: 0x20000000
   Loading Ramdisk to 1faf6000, end 1ffff3c1 ... OK
   Loading Device Tree to 1faed000, end 1faf5d7a ... OK
libfdt fdt_path_offset() returned FDT_ERR_NOTFOUND
libfdt fdt_path_offset() returned FDT_ERR_NOTFOUND

Starting kernel ...
```

as-shura, 2023-03-21:
There is a problem with the the device I have an external HD connected and it doesn't get recognized at boot.

as-shura, 2023-03-21:
> Okay so far so good. Since we used FAT32 before, still no idea why now FAT16 is needed, or if it was just a coincidence.
> 
> I'll generate another ext4 image, but at least we have one that works again.

Yes i wanted to ask you to generate a normal ext4 partition... I sometimes have corruption errors on sdcard while the os is running. How complicated will it be to test with the 6.x mainline serie? I think a lot of bugs were fixed for that branch.

as-shura, 2023-03-21:
> The Bookworm image is just the same with FAT16 partition. I just wanted to have it with this filesystem for our stable downloads, as it is now know to boot.
> 
> ext4 partition images currently build, ready in ~30 minutes.

Ok good to know... but I am still testing it for validation 😄

There is one thing which is not like the previous versions. There is no details for the kernel's loading. Normally it should list all the modules one by one with green text but in this case it is just a cold login and pass screen.

as-shura, 2023-03-21:
> Ok good to know... but I am still testing it for validation

It works for good.

as-shura, 2023-03-21:
I am flashing and testing soon DietPi_OdroidC1-ARMv7-Bullseye.7z | 2023-03-21 18:21 | 105M

MichaIng, 2023-03-21:
> Normally it should list all the modules one by one with green text but in this case it is just a cold login and pass screen.

Boot log details are done to local console `/dev/tty1`, not serial console. But now that you mention it, since HDMI does not work anyway, we should switch it around to use the serial console for main boot logs. It's just exchanging the two `console=` entries in `/boot/boot.ini`:
```ini
setenv condev "console=ttyAML0,115200n8 console=tty1"
```
=>
``ìni
setenv condev "console=console=tty1 ttyAML0,115200n8"
```

as-shura, 2023-03-21:
Result with EXT4 partition

```
QA5:A;SVN:B72;POC:17F;STS:0;BOOT:0;INIT:10;BOOT:1;INIT:0;READ:0;CHECK:0;PASS:1;
-----------------------------------------------------------------------
* Welcome to Hardkernel's ODROID-C... (Built at 19:33:00 Dec  8 2014) *
-----------------------------------------------------------------------
CPU : AMLogic S805
MEM : 1024MB (DDR3@792MHz)
BID : HKC1310001
S/N : HKC11122F37DE839
0x0000009f
check SD_boot_type:0x1   card_type:0x1
Loading U-boot...success.

U-boot-00000-gb7b8dc2-dirty(odroidc@b7b8dc21) (Mar 08 2021 - 18:45:29)

I2C:   clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[20]=0
set output val 0xc1108058[20]=0
clear pinmux reg1[25]=0
clear pinmux reg8[12]=0
clear pinmux reg1[3]=0
clear pinmux reg1[2]=0
set output en 0xc1108054[20]=1
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=fffcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
clear pinmux reg1[1]=0
out reg=c1108058,value=ffdcfa00
set output en 0xc1108054[21]=0
set output val 0xc1108058[21]=0
clear pinmux reg1[24]=0
[... 115 lines cut ...]
There is no valid bmp file at the given address
============================================================
Vendor: Man 035344 Snr dcee2573 Rev: 8.0 Prod: SC32G
            Type: Removable Hard Disk
            Capacity: 30436.5 MB = 29.7 GB (62333952 x 512)
------------------------------------------------------------
Partition     Start Sector     Num Sectors     Type
    1                 8192          843664      83
============================================================
Net:   Meson_Ethernet
init suspend firmware done. (ret:0)
Hit Enter key to stop autoboot -- :  0
exit abortboot: 0

** Unable to use mmc 0:1 for fatload **
Loading file "/boot/boot.ini" from mmc device 0:1 xxa1
** File not found /boot/boot.ini
Loading file "boot.ini" from mmc device 0:1 xxa1
** File not found boot.ini
Failed to load boot.init, maybe empty file?
MMC read: dev # 0, block # 1216, count 16384 ... 16384 blocks read: OK
MMC read: dev # 0, block # 1088, count 128 ... 128 blocks read: OK
Wrong Image Format for bootm command
ERROR: can't get kernel image!
```

MichaIng, 2023-03-21:
So regarding ext4 its just as before. While the bootloader tries to use `ext4load`, it just isn't able to, for whichever reason.

Found the commit about the switch to FAT: https://github.com/armbian/build/commit/f054cac
> U-boot FAT driver is very unstable and there is no wish to dig into u-boot.

Typo, "ext4" was meant, and then it basically matches your tests: Theoretically ext4 should work, practically it does not, or at least not reliable. So for now I'll leave it like it is, probably testing an own U-Boot build (from same sources).

Btw, current kernel should be upgraded to Linux 6.1 soon: https://github.com/armbian/build/commit/41709dd
Can be done with "edge" kernel already:
```sh
apt install linux-{image,dtb}-edge-meson
```

as-shura, 2023-03-21:
> 

You mean replacing
```
setenv condev "console=ttyAML0,115200n8 console=tty1"
```
by this
```
setenv condev "console=console=tty1 ttyAML0,115200n8"
```

MichaIng, 2023-03-21:
> You mean replacing

Whoops, I meant replacing it by this:
```ini
setenv condev "console=tty1 console=ttyAML0,115200n8"
```
The last `console` entry is what the kernel uses for its primary logs.

> I tried the edge kernel and this is what I got

😞

Then lets hope that the build that is installed with "current" for everyone soon will not break it as well. I'm not keen to start maintaining own kernel builds for C1, as this is a very old device, used by a very small number of people, has no proper U-Boot support (obviously) and lacks major features already (HDMI, eMMC, ...). And of course there is not guarantee that it will ever work or just be a waste of time after all.

... interesting: https://github.com/armbian/build/blob/main/patch/kernel/archive/meson-6.1/board_odroidc1/dts-Enable-HDMI.patch
That patch was added by Armbian with v5.18, hence was never part of v5.15 or below. While its nice that someone obviously takes care the C1 on Armbian end, likely one of these added patches actually broke it. Probably this repaired it, not sure whether the latest kernel build is based on latest commit: https://github.com/armbian/build/commit/c6b9135

as-shura, 2023-03-22:
For the use I wanted it seems to be working without any error with docker+portainer+traefik in headless mode. I tried a complicated container setup and the board is able to deal with all networking without a glitch.

MichaIng, 2023-03-22:
A lot of think were stated to be fixed and broken again, or the fixes actually broke things. Did you have functional USB hot plug detection with any recent kernel? This SBC has neither proper mainline support, nor official Armbian support, so things are perfectly possible to break at any time, sadly.

as-shura, 2023-03-22:
Well now I am having issues with the board it seems that they also broke the sd card driver...

```
2: comm (agetty): reading directory lblock 0
[   83.893264] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #                                                         2: comm (agetty): reading directory lblock 0
[   83.909183] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #                                                         2: comm (agetty): reading directory lblock 0
[   83.918563] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #                                                         2: comm (agetty): reading directory lblock 0
[   88.858183] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm (modprobe): reading directory lblock 0
[   88.903315] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm systemd: reading directory lblock 0
[   88.912551] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm systemd: reading directory lblock 0
[   88.926312] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm systemd: reading directory lblock 0
[   88.934310] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm systemd: reading directory lblock 0
[   89.132229] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm systemd: reading directory lblock 0
[   89.142671] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm systemd: reading directory lblock 0
[   89.158675] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm systemd: reading directory lblock 0
[   89.167703] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm systemd: reading directory lblock 0
[   90.481864] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm (agetty): reading directory lblock 0
[   95.248476] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm (modprobe): reading directory lblock 0
[   99.849708] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #15234: comm systemd-udevd: reading directory lblock 0
[   99.858224] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #838: comm systemd-udevd: reading directory lblock 0
[  100.724925] debugfs: Directory 'mmcblk0' with parent 'block' already present!
[  125.599175] blk_update_request: I/O error, dev mmcblk0, sector 560 op 0x0:(READ) flags 0x84700 phys_seg 1 prio class 0
[  125.652291] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #15234: comm systemd-udevd: reading directory lblock 0
[  125.663098] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #838: comm systemd-udevd: reading directory lblock 0
[  125.834224] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm kworker/u8:3: reading directory lblock 0
[  125.844638] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1665: inode #2: comm kworker/u8:1: reading directory lblock 0
```

MichaIng, 2023-03-22:
> it seems that they also broke the sd card driver...

This looks like a common filesystem corruption, e.g. caused by unclean shutdown or crash. Try to fix it on reboot:
```sh
touch /forcefsck
reboot
cat /run/initramfs/fsck.log
```

> Could you make a new release with the latest mainline kernel 6.1

We are not doing these kernel releases, but Armbian does. You did already test Linux v6.1 from "edge" branch, and soon you will be offered the upgrade to v6.1 on "current" branch. As said, I'm currently not considering own kernel builds, as long as not something _definitely_ broke with Armbian's "current" branch. Currently we only see limitations with U-Boot and the kernel.

I read somewhere in the Armbian forum that WiFi usage causes SD card corruption. Not sure how WiFi and SD card can be related, but if you currently use WiFi, that would be something to test.

MichaIng, 2023-04-02:
New images built: https://dietpi.com/downloads/images/testing/
But without major kernel upgrade. However, the installer received some cleanup in the meantime.

closed 2023-04-11 by MichaIng