MichaIng/DietPi #6252: │ [FAILED] Unknown install state/First run setup failed                        │

Vinodshps, 2023-03-16:
[image]

Joulinar, 2023-03-16:
Was there an issue with filling the bug report template? Or any other reason to not fill it? At least you could have spent some words on what exactly you did.

Did you tried to hit `<Ok>`?

Vinodshps, 2023-03-17:
Hi, while trying to update the dietpi os i got that error message.

Joulinar, 2023-03-17:
This message is shown after an incorrect initial setup but not on update. Still you are missing to share basis information on your system.

MichaIng, 2023-03-17:
First try to solve it by hitting `<Ok>` to retry. If it fails again, hit `<Cancel>` and provide all logs you see on the screen/SSH client, so we can see the actual error.

And please fill the bug report template, so we know which hardware and image you used, to be able to replicate the issue, if it is not something related to your network setup.

Joulinar, 2023-03-17:
You can't add pictures using mail. Pls login to GitHub. And pla share your system details. There was a reason why you should have filled the bug report template. This is not for fun.

Vinodshps, 2023-03-17:
any solution plz

Vinodshps, 2023-03-17:
Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version

Joulinar, 2023-03-17:
As of now, you did not share a real error message. Therefore, we are not able to check your issue. Pls share all logs from `/var/tmp/dietpi/logs`. You can attach them as files directly to Github.

Vinodshps, 2023-03-17:
can you plz tell me command to skip update

closed 2023-03-17 by Vinodshps

Joulinar, 2023-03-17:
this is a mandatory process you need to complete. And again, without logs and error message, we are not able to assist.