MichaIng/DietPi #6261: Readarr service fails to start

Gabba35, 2023-03-20:
Readarr service fails to start. Sonarr running fine but was installed before upgrading to bullseye.

Appreciate any support on this, is Readarr not supported or has an issue with mono?

- DietPi version | version 5.1.4(1)-release (arm-unknown-linux-gnueabihf)
- Distro version | bullseye 1
- Kernel version | Linux DietPi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
- SBC model | RPi4
- Power supply used | 5.1 Volts · 3.0 Amps · 15.3 Watts 
- SD card used | Sandisk

#### Additional Information
- Software title | Readarr
- Was the software title installed freshly or updated/migrated? Installed freshly
- Can this issue be replicated on a fresh installation of DietPi? Not attempted

#### Steps to reproduce
1. Completed reinstall of software

#### Expected behaviour
Should start when rebooting or loading using dietpi-services

#### Actual behaviour
 ● readarr.service - Readarr (DietPi)                                                                                 │
                        │      Loaded: loaded (/etc/systemd/system/readarr.service; disabled; vendor preset: enabled)                          │
                        │      Active: failed (Result: exit-code) since Mon 2023-03-20 11:04:05 AEDT; 2h 31min ago                             │
                        │     Process: 7159 ExecStart=/opt/readarr/Readarr -nobrowser -data=/mnt/dietpi_userdata/readarr (code=exited,         │
                        │ status=203/EXEC)                                                                                                     │
                        │    Main PID: 7159 (code=exited, status=203/EXEC)                                                                     │
                        │         CPU: 79ms                                                                                                    │
                        │                                                                                                                      │
                        │ Mar 20 11:04:05 DietPi systemd[1]: readarr.service: Scheduled restart job, restart counter is at 5.                  │
                        │ Mar 20 11:04:05 DietPi systemd[1]: Stopped Readarr (DietPi).                                                         │
                        │ Mar 20 11:04:05 DietPi systemd[1]: readarr.service: Start request repeated too quickly.                              │
                        │ Mar 20 11:04:05 DietPi systemd[1]: readarr.service: Failed with result 'exit-code'.                                  │
                        │ Mar 20 11:04:05 DietPi systemd[1]: Failed to start Readarr (DietPi). 

#### Extra details
- Previously upgraded to Bullseye using instructions from Dietpi. Sonarr still operating correctly.

Joulinar, 2023-03-20:
pls reboot your system and share following afterwards

```
journalctl -u readarr.service
```

Gabba35, 2023-03-20:
Thanks for reaching our Joulinar!

root@DietPi:~# journalctl -u readarr.service
Journal file /var/log/journal/365bf3506715487e91db1ed3418110c5/system.journal is truncated, ignoring file.
-- Journal begins at Mon 2023-03-20 18:35:40 AEDT, ends at Mon 2023-03-20 18:38:22 AEDT. --
Mar 20 18:35:52 DietPi systemd[1]: Started Readarr (DietPi).
Mar 20 18:35:52 DietPi systemd[713]: readarr.service: Failed to execute /opt/readarr/Readarr: No such file or directory
Mar 20 18:35:52 DietPi systemd[713]: readarr.service: Failed at step EXEC spawning /opt/readarr/Readarr: No such file or directory
Mar 20 18:35:52 DietPi systemd[1]: readarr.service: Main process exited, code=exited, status=203/EXEC
Mar 20 18:35:52 DietPi systemd[1]: readarr.service: Failed with result 'exit-code'.
Mar 20 18:35:52 DietPi systemd[1]: readarr.service: Scheduled restart job, restart counter is at 1.
Mar 20 18:35:52 DietPi systemd[1]: Stopped Readarr (DietPi).
Mar 20 18:35:52 DietPi systemd[1]: Started Readarr (DietPi).
Mar 20 18:35:53 DietPi systemd[724]: readarr.service: Failed to execute /opt/readarr/Readarr: No such file or directory
Mar 20 18:35:53 DietPi systemd[724]: readarr.service: Failed at step EXEC spawning /opt/readarr/Readarr: No such file or directory
Mar 20 18:35:53 DietPi systemd[1]: readarr.service: Main process exited, code=exited, status=203/EXEC
Mar 20 18:35:53 DietPi systemd[1]: readarr.service: Failed with result 'exit-code'.
Mar 20 18:35:53 DietPi systemd[1]: readarr.service: Scheduled restart job, restart counter is at 2.
Mar 20 18:35:53 DietPi systemd[1]: Stopped Readarr (DietPi).
Mar 20 18:35:53 DietPi systemd[1]: Started Readarr (DietPi).
Mar 20 18:35:53 DietPi systemd[730]: readarr.service: Failed to execute /opt/readarr/Readarr: No such file or directory
Mar 20 18:35:53 DietPi systemd[730]: readarr.service: Failed at step EXEC spawning /opt/readarr/Readarr: No such file or directory
Mar 20 18:35:53 DietPi systemd[1]: readarr.service: Main process exited, code=exited, status=203/EXEC
Mar 20 18:35:53 DietPi systemd[1]: readarr.service: Failed with result 'exit-code'.
Mar 20 18:35:53 DietPi systemd[1]: readarr.service: Scheduled restart job, restart counter is at 3.
Mar 20 18:35:53 DietPi systemd[1]: Stopped Readarr (DietPi).
Mar 20 18:35:53 DietPi systemd[1]: Started Readarr (DietPi).
Mar 20 18:35:53 DietPi systemd[731]: readarr.service: Failed to execute /opt/readarr/Readarr: No such file or directory
Mar 20 18:35:53 DietPi systemd[731]: readarr.service: Failed at step EXEC spawning /opt/readarr/Readarr: No such file or directory
Mar 20 18:35:53 DietPi systemd[1]: readarr.service: Main process exited, code=exited, status=203/EXEC
Mar 20 18:35:53 DietPi systemd[1]: readarr.service: Failed with result 'exit-code'.
Mar 20 18:35:53 DietPi systemd[1]: readarr.service: Scheduled restart job, restart counter is at 4.
Mar 20 18:35:53 DietPi systemd[1]: Stopped Readarr (DietPi).
Mar 20 18:35:53 DietPi systemd[1]: Started Readarr (DietPi).
Mar 20 18:35:54 DietPi systemd[733]: readarr.service: Failed to execute /opt/readarr/Readarr: No such file or directory
Mar 20 18:35:54 DietPi systemd[733]: readarr.service: Failed at step EXEC spawning /opt/readarr/Readarr: No such file or directory
Mar 20 18:35:54 DietPi systemd[1]: readarr.service: Main process exited, code=exited, status=203/EXEC
Mar 20 18:35:54 DietPi systemd[1]: readarr.service: Failed with result 'exit-code'.
Mar 20 18:35:54 DietPi systemd[1]: readarr.service: Scheduled restart job, restart counter is at 5.
Mar 20 18:35:54 DietPi systemd[1]: Stopped Readarr (DietPi).
Mar 20 18:35:54 DietPi systemd[1]: readarr.service: Start request repeated too quickly.
Mar 20 18:35:54 DietPi systemd[1]: readarr.service: Failed with result 'exit-code'.
Mar 20 18:35:54 DietPi systemd[1]: Failed to start Readarr (DietPi).

Joulinar, 2023-03-20:
can you share following as well

```
systemctl cat readarr.service
```

Gabba35, 2023-03-20:
# /etc/systemd/system/readarr.service
[Unit]
Description=Readarr (DietPi)
Wants=network-online.target
After=network-online.target

[Service]
SyslogIdentifier=Readarr
User=readarr
UMask=002
LogsDirectory=readarr
ExecStart=/opt/readarr/Readarr -nobrowser -data=/mnt/dietpi_userdata/readarr
TimeoutStopSec=20
KillMode=process
Restart=on-failure

# Hardening
ProtectSystem=strict
ProtectHome=true
PrivateDevices=true
ProtectKernelTunables=true
ProtectControlGroups=true
ReadWritePaths=-/opt/readarr -/mnt -/media -/var/log/readarr -/tmp

[Install]
WantedBy=multi-user.target

Joulinar, 2023-03-20:
ok let's check if following folder are existing

```
ls -la /opt/readarr/
ls -la /var/log/readarr
```

Gabba35, 2023-03-20:
root@DietPi:~# ls -la /opt/readarr/
ls -la /var/log/readarr
total 130320
drwxr-xr-x 7 readarr dietpi    32768 Feb 26 06:17 .
drwxr-xr-x 4 root    root       4096 Mar 20 09:20 ..
-rw-r--r-- 1 readarr dietpi  3318504 Feb 26 06:17 BouncyCastle.Crypto.dll
-rw-r--r-- 1 readarr dietpi   199424 Feb 26 06:17 createdump
-rw-r--r-- 1 readarr dietpi   202752 Feb 26 06:17 Dapper.dll
drwxr-xr-x 2 readarr dietpi     4096 Feb 26 06:17 de
drwxr-xr-x 2 readarr dietpi     4096 Feb 26 06:17 de-DE
-rw-r--r-- 1 readarr dietpi   773632 Feb 26 06:17 DryIoc.dll
-rw-r--r-- 1 readarr dietpi    15872 Feb 26 06:17 DryIoc.Microsoft.DependencyInjection.dll
-rw-r--r-- 1 readarr dietpi   167424 Feb 26 06:17 Dynamitey.dll
-rw-r--r-- 1 readarr dietpi    18432 Feb 26 06:17 Equ.dll
-rw-r--r-- 1 readarr dietpi    96768 Feb 26 06:17 FluentMigrator.Abstractions.dll
-rw-r--r-- 1 readarr dietpi    57856 Feb 26 06:17 FluentMigrator.dll
-rw-r--r-- 1 readarr dietpi     9728 Feb 26 06:17 FluentMigrator.Extensions.Oracle.dll
-rw-r--r-- 1 readarr dietpi    20480 Feb 26 06:17 FluentMigrator.Extensions.Postgres.dll
-rw-r--r-- 1 readarr dietpi     9216 Feb 26 06:17 FluentMigrator.Extensions.SqlAnywhere.dll
-rw-r--r-- 1 readarr dietpi    14848 Feb 26 06:17 FluentMigrator.Extensions.SqlServer.dll
-rw-r--r-- 1 readarr dietpi   126464 Feb 26 06:17 FluentMigrator.Runner.Core.dll
-rw-r--r-- 1 readarr dietpi    33792 Feb 26 06:17 FluentMigrator.Runner.Db2.dll
-rw-r--r-- 1 readarr dietpi   105984 Feb 26 06:17 FluentMigrator.Runner.dll
-rw-r--r-- 1 readarr dietpi    75776 Feb 26 06:17 FluentMigrator.Runner.Firebird.dll
-rw-r--r-- 1 readarr dietpi    26112 Feb 26 06:17 FluentMigrator.Runner.Hana.dll
-rw-r--r-- 1 readarr dietpi    26112 Feb 26 06:17 FluentMigrator.Runner.MySql.dll
-rw-r--r-- 1 readarr dietpi    43008 Feb 26 06:17 FluentMigrator.Runner.Oracle.dll
-rw-r--r-- 1 readarr dietpi    46592 Feb 26 06:17 FluentMigrator.Runner.Postgres.dll
-rw-r--r-- 1 readarr dietpi    27648 Feb 26 06:17 FluentMigrator.Runner.Redshift.dll
-rw-r--r-- 1 readarr dietpi    37376 Feb 26 06:17 FluentMigrator.Runner.SqlAnywhere.dll
-rw-r--r-- 1 readarr dietpi    23040 Feb 26 06:17 FluentMigrator.Runner.SQLite.dll
-rw-r--r-- 1 readarr dietpi    20480 Feb 26 06:17 FluentMigrator.Runner.SqlServerCe.dll
-rw-r--r-- 1 readarr dietpi    55808 Feb 26 06:17 FluentMigrator.Runner.SqlServer.dll
-rw-r--r-- 1 readarr dietpi   339456 Feb 26 06:17 FluentValidation.dll
-rw-r--r-- 1 readarr dietpi   189440 Feb 26 06:17 Ical.Net.dll
-rw-r--r-- 1 readarr dietpi   204800 Feb 26 06:17 ICSharpCode.SharpZipLib.dll
-rw-r--r-- 1 readarr dietpi    71168 Feb 26 06:17 ImpromptuInterface.dll
-rw-r--r-- 1 readarr dietpi    24064 Feb 26 06:17 LazyCache.dll
-rw-r--r-- 1 readarr dietpi  2433384 Feb 26 06:17 libclrjit.so
-rw-r--r-- 1 readarr dietpi  6929448 Feb 26 06:17 libcoreclr.so
-rw-r--r-- 1 readarr dietpi   858808 Feb 26 06:17 libcoreclrtraceptprovider.so
-rw-r--r-- 1 readarr dietpi   791616 Feb 26 06:17 libdbgshim.so
-rw-r--r-- 1 readarr dietpi   461432 Feb 26 06:17 libhostfxr.so
-rw-r--r-- 1 readarr dietpi   461448 Feb 26 06:17 libhostpolicy.so
-rw-r--r-- 1 readarr dietpi   312464 Feb 26 06:17 libMonoPosixHelper.so
-rw-r--r-- 1 readarr dietpi  2589248 Feb 26 06:17 libmscordaccore.so
-rw-r--r-- 1 readarr dietpi  1774960 Feb 26 06:17 libmscordbi.so
-rw-r--r-- 1 readarr dietpi   199280 Feb 26 06:17 libSystem.Globalization.Native.so
-rw-r--r-- 1 readarr dietpi   854464 Feb 26 06:17 libSystem.IO.Compression.Native.so
-rw-r--r-- 1 readarr dietpi   198944 Feb 26 06:17 libSystem.Native.so
-rw-r--r-- 1 readarr dietpi   199080 Feb 26 06:17 libSystem.Net.Security.Native.so
-rw-r--r-- 1 readarr dietpi   198960 Feb 26 06:17 libSystem.Security.Cryptography.Native.OpenSsl.so
-rw-r--r-- 1 readarr dietpi    35141 Feb 26 06:17 LICENSE.md
drwxr-xr-x 3 readarr dietpi     4096 Feb 26 06:17 Localization
-rw-r--r-- 1 readarr dietpi   869888 Feb 26 06:17 MailKit.dll
-rw-r--r-- 1 readarr dietpi   113280 Feb 26 06:17 Microsoft.AspNetCore.Antiforgery.dll
-rw-r--r-- 1 readarr dietpi    58520 Feb 26 06:17 Microsoft.AspNetCore.Authentication.Abstractions.dll
-rw-r--r-- 1 readarr dietpi   117888 Feb 26 06:17 Microsoft.AspNetCore.Authentication.Cookies.dll
-rw-r--r-- 1 readarr dietpi    88704 Feb 26 06:17 Microsoft.AspNetCore.Authentication.Core.dll
-rw-r--r-- 1 readarr dietpi   161920 Feb 26 06:17 Microsoft.AspNetCore.Authentication.dll
-rw-r--r-- 1 readarr dietpi   106624 Feb 26 06:17 Microsoft.AspNetCore.Authentication.OAuth.dll
-rw-r--r-- 1 readarr dietpi    98456 Feb 26 06:17 Microsoft.AspNetCore.Authorization.dll
-rw-r--r-- 1 readarr dietpi    63128 Feb 26 06:17 Microsoft.AspNetCore.Authorization.Policy.dll
-rw-r--r-- 1 readarr dietpi    51840 Feb 26 06:17 Microsoft.AspNetCore.Components.Authorization.dll
-rw-r--r-- 1 readarr dietpi   631424 Feb 26 06:17 Microsoft.AspNetCore.Components.dll
-rw-r--r-- 1 readarr dietpi    68760 Feb 26 06:17 Microsoft.AspNetCore.Components.Forms.dll
-rw-r--r-- 1 readarr dietpi   805504 Feb 26 06:17 Microsoft.AspNetCore.Components.Server.dll
-rw-r--r-- 1 readarr dietpi   331904 Feb 26 06:17 Microsoft.AspNetCore.Components.Web.dll
-rw-r--r-- 1 readarr dietpi    71320 Feb 26 06:17 Microsoft.AspNetCore.Connections.Abstractions.dll
-rw-r--r-- 1 readarr dietpi    53376 Feb 26 06:17 Microsoft.AspNetCore.CookiePolicy.dll
[... 326 lines cut ...]
-rw-r--r-- 1 readarr dietpi    16024 Feb 26 06:17 System.Threading.Thread.dll
-rw-r--r-- 1 readarr dietpi    16000 Feb 26 06:17 System.Threading.ThreadPool.dll
[... cut ...]
-rw-r--r-- 1 readarr dietpi    18056 Feb 26 06:17 System.Xml.XmlSerializer.dll
-rw-r--r-- 1 readarr dietpi    16016 Feb 26 06:17 System.Xml.XPath.dll
-rw-r--r-- 1 readarr dietpi    18560 Feb 26 06:17 System.Xml.XPath.XDocument.dll
-rw-r--r-- 1 readarr dietpi   492032 Feb 26 06:17 TagLibSharp.dll
drwxr-xr-x 3 readarr dietpi     4096 Feb 26 06:17 UI
-rw-r--r-- 1 readarr dietpi    17024 Feb 26 06:17 WindowsBase.dll
total 0
drwxr-xr-x  2 readarr dietpi  40 Mar 20 09:20 .
drwxr-xr-t 10 root    root   380 Mar 20 09:20 ..

Joulinar, 2023-03-20:
did you already tried to reinstall? If not, you can do following:

```
systemctl stop readarr.service
rm -Rf /opt/readarr
dietpi-software reinstall 203
```

Gabba35, 2023-03-20:
Yeah, tried that before posting but no luck on it.

The repos:

Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Hit:3 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:4 https://apt.sonarr.tv/debian buster InRelease

Readarr.develop.0.1.4.1596.linux-core-arm64.tar.gz

No luck, still 'failed' as a service.

Joulinar, 2023-03-20:
Repositories doesn't matter as Readarr is downlaod directly from GitHub. One last attend would be to remove Readarr, reboot and do a fresh install.

Gabba35, 2023-03-20:
I now have no video signal after dieting shows one line of text on reboot (audio still works..). Something unhappy. It's a lot do a fresh install with everything configured at the moment, might have to give up on Readarr. Thanks for all of your help, it's much appreciated.

Joulinar, 2023-03-20:
can you try to access the system using SSH?

On my test RPi4B Readarr is working without issues. I guess it's something on your dedicated system.

MichaIng, 2023-03-20:
You have 64-bit kernel mode enabled on a Raspbian (ARMv6) userland. Please remove the `arm_64bit` setting from your `/boot/config.txt`, reboot and reinstall Readarr.

There is zero benefit to enable the 64-bit kernel mode if the whole OS is 32-bit. It just causes confusion among any installer which selects binaries based on `uname -m` output, expecting this to match the installed OS.

Gabba35, 2023-03-21:
Had a look, no arm_64bit setting in my config.txt. I'm not sure how I'd changed it. I'm thinking I'll go a new install with a new SD card since it's been a while.

Joulinar, 2023-03-21:
can you share your configuration file pls:

```
cat /boot/config.txt
```

MichaIng, 2023-03-21:
Strange, please also show again:
```sh
uname -m
dpkg --print-architecture
cat /etc/os-release
```

Gabba35, 2023-03-22:
I've done a fresh install. Readarr works. So the current errors are a different problem.

I don't get HDMI output if I have the GPU split at 512mb(384mb works) - it cuts out during the console boot text otherwise. This was happening before fresh install too - was there anything in the last update? It didn't happen before it. It boots into 4k for some reason, I've got the below in dietpi-config display resolution.

Hardware: RPi 4 Model B (aarch64)                                           
Current resolution: vc4-fkms-v3d 

Fresh install config.txt has arm_64bit 1 in it. Is it Kodi or another software causing this? Kodi still runs if I make it '#arm_64bit 1'. 
Running 'kodi --standalone' or  'startkodi' via console. It did give me an error about 64 vs arm7 or something though with '#arm_64bit 1'.

I get these errors when running Kodi.

error: XDG_RUNTIME_DIR not set in the environment.
context.c:56: warning: mpd_setminalloc: ignoring request to set MPD_MINALLOC a second time

root@DietPi:~# uname -m
dpkg --print-architecture
cat /etc/os-release
aarch64
arm64
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Fresh install:
# Docs: https://www.raspberrypi.com/documentation/computers/config_txt.html
# Overlays: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README

#-------Display---------
# Max allocated framebuffers: Set to "0" in headless mode to reduce memory usage
# - Defaults to "2" on RPi4 and "1" on earlier RPi models
#max_framebuffers=0

# If you get no picture, set the following to "1" to apply most compatible HDMI settings.
#hdmi_safe=1

# Uncomment to adjust the HDMI signal strength if you have interferences, blanking, or no display.
# - Ranges from "0" to "11", use values above "7" only if required, e.g. with very long HDMI cable.
# - Default on first RPi1 A/B is "2", else "5", on RPi4 this setting is ignored.
#config_hdmi_boost=11

# Uncomment if HDMI display is not detected and composite is being outputted.
hdmi_force_hotplug=1

# Uncomment to disable HDMI even if plugged, e.g. to force composite output.
#hdmi_ignore_hotplug=1

# Uncomment to force a console size. By default it will be display's size minus overscan.
#framebuffer_width=1920
#framebuffer_height=1080

# Uncomment to enable SDTV/composite output on RPi4. This has no effect on previous RPi models.
#enable_tvout=0
# SDTV mode
#sdtv_mode=0

# Uncomment to force a specific HDMI mode (this will force VGA).
#hdmi_group=1
#hdmi_mode=1

# Uncomment to force an HDMI mode rather than DVI. This enables HDMI audio in DMT modes.
#hdmi_drive=2

# Set "hdmi_blanking=1" to allow the display going into standby after 10 minutes without input.
[... 132 lines cut ...]
#-------I2C-------------
dtparam=i2c_arm=off
#dtparam=i2c_arm_baudrate=100000

#-------SPI-------------
dtparam=spi=off

#-------Serial/UART-----
# NB: Enabled for 1st run only. If you want to keep this setting, please set CONFIG_SERIAL_CONSOLE_ENABLE=1 in dietpi.txt before 1st boot.
#core_freq=500
enable_uart=1

#-------Overclock-------
temp_limit=82
initial_turbo=20

#over_voltage=0
#arm_freq=1500
#core_freq=500

#arm_freq_min=300
#core_freq_min=250
#sdram_freq_min=400
#max_framebuffers=2
dtoverlay=vc4-fkms-v3d

Gabba35, 2023-03-22:
I've done a fresh install. Readarr works. So the current errors are a different problem.

I don't get HDMI output if I have the GPU split at 512mb(384mb works) - it cuts out during the console boot text otherwise. This was happening before fresh install too - was there anything in the last update? It didn't happen before it. It boots into 4k for some reason, I've got the below in dietpi-config display resolution.

Hardware: RPi 4 Model B (aarch64)                                           
Current resolution: vc4-fkms-v3d 

Fresh install config.txt has arm_64bit 1 in it. Is it Kodi or another software causing this? Kodi still runs if I make it '#arm_64bit 1'. 
Running 'kodi --standalone' or  'startkodi' via console. It did give me an error about 64 vs arm7 or something though with '#arm_64bit 1'.

I get these errors when running Kodi.

error: XDG_RUNTIME_DIR not set in the environment.
context.c:56: warning: mpd_setminalloc: ignoring request to set MPD_MINALLOC a second time

root@DietPi:~# uname -m
dpkg --print-architecture
cat /etc/os-release
aarch64
arm64
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Fresh install:
```
# Docs: https://www.raspberrypi.com/documentation/computers/config_txt.html
# Overlays: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README

#-------Display---------
# Max allocated framebuffers: Set to "0" in headless mode to reduce memory usage
# - Defaults to "2" on RPi4 and "1" on earlier RPi models
#max_framebuffers=0

# If you get no picture, set the following to "1" to apply most compatible HDMI settings.
#hdmi_safe=1

# Uncomment to adjust the HDMI signal strength if you have interferences, blanking, or no display.
# - Ranges from "0" to "11", use values above "7" only if required, e.g. with very long HDMI cable.
# - Default on first RPi1 A/B is "2", else "5", on RPi4 this setting is ignored.
#config_hdmi_boost=11

# Uncomment if HDMI display is not detected and composite is being outputted.
hdmi_force_hotplug=1

# Uncomment to disable HDMI even if plugged, e.g. to force composite output.
#hdmi_ignore_hotplug=1

# Uncomment to force a console size. By default it will be display's size minus overscan.
#framebuffer_width=1920
#framebuffer_height=1080

# Uncomment to enable SDTV/composite output on RPi4. This has no effect on previous RPi models.
#enable_tvout=0
# SDTV mode
#sdtv_mode=0

# Uncomment to force a specific HDMI mode (this will force VGA).
#hdmi_group=1
#hdmi_mode=1

# Uncomment to force an HDMI mode rather than DVI. This enables HDMI audio in DMT modes.
#hdmi_drive=2

[... 136 lines cut ...]
dtparam=i2c_arm=off
#dtparam=i2c_arm_baudrate=100000

#-------SPI-------------
dtparam=spi=off

#-------Serial/UART-----
# NB: Enabled for 1st run only. If you want to keep this setting, please set CONFIG_SERIAL_CONSOLE_ENABLE=1 in dietpi.txt before 1st boot.
#core_freq=500
enable_uart=1

#-------Overclock-------
temp_limit=82
initial_turbo=20

#over_voltage=0
#arm_freq=1500
#core_freq=500

#arm_freq_min=300
#core_freq_min=250
#sdram_freq_min=400
#max_framebuffers=2
dtoverlay=vc4-fkms-v3d
```

Joulinar, 2023-03-22:
> Fresh install config.txt has arm_64bit 1 in it.

This if fully correct because you used the ARMv8 64bit image for the new installation. Therefore the flag has been set correctly 😉

MichaIng, 2023-03-22:
> I don't get HDMI output if I have the GPU split at 512mb(384mb works)

What for do you assign so much GPU memory? Note that modern RPi kernel, especially on RPi 4, does make use of the GPU memory very rarely. Most things have been moved to CMA and hence common system memory. You will not need more than the default in very most cases, up to 128 MiB if you have specific applications which make dedicated use of the GPU memory for decoding or encoding. Everything above that, is just wasted an weakens performance as less system RAM is available.

Uff I just see with Kodi we still suggest 256 MiB, but that should be actually lowered to 128 MiB or no raise at all. At least someone should verify that there are cases where raising the GPU memory above default has any positive effect on Kodi video playback performance.

MichaIng, 2023-04-03:
Marking this as closed as the original issue has been solved. Feel free to reopen if required.

closed 2023-04-03 by MichaIng