MichaIng/DietPi #6265: DietPi update 8.8.1 -> 8.15.2

b2beda, 2023-03-21:
#### Details:

### Creating a bug report/issue
#### Details:
- Date           | Wed Mar 22 00:13:23 CET 2023
- DietPi version | v8.8.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Radxa
- Hardware       | ROCK Pi S (aarch64) (ID=73)
- Kernel version | `Linux test 4.4.143-69-rockchip-g8ccef796d27d #1 SMP PREEMPT Tue Jun 14 12:32:20 UTC 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y update`
- Exit code      | 255
- Software title | DietPi-Update
#### Steps to reproduce:

1. ... Apply update to: v8.15.2 
2. ...
#### Expected behaviour:

update 8.8.1 -> 8.15.2
- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Get:5 https://apt.radxa.com/buster-stable buster InRelease [2363 B]
Ign:6 https://download.webmin.com/download/repository sarge InRelease
Hit:7 https://download.webmin.com/download/repository sarge Release
Err:5 https://apt.radxa.com/buster-stable buster InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9B98116C9AA302C7
Fetched 2363 B in 4s (653 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://apt.radxa.com/buster-stable buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9B98116C9AA302C7
W: Failed to fetch https://apt.radxa.com/buster-stable/dists/buster/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9B98116C9AA302C7
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- root@test:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- bullseye

- Kernel version | `uname -a`
- Linux test 4.4.143-69-rockchip-g8ccef796d27d #1 SMP PREEMPT Tue Jun 14 12:32:20 UTC 2022 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- ROCK Pi S (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- 5V 2.4A
- SD card used | (EG: SanDisk ultra)
- SanDisk Ultra 32 GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2023-03-21:
looks like one of your apt package source files needs to be switched to Bullseye as well, as it is still pointing to Buster.

> `Err:5 https://apt.radxa.com/buster-stable buster InRelease`

MichaIng, 2023-03-21:
It's one of the old images which use the Radxa repository, and obviously their key expired. As you are on Bullseye, you can also update the repo to Bullseye. They didn't have a Bullseye suite for a long time but it's there now:
```sh
curl -sSfL 'https://apt.radxa.com//bullseye-stable/public.key' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-radxa.gpg --yes
echo 'deb https://apt.radxa.com/bullseye-stable/ bullseye main' > /etc/apt/sources.list.d/dietpi-radxa.list
apt update
```

MichaIng, 2023-04-03:
Marking this as closed as the above commands should solve the issue. Feel free to reopen if required.