MichaIng/DietPi #6266: wrong php (8.1) after update buster to bullseye + 503 Service unavailable at pihole web gui

strupp666, 2023-03-22:
### Creating a bug report/issue
wrong php (8.1) after update buster to bullseye + 503 Service unavailable at pihole web gui
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=15
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye
- Kernel version | `uname -a`
Linux DietPi 4.14.294+ #1 SMP PREEMPT Sat Sep 24 10:24:15 UTC 2022 armv7l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` 
Odroid XU3/XU4/MC1/HC1/HC2 (armv7l) - HC2
- Power supply used | (EG: 5V 1A RAVpower)   i. O
- SD card used | (EG: SanDisk ultra)   i.O

#### Additional Information (if applicable)
- Software title | pihole, lighttpd, php
- Was the software title installed freshly or updated/migrated?  updated
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Update from buster to bullseye
2. ...

#### Expected behaviour
Dashboard of pihole must be reachable
- ...

#### Actual behaviour
503 service unavailable
- ...

#### Extra details
- 1 -

root@DietPi:~# pihole -up
  [✓] Update local cache of available packages
  [i] Existing PHP installation detected : PHP version 8.1.2
  [✓] Checking for git
  [✓] Checking for iproute2
  [✓] Checking for dialog
  [✓] Checking for ca-certificates

  [i] Checking for updates...
  [i] Pi-hole Core:     update available
  [i] Web Interface:    update available
  [i] FTL:              update available

  [i] Pi-hole core files out of date, updating local repo.
  [✓] Check for existing repository in /etc/.pihole
  [i] Update repo in /etc/.pihole...HEAD ist jetzt bei d86b325d Only search for "OVERWRITTEN BY PI-HOLE" when checking inside `lighttpd.conf` (#5167)
  [✓] Update repo in /etc/.pihole

  [i] If you had made any changes in '/etc/.pihole/', they have been stashed using 'git stash'

  [i] Pi-hole Web Admin files out of date, updating local repo.
  [✓] Check for existing repository in /var/www/html/admin
  [i] Update repo in /var/www/html/admin...HEAD ist jetzt bei b29a423b Pi-hole web v5.18.4 (#2523)
  [✓] Update repo in /var/www/html/admin

  [i] If you had made any changes in '/var/www/html/admin/', they have been stashed using 'git stash'

  [i] FTL out of date, it will be updated by the installer.

[... 173 lines cut ...]
  [✓] Update complete! 

  Current Pi-hole version is v5.15.5
  Current AdminLTE version is v5.18.4
  Current FTL version is v5.21
- ...
- 2 -

root@DietPi:~# journalctl -u lighttpd
tail -10 /var/log/lighttpd/error.log
Journal file /var/log/journal/e06a67a4ff8e40c2a988f11152e0f6d3/system@da4c718c35cd4bc3abea1192c6070263-00000000000042d6-0005f76accaa8a1b.journal is truncated, ignoring file.
-- Journal begins at Wed 2023-03-22 07:31:09 CET, ends at Wed 2023-03-22 08:58:28 CET. --
-- No entries --
2023-03-22 08:54:19: gw_backend.c.238) establishing connection failed: socket: unix:/run/php/php7.4-fpm.sock: No such file or directory
2023-03-22 08:54:19: gw_backend.c.970) all handlers for /admin/index.php?login on .php are down.
2023-03-22 08:54:21: gw_backend.c.315) gw-server re-enabled: unix:/run/php/php7.4-fpm.sock  0 /run/php/php7.4-fpm.sock
root@DietPi:~# getent hosts doubleclick.com
192.168.168.22  doubleclick.com
root@DietPi:~# ls -l "$(which php)"
php -v
lrwxrwxrwx 1 root root 21 27. Feb 2020  /usr/bin/php -> /etc/alternatives/php
PHP 8.1.2 (cli) (built: Jan 27 2022 12:19:23) (NTS)
Copyright (c) The PHP Group
Zend Engine v4.1.2, Copyright (c) Zend Technologies
    with Zend OPcache v8.1.2, Copyright (c), by Zend Technologies

Joulinar, 2023-03-22:
Did you upgrade from Debian Stretch before? Because PHP 8 is not available on default Debian Bullseye apt package repository and would need to be installed from 3rd party location. Do you know if Ondrej's PHP repository is setup? Can be checked as follow

```
apt update
```

Joulinar, 2023-03-22:
Did you follow our update guide to switch from Stretch to Buster? https://dietpi.com/docs/usage/#how-to-upgrade-to-buster
If yes, the repository has been removed. But I guess the PHP version has not been set back to Debian repository. 

Can you check which packages are installed actually

```
dpkg -l *php*
```

strupp666, 2023-03-22:
Gewünscht=Unbekannt/Installieren/R=Entfernen/P=Vollständig Löschen/Halten
| Status=Nicht/Installiert/Config/U=Entpackt/halb konFiguriert/
         Halb installiert/Trigger erWartet/Trigger anhängig
|/ Fehler?=(kein)/R=Neuinstallation notwendig (Status, Fehler: GROSS=schlecht)
||/ Name                  Version                                           Arch
itektur  Beschreibung
+++-=====================-=================================================-====
========-===============================================================
un  libapache2-mod-php    <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
ii  libapache2-mod-php7.4 7.4.33-1+deb11u3                                  armh
f        server-side, HTML-embedded scripting language (Apache 2 module)
un  php                   <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
ii  php-apcu              5.1.21+4.0.11-7+0~20220117.31+debian9~1.gbp9a887a armh
f        APC User Cache for PHP
un  php-apcu-bc           <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-bz2               <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-calendar          <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-cgi               <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-cli               <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
ii  php-common            2:92+0~20220117.43+debian9~1.gbpe0d14e            all
         Common files for PHP packages
un  php-crypt-gpg         <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-ctype             <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-curl              <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-dom               <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-exif              <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-ffi               <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-fileinfo          <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-fpm               <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-ftp               <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-gd                <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-iconv             <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-intl              <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-json              <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-mbstring          <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-mysqli            <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-mysqlnd           <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-opcache           <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-pdo               <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-pdo-mysql         <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-pear              <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php-phar              <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
[... 153 lines cut ...]
ne>      (keine Beschreibung vorhanden)
un  php8.1-phar           <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
ii  php8.1-phpdbg         8.1.2-1+0~20220127.12+debian9~1.gbpe46588         armh
f        server-side, HTML-embedded scripting language (PHPDBG binary)
un  php8.1-posix          <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
ii  php8.1-readline       8.1.2-1+0~20220127.12+debian9~1.gbpe46588         armh
f        readline module for PHP
un  php8.1-shmop          <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php8.1-sockets        <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php8.1-sysvmsg        <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php8.1-sysvsem        <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php8.1-sysvshm        <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  php8.1-tokenizer      <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  phpapi-20190902       <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)
un  phpapi-20210902       <keine>                                           <kei
ne>      (keine Beschreibung vorhanden)

Joulinar, 2023-03-22:
@MichaIng could you give a helping hand pls before don't something wrong. I guess PHP needs to be purged right?

strupp666, 2023-03-22:
root@DietPi:~# apt autopurge $(dpkg --get-selections 'php8.2-*' | mawk '{print $1}')
dpkg: Kein Paket gefunden, das auf php8.2-* passt
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

strupp666, 2023-03-22:
Maybe no minus?

MichaIng, 2023-03-22:
PHP 8.1, of course:
```sh
apt autopurge $(dpkg --get-selections 'php8.1-*' | mawk '{print $1}')
```

strupp666, 2023-03-22:
After reboot, 503 fail still appears.

strupp666, 2023-03-22:
root@DietPi:~# dpkg -l *php*
Gewünscht=Unbekannt/Installieren/R=Entfernen/P=Vollständig Löschen/Halten
| Status=Nicht/Installiert/Config/U=Entpackt/halb konFiguriert/
         Halb installiert/Trigger erWartet/Trigger anhängig
|/ Fehler?=(kein)/R=Neuinstallation notwendig (Status, Fehler: GROSS=schlecht)
||/ Name                  Version                                Architektur  Be
schreibung
+++-=====================-======================================-============-==
=============================================================
un  libapache2-mod-php    <keine>                                <keine>      (k
eine Beschreibung vorhanden)
ii  libapache2-mod-php7.4 7.4.33-1+deb11u3                       armhf        se
rver-side, HTML-embedded scripting language (Apache 2 module)
un  php                   <keine>                                <keine>      (k
eine Beschreibung vorhanden)
un  php-bz2               <keine>                                <keine>      (k
eine Beschreibung vorhanden)
un  php-calendar          <keine>                                <keine>      (k
eine Beschreibung vorhanden)
un  php-cgi               <keine>                                <keine>      (k
eine Beschreibung vorhanden)
un  php-cli               <keine>                                <keine>      (k
eine Beschreibung vorhanden)
ii  php-common            2:92+0~20220117.43+debian9~1.gbpe0d14e all          Co
mmon files for PHP packages
un  php-crypt-gpg         <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-ctype             <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-curl              <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-dom               <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-exif              <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-ffi               <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-fileinfo          <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-fpm               <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-ftp               <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-gd                <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-iconv             <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-intl              <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-json              <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-mbstring          <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-mysqli            <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-mysqlnd           <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-opcache           <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-pdo               <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-pdo-mysql         <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-pear              <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-phar              <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-posix             <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php-readline          <keine>                                <keine>      (keine Beschre
[... 94 lines cut ...]
ibung vorhanden)
un  php7.4-simplexml      <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php7.4-sockets        <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php7.4-sysvmsg        <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php7.4-sysvsem        <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php7.4-sysvshm        <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php7.4-tokenizer      <keine>                                <keine>      (keine Beschre
ibung vorhanden)
ii  php7.4-xml            7.4.33-1+deb11u3                       armhf        DOM, SimpleXML
, XML, and XSL module for PHP
un  php7.4-xmlreader      <keine>                                <keine>      (keine Beschre
ibung vorhanden)
un  php7.4-xmlwriter      <keine>                                <keine>      (keine Beschre
ibung vorhanden)
ii  php7.4-xsl            7.4.33-1+deb11u3                       all          XSL module for
 PHP (dummy)
ii  php7.4-zip            7.4.33-1+deb11u3                       armhf        Zip module for
 PHP
un  phpapi-20190902       <keine>                                <keine>      (keine Beschre
ibung vorhanden)

strupp666, 2023-03-22:
Now it is running.
TY very much.
Maybe I had some bees in my fingers. I follow all steps...

MichaIng, 2023-03-22:
Okay great. I'll mark this as closed. Feel free to reopen if required.

closed 2023-03-22 by MichaIng