MichaIng/DietPi #6267: unable to update to v8.15.2

dalius415, 2023-03-22:
#### Details:
- Date           | Wed Mar 22 12:11:33 EET 2023
- DietPi version | v8.14.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux bonjour 5.15.84-v7+ #1613 SMP Thu Jan 5 11:59:48 GMT 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.nodesource.com/node_12.x bullseye InRelease
Get:2 https://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Get:4 https://archive.raspberrypi.org/debian bullseye/main armhf Packages [315 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages [13.2 MB]
Err:6 http://pkg.cloudflare.com bullseye InRelease
  Connection failed [IP: 104.18.0.118 80]
Fetched 13.6 MB in 1min 1s (223 kB/s)
Reading package lists...
E: Failed to fetch http://pkg.cloudflare.com/dists/bullseye/InRelease  Connection failed [IP: 104.18.0.118 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Joulinar, 2023-03-22:
You habe added a 3rd party repository from Cloudflare which seems not available. Are you able to open the location on a browser?

```
http://pkg.cloudflare.com/dists/bullseye/InRelease
```

dalius415, 2023-03-22:
ouch, silly me :)
curl not working...removed repo, all works fine. Thank you!

closed 2023-03-22 by Joulinar