MichaIng/DietPi #6268: NextCloud error, due to missing PHP > 8.x.x for composer

robyscar, 2023-03-24:
### Creating a bug report/issue
**While installing from Dietpi-software NextCloud starts and works, after updating it, and adding modules/apps, it fails.**

1. Removed  NextCloud (via Dietpi-software)
2. installed php 8 (via https://packages.sury.org/php)
3. reinstalled NextCloud (via Dietpi-software) >> during installation 
3.a) the script removed the user installed PHP8 modules (apache)
3.b) fails

#### Details:
- Date           | Fri Mar 24 14:16:07 +07 2023
- Bug report     | f33b7bec-62e5-40eb-af7f-f31540b8fba1
- DietPi version | v8.15.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux R1PeeHole 6.1.19-v7+ #1637 SMP Tue Mar 14 11:04:52 GMT 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ncc maintenance:mode --off`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.4.
```
 

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2023-03-24:
This is a limitation of NC software not supporting PHP 8.2 atm. We added a small hack for Debian Bookworm to allow NC bypassing the compatibility check. You could try to adjust manually. 

```
sed -i 's/>= 80200/>= 80300/' /var/www/nextcloud/lib/versioncheck.php
```

MichaIng, 2023-03-24:
Nextcloud 26 was just released btw, so PHP 8.2 support is there now.