MichaIng/DietPi #6270: Reacquainting myself with the Rock64 and want to play with containers...

madmedix, 2023-03-24:
It's been awhile for this little board and dietpi and I wanted to reacquaint myself with it and containers. Can't get past the installation through. I've used this board and many other dietpi apps in the recent past without much issue but this one...

### Creating a bug report/issue
 Reference code: 9d6fb307-0942-4f1a-9c94-3117b3589343
#### Details:
- Date           | Fri Mar 24 11:48:32 EDT 2023
- Bug report     | 9d6fb307-0942-4f1a-9c94-3117b3589343
- DietPi version | v8.15.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | ROCK64 (aarch64) (ID=43)
- Kernel version | `Linux DietPi 5.15.93-rockchip64 #23.02.2 SMP PREEMPT Fri Feb 17 23:48:36 UTC 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Fresh image on fresh microSD.
2. During initial boot; to complete install: dashboard installed
3. Finish initial boot and restart
4. transferred /swap and /user to a usb stick after formatting to EXT4 and mounting
5. install docker/docker compose using software mgr
6. all ok
7. proceed to install Portainer = error.

#### Expected behaviour:
Installation without error of docker, docker compose & portainer

#### Actual behaviour:
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

#### Extra details:

#### Additional logs:
```
Hit:1 https://download.docker.com/linux/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye InRelease
Hit:3 https://deb.debian.org/debian bullseye-updates InRelease
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Err:6 https://apt.armbian.com bullseye InRelease
  502  Bad Gateway [IP: 2602:fea7:e0d:1001::12 443]
Reading package lists...
E: Failed to fetch https://apt.armbian.com/dists/bullseye/InRelease  502  Bad Gateway [IP: 2602:fea7:e0d:1001::12 443]
E: Some index files failed to download. They have been ignored, or old ones used instead.

Joulinar, 2023-03-24:
Looks like an issue with external Armbian apt package repository.

> Err:6 https://apt.armbian.com bullseye InRelease
  502  Bad Gateway [IP: 2602:fea7:e0d:1001::12 443]
Reading package lists...

Would it be possible to switch of IPv6 for testing and run `apt update` afterwards?

madmedix, 2023-03-24:
Did the trick - thank you.

closed 2023-03-24 by madmedix