MichaIng/DietPi #6277: Invidious (arm64) docker installation problem

bbsixzz, 2023-03-29:
### Creating a bug report/issue

Original report: https://github.com/iv-org/invidious/issues/3712#event-8877575822
[Bug] Raspberry Pi4 DietPi - Docker (linux/arm64) does not match the detected host platform (linux/arm/v8)

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=15
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye 1
- Kernel version | `uname -a`
Linux DietPi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
Original Pi4 PSU
- SD card used | (EG: SanDisk ultra)
SanDisk ultra

#### Additional Information (if applicable)
git clone https://github.com/iv-org/invidious.git
cd invidious
docker compose up -d

[image]

`! invidious The requested image's platform (linux/arm64) does not match the detected host platform (linux/arm/v8) and no specific platform was requested`

invidious-db is up but invidious keeps restarting and log is empty.

I know this isn't a DietPi package but the same installation worked flawlessly on my x86 debian server.

Any help would be much appreciated, Invidious is great for watching YouTube without ads/tracking/javascript.

MichaIng, 2023-03-29:
Many thanks for reporting. It seems we found the first case of external installers/software which has issues with the now by default enabled 64-bit kernel on RPi 4 with 32-bit userland/OS.

Can you verify that it works fine when you add `arm_64bit=0` to `/boot/config.txt`, `reboot` and reinstall that Docker image?

For reference, here the related release notes and discussion: https://forums.raspberrypi.com/viewtopic.php?p=2093749#p2093749

bbsixzz, 2023-03-29:
Alright, so I tried defining linux/arm/v7 as explicit platform for the Docker image in docker-compose.yml by adding `platform: linux/arm/v7` and also `platform: linux/arm`

This resulted in:

```
Error response from daemon: image with reference quay.io/invidious/invidious:latest-arm64 was found but does not match the specified platform: wanted linux/arm/v7, actual: linux/arm64

Error response from daemon: image with reference quay.io/invidious/invidious:latest-arm64 was found but does not match the specified platform: wanted linux/arm, actual: linux/arm64
```

After that I closed the containers and cleaned up with `docker system prune -a `

I edited `/boot/config.txt` and added `arm_64bit=0` followed by a reboot.

I then reinstalled (tried with and without `platform: linux/arm/v7`):

`! invidious The requested image's platform (linux/arm64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested`

Thanks again for your excellent support.

bbsixzz, 2023-03-29:
Seems also relevant: https://github.com/iv-org/invidious/issues/361#issuecomment-881047911

> Hi!
> 
> Thanks to the work of @nemunaire, we are finally officially supporting ARM(v8) Docker image!
> 
> It is available under this Docker tag: `quay.io/invidious/invidious:latest-arm64` In order to test it, you need to use ARM64/ARMv8 on your device. **Raspbian doesn't support ARMv8 by default, you may use Ubuntu ARMv8 instead!**
> 
> Please test it and give your feedback to us in this GitHub issue!
> 
> ## Some Q&A
> ### Why aren't we supporting ARMv7?
> Unfortunately, Crystal doesn't really support ARMv7 for musl (a C standard library), see https://crystal-lang.org/reference/platform_support.html, so there is no `crystal` package for ARMv7 on Alpine Linux.
> 
> ### Why is the Docker image not available under the Docker tag `latest`?
> Building a multi architecture Docker image requires to use the same Dockerfile, but Crystal doesn't officially offer an ARM64 docker image, so we have to rely on the [`crystal` package from Alpine Linux](https://pkgs.alpinelinux.org/package/edge/community/x86_64/crystal). The version of this package may change and result in the inability of building the Docker image because Invidious frequently fail to work on newer version of Crystal. That's why we prefer for the moment to rely on the official Crystal docker image in order to build our Docker image for x86/AMD64 and thus having a separate Dockerfile for each CPU architecture.

MichaIng, 2023-03-29:
> image with reference quay.io/invidious/invidious:latest-arm64 was found

You need to remove the image via `docker image rm <ID>` before you can pull a new one.

> The requested image's platform (linux/arm64)

So it is still looking for `linux/arm64`? But probably the old image is still there:
```sh
docker image ls
```

MichaIng, 2023-03-29:
Okay, if there is no 32-bit ARM image, then of course it cannot work anyway. However, currently the error message indicates that there is still the `linux/arm64` image installed, otherwise I'd expect something like "linux/arm/v7 image could not be found".

However, seems you'd need to flash our ARMv8 64-bit image to make this work.

bbsixzz, 2023-03-29:
`docker image ls` or `docker images` is same, images were gone every time after running `docker system prune -a`:

```
WARNING! This will remove:
  - all stopped containers
  - all networks not used by at least one container
  - all images without at least one container associated to them
  - all build cache
```

I've also removed images with `docker rmi <ID>` which is same as `docker image rm <ID>`.

Bottom line the images were gone before pulling, I made sure.

bbsixzz, 2023-03-29:
> But seems you'd need to flash our ARMv8 64-bit image to make this work.

Alright, is there a quick and painless way to do this while keeping everything intact?

Joulinar, 2023-03-31:
> I wonder whether Docker in general tries to pull the ARMv8 images in these cases. @Joulinar, if you find time, could you test this with e.g. Portainer or even just docker run hello-world?

Ok did some test on my RPi4B running ARMv7 image. Docker themselves seems to be able to deal with this, and is pulling correct image.

```
root@DietPi4:~# dpkg --print-architecture
armhf
root@DietPi4:~#
```

```
root@DietPi4:~# uname -a
Linux DietPi4 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
root@DietPi4:~#
```

```
root@DietPi4:~# uname -m
aarch64
root@DietPi4:~#
```

```
root@DietPi4:~# docker image inspect portainer/portainer-ce:latest | grep 'Architecture\|Variant'
        "Architecture": "arm",
        "Variant": "v7",
root@DietPi4:~#
```

MichaIng, 2023-03-31:
Okay thanks for testing. Then the only reason was that crystal does not provide 32-bin ARM images.

MichaIng, 2023-04-02:
I'll mark it as closed then. Feel free to reopen if required.

closed 2023-04-02 by MichaIng