MichaIng/DietPi #6281: Open Beta v8.16 | Please help testing and hardening our upcoming release

MichaIng, 2023-04-01:
RC version | v8.16.0
----------- | -------
v8.16.0 PR | #6258
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2023-04-08**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.16
____
### Beta v8.16.0
_(2023-04-01)_

#### New software
- microblog.pub | A self-hosted, single-user, ActivityPub powered microblog has been added to our software catalogue. Many thanks to @mtekman for implementing this software option: https://github.com/MichaIng/DietPi/pull/6246

#### Enhancements
- General | Reworked GPU driver installations for x86 systems. This provides better control of GPU drivers for DietPi, adding support for Vulkan and resolves issues with Steam Proton. Users will be prompted to select a driver package during DietPi-Update.: https://github.com/MichaIng/DietPi/issues/6262
- NanoPi R5S/R6S | The kernel will be upgraded during the DietPi update to disable a writable clk DebugFS flag it was previously shipped with. Many thanks to @thinkpanzer for beinging this to our attention: https://github.com/MichaIng/DietPi/issues/6136
- Orange Pi 5/ROCK 5B | The hardware random generator daemon rngd is used now in favour of the software random generator haveged. It will be replaced on DietPi update.
- First run setup | When connected via local console and the keyboard layout has not been changed in dietpi.txt, an additional prompt allows you now to change the keyboard layout before you are offered to change passwords. This is obviously helpful since the password input fields are masked, hence false characters due to false keyboard layout may not be recognised but bite you later when you cannot login into your installed application. Many thanks to @LeFish1 for doing this suggestion: https://github.com/MichaIng/DietPi/issues/5925
- DietPi-Services | The service handling for software installed via dietpi-software has received a major rework. Previously, services were "disabled" by default, so that the systemd init system did not start them automatically at boot. Instead, they were enabled at a late boot stage as part of the dietpi-postboot service via dietpi-services. The idea was to assure fast undisturbed early boot process and to have all services started up in a specific order, which guarantees that dependencies are running before dependants. But it had several negative implications: systemd is designed to speed overall services startup via parallelism, while our serial startup broke this benefit. "systemctl status SERVICE" showed services as "disabled", which caused confusion as they were still started at boot. Consequently the only way to prevent them from being started was to mask services or exclude them with our script or its config file, as "systemctl disable SERVICE" had no effect. Also it was not uncommon that APT package upgrades left the service stopped, respecting the "disabled" state. DietPi has now been aligned with how Linux distros with systemd handle services commonly. The most important startup ordering is instead done via systemd unit "After=" ordering directives to have e.g. webservers starting after PHP, and PHP after databases.
- DietPi-RAMlog | For unknown historical reasons, the /var/log tmpfs mount had 1777 mode, i.e. was world-writable with sticky bit. Since this does not match the Debian default for this directory, this mode has been removed from the mount options, so that the default mode 0755 is now used.

#### Bug fixes
- General | Added more generic support for obtaining the CPU temperature, covering some thin clients and potentially other systems where no CPU temperature was shown when running the "cpu" command or enabling it for the login banner.
- Orange Pi 5/ROCK 5B | Resolved an issue where the wrong serial console device name was expected. The correct /dev/ttyFIQ0 will be enabled on DietPi update and patched for a baudrate of 1,500,000.
- Raspberry Pi | With the recent official Raspberry Pi kernel upgrade, Raspberry Pi 4 systems do now boot with the 64-bit kernel by default, even if the underlying OS/userland is 32-bit. While we doubt in any practical benefits, dietpi-software has been patched to correctly deal with this case, i.e. installing 32-bit packages and binaries instead of relying on the "uname -m" kernel architecture output. Some 3rd party software installers may still fail to deal with this case, but we give it a chance for now and hope that affected installers can be relatively easily made compatible.
- DietPi-Config | Resolved an issue where selecting the Nvidia driver on x86_64 systems did not actually install the needed kernel module if kernel headers were not installed before. Headers matching your kernel package are now installed automatically when selecting this option, to have the driver compiled and installed.
- DietPi-Software | Fail2Ban: Resolved a DietPi v8.11 regression where moving logs to systemd journal on install fails due to a typo.
- DietPi-Software | Portainer: Disabled the install option on ARMv6 RPi models since no Portainer Docker image is available for this architecture anymore. Many thanks to `@intiplink` for making us aware of this: https://dietpi.com/forum/t/armv6-bookworm-software-test-matrix-question/16380/11

MichaIng, 2023-04-07:
### DietPi beta v8.16.1 has been merged: #6285

Fourdee, 2023-04-08:
**Testing of https://github.com/MichaIng/DietPi/issues/6262 and https://github.com/MichaIng/DietPi/pull/6288**
- ASUS Zenbook 13 | 11th Gen I5 with Intel Iris Xe graphics.
- DietPi UEFI fresh installation. Setting Beta before network was connected.
- DietPi-Software | LXDE + Steam. :u6307: GPU Driver prompt appeared correctly. 
- DietPi-Autostart | Prompt did not appear for desktop installation. I cant be 100% sure on this, was distracted at the time, however, a retest would be needed my end (when time permits). 
- :u6307: Display brightness (once updated to 5.18 kernel)
- :u6307: GPU Driver
- :u6307: Steam + Proton + Sonic Mania [image]

In relation to above tickets, passed fine here.

Only issue I did noticed was ```apt install firmware-sof-signed``` was required for soundcard to function on my device. Should add this to Intel firmware installs?
```
[    5.164614] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:18:1 Kernel ABI 3:19:1
[    5.232336] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input34
```

MichaIng, 2023-04-08:
https://packages.debian.org/firmware-sof-signed

Available from Bullseye on, "needed for audio functionality on some Intel system". 

For https://packages.debian.org/firmware-intel-sound we have a dedicated sound card option if an Intel CPU was detected. But it may be even needed that both packages are needed in some cases? Not sure if there is a clever way to detect whether either of these firmware is needed, e.g. checking for related error messages in `dmesg`?

However, dedicated topic which needs to wait for next release 🙂.

MichaIng, 2023-04-08:
### DietPi v8.16 has been released: #6282
Many thanks to all testers ❤️!

closed 2023-04-08 by MichaIng