MichaIng/DietPi #6284: DietPi-Config | Support setting OLED display brightness

Fourdee, 2023-04-02:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...

#### Describe the solution you'd like:

- ...

#### Describe alternatives you've considered:

- ...

#### Additional context
Notes:
```
                        MIN_VALUE=0
                        MAX_VALUE=100
                        local fp_brightness current_brightness
                        for fp_brightness in "${afp_current_set_brightness[@]}"
                        do
                                [[ -f $fp_brightness/brightness ]] || continue
                                current_brightness=$(<"$fp_brightness/brightness")
                                [[ -f $fp_brightness/max_brightness ]] && MAX_VALUE=$(<"$fp_brightness/max_brightness")
                                break
                        done

                        if disable_error=1 G_CHECK_VALIDINT "$current_brightness"; then

                                G_WHIP_MENU_ARRAY=()
                                local i=0
                                for i in {1..10}
                                do
                                        G_WHIP_MENU_ARRAY+=("$(( $i * 10 ))" '%')
                                done

                                local current_brightness_percent="$(xrandr --verbose | awk '/Brightness/ { printf "%.0f\n", $2 * 100; exit }')"
                                disable_error=1 G_CHECK_VALIDINT "$current_brightness_percent" && G_WHIP_DEFAULT_ITEM="$current_brightness_percent"
                                G_WHIP_MENU "Please select a Brightness level." || { Back_or_Exit 1; return 0; }

                                local brightness_new_value=$(( ( $MAX_VALUE / 100 ) * $G_WHIP_RETURNED_VALUE ))
                                local brightness_new_oled=$(echo $G_WHIP_RETURNED_VALUE | awk '{printf "%.1f\n", $1 / 100}')

                                echo $brightness_new_value > "$fp_brightness/brightness"
                                G_EXEC xrandr --output eDP-1 --brightness "$brightness_new_oled"

                        else

                                Info_HW_OptionNotSupported

                        fi
```

MichaIng, 2023-04-02:
> Convert previous system (eg: 159200 = Max brightness) to percent based values for WHIP menus.

👍 👍

> Should we move to only using xrandr --output eDP-1 --brightness instead of $fp_brightness/brightness to set brightness? Can it work for all devices?

But only for X sessions, and if executed outside of an X session, but one is active the screen number or name needs to be defined, isn't it? <https://manpages.debian.org/xrandr#d,>
Don't desktops usually have a brightness setting in their GUI which works for OLEDs as well?

Probably we can detect whether the currently used screen has sysfs brightness support or not, and otherwise whether `xrandr` is installed (and hence X11). In the latter case, is it possible to set this somehow in `xorg.conf` instead? Otherwise we could (additionally) add the `xrandr` call to one of the scripts invoked by `startx` to make the change persistent. Same could be done via udev rules for the sysfs brightness setting.

Fourdee, 2023-04-02:
@MichaIng 

> Don't desktops usually have a brightness setting in their GUI which works for OLEDs as well?

I've only ever used LXDE, no current support for brightness that I can see. Always used (and prefered) the DietPi-Config option. Just doesnt work for OLED panels (at least the one i'am using, selfish me lol).

> Probably we can detect whether the currently used screen has sysfs brightness support or not, and otherwise whether xrandr is installed (and hence X11).

Yep, was thinking this aswell. However, currently on my laptop: 
- the sysfs options exist
- setting the value has no effect because OLED has no backlight. I believe its PWM based.
```
root@DietPi-Zen:/home/dietpi# cat /sys/class/backlight/intel_backlight/max_brightness > /sys/class/backlight/intel_backlight/brightness; cat /sys/class/backlight/intel_backlight/brightness
19200
```

Could use this to detect primary display (```eDP-1```). But still ```xrandr``` only.
```
root@DietPi-Zen:/home/dietpi# xrandr --verbose | grep 'connected primary'
eDP-1 connected primary 1920x1080+0+0 (0x48) normal (normal left inverted right x axis y axis) 294mm x 165mm
```

> is it possible to set this somehow in xorg.conf instead?

Will look into it, could work in addition to sysfs? :+1:.

---

Other options?
- https://github.com/Hummer12007/brightnessctl

MichaIng, 2023-04-02:
I wonder what happens if we just always set both, sysfs + `xrandr` if both are supported by the display. Does one overwrite the other or do both stack up? Otherwise it would be probably simplest to just always set both if present/installed.

> Could use this to detect primary display (`eDP-1`). But still `xrandr` only.

👍

Or probably the case that one sets the screen brightness from SSH or another display is sufficiently rare so that we can assume it is wanted for the screen that is currently worked on (if any) and hence call `xrandr` without an identifier (to apply to current screen).
- `[[ $DISPLAY ]]` can be used to identify whether we are in an X session and hence call just `xrandr --brightness ...`.
- If we additionally add it to `xorg.conf` or an X startup script, it then applies to any next X session, regardless which screen.

Nice that there is native Linux support from 5.12 on. The SBCs with lower kernel versions are very rare in the meantime. The once which will never get an update must loose support with Bookworm for several other reasons (Sparky SBC, NanoPi 2 and 3 series (excluding NEO3)). Debian Bullseye kernel (`x86_64`) is 5.10 there is a backport of 6.1 which we could apply on all `x86_64` systems as it brings so many other benefits as well.

Fourdee, 2023-04-02:
Will leave the OLED stuff for now, until I can test latest kernel. Something I can do in own time.

But, ill convert DietPi-Config to percent based system. Might even add another submenu to select display if multiple sysfs entries found.

Fourdee, 2023-04-03:
[image]
Just need to figure out how to get percent of current setting, against max. Math aint as good as it used to be lol :smile:

Fourdee, 2023-04-08:
Will mark this as completed. OLED support is now functional with 5.12 kernel via sysfs. DietPi-Config display brightness changes are now also completed.

closed 2023-04-08 by Fourdee