MichaIng/DietPi #6286: Error H3 Generic Build

actraiser30, 2023-04-03:
Hello

Im trying to build an Updated DietPi image for my OrangePi One AllWinner H3 SBC

using this image "https://fi.mirror.armbian.de/dl/orangepione/archive/Armbian_23.02.2_Orangepione_bullseye_current_5.15.93_minimal.img.xz"

and i get this error in the photo down,do you know whats wrong??

https://ibb.co/VTvKv0z

thank you

actraiser30, 2023-04-03:
sorry how i make this?

MichaIng, 2023-04-03:
It was probably a temporary issue with your network or the used DNS resolver/server, or a DNS settings issue. Can you simply retry? If it fails the same way, can you test this:
```sh
getent hosts deb.debian.org
```
And can you please also show:
```sh
cat /etc/resolv.conf
```

actraiser30, 2023-04-03:
The cat /etc/resolv.conf 
returns: nameserver 192.168.10.1

The getent hosts deb.debian.org
did not work still the same error

Joulinar, 2023-04-03:
192.168.10.1 is your router? Do you use STATIC IP or DHCP? If DHCP, can you switch to STATIC and change DNS server to a global public one like Quad9 or Cloudflare.

Joulinar, 2023-04-03:
could you try to disable IPv6 and check following again

```
getent hosts deb.debian.org
```

MichaIng, 2023-04-03:
> last time i setup dietpi was the normal Armbian cli image not the minimal Armbian cli,can that be the issue??

Only if the minimal image would come without network stack so that `/etc/resolv.conf` could contain invalid data.

> if i exit i have to start all over,flash the image etc

Shouldn't be needed, but you just need to fix the DNS nameserver entry. E.g. you can use
```sh
echo 'nameserver 9.9.9.9' > /etc/resolv.conf
```
To use the public Quad9 DNS server. If DHCP is used (and indeed your router for whatever reason isn't able to resolve `deb.debian.org`), this lasts only until next DHCP lease, but is usually sufficient.

> could you try to disable IPv6

By doing:
```sh
sysctl -w net.ipv6.conf.all.disable_ipv6=1
```

actraiser30, 2023-04-03:
thanks but i dont have ipv6 only ipv4

Joulinar, 2023-04-03:
and you have it disabled actively?

```
sysctl -w net.ipv6.conf.all.disable_ipv6=1
```

actraiser30, 2023-04-03:
nothing is working,what Armbian image you use to make  the armv7 builds for DietPi ?? the minimal cli or the normal cli?

Joulinar, 2023-04-03:
Did you tried to change the DNS server as recommended by @MichaIng ?

MichaIng, 2023-04-03:
Is the network connection up at all at this point?
```sh
ip a
ip r
ping -w 1 9.9.9.9
```

> what Armbian image you use to make the armv7 builds for DietPi ?

We do not use an Armbian image but generate everything from scratch via `debootstrap`: https://github.com/MichaIng/DietPi/blob/dev/.build/images/dietpi-build

actraiser30, 2023-04-03:
network is unreachable it says
i dont understand before it setup packages fine when it reaches a point it stops

i tried also the standard cli image got the same error

Joulinar, 2023-04-03:
It would be helpful to share the entire log from install script

actraiser30, 2023-04-03:
now leave it i wiped the card,does it wants direct connection to the modem? 

i have the orangepi connected to a switch first

MichaIng, 2023-04-03:
It's may be related to these issues:
- #2054
- #2739

I NetworkManager is also used on the minimal image, when removing it, the network is completely de-configured. It was previously used only on the desktop images, but probably that changed. The whole output of the script would be great indeed, e.g. when you execute the installer from SSH and then copy the log from there.

symbios24, 2023-04-03:
So it needs to be removed without re-configure the network??

Joulinar, 2023-04-03:
Probably a misunderstanding. Network Manager is a complete different toll as we use on DietPi. To manage the network we use `ifupdown`. Both tools conflicting and theoretically you would need to switch before running the DietPi install script.

MichaIng, 2023-04-03:
It must leave the active network connections active instead of killing network completely on uninstall. That is a flaw of NetworkManager and Connection Manager that they kill the network and hence it is impossible to uninstall them via SSH or within scripts (without immediately setting up a new network connection). It does another bad thing:
```sh
        # Replace /etc/resolv.conf symlink with an empty file on package
        # removal to signal other packages that NetworkManager is no longer
        # controlling the file. (See: #826366)
        if [ "$(readlink /etc/resolv.conf)" = /run/NetworkManager/resolv.conf ] ; then
            rm -f /etc/resolv.conf
            touch /etc/resolv.conf
        fi
```
So it breaks DNS resolving as well. Not sure what this "signal other packages" means as other network tools shouldn't care about the content of the file when being asked to configure network/set a DNS nameserver. Better would be
```sh
        if [ "$(readlink /etc/resolv.conf)" = /run/NetworkManager/resolv.conf ] ; then
            rm -f /etc/resolv.conf
            if [ -f /run/NetworkManager/resolv.conf ]; then
                mv /run/NetworkManager/resolv.conf /etc/resolv.conf
            else
                touch /etc/resolv.conf
            fi
        fi
```
to just leave the DNS config untouched until something else changes it. But I think what de-configures the network is the service stop done in `prerm`.

MichaIng, 2023-04-03:
Okay strange, I just played around with it:
- Stopping the service does not de-configure the network
- It doesn't erase `/etc/resolv.conf` either, because it is no symlink. It is neither created as symlink on package install, nor on service start, nor on any specific `/etc/NetworkManager/NetworkManager.conf`, nor is it a symlink in the Armbian image and I also used the same NetworkManager config.

Tested on Debian Bullseye. Not sure whether something changed in the meantime, but NetworkManager is not an issue at the moment and I've not idea why the network broke in your case.

Would be great if you could do the following:
- Flash again Armbian
- `apt install isc-dhcp-server`
- Run our installer (on local console, not SSH)
- When you hit this error again, select "Open subshell"

From there we can bring up the network again:
```sh
echo 'iface eth0 inet dhcp' > /etc/network/interfaces
ifup eth0
```

If you can, post a picture of the last output lines so we may be able to catch the important bits. And also:
```sh
ip a s eth0
ip r l dev eth0
```
_(at least whether those outputs contain the expected IP address of the Orange Pi and the router as default route)_

symbios24, 2023-04-04:
[image]

I got this error when installing isc-dhcp-server
Shall i continue with the Dietpi installer ??

Joulinar, 2023-04-04:
That's fine in first install. Usually, the DHCP server should become available on next reboot.

MichaIng, 2023-04-04:
So a valid IPv4 address is still assigned 🤔. Let's see about the route.

> The command ip r 1 dev eth0 returns nothing

That is a lower case `L`, no "one" 😉. Stands for "list". Can also use the long form:
```sh
ip route list dev eth0
```

> The dhcp client is already installed

Ah great, then better to make use of it:
```sh
echo 'iface eth0 inet dhcp' > /etc/network/interfaces
ifup eth0
exit
```

symbios24, 2023-04-04:
[image]

The command ip r l dev eth0 returns this up

MichaIng, 2023-04-04:
So all looks fine, the correct default route and LAN route are assigned. But network does not work?
```sh
apt update
```
This "device or resource busy" error you might indicate a different problem.

symbios24, 2023-04-04:
The apt update works before i start the DietPi installation, afterwards while installing and configure things it stops working

Should i do the: 

echo 'iface eth0 inet dhcp' > /etc/network/interfaces
ifup eth0
exit

Before starting the DietPi installation??

MichaIng, 2023-04-04:
Can you show the output of the following:
```sh
ls -l /etc/hosts
cat /etc/hosts
```

symbios24, 2023-04-04:
[image]

.

MichaIng, 2023-04-04:
Both no symlinks and content looks fine.

And
```sh
df -a
```

symbios24, 2023-04-04:
[image]

.

MichaIng, 2023-04-04:
I see you use a non-root user. Probably there is somehow an issue when its home directory is removed, which is done directly before the `base-files` reinstall attempt.
```sh
pwd
sudo mkdir -m 0700 /root
sudo mkdir /home
mkdir -m 0700 ~
cd ~
apt update
```

symbios24, 2023-04-04:
I do this after the the error or before i start the installation??

symbios24, 2023-04-04:
The installation is completed without the error,i did not had to put the:

pwd
sudo mkdir -m 0700 /root
sudo mkdir /home
mkdir -m 0700 ~

But i did put them before and i dont think they did anything because i got a message that all the folders exist

MichaIng, 2023-04-04:
Okay that is great. I've no idea either what changed. I'll do a test with Armbian minimal image as well when I find time, but I also couldn't find a problem with NetworkManager when browsing the image content and the package maintainer scripts. Need to verify this as well with WiFi, so we could close the related issues I linked above.

symbios24, 2023-04-04:
Yes no problem i will try one more time to install the minimal image to see if it works again and let you know tomorrow

Thank a lot

symbios24, 2023-04-04:
An off topic question,what wifi chipsets(usb) are working better/are most compatible with dietpi??

MichaIng, 2023-04-04:
Good, I already wanted to call the Codebusters and the Codechester brothers. What a great business idea, actually 😎.

> An off topic question,what wifi chipsets(usb) are working better/are most compatible with dietpi??

Generally safest is if they have mainline kernel support/drivers. Debian and Raspberry Pi do not add 3rd party drivers to their kernel builds: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless

But most WiFi 5 and WiFi 6 chips are not there. Armbian (the kernel we use for most other SBCs) adds some 3rd party drivers: https://github.com/armbian/build/blob/main/lib/functions/compilation/patch/drivers_network.sh

So depends on your board.

symbios24, 2023-04-05:
Atheros is the best choice for wifi hotspot i believe,they work good in every board i tried,they are hard to find though.

MichaIng, 2023-04-05:
I think it cannot be said/generalised for a specific manufacturer/brand. Either Linux has drivers+firmware for it, then it works, or not. For very new chips, it usually has not, but it's best to have a look into the source code or search the web to verify this.

And among them which have drivers, I'd just check the stats and benchmarks to know whether it generally suits your needs or not, which WiFi version you need, wide range (proper antenna) or not (mini dongle sufficient) and such things.

actraiser30, 2023-04-06:
Hello, i tried one more time to install the minimal image to see if it works now and got the same error in the first post
i opened a subshell and typed the:

echo 'iface eth0 inet dhcp' > /etc/network/interfaces
ifup eth0
exit

and the installation continued and finished successfully

you need to push a commit to fix the issue?

thanks

MichaIng, 2023-04-06:
First we need to find out why the network connection is actually dropped, and why this seems to not always happen (last time it did not?). Just adding these commands which help in your case can exactly break the network in other cases, so it is not that simple. Best really is to just not touch the network at all and eliminate the reason which kills the network in cases like yours.

actraiser30, 2023-04-07:
ok no problem

Is it possible to release a generic image for H3 boards with armbian cli minimal in the downloads section,orange pi one has not discontinued yet they still making that sbc since 2016

thank you

MichaIng, 2023-04-07:
A generic H3 image is not possible since every SBC requires its own bootloader 😉. The choice in the `dietpi-installer` sets only a flag for our scripts to use to do/show SoC-specific config and install options, but leaves bootloader and kernel of the base image untouched.

actraiser30, 2023-04-07:
Its not possible to add a variety of bootloader's in a single image??

actraiser30, 2023-04-07:
Or at least add a pre-made image for Orange pi one in the downloads as it
turns out its still a Popular sbc

MichaIng, 2023-04-07:
> Its not possible to add a variety of bootloader's in a single image??

Nope. There is always just a single fixed address where the SPL and bootloader image is stored.

I'll reconsider adding back some older Orange Pi SBCs, since we can now easily create them via CI pipeline. At best open a request about it here: https://github.com/MichaIng/DietPi/discussions/categories/image-request

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen if the problem persists.

closed 2025-05-31 by MichaIng