MichaIng/DietPi #6291: initial boot set-up failed

mrfoxie, 2023-04-07:
i flashed os in pi 0w and connect to wifi after that installation process start and stuck on this error

[image]

Joulinar, 2023-04-07:
Looks like an issue with SD card. Can you flash a new image again or try another one?

mrfoxie, 2023-04-07:
let me try on new sdcard

mrfoxie, 2023-04-07:
still the same issue but that same sdcard works on raspberry Pi 4 perfectly without any errors

mrfoxie, 2023-04-07:
i this it maybe having with repository

Joulinar, 2023-04-07:
The repository is the global Raspberry one but your error is a timeout on mmc1, which would be your SD card.

mrfoxie, 2023-04-07:
the same sdcard worked fine on raspberry Pi 4

mrfoxie, 2023-04-07:
[image]

now getting this error 🤦🏻‍♂️🤦🏻‍♂️🤦🏻‍♂️

Joulinar, 2023-04-07:
looks like your system is trying to connect to some Raspbian mirrors in India. Not sure if they have some server issues? Or if it is a local network problem.

mrfoxie, 2023-04-10:
well, I tried to change the network but still got the same issue. I change the repo to USA and it worked perfectly fine. I had to change the repo multiple times to check the perfect one but I got it.

MichaIng, 2023-04-10:
Generally the mirror is accessible: https://raspbian.mirror.net.in/raspbian/raspbian/

And at least the first package ia as well here: https://raspbian.mirror.net.in/raspbian/raspbian/pool/main/libe/libedit/libedit2_3.1-20191231-2_armhf.deb

Probably a temporary server issue 🤔.

Joulinar, 2023-04-10:
Or some geo blocking?

MichaIng, 2023-04-10:
Although the apt update seems to have succeeded and only the DEB download timed out.

mrfoxie, 2023-04-10:
may be its due to geo blocking

MichaIng, 2023-04-11:
I'll mark this issue as closed. Feel free to reopen if required.