MichaIng/DietPi #6295: ability to wipe everything and have base image

G2G2G2G, 2023-04-07:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
no

#### Describe the solution you'd like:
Dietpi runs a script to wipe all the bloat from a normal debian install, and then installs it's own scripts and utilities.. I'd like to remove those or never install them in the first place

https://dietpi.com/docs/hardware/#make-your-own-distribution
I don't see any mention of it here either, but the idea would be a truly minimal install, with dietpi deleting everything and not re-adding anything.

Joulinar, 2023-04-08:
> Dietpi runs a script to wipe all the bloat from a normal debian install

This is what our install script does already. It will remove all packages we don't need.

MichaIng, 2023-04-08:
> a truly minimal install, with dietpi deleting everything and not re-adding anything.

Define "truly minimal". I'd say that skipping any further package would make the result barely usable as general purpose system, if you select to skip WiFi packages/firmware already (and do not use WiFi).

Note that what the installer "adds" is sometimes a replacement, like that the network stack is in case replaced with ifupdown + isc-dhcp-client. Same may be the case for kernel and bootloader. The extra files/script are btw <1.5 MiB in size.

While everyone has a different opinion about what is "minimal" and which features are mandatory for general purpose images, ours are at least still about half in size compared to other "lite" and "minimal" images out there. But we are a few things we want to guarantee:
- Network access with (nearly) any Ethernet and WiFi adapter, as without initial network one has a bootstrap issue to install further requirements. WiFi however can be de-selected already.
- The mandatory parts of the systemd init system (including udev and journal). This is quite debatable since systemd is huge compared to SysV or others. But we see it as too essential nowadays to run the wide range of server software we offer. Also it is much easier for newbies to find info/guides/templates for systemd, as nearly every distro uses it by default.
- Full APT capability to access the largest (I think?) Linux binary package repository out there. That is basically a question about the base distro: There are much lighter alternatives to Debian, like Alpine, but one will face much more often the case that software needs to be compiled from source. And the needed toolchain and headers quickly eats up all the initial size benefit such a distro might have.

You could run a Linux kernel booting into busybox only which can be less than 10 MiB in size. An example with some minimal package management on top: http://www.tinycorelinux.net/
But this is not what we aim to offer as DietPi shall be easy to setup a wide range of server applications for Linux newcomers as well.

With this as background, I think it would be more productive if you could name the package(s) or files which are installed but which you would like to not have installed. We might consider adding further options or skip some in certain circumstances. What I already thought about:
- Remove bzip2 and p7zip and instead install them on demand in our scripts.
- Skip `console-setup` on SBCs with no display port and add a headless option for this as well: HDMI and keyboard would still work, but e.g. no keyboard layout and (sometimes locale related) console fonts/sizes are/can be applied.

G2G2G2G, 2023-04-08:
>This is what our install script does already. It will remove all packages we don't need.

it adds /boot/dietpi and several other things. Was looking to not have any of those

>Define "truly minimal".

min·i·mal
adjective
adjective: minimal
    1.    of a minimum amount, quantity, or degree; negligible

>I'd say that skipping any further package would make the result barely usable as general purpose system, if you select to skip WiFi packages/firmware already (and do not use WiFi).

as stated in my first post, looking to not reinstall any of the dietpi stuff, just delete all of what you initially delete. When executing `bash` for example you have scripts that are ran every time. I am inquiring about either having a script that can remove all of it or it is just never added in the first place
```
rm /etc/bashrc.d/dietpi-bash_completion.sh
rm /etc/bashrc.d/dietpi.bash
rm /etc/cron.hourly/dietpi
rm /etc/cron.daily/dietpi
rm -R /boot/dietpi
```
...and many others

> Full APT capability to access the largest (I think?) Linux binary package repository out there. That is basically a question about the base distro: There are much lighter alternatives to Debian, [...]

not really related to what I said, I said remove what you do and not add dietpi scripts back in. apt is not a dietpi script and it is not wiped, therefore it'd remain..

Joulinar, 2023-04-08:
Honestly I'm not sure what exactly you expect from us. If you are looking for a system without DietPi, you need to look for other systems like Alpine Linux. Our files/script are <1.5 MiB in size, and removing them would break our images.

G2G2G2G, 2023-04-09:
I removed them just fine, only issues that can even occur are dbus and wifi related.
Was just asking for a bare image of what you wiped, it isn't that complicated... not sure what there is to understand.
My entire student base uses these as learning systems so I figured it was useful to others. If you're uninterested that's fine...

closed 2023-04-09 by G2G2G2G

MichaIng, 2023-04-09:
May I ask for the reasons to remove the scripts? If it's about disk usage, taking care about DEB packages has a multiple times larger effect.

Note that some of the tasks our services/cron jobs/scripts do are usually seen as needed and you hence may want to install alternatives to take care of those tasks:
- Enable the `systemd-timesyncd.service` or install another time sync daemon
- Install `cpufrequtils` or something similar to apply a proper CPU governor which does not keep the CPU frequency at minimum (or depending on SBC maximum) all the time
- Install `unattended-updates` or something similar to keep security update notifications up

You should also disable and remove our systemd services as currently they are enabled but doomed to fail. And you should remove the tmpfs from /var/log as now nothing is cleaning that directory and it can hence fill completely.

G2G2G2G, 2023-04-11:
nothing to do with storage, as joulinar said it's minimal... was never even a consideration
all of the reasons are because of the execution and automation and class being a learning environment for these things. 
all the stuff dietpi has is good for anyone not having/wanting to deal with any of it, but this is the opposite we have here.
- running bash runs a script (which also seems to check for dietpi updates?) 
- the automated installer and your use of /mnt/dietpi some stuff installs there, so any manual editing is non-standard

see the minimal install, removing all this, allows to learn the systems administration required for installing packages manually. (something as simple as: `apt install nginx`, for example. and then configuring it.. in /etc/nginx/)

syncthing for example, you could apt install it but it's ideal to find this page: https://apt.syncthing.net/ and add actual recent ones (for them)

**All of these aren't actual issues, dietpi is just the easiest vector to getting a bare system for most SBCs. The second best is armbian for this.. the official ones take a lot more work to remove all the crap from.**

>Enable the systemd-timesyncd.service or install another time sync daemon

this seems to automatically do that itself or something. There are several things we need to initially do (like the cpu governor) but never had to do this

>Install cpufrequtils or something similar to apply a proper CPU governor which does not keep the CPU frequency at minimum (or depending on SBC maximum) all the time

We don't install anything like that. This is a learning thing on "how to create a systemd service" a basic script that echos "schedutil" into `/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor`

>Install unattended-updates or something similar to keep security update notifications up

`apt install update && apt install upgrade` 

The only thing to do with services is `apt install libpam-systemd dbus` which isn't a "HAVE TO DO" but it is nice to have for SSH to not give errors/hang. System is fine without it.

>You should also disable and remove our systemd services as currently they are enabled but doomed to fail. And you should remove the tmpfs from /var/log as now nothing is cleaning that directory and it can hence fill completely.

yea everything was removed.
we keep /tmp and /var/log as ram, enabled zram etc. I think if someone did fill their /var/log that's a good learner too

**Anyway it's nothing negative to dietpi, it is just the fact that the minimal images elsewhere take more work, especially like the "raspberry pi lite" which is far from "lite".. it should be renamed "headless" or "CLI" because it has tons of crap by default.** I figured not adding the dietpi scripts would be as simple as having the "dietpi-prepare-installer-script" just stop after it removes all the garbage from raspberry pi foundation image.

The ONLY negatives to dietpi I have are (and they don't really matter that much)
- the dietpi-update/upgrade turns off stuff before updating so it usually is off longer than a regular apt upgrade which only restarts stuff after it's already replaced. Generally even if that thing isn't being updated.. except for pi-hole FTL or whatever it's called 

and maybe a privacy complaint
- automatic survey thing, automatically checking for newer versions, etc

Joulinar, 2023-04-11:
> allows to learn the systems administration required for installing packages manually. (something as simple as: apt install nginx, for example. and then configuring it.. in /etc/nginx/)

but you can use/install everything manual as you the wish using `apt install`. There is no need to use our software catalogue. 

> and maybe a privacy complaint
>
> automatic survey thing, automatically checking for newer versions, etc

all this can be disabled if you like.

> the dietpi-update/upgrade turns off stuff before updating 

Services can be excluded from being shutdown before updates happen

MichaIng, 2023-04-11:
> running bash runs a script (which also seems to check for dietpi updates?)

Running bash always runs several scripts on every distro:
- `/etc/profile`, `/etc/profile.d`, `~/.profile`
- `/etc/bash.bashrc`, `~/.bashrc`
- `/etc/update-motd.d`, `/etc/motd`
- `/etc/pam.d`

DietPi adds `/etc/bashrc.d` (non-login bash shell counterpart to `/etc/profile.d`) and a script which adds the `dietpi-*` aliases and the login banner (alternative to `/etc/update-motd.d` > `/etc/motd` while I aim to merge this into `/etc/update-motd.d` when I find time). The login script does not imply the update check: This is done as systemd service at boot and via daily (nightly) cron job.

> the automated installer and your use of /mnt/dietpi some stuff installs there, so any manual editing is non-standard

But it does not prevent you to install things different. We do not offer install options for each and every software that exists, so one often needs to still install software manually the common way. We use `/etc/dietpi_userdata` along with a simple way to move/mount that on an external drive to reduce writes to the often used SD card rootfs on SBCs and enhance performance by splitting OS and service I/O. But that does not mean that not using it (for manually installed software) would be any issue. As Joulinar said, using DietPi tools and "ways" is an offer, but you can setup everything like you would do on any other Debian based distro without issues.

However, I see you have the essential tasks in view (like CPU governor and time sync, which were by main concerns) and aim to have a system where such explicitly needs to be setup manually. The existence of scripts which automate/abstract things contradicts such aim, of course 😄.

> syncthing for example, you could apt install it but it's ideal to find this page: https://apt.syncthing.net/ and add actual recent ones (for them)

If I remember right, this is x86 only, or does it support ARM as well in the meantime?

G2G2G2G, 2023-04-12:
>If I remember right, this is x86 only, or does it support ARM as well in the meantime?

it has always supported ARM v6 - v8 (aarch64).
Syncthing is written in Go all you have to do to compile to those is `GOARCH=arm GOARM=7` or `GOARCH=arm64`

MichaIng, 2023-04-16:
I think we decided against those APT packages since they implement the systemd service on a per-user basis instead of as single system service. So we'd need to our implementation and add support for instantiated systemd services to `dietpi-services`, which currently do not fully work there.

Hmm, or we take the package but just keep using our `syncthing.service` 🤔.

Ah, it does not support RISC-V yet, and not sure whether those armhf packages are compatible with ARMv6 RPi models as well.

Since Syncthing has an internal updater, the security benefit I first thought about is not really given, so I think for now our implementation, which takes the very latest release as tarball from GitHub, is fine.