MichaIng/DietPi #6297: Home Assistant Install Fail

holocronology, 2023-04-09:
#### Details:
- Date           | Sun Apr  9 02:30:18 BST 2023
- Bug report     | 634f2cf4-58f9-459c-9da3-89b194efb257
- DietPi version | v8.16.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `sudo -u homeassistant dash -c
. /home/homeassistant/pyenv-activate.sh
pyenv install 3.10.11
pyenv local 3.10.11
pip3 install -U pip setuptools wheel
[ -z '' ] || pip3 install
exec pip3 install homeassistant`
- Exit code      | 127
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Downloading Python-3.10.11.tar.xz...
-> https://www.python.org/ftp/python/3.10.11/Python-3.10.11.tar.xz
/home/homeassistant/.pyenv/plugins/python-build/bin/python-build: line 248: pushd: Python-3.10.11: No such file or directory

BUILD FAILED (Debian GNU/Linux 11 using python-build 20180424)

Inspect or clean up the working tree at /tmp/python-build.20230409023017.3629
Results logged to /tmp/python-build.20230409023017.3629.log

Last 10 log lines:
/tmp/python-build.20230409023017.3629 ~
warning: xz not found; consider installing `xz` package
tar (child): xz: Cannot exec: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
pyenv: version `3.10.11' not installed
dash: 5: pip3: not found
dash: 7: exec: pip3: not found
```

Joulinar, 2023-04-09:
That sounds like file system issues. Can you check kernel messages.

```
dmesg -l 0,1,2,3
```

holocronology, 2023-04-09:
`[    7.100084] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped`

MichaIng, 2023-04-09:
```sh
apt install xz-utils
```
Just strage that pyenv downloads Python as xz archive in the first place. This seems to be new.

Joulinar, 2023-04-09:
Another one https://dietpi.com/forum/t/home-assistant-update-failed-with-pushd-python-3-10-11-no-such-file-or-directory/16515

josh3003, 2023-04-11:
Also happened to me, 

`apt install xz-utils`

seems to of fixed it as well with me, currently progressing with the install.

MichaIng, 2023-04-11:
Was a pyenv bug solved upstream 4 hours ago: https://github.com/pyenv/pyenv/commit/7770894

closed 2023-04-11 by MichaIng