MichaIng/DietPi #6304: DietPi-Software | microblog.pub | new install fails with error: can't find Rust compiler

kinoushe, 2023-04-11:
#### Details:
- Date           | Tue Apr 11 22:54:03 AEST 2023
- DietPi version | v8.16.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux dietPi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `sudo -u microblog-pub dash -c 
. /opt/microblog-pub/pyenv-activate.sh || exit 1
pyenv install 3.10.9 || exit 1
pyenv local 3.10.9 || exit 1
pip3 install -U pip setuptools wheel || exit 1
pip3 install poetry || exit 1
poetry install || exit 1
# Skip these for now:
#poetry run inv configuration-wizard
#poetry run inv migrate-db
poetry env info | grep -oP '(?<=^Path:\s).*/virtualenvs/.*' > '/opt/microblog-pub/.virtual_path'`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. Just testing the new microblog.pub software install, and it's failing on install for me:
#### Actual behaviour:
```error: can't find Rust compiler```
#### Additional logs:
```
Downloading Python-3.10.9.tar.gz...
-> https://www.python.org/ftp/python/3.10.9/Python-3.10.9.tgz
Installing Python-3.10.9...
Installed Python-3.10.9 to /opt/microblog-pub/versions/3.10.9
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /opt/microblog-pub/versions/3.10.9/lib/python3.10/site-packages (22.3.1)
Collecting pip
  Downloading https://www.piwheels.org/simple/pip/pip-23.0.1-py3-none-any.whl (2.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 165.3 kB/s eta 0:00:00
Requirement already satisfied: setuptools in /opt/microblog-pub/versions/3.10.9/lib/python3.10/site-packages (65.5.0)
Collecting setuptools
  Downloading https://www.piwheels.org/simple/setuptools/setuptools-67.6.1-py3-none-any.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 176.2 kB/s eta 0:00:00
Collecting wheel
  Downloading https://www.piwheels.org/simple/wheel/wheel-0.40.0-py3-none-any.whl (64 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 64.5/64.5 kB 161.3 kB/s eta 0:00:00
Installing collected packages: wheel, setuptools, pip
  Attempting uninstall: setuptools
    Found existing installation: setuptools 65.5.0
    Uninstalling setuptools-65.5.0:
      Successfully uninstalled setuptools-65.5.0
  Attempting uninstall: pip
    Found existing installation: pip 22.3.1
    Uninstalling pip-22.3.1:
      Successfully uninstalled pip-22.3.1
Successfully installed pip-23.0.1 setuptools-67.6.1 wheel-0.40.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting poetry
  Downloading https://www.piwheels.org/simple/poetry/poetry-1.4.2-py3-none-any.whl (222 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 222.7/222.7 kB 128.1 kB/s eta 0:00:00
Collecting lockfile<0.13.0,>=0.12.2
  Downloading https://www.piwheels.org/simple/lockfile/lockfile-0.12.2-py2.py3-none-any.whl (13 kB)
Collecting requests<3.0,>=2.18
  Downloading https://www.piwheels.org/simple/requests/requests-2.28.2-py3-none-any.whl (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.8/62.8 kB 150.6 kB/s eta 0:00:00
Collecting trove-classifiers>=2022.5.19
  Downloading https://www.piwheels.org/simple/trove-classifiers/trove_classifiers-2023.3.9-py3-none-any.whl (13 kB)
Collecting pkginfo<2.0.0,>=1.9.4
  Downloading https://www.piwheels.org/simple/pkginfo/pkginfo-1.9.6-py3-none-any.whl (30 kB)
Collecting tomlkit!=0.11.2,!=0.11.3,<1.0.0,>=0.11.1
  Downloading https://www.piwheels.org/simple/tomlkit/tomlkit-0.11.7-py3-none-any.whl (35 kB)
Collecting poetry-core==1.5.2
[... 360 lines cut ...]
      error: can't find Rust compiler
      
      If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
      
      To update pip, run:
      
          pip install --upgrade pip
      
      and then retry package installation.
      
      If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
      
      This package requires Rust >=1.48.0.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cryptography
  Building wheel for cffi (setup.py): started
  Building wheel for cffi (setup.py): finished with status 'done'
  Created wheel for cffi: filename=cffi-1.15.1-cp310-cp310-linux_armv7l.whl size=402526 sha256=bceddc7c761d730fb8254b7c8ded30d1b798b3abab18b7a129b5019d5fec5ba1
  Stored in directory: /tmp/pip-ephem-wheel-cache-izkzs3nu/wheels/9a/ee/45/194003e40379492794fdf4ff200b86205f5f63115b85864251
Successfully built dulwich msgpack rapidfuzz cffi
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects
```

Joulinar, 2023-04-12:
@MichaIng I guess we would need to do it similar to how we install Rust for Home Assistant.

MichaIng, 2023-04-12:
32-bit ARM cryptography, so also `libssl-dev` and `pkg-config` is needed.

closed 2023-04-21 by MichaIng