MichaIng/DietPi #6305: dietpi-software reinstall 114 (NextCloud) >>> This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.4.

robyscar, 2023-04-12:
#### Details:
- Date           | Wed Apr 12 15:29:46 +07 2023
- DietPi version | v8.16.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux R1PeeHole 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ncc maintenance:install --no-interaction --database=mysql --database-name=nextcloud --database-user=tmp_root --database-pass=******* --admin-user=admin --admin-pass=******* --data-dir=/mnt/5TB/nextcloud_data`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.4.
```

Joulinar, 2023-04-12:
Did you install PHP8 yourself? Because on Debian Bullseye, highest version should be PHP7.4.

robyscar, 2023-07-28:
yes i did of course.

DietPi is my dstro of choice, but i need also the "freedom" to do something on my own. tweaking it sometime.

that's why i am also opening an "issue" (suggestion) asking to simplify the dietpi-sofware script. As is - other than INCREDIBLE COMPLEX TO MANAGE - is not easily tweakable.

So, improvements and new ideas are difficult to come into the main stream. and you are not 10000 of you, but a dozen.

R1

Joulinar, 2023-07-28:
> So, improvements and new ideas are difficult to come into the main stream. and you are not 10000 of you, but a dozen.

Feel free to join and contribute 😉

MichaIng, 2023-07-28:
> DietPi is my dstro of choice, but i need also the "freedom" to do something on my own. tweaking it sometime.

Of course you are free to do anything on your own. But in some cases you may then also need to do the related software install on your own. There are ways and guides to do this with Nextcloud. You got it right that `dietpi-software` is complex already, handling all architectures, Debian versions, in case individual SBCs correctly. If we now started to also cover all possible custom runtime system (PHP, Python, Java, ...) versions one can (of course) compile manually, it would become much worse, impossible to maintain, at least not with our limited man power.

For PHP in particular I played with the idea to check for and handle things generically based on the actually installed PHP version, but did not go for it yet since this can cause a lot of other issues anyway. There are some things we must expect to be able to automate installs and do things correctly, and for PHP this includes the package(s) Debian provides, including its config files, defaults, dir structure etc.

However, to allow the install of the correct Nextcloud version in your case: https://github.com/MichaIng/DietPi/blob/3d8bafd/dietpi/dietpi-software#L4667
```sh
sed -i 's/latest-25/latest/' /boot/dietpi/dietpi-software
```
A remaining problem is that the PHP configuration we do is done for PHP 7.4 only, including the installed/enabled extra modules and config files. So you may need to do that for your PHP version as well. Check the install log for what is done to PHP 7.4, or just copy the created/edited config files from `/etc/php/7.4` over and enable the related modules.

Joulinar, 2024-01-01:
@MichaIng should be close this one? In meantime Debian Bookworm has been released shipping PHP8 by default.

closed 2024-01-08 by MichaIng