MichaIng/DietPi #6306: Installing Home Assistant fails with "could not build wheels for cryptography"

josh3003, 2023-04-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=16
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | bullseye 1

- Kernel version |Linux DietPi 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux
- SBC model | `RPI4)
- Power supply used |Official rpi4 power supply )
- SD card used | (EG: SanDisk high endurance)

#### Additional Information (if applicable)
- Software title | Home Assistant)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | e1ad757d-e820-4a1c-a450-aeeda49bf7fb

#### Steps to reproduce

1. ... run dietpi-software install 157

#### Expected behaviour

- Install home assistant

#### Actual behaviour

-errors out and cancels install
Failed to build bcrypt cryptography
ERROR: Could not build wheels for bcrypt, cryptography, which is required to install pyproject.toml-based projects

#### Extra details

- ...
`
---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Wed Apr 12 14:59:49 AEST 2023
- DietPi version | v8.16.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `sudo -u homeassistant dash -c
. /home/homeassistant/pyenv-activate.sh
pyenv install 3.10.11
pyenv local 3.10.11
pip3 install -U pip setuptools wheel
[ -z '' ] || pip3 install
exec pip3 install homeassistant`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
[... 528 lines cut ...]
  Stored in directory: /tmp/pip-ephem-wheel-cache-q7z41pz7/wheels/6a/47/39/0658f195a3ad6a90f063e0d52e2388bd5203832134250ae045
  Building wheel for yarl (pyproject.toml): started
  Building wheel for yarl (pyproject.toml): finished with status 'done'
  Created wheel for yarl: filename=yarl-1.8.1-cp310-cp310-linux_aarch64.whl size=248541 sha256=27ca44171caf845b10e7b470ad7a4e29f4c2d7466de20d55914c3f934933f4b9
  Stored in directory: /tmp/pip-ephem-wheel-cache-q7z41pz7/wheels/55/3c/8b/5668641fde6efc8aed9afe6e01ca9981f314fc8ab320472219
  Building wheel for cffi (setup.py): started
  Building wheel for cffi (setup.py): finished with status 'done'
  Created wheel for cffi: filename=cffi-1.15.1-cp310-cp310-linux_aarch64.whl size=402540 sha256=2027a5e6b4230d4b86f5e68e76783a7d3d2429a1d792e7cccc46a2b725ecc248
  Stored in directory: /tmp/pip-ephem-wheel-cache-q7z41pz7/wheels/9a/ee/45/194003e40379492794fdf4ff200b86205f5f63115b85864251
  Building wheel for frozenlist (pyproject.toml): started
  Building wheel for frozenlist (pyproject.toml): finished with status 'done'
  Created wheel for frozenlist: filename=frozenlist-1.3.3-cp310-cp310-linux_aarch64.whl size=141429 sha256=76063895fef5edb21ed572cd6b13ef3c50a5b698e7564f19db8508d8b5c63017
  Stored in directory: /tmp/pip-ephem-wheel-cache-q7z41pz7/wheels/75/c0/b3/57ac2234203df002195d6d093179dc847a8694fb02fe1aefa8
  Building wheel for MarkupSafe (setup.py): started
  Building wheel for MarkupSafe (setup.py): finished with status 'done'
  Created wheel for MarkupSafe: filename=MarkupSafe-2.1.2-cp310-cp310-linux_aarch64.whl size=24954 sha256=639d38e717001f0059471c9b737e33b5663f756e4cb9c46f4d655ca7d03a7ecc
  Stored in directory: /tmp/pip-ephem-wheel-cache-q7z41pz7/wheels/54/8e/98/3c8a462676f35ac84e7e2a886d2e90c50c796b73f4cd1f351f
  Building wheel for multidict (pyproject.toml): started
  Building wheel for multidict (pyproject.toml): finished with status 'done'
  Created wheel for multidict: filename=multidict-6.0.4-cp310-cp310-linux_aarch64.whl size=108534 sha256=852677d32048b3ff2f1e64870847e75520a43c0433cca23be09eb32d0723c337
  Stored in directory: /tmp/pip-ephem-wheel-cache-q7z41pz7/wheels/d3/0f/ad/bd2376f99105cc211cfe25f50868fdbddedcf5bab5a699e47b
Successfully built aiohttp ciso8601 home-assistant-bluetooth lru-dict pyyaml ulid-transform yarl cffi frozenlist MarkupSafe multidict
Failed to build bcrypt cryptography
ERROR: Could not build wheels for bcrypt, cryptography, which is required to install pyproject.toml-based projects
`

Joulinar, 2023-04-12:
ok some questions,

> Kernel version |Linux DietPi 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux

your kernel version `5.10.103` is not the last one, as this is 6.1.21. Usually this should have been update during DietPi-Update. Are you running this old version intentionally? Did you reboot your system after finishing DietPi-Update?

> Pre-image | Raspbian Lite
> Hardware | RPi 4 Model B (aarch64) (ID=4)

Your kernel is within 64bit mode, however you are running a 32bit image. Is this intentionally?

Was this a fresh install of DietPi image? If yes, I would recommend to start with a ARMv8 image.

josh3003, 2023-04-12:
> your kernel version `5.10.103` is not the last one, as this is 6.1.21. Usually this should have been update during DietPi-Update. Are you running this old version intentionally? Did you reboot your [...]
 I have done a reboot, might do another for good measure.
> > Pre-image | Raspbian Lite
> > Hardware | RPi 4 Model B (aarch64) (ID=4)
> 
> Your kernel is within 64bit mode, however you are running a 32bit image. Is this intentionally?
 I think this was a weird workaround to a OOM reaper comment I found. I will roll that back to 32 bit thanks for bringing it to my attention, I think it was either faulty hdd or memory leak issue. https://github.com/MichaIng/DietPi/issues/4622
> Was this a fresh install of DietPi image? If yes, I would recommend to start with a ARMv8 image.
No it isn't, but might try the above and see if there is any change.

josh3003, 2023-04-12:
Ok so have removed the 64bit and ran a dietpi update and as there is no available update even after a couple reboots it is still stuck on 

`Linux DietPi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux`

Any other way to force update the kernel? Should this resolve the issue do we think?

Joulinar, 2023-04-12:
No, the kernel update will not solve the problem, as these are two different issues. But you should be able to do the installation now. Ok let's clean Rust to get correct version and install one package in addition.

```
apt install pkg-config -y
rm -R /home/homeassistant/.cargo/
rm -R /home/homeassistant/.rustup/
dietpi-software install 157
```

MichaIng, 2023-04-12:
However, let's also check why the kernel is stuck at a version of Buster while you're on Bullseye:
```sh
cat /etc/apt/sources.list.d/raspi.list
apt policy raspberrypi-kernel
```

josh3003, 2023-04-12:
> However, let's also check why the kernel is stuck at a version of Buster while you're on Bullseye:
> ```sh
> cat /etc/apt/sources.list.d/raspi.list
> apt policy raspberrypi-kernel
> ```

Ok it looks like that file doesn't exist. What do I need to populate the raspi.list with? 

cat: /etc/apt/sources.list.d/raspi.list: No such file or directory
root@DietPi:~# apt policy raspberrypi-kernel
raspberrypi-kernel:
  Installed: 1:1.20220308~buster-1
  Candidate: 1:1.20220308~buster-1
  Version table:
 *** 1:1.20220308~buster-1 100
        100 /var/lib/dpkg/status

MichaIng, 2023-04-12:
```sh
echo 'deb https://archive.raspberrypi.org/debian/ bookworm main' > /etc/apt/sources.list.d/raspi.list
```

Joulinar, 2023-04-12:
> bookworm main

shouldn't it be `bullseye`?

MichaIng, 2023-04-12:
Right (fixed).

josh3003, 2023-04-12:
> ```sh
> echo 'deb https://archive.raspberrypi.org/debian/ bullseye main' > /etc/apt/sources.list.d/raspi.list
> apt update
> apt full-upgrade
> ```

Okay updating now then I will try the install afterwards, thank you for your assistance

josh3003, 2023-04-12:
Still having the problem

```
0:00:00
Building wheels for collected packages: aiohttp, bcrypt, ciso8601, cryptography, home-assistant-bluetooth, lru-dict, pyyaml, ulid-transform, yarl, cffi, frozenlist, MarkupSafe, multidict
  Building wheel for aiohttp (pyproject.toml): started
  Building wheel for aiohttp (pyproject.toml): finished with status 'done'
  Created wheel for aiohttp: filename=aiohttp-3.8.4-cp310-cp310-linux_aarch64.whl size=1000389 sha256=b0a3c174ab51f6be8a16a44134b0e5837921fd5f449512738e014078d79a6d6f  Stored in directory: /tmp/pip-ephem-wheel-cache-2tckb9o_/wheels/45/f9/f2/66c145c2f6d7b35277d17b15c7a875b85b5b69ab0f3b3fad38
  Building wheel for bcrypt (pyproject.toml): started
  Building wheel for bcrypt (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error

  × Building wheel for bcrypt (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [59 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-310
      creating build/lib.linux-aarch64-cpython-310/bcrypt
      copying src/bcrypt/__init__.py -> build/lib.linux-aarch64-cpython-310/bcrypt
      copying src/bcrypt/__about__.py -> build/lib.linux-aarch64-cpython-310/bcrypt      running egg_info
      writing src/bcrypt.egg-info/PKG-INFO
      writing dependency_links to src/bcrypt.egg-info/dependency_links.txt
      writing requirements to src/bcrypt.egg-info/requires.txt
      writing top-level names to src/bcrypt.egg-info/top_level.txt
      reading manifest file 'src/bcrypt.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      warning: no previously-included files found matching 'requirements.txt'
      warning: no previously-included files found matching 'release.py'
      warning: no previously-included files found matching 'mypy.ini'
      warning: no previously-included files matching '*' found under directory '.github'
      warning: no previously-included files matching '*' found under directory '.circleci'
      warning: no previously-included files found matching 'src/_bcrypt/target'
      warning: no previously-included files matching '*' found under directory 'src/_bcrypt/target'
      adding license file 'LICENSE'
      writing manifest file 'src/bcrypt.egg-info/SOURCES.txt'
      copying src/bcrypt/_bcrypt.pyi -> build/lib.linux-aarch64-cpython-310/bcrypt
      copying src/bcrypt/py.typed -> build/lib.linux-aarch64-cpython-310/bcrypt
      running build_ext
      running build_rust
      error: command failed: 'rustc': No such file or directory (os error 2)

          =============================DEBUG ASSISTANCE=============================
          If you are seeing a compilation error please try the following steps to
          successfully install bcrypt:
          1) Upgrade to the latest pip and try again. This will fix errors for most             users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
          2) Ensure you have a recent Rust toolchain installed. bcrypt requires
             rustc >= 1.56.0.

          Python: 3.10.11
          platform: Linux-6.1.21-v8+-aarch64-with-glibc2.31
          pip: n/a
          setuptools: 67.6.1
          setuptools_rust: 1.5.2
          rustc: n/a
          =============================DEBUG ASSISTANCE=============================

      error: can't find Rust compiler

      If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.

      To update pip, run:

          pip install --upgrade pip

      and then retry package installation.

      If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
[... 249 lines cut ...]
  Created wheel for yarl: filename=yarl-1.8.1-cp310-cp310-linux_aarch64.whl size=248558 sha256=9a8b73ba48028b034dfca2baa9bbe09d17d63d38b8d82848423bf989fc8c3921
  Stored in directory: /tmp/pip-ephem-wheel-cache-2tckb9o_/wheels/55/3c/8b/5668641fde6efc8aed9afe6e01ca9981f314fc8ab320472219
  Building wheel for cffi (setup.py): started
  Building wheel for cffi (setup.py): finished with status 'done'
  Created wheel for cffi: filename=cffi-1.15.1-cp310-cp310-linux_aarch64.whl size=402522 sha256=553bf08c8d87f8c30c6bfabf7e8418b86b4271fe243d7db674951311d8e1361f
  Stored in directory: /tmp/pip-ephem-wheel-cache-2tckb9o_/wheels/9a/ee/45/194003e40379492794fdf4ff200b86205f5f63115b85864251
  Building wheel for frozenlist (pyproject.toml): started
  Building wheel for frozenlist (pyproject.toml): finished with status 'done'
  Created wheel for frozenlist: filename=frozenlist-1.3.3-cp310-cp310-linux_aarch64.whl size=141423 sha256=0f28f31991b2d263b1120af8b57e6f34c3ea15b5b40abf19a7a9c88c229b4a4d
  Stored in directory: /tmp/pip-ephem-wheel-cache-2tckb9o_/wheels/75/c0/b3/57ac2234203df002195d6d093179dc847a8694fb02fe1aefa8
  Building wheel for MarkupSafe (setup.py): started
  Building wheel for MarkupSafe (setup.py): finished with status 'done'
[... cut ...]
  Stored in directory: /tmp/pip-ephem-wheel-cache-2tckb9o_/wheels/d3/0f/ad/bd2376f99105cc211cfe25f50868fdbddedcf5bab5a699e47b
Successfully built aiohttp ciso8601 home-assistant-bluetooth lru-dict pyyaml ulid-transform yarl cffi frozenlist MarkupSafe multidict
Failed to build bcrypt cryptography
ERROR: Could not build wheels for bcrypt, cryptography, which is required to install pyproject.toml-based projects
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.
```

MichaIng, 2023-04-12:
It tries to install `aarch64` wheels while the userland is Raspbian and even the 32-bit kernel is loaded now (while this is actually not an issue). Can you please check all this:
```sh
cat /etc/os-release
cat /etc/apt/sources.list
dpkg --print-architecture
```

josh3003, 2023-04-12:
cat /etc/os-release

`PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

cat /etc/apt/sources.list

`deb http://raspbian.raspberrypi.org/raspbian bullseye main contrib non-free rpi`

dpkg --print-architecture

`armhf`

Joulinar, 2023-04-13:
Did you perform all steps to clean the install? https://github.com/MichaIng/DietPi/issues/6306#issuecomment-1505499484

josh3003, 2023-04-13:
I will do that now, rebooting and will come back with an update. Thanks

josh3003, 2023-04-13:
> 

> Pls can you add following to the end of your `/boot/config.txt` and reboot your system afterwards

`
Downloading Python-3.10.11.tar.xz...
-> https://www.python.org/ftp/python/3.10.11/Python-3.10.11.tar.xz
Installing Python-3.10.11...
Installed Python-3.10.11 to /home/homeassistant/.pyenv/versions/3.10.11
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Requirement already satisfied: pip in ./.pyenv/versions/3.10.11/lib/python3.10/site-packages (23.0.1)
Requirement already satisfied: setuptools in ./.pyenv/versions/3.10.11/lib/python3.10/site-packages (65.5.0)
Collecting setuptools
  Downloading https://www.piwheels.org/simple/setuptools/setuptools-67.6.1-py3-none-any.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 801.4 kB/s eta 0:00:00
Collecting wheel
  Downloading https://www.piwheels.org/simple/wheel/wheel-0.40.0-py3-none-any.whl (64 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 64.5/64.5 kB 36.8 MB/s eta 0:00:00
Installing collected packages: wheel, setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 65.5.0
    Uninstalling setuptools-65.5.0:
      Successfully uninstalled setuptools-65.5.0
Successfully installed setuptools-67.6.1 wheel-0.40.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Collecting homeassistant
  Downloading homeassistant-2023.4.3-py3-none-any.whl (23.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 23.2/23.2 MB 40.7 MB/s eta 0:00:00
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/aiohttp/
Collecting aiohttp==3.8.4
  Downloading aiohttp-3.8.4.tar.gz (7.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 7.3/7.3 MB 35.9 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Installing backend dependencies: started
  Installing backend dependencies: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting PyJWT==2.6.0
  Downloading https://www.piwheels.org/simple/pyjwt/PyJWT-2.6.0-py3-none-any.whl (20 kB)
Collecting voluptuous==0.13.1
  Downloading https://www.piwheels.org/simple/voluptuous/voluptuous-0.13.1-py3-none-any.whl (35 kB)
Collecting attrs==22.2.0
  Downloading https://www.piwheels.org/simple/attrs/attrs-22.2.0-py3-none-any.whl (60 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60.0/60.0 kB 199.4 kB/s eta 0:00:00
Collecting voluptuous-serialize==2.6.0
  Downloading https://www.piwheels.org/simple/voluptuous-serialize/voluptuous_serialize-2.6.0-py3-none-any.whl (6.8 kB)
Collecting atomicwrites-homeassistant==1.4.1
  Downloading https://www.piwheels.org/simple/atomicwrites-homeassistant/atomicwrites_homeassistant-1.4.1-py2.py3-none-any.whl (7.1 kB)
Collecting typing-extensions<5.0,>=4.5.0
  Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-4.5.0-py3-none-any.whl (27 kB)
Collecting awesomeversion==22.9.0
  Downloading https://www.piwheels.org/simple/awesomeversion/awesomeversion-22.9.0-py3-none-any.whl (12 kB)
Collecting cryptography==40.0.1
  Downloading cryptography-40.0.1.tar.gz (624 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 625.0/625.0 kB 43.7 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting astral==2.2
  Downloading https://www.piwheels.org/simple/astral/astral-2.2-py2.py3-none-any.whl (30 kB)
Collecting orjson==3.8.7
  Downloading orjson-3.8.7-cp310-cp310-manylinux_2_17_armv7l.manylinux2014_armv7l.whl (149 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 149.8/149.8 kB 45.0 MB/s eta 0:00:00
Collecting jinja2==3.1.2
[... 459 lines cut ...]
  Created wheel for yarl: filename=yarl-1.8.1-cp310-cp310-linux_armv7l.whl size=248538 sha256=7a51df385a541a8b3a659ea714a4c5cf406ad5feaf78b581ae05c7d463a0c2f4
  Stored in directory: /tmp/pip-ephem-wheel-cache-egqdtqro/wheels/55/3c/8b/5668641fde6efc8aed9afe6e01ca9981f314fc8ab320472219
  Building wheel for cffi (setup.py): started
  Building wheel for cffi (setup.py): finished with status 'done'
  Created wheel for cffi: filename=cffi-1.15.1-cp310-cp310-linux_armv7l.whl size=402512 sha256=a2f6c8012067a6111936a743380f9fe621eaf10a9a640d4e95def17268f7b257
  Stored in directory: /tmp/pip-ephem-wheel-cache-egqdtqro/wheels/9a/ee/45/194003e40379492794fdf4ff200b86205f5f63115b85864251
  Building wheel for frozenlist (pyproject.toml): started
  Building wheel for frozenlist (pyproject.toml): finished with status 'done'
  Created wheel for frozenlist: filename=frozenlist-1.3.3-cp310-cp310-linux_armv7l.whl size=141424 sha256=bbc1e828719ba44b5b58819c70f50b8227cc509011767a5d87cec97a097cb299
  Stored in directory: /tmp/pip-ephem-wheel-cache-egqdtqro/wheels/75/c0/b3/57ac2234203df002195d6d093179dc847a8694fb02fe1aefa8
  Building wheel for MarkupSafe (setup.py): started
[... cut ...]
  Created wheel for multidict: filename=multidict-6.0.4-cp310-cp310-linux_armv7l.whl size=108533 sha256=031f040885406215b7cf8d12a39e7e09f98e3f72a2b195c9c5df00c3afe444dc
  Stored in directory: /tmp/pip-ephem-wheel-cache-egqdtqro/wheels/d3/0f/ad/bd2376f99105cc211cfe25f50868fdbddedcf5bab5a699e47b
Successfully built aiohttp ciso8601 home-assistant-bluetooth lru-dict pyyaml ulid-transform yarl cffi frozenlist MarkupSafe multidict
Failed to build bcrypt cryptography
ERROR: Could not build wheels for bcrypt, cryptography, which is required to install pyproject.toml-based projects
```
`

Still no luck...

josh3003, 2023-04-13:
Looks to of been resolved. Will mark this as fixed. Must've been the 64bit causing the issue.

Joulinar, 2023-04-13:
> can you please let me know the login details

Which login details you are looking for? For Home Assistance?

josh3003, 2023-04-13:
> > can you please let me know the login details
> 
> Which login details you are looking for? For Home Assistance?

Yes Home Assistant app

josh3003, 2023-04-13:
> Once done, login to web interface and create your own individual admin account. There are no default credentials.

I've ran the `systemctl restart home-assistant`
htop shows low cpu usage but it isn't letting me create an account just to sign in. Maybe I'll try restart the service again shortly?

Joulinar, 2023-04-13:
you are connecting via web interface or some app?

josh3003, 2023-04-13:
> you are connecting via web interface or some app?

via the web port 8123, <localaddress>:8123

josh3003, 2023-04-13:
See here?
[image]

josh3003, 2023-04-13:
Just going to do a reinstall and let it sit a while to see if there's any changes.

Joulinar, 2023-04-13:
You can check for available user

```
grep username /mnt/dietpi_userdata/homeassistant/.storage/auth
```

And to reset login as root via SSH/CLI

```
sudo -u homeassistant bash
. /home/homeassistant/pyenv-activate.sh
exec hass --script auth change_password your_username new_password
systemctl restart home-assistant.service
```

josh3003, 2023-04-13:
Thanks that got the username all sorted now.

steals, 2023-04-16:
I have similar issue when trying to update homeassistant on 32bit dietpi installation.

Python: 3.10.11
          platform: Linux-6.1.21+-armv6l-with-glibc2.31
          pip: n/a
          setuptools: 67.6.1
          setuptools_rust: 1.5.2
          rustc: n/a
          =============================DEBUG ASSISTANCE=============================

      error: `cargo rustc --lib --message-format=json-render-diagnostics --manifest-path src/rust/Cargo.toml --release -v --features 'pyo3/extension-module pyo3/abi3-py36' -- --crate-type cdylib` failed with code 101
      [end of output]

Joulinar, 2023-04-16:
pls open an own issue and fill the related bug report. As well, you would need to attach the entire output of HA update process (not just some lines)

josh3003, 2023-04-18:
Yeah that's correct

MichaIng, 2023-04-19:
Strange: The rustup installer pulls 32-bit `rustup`, `cargo` and `rustc` binaries, but the default toolchain is indeed `stable-aarch64-unknown-linux-gnu`, only based on the CPU architecture, despite 32-bit userland. This can be done to switch the default toolchain:
```sh
rustup default stable-armv7-unknown-linux-gnueabihf
```
Or directly on install:
```sh
curl -sSf https://sh.rustup.rs | sh -s -- --profile minimal --default-host armv7-unknown-linux-gnueabihf
```
Not sure why but "syncing channel updates" currently takes ages. With this, hopefully also `pip` compiles wheels with Rust code correctly when the 64-bit kernel is enabled on 32-bit OS.

MichaIng, 2023-04-19:
Dammit, this is the reason why it hangs:
```
[ 3392.600164] cp15barrier_handler: 26605006 callbacks suppressed
[ 3392.600172] "reqwest-interna" (4115) uses deprecated CP15 Barrier instruction at 0xb12d14
[ 3392.600178] "reqwest-interna" (4115) uses deprecated CP15 Barrier instruction at 0xb12d38
[ 3392.600180] "reqwest-interna" (4115) uses deprecated CP15 Barrier instruction at 0xb12d14
[ 3392.600182] "reqwest-interna" (4115) uses deprecated CP15 Barrier instruction at 0xb12d38
[ 3392.600184] "reqwest-interna" (4115) uses deprecated CP15 Barrier instruction at 0xb12d14
[ 3392.600187] "reqwest-interna" (4115) uses deprecated CP15 Barrier instruction at 0xb12d38
[ 3392.600189] "reqwest-interna" (4115) uses deprecated CP15 Barrier instruction at 0xb12d14
[ 3392.600190] "reqwest-interna" (4115) uses deprecated CP15 Barrier instruction at 0xb12d38
[ 3392.600192] "reqwest-interna" (4115) uses deprecated CP15 Barrier instruction at 0xb12d14
[ 3392.600194] "reqwest-interna" (4115) uses deprecated CP15 Barrier instruction at 0xb12d38
```
This repeats every 5 seconds. Could one of you test this on RPi 4 (with 32-bit userland + 64-bit kernel), the second command posted above? I'm currently testing it on Odroid N2. Hopefully this is a platform specific issue.

MichaIng, 2023-04-19:
PR up: #6325
Would be great if someone could test whether this `abi.cp15_barrier=2` is required on RPi as well.

closed 2023-04-21 by MichaIng